cyring/CoreFreq #229: System hang when registering Clock Source twice

ppascher, 2021-04-28:
I registered "Clock Source" in the settings menu which worked fine. Next I wanted to "unregister" but accidentally hit "register" again. This caused my system to hang. Not sure if this is due to corefreq or a bug with amdgpu? Journal only shows me this:
```syslog
Apr 28 16:53:24 kernel: clocksource: corefreq: mask: 0xffffffffffffffff max_cycles: 0x3102c246e1e, max_idle_ns: 440795343192 ns
Apr 28 16:57:34 kernel: clocksource: corefreq: mask: 0xffffffffffffffff max_cycles: 0x3102c2dd02c, max_idle_ns: 440795323298 ns
Apr 28 16:57:38 kernel: [drm:amdgpu_dm_atomic_commit_tail [amdgpu]] *ERROR* Waiting for fences timed out!
Apr 28 16:57:39 kernel: [drm:amdgpu_dm_atomic_commit_tail [amdgpu]] *ERROR* Waiting for fences timed out!
```
after which I hit the reset button since switching tty was not possible. This is on kernel 5.12 with a 5950x CPU and 290x GPU using corefreq-git (split packages) on ArchLinux.

Thank you for your work cyring.

cyring, 2021-04-28:
Hello

* Do you build with my TSC implementation ?
```
make DELAY_TSC=1
```

* Is CoreFreq registered as idle, governor and cpufreq driver ?
```
insmod corefreqk.ko  Register_Governor=1 Register_CPU_Freq=1 Register_CPU_Idle=1
```

* Above setup is not provided by package.
You will have to build and launch CoreFreq manually.
It will also required to make some room in kernel space : blacklist the drivers automatically loaded which prevent CoreFreq to register instead.
Somehow it would be an almost bare-metal kernel.

* At the end of all this preparation, CoreFreq should be in total control of Processor, which is confirmed in the Client: kernel window

* Employing all these prerequisites, please let me know if you are still facing the issue.
* Meanwhile I will check for any bug with consecutive clock source registrations.

ppascher, 2021-04-29:
No, I did not build with DELAY_TSC (using AUR corefreq-dkms-git). Neither did I register as idle, governor or cpufreq.
I should have checked the README. Closing this issue.

closed 2021-04-29 by ppascher

cyring, 2021-04-29:
> No, I did not build with DELAY_TSC (using AUR corefreq-dkms-git). Neither did I register as idle, governor or cpufreq.
> I should have checked the README. Closing this issue.

Just to make code better: you mean that with prerequisites fully set up, you don't encounter the issue anymore ?

If true then I will add those registrations check before any attempt to enable my clock source algorithm.