cyring/CoreFreq #236: Zen/Whitehaven - Inverted TDP values

h1z1, 2021-05-04:
Just noticed this, not sure if it's a bug in Corefreq or platform?

[image]

cyring, 2021-05-04:
> Just noticed this, not sure if it's a bug in Corefreq or platform?

Not a bug but experimental with your Zen model:
https://github.com/cyring/CoreFreq/blob/717e444e89caa0c74038f9c0d09ae68b725007cd/corefreqk.c#L5682

So these SMU addresses only work with Zen 2 & 3:
https://github.com/cyring/CoreFreq/blob/717e444e89caa0c74038f9c0d09ae68b725007cd/corefreqk.c#L5618

Zen1: I believe TDP has to be queried on the PM table with mailbox protocol. 
Do you know any SMU single address to read your TDP ?

What about the others: Min, Max, PPT, EDC, TDC: do you confirm values are also wrong ?

h1z1, 2021-05-06:
No idea tbh.  Wasn't aware TDP could be read, thought it was inferred and even that varied with vendors, implementations and phases of the moon.  AMD goes so far as to conflate Thermal TDP with Electrical.   GN did an excellent job covering it iirc.

cyring, 2021-05-14:
Hello,

Can you please show me what values you are getting from this project ?

https://gitlab.com/CyrIng/ryzen_smu

h1z1, 2021-05-19:
Hi, sorry not sure why this didn't send a notification... hmm

Which values are you looking for, from syslog?

```
[1343745.127347] ryzen_smu: CPUID: family 0x17, model 0x1, stepping 0x1, package 0x7
[1343745.127917] ryzen_smu: SMU v4.25.118.0
[1343745.128734] ryzen_smu: SMU v4.25.118.0
[1343745.128759] sysfs: cannot create duplicate filename '/kernel/ryzen_smu_drv'
[1343745.128761] CPU: 8 PID: 19460 Comm: insmod Tainted: P           O      5.4.70 #1
[1343745.128763] Hardware name: Gigabyte Technology Co., Ltd. X399 AORUS Gaming 7/X399 AORUS Gaming 7, BIOS F12 12/11/2019
```

```

cyring, 2021-05-19:
> Hi, sorry not sure why this didn't send a notification... hmm
> 
> Which values are you looking for, from syslog?
> 
> ```
> [1343745.127347] ryzen_smu: CPUID: family 0x17, model 0x1, stepping 0x1, package 0x7
> [1343745.127917] ryzen_smu: SMU v4.25.118.0
> [1343745.128734] ryzen_smu: SMU v4.25.118.0
> [1343745.128759] sysfs: cannot create duplicate filename '/kernel/ryzen_smu_drv'
> [1343745.128761] CPU: 8 PID: 19460 Comm: insmod Tainted: P           O      5.4.70 #1
> [1343745.128763] Hardware name: Gigabyte Technology Co., Ltd. X399 AORUS Gaming 7/X399 AORUS Gaming 7, BIOS F12 12/11/2019
> ```
> 
> /sys/kernel/ryzen_smu_drv/codename:05
> /sys/kernel/ryzen_smu_drv/drv_version:0.1.0
> /sys/kernel/ryzen_smu_drv/mp1_if_version:0
> /sys/kernel/ryzen_smu_drv/version:4.25.118.0

Because ryzen_smu implements the mailbox protocol, I'm interested in the TDP, PPT, EDC, TDC 
In fact the whole output from its monitor UI will be fine.

But apparently you are facing issues to start it up ?

h1z1, 2021-05-19:
Not sure what you mean by start up.   Module appeared to load despite the error though none of the utils run...

```
# ./userspace/monitor_cpu 
rd_buf: 0.1.0
PM Tables are not supported on this platform.
# ./scripts/test.py 
Failed to write SMU arguments
# ./scripts/
cpuid.py          dump_pm_table.py  monitor_cpu.py    __pycache__/      read_dump.py      test.py
# ./scripts/monitor_cpu.py 
PM Table: Unsupported
# ./scripts/dump_pm_table.py 
PM Tables are not supported for this model of processor.
# 
```

cyring, 2021-05-19:
> Not sure what you mean by start up. Module appeared to load despite the error though none of the utils run...
> 
> ```
> # ./userspace/monitor_cpu 
> rd_buf: 0.1.0
> PM Tables are not supported on this platform.
> # ./scripts/test.py 
> Failed to write SMU arguments
> # ./scripts/
> cpuid.py          dump_pm_table.py  monitor_cpu.py    __pycache__/      read_dump.py      test.py
> # ./scripts/monitor_cpu.py 
> PM Table: Unsupported
> # ./scripts/dump_pm_table.py 
> PM Tables are not supported for this model of processor.
> # 
> ```
> 
> Edit: Just realized what you mean, some sysfs files did not in fact get created in ryazen_smu_drv. pm_table being one.
> 
> ```
> drwxr-xr-x  2 root root    0 May 19 01:27 .
> drwxr-xr-x 14 root root    0 May  3 11:35 ..
> -r--------  1 root root 4096 May 19 01:27 codename
> -r--------  1 root root 4096 May 19 01:27 drv_version
> -r--------  1 root root 4096 May 19 01:27 mp1_if_version
> -rw-------  1 root root 4096 May 19 01:27 mp1_smu_cmd
> -rw-------  1 root root 4096 May 19 00:52 rsmu_cmd
> -rw-------  1 root root 4096 May 19 01:27 smn
> -rw-------  1 root root 4096 May 19 01:30 smu_args
> -r--------  1 root root 4096 May 19 01:27 version
> ```
> 
> Couple lines were truncated above too _sigh_
> 
> ```
> [1343745.127347] ryzen_smu: CPUID: family 0x17, model 0x1, stepping 0x1, package 0x7
> [1343745.127917] ryzen_smu: SMU v4.25.118.0
> [1343745.128734] ryzen_smu: SMU v4.25.118.0
> [1343745.128759] sysfs: cannot create duplicate filename '/kernel/ryzen_smu_drv'
> [1343745.128789]  ryzen_smu_probe+0x114/0x390 [ryzen_smu]
> [1343745.128812]  ryzen_smu_driver_init+0x23/0x1000 [ryzen_smu]
> [1343745.128841] kobject_add_internal failed for ryzen_smu_drv with -EEXIST, don't try to register things with the same name in the same directory.
> [1343745.128843] ryzen_smu: Unable to create sysfs interface
> [1343745.128846] ryzen_smu: probe of 0000:40:00.0 failed with error -12
> ```

With my 3950X I have to force loading the software with option `-f`
_make sure no other instance of its driver is running or any SMU software is already present_
```
insmod ryzen_smu.ko
```
---
```
ryzen_smu: CPUID: family 0x17, model 0x71, stepping 0x0, package 0x2
```
---
```
./userspace/monitor_cpu -f
```
---
```
╭────────────────────────────────────────────────┬─────────────────────────────────────────────────╮
│                                      CPU Model │             AMD Ryzen 9 3950X 16-Core Processor │
│                            Processor Code Name │                                         Matisse │
│                                          Cores │                                              16 │
│                                      Core CCDs │                                               2 │
│                                      Core CCXs │                                               4 │
│                                  Cores Per CCX │                                               4 │
│                                 SMU FW Version │                                        v46.63.0 │
│                                 MP1 IF Version │                                             v11 │
[... 47 lines cut ...]
│                                      cLDO_VDDM │                                        0.9504 V │
│                                      cLDO_VDDP │                                        0.9976 V │
│                                      cLDO_VDDG │                                        1.0477 V │
╰────────────────────────────────────────────────┴─────────────────────────────────────────────────╯
```
* Comparing with the _CoreFreq_ TDP section
```
Power, Current & Thermal                                                        
|- Junction Temperature                                        TjMax   [49: 95C]
|- Digital Thermal Sensor                                        DTS   [Capable]
|- Power Limit Notification                                      PLN   [Missing]
|- Package Thermal Management                                    PTM   [Missing]
|- Thermal Monitor 1                                             TTP   [Capable]
|- Thermal Monitor 2                                             HTC   [Capable]
|- Thermal Design Power                                          TDP   [  105 W]
   |- Minimum Power                                              Min   [  105 W]
   |- Maximum Power                                              Max   [  105 W]
|- Package Power Tracking                                        PPT   [  142 W]
|- Electrical Design Current                                     EDC   [  140 A]
|- Thermal Design Current                                        TDC   [   95 A]
|- Units                                                                        
   |- Power                                               watt   [  0.125000000]
   |- Energy                                             joule   [  0.000015259]
   |- Window                                            second   [  0.000976562]
```

h1z1, 2021-05-19:
No change with -f.   Doesn't appear that supports Threadripper?  There is no case for it in smu_init

cyring, 2021-05-19:
> No change with -f. Doesn't appear that supports Threadripper? There is no case for it in smu_init
* I'm finding two [`CODENAME_THREADRIPPER`](https://gitlab.com/CyrIng/ryzen_smu/-/blob/smu_debug/smu.c#L212) cases of SMU interface detection:
1. case of [`RSMU`](https://gitlab.com/CyrIng/ryzen_smu/-/blob/smu_debug/smu.c#L302)
2. case of [`MP1`](https://gitlab.com/CyrIng/ryzen_smu/-/blob/smu_debug/smu.c#L334)

* Aren't you sure you are not facing this issue
```
kobject_add_internal failed for ryzen_smu_drv with -EEXIST, don't try to register things with the same name in the same directory.
ryzen_smu: Unable to create sysfs interface
```
Presuming `"ryzen_smu_drv"` was uncleanly unloaded (kernel reboot may be required)
There are various reasons which _lead_ to those [error messages](https://gitlab.com/CyrIng/ryzen_smu/-/blob/smu_debug/drv.c#L373).

Perhaps, you may have a better chance with the [original ryzen_smu project](https://gitlab.com/leogx9r/ryzen_smu) ?

* The SMU mailbox protocol has to be possible with Whitehaven Threadripper, it is reversed from Ryzen-Master. Without that Windows tool, I don't see another mean to have the TDP reference values to compare with; except datasheets if available.

h1z1, 2021-05-19:
>     * [`CODENAME_THREADRIPPER`](https://gitlab.com/CyrIng/ryzen_smu/-/blob/smu_debug/smu.c#L212)

Neither of the references I could find actually initialize it though:  int smu_resolve_cpu_class 

smu_init has no case:

```
 switch (g_smu.codename) {
        case CODENAME_SUMMITRIDGE:
        case CODENAME_THREADRIPPER:
        case CODENAME_PINNACLERIDGE:
            g_smu.addr_rsmu_mb_cmd  = 0x3B1051C;
            g_smu.addr_rsmu_mb_rsp  = 0x3B10568;
            g_smu.addr_rsmu_mb_args = 0x3B10590;
            goto LOG_RSMU;

```

cyring, 2021-05-26:
Hello,

Latest `develop` branch is bringing the HSMP mailbox which is specified for the AMD Family 19h Model 01h
Fortunately it works with my `3950X`. See this [discussion](https://github.com/cyring/CoreFreq/discussions/242).

I would like to know if it works with your Threadripper ?

h1z1, 2021-05-31:
Hi

Would depend on what you mean by works :)   TDP is reported as missing now

[image]

cyring, 2021-05-31:
Sorry I forgot to add you need to start the driver with the Experimental mode.

```
insmod corefreqk.ko Experimental=1
```

h1z1, 2021-05-31:
Eeep thought I did but nope.  Sorry.

Values still appear backward though

[image]

cyring, 2021-05-31:
I really don't find any good registers for Ryzen, Threadripper, EPYC of generations Zen1 & Zen+

I was hoping that the HSMP protocol to the SMU will provide a universal mean to query those Power values. Unfortunately, HSMP is specified within the **Zen3** PPR documentation. I'm lucky it works with Zen2/Matisse, however I still have not received any report from CoreFreq Users, owner of  others APU, TR, EPYC generation 2 and 3

It's up to you to close or not this issue, but I have to admit that I'm facing the limits of the AMD public specifications.

cyring, 2021-06-23:
I'm still searching on the subject: do you see any option in your BIOS to activate HSMP or names like BMC ?

cyring, 2021-07-01:
According to this readme https://github.com/AMDESE/libhsmp/commit/a3bb6efac327645a18835b2efcc542a12d49012c
can you confirm the HSMP state ?

cyring, 2021-07-16:
[image]

h1z1, 2021-07-26:
I don't see anything about those but then again it's not exactly a server board (especially wrt the BMC).

cyring, 2021-07-30:
I have updated BIOS to 3703 and I now see a value change in `PL2`
[image]

h1z1, 2021-07-30:
Which motherboard is that with?

cyring, 2021-07-30:
> Which motherboard is that with?

This 395OX setup 
https://github.com/cyring/CoreFreq/wiki/CoreFreq-Lab

cyring, 2021-08-15:
I tried today Ryzen Master (Windows) and learned from it that `395W` is the Max power `PPT`
[image]
This confirmed for my `3950X` that `PL2` = `395 W` is the value read from HSMP when `PPT` is left as AUTO into BIOS.

Your `WhiteHaven` should have other Max value(s) when `PPT` is AUTO. You may read this using  Ryzen Master. Please let me know what you get ?

cyring, 2021-09-16:
> Just noticed this, not sure if it's a bug in Corefreq or platform?
> 
> [image]

I have noticed with PBO enabled and Power Limits as AUTO that my 3950X is showing some specific constant values:`395W` for `PL1` and `PL2`

[image]

I wonder if you are getting the same behavior but with different constants ?

cyring, 2021-09-25:
HSMP is now a per model feature.

closed 2021-09-25 by cyring

h1z1, 2021-11-11:
Still haven't had a chance to install windows on these boards to test sorry.  With latest head though they appear the same ?

[image]

h1z1, 2021-11-30:
Was chasing some crashes that I"m not entirely sure are directly related to corefreq or not but I can confirm at least that no amount of disabling c6/power states appear to have any impact.  Values are still inverted *shrug*

The other crashes happen with kernel tracing  and corefreq.   It's unfortunately a hard lock and only appears to happen in X.  Maybe a race somewhere

cyring, 2021-11-30:
> Was chasing some crashes that I"m not entirely sure are directly related to corefreq or not but I can confirm at least that no amount of disabling c6/power states appear to have any impact. Values [...]

* I have been chasing an incompatibility with `kdump`
See [Wiki](https://github.com/cyring/CoreFreq/wiki/Software-incompatibilities-and-workarounds#kernel-crash-dump) if it is part of the issue.
Adding `crashkernel=0` in the boot command line had solved the problem; but I've not found its root cause.

* `perf top` is also killed if started **after** _CoreFreq_, and not the reverse.

* `selinux` is also a no go with _CoreFreq_

* `nmi_watchdog` and fixed performance counters can not be re-enabled

* About the inverted values, HSMP is not doing great with Zen v1 families. I'm developing the RMI protocol for Naples but not sure if it is available in Threadripper ?

h1z1, 2021-12-02:
Don't think so from what I gather of [varorum](https://github.com/LLNL/variorum) and a tool from [amd](https://github.com/amd/amd_hsmp).   NGL, Corefreq runs circles around both.   https://github.com/amd/esmi_oob_library  looks neat too but limited.

cyring, 2021-12-02:
> Don't think so from what I gather of [varorum](https://github.com/LLNL/variorum) and a tool from [amd](https://github.com/amd/amd_hsmp). NGL, Corefreq runs circles around both. https://github.com/amd/esmi_oob_library looks neat too but limited.

Can you plz post the output of what you're getting from those software ?

cyring, 2025-05-10:
@h1z1 Hello,
Do you mind to verify if latest CoreFreq is still working with your ThreadRipper ?
CLI outputs are welcomed to verify frequencies, thermal, voltage, powet.
Thank you