cyring/CoreFreq #244: Update AUR links in readme

RealStickman, 2021-05-22:
The packages `corefreq` and `corefreq-git` don't exist any more in the AUR.  
Instead, they have been split into `corefreq-client`, `corefreq-server` and `corefreq-dkms` with git-versions available as well.  
`corefreq-client` provides the full program as it pulls the other two as dependencies and is probably what most people want.

cyring, 2021-05-22:
Indeed CoreFreq has a new AUR package maintainer.
Thanks to @inglor who I believe may help with this issue.

inglor, 2021-05-23:
@RealStickman What's the issue though? Readme already refers to them and has links. The split package is only one PKGBUILD and the link points to the `pkgbase`.

RealStickman, 2021-05-23:
I was confused because the link in the readme points to a package base instead of an actual package.  
The AUR helpers I tried can't install the package with the name `corefreq`.  
The proper, `makepkg` way works though, so feel free to close this issue.

closed 2021-05-23 by cyring