cyring/CoreFreq #256: Intel i5-10400 on ASRock - H470M-ITX/ac

ich777, 2021-06-19:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

[lspci.txt](https://github.com/cyring/CoreFreq/files/6680147/lspci.txt)
[motherboard.txt](https://github.com/cyring/CoreFreq/files/6680148/motherboard.txt)
[lscpu.txt](https://github.com/cyring/CoreFreq/files/6680149/lscpu.txt)

`dmidecode -t memory`:
```# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x000A, DMI type 16, 23 bytes
Physical Memory Array
        Location: System Board Or Motherboard
        Use: System Memory
        Error Correction Type: None
        Maximum Capacity: 32 GB
        Error Information Handle: Not Provided
        Number Of Devices: 2

Handle 0x000B, DMI type 17, 84 bytes
Memory Device
        Array Handle: 0x000A
        Error Information Handle: Not Provided
        Total Width: 64 bits
        Data Width: 64 bits
        Size: 16 GB
        Form Factor: DIMM
        Set: None
        Locator: ChannelA-DIMM0
        Bank Locator: BANK 0
        Type: DDR4
        Type Detail: Synchronous
        Speed: 2667 MT/s
        Manufacturer: Crucial
        Serial Number: E38A9CF5
        Asset Tag: 9876543210
        Part Number: CT16G4DFRA266.C8FE
        Rank: 1
        Configured Memory Speed: 2667 MT/s
        Minimum Voltage: 1.2 V
        Maximum Voltage: 1.2 V
        Configured Voltage: 1.2 V
        Memory Technology: DRAM
        Memory Operating Mode Capability: Volatile memory
        Firmware Version: Not Specified
        Module Manufacturer ID: Bank 6, Hex 0x9B
        Module Product ID: Unknown
        Memory Subsystem Controller Manufacturer ID: Unknown
        Memory Subsystem Controller Product ID: Unknown
        Non-Volatile Size: None
        Volatile Size: 16 GB
        Cache Size: None
        Logical Size: None

Handle 0x000C, DMI type 17, 84 bytes
Memory Device
        Array Handle: 0x000A
        Error Information Handle: Not Provided
        Total Width: 64 bits
        Data Width: 64 bits
        Size: 16 GB
        Form Factor: DIMM
        Set: None
        Locator: ChannelB-DIMM0
        Bank Locator: BANK 2
        Type: DDR4
        Type Detail: Synchronous
        Speed: 2667 MT/s
        Manufacturer: Crucial
        Serial Number: E38A9CF1
        Asset Tag: 9876543210
        Part Number: CT16G4DFRA266.C8FE
        Rank: 1
        Configured Memory Speed: 2667 MT/s
        Minimum Voltage: 1.2 V
        Maximum Voltage: 1.2 V
        Configured Voltage: 1.2 V
        Memory Technology: DRAM
        Memory Operating Mode Capability: Volatile memory
        Firmware Version: Not Specified
        Module Manufacturer ID: Bank 6, Hex 0x9B
        Module Product ID: Unknown
        Memory Subsystem Controller Manufacturer ID: Unknown
        Memory Subsystem Controller Product ID: Unknown
        Non-Volatile Size: None
        Volatile Size: 16 GB
        Cache Size: None
        Logical Size: None```

cyring, 2021-06-19:
Thanks a lot.
I see many small bits to fix, starting from specs of generation 10 and 11.
Stay tuned ...

cyring, 2021-06-20:
Hello,

For testings `develop` is updated for IMC.

cyring, 2021-09-06:
Can you plz test the Comet Lake against latest development branch ?

ich777, 2021-09-06:
Oh sorry, yes of course. Will report back.

ich777, 2021-09-07:
Is this what you wanted:
[image]

```
                             Intel Z490  [ 685]                            
Controller #0                                                Dual Channel  
 Bus Rate  8000 MT/s      Bus Speed 8028 MT/s          DRAM Speed 2667 MHz 
                                                                           
 Cha    CL  RCD   RP  RAS  RRD  RFC   WR RTPr WTPr  FAW  B2B  CWL CMD  REFI
  #0    16   18   18   35   24  734   21   10   40   28    0   15  2T 10400
  #1    16   18   18   35   24  734   21   10   40   28    0   15  2T 10400
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4    6    7        10   10   10   11        32   27    7    7
  #1     7    4    6    7        10   10   10   11        32   27    7    7
      sgWW dgWW drWW ddWW                                         CKE   ECC
  #0     7    4    7    7                                           4    0 
  #1     7    4    7    7                                           4    0 
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    1     65536      1024          16384                    
       #1    16    1     65536      1024          16384                    
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    1     65536      1024          16384                    
       #1    16    1     65536      1024          16384
```

[image]

Should I also test this on my 6700k?

cyring, 2021-09-07:
> Is this what you wanted:
> [image]
>
> ```
>                              Intel Z490  [ 685]
> Controller #0                                                Dual Channel
>  Bus [...]

Everything looks working great.

`rank` may however be wrong b/c equation to DIMM size is not true.

Yes for 6700K as I don't have anymore access to Skylake family for non-regression tests.

ich777, 2021-09-07:
> Yes for 6700K as I don't have anymore access to Skylake family for non-regression tests.

Will create a dedicated issue for that with all pictures included. :)

Before I forgot to say this was from my i5-10600, @sonic6 will also post a picture and output from IMC here in this issue for his 10400 that this thread is for.

soonic6, 2021-09-10:
hello,

like @ich777 said, here a the screenshots:
[image]

```
root@Unraid-1:/tmp/corefreq# corefreq-cli -M
                             Intel H470  [ 684]                            
Controller #0                                                Dual Channel  
 Bus Rate  8000 MT/s      Bus Speed 8011 MT/s          DRAM Speed 2667 MHz 
                                                                           
 Cha    CL  RCD   RP  RAS  RRD  RFC   WR RTPr WTPr  FAW  B2B  CWL CMD  REFI
  #0    19   19   19   43    7  467   20   10   42   28    0   18  2T 10400
  #1    19   19   19   43    7  467   20   10   42   28    0   18  2T 10400
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4    6    7         9    9   10   11        33   28    6    6
  #1     7    4    6    7         9    9   10   11        33   28    6    6
      sgWW dgWW drWW ddWW                                         CKE   ECC
  #0     7    4    7    7                                           4    0 
  #1     7    4    7    7                                           4    0 
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    1     65536      1024          16384                    
       #1                                                                  
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    1     65536      1024          16384                    
       #1                                                                  
```

cyring, 2021-09-10:
@soonic6 Thank you. 
Intel H470 and IMC of this CML i5 10400 have been detected.

closed 2021-09-25 by cyring