cyring/CoreFreq #258: AMD Ryzen 9 3950X on MSI MEG X570 UNIFY

ich777, 2021-06-20:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

[motherboard.txt](https://github.com/cyring/CoreFreq/files/6682747/motherboard.txt)
[lscpu.txt](https://github.com/cyring/CoreFreq/files/6682748/lscpu.txt)
[lspci.txt](https://github.com/cyring/CoreFreq/files/6682749/lspci.txt)

`dmidecode -t memory`:

```# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 2.8 present.

Handle 0x0010, DMI type 16, 23 bytes
Physical Memory Array
        Location: System Board Or Motherboard
        Use: System Memory
        Error Correction Type: None
        Maximum Capacity: 128 GB
        Error Information Handle: 0x000F
        Number Of Devices: 4

Handle 0x0018, DMI type 17, 40 bytes
Memory Device
        Array Handle: 0x0010
        Error Information Handle: 0x0017
        Total Width: 64 bits
        Data Width: 64 bits
        Size: 8 GB
        Form Factor: DIMM
        Set: None
        Locator: DIMM 0
        Bank Locator: P0 CHANNEL A
        Type: DDR4
        Type Detail: Synchronous Unbuffered (Unregistered)
        Speed: 3200 MT/s
        Manufacturer: Kingston
        Serial Number: 1104E1D4
        Asset Tag: Not Specified
        Part Number: KHX3200C16D4/8GX    
        Rank: 1
        Configured Memory Speed: 3200 MT/s
        Minimum Voltage: 1.2 V
        Maximum Voltage: 1.2 V
        Configured Voltage: 1.2 V

Handle 0x001B, DMI type 17, 40 bytes
Memory Device
        Array Handle: 0x0010
        Error Information Handle: 0x001A
        Total Width: 64 bits
        Data Width: 64 bits
        Size: 8 GB
        Form Factor: DIMM
        Set: None
        Locator: DIMM 1
        Bank Locator: P0 CHANNEL A
        Type: DDR4
        Type Detail: Synchronous Unbuffered (Unregistered)
        Speed: 3200 MT/s
        Manufacturer: Kingston
        Serial Number: 120498D4
        Asset Tag: Not Specified
        Part Number: KHX3200C16D4/8GX    
        Rank: 1
        Configured Memory Speed: 3200 MT/s
        Minimum Voltage: 1.2 V
        Maximum Voltage: 1.2 V
        Configured Voltage: 1.2 V

Handle 0x001E, DMI type 17, 40 bytes
Memory Device
        Array Handle: 0x0010
        Error Information Handle: 0x001D
        Total Width: 64 bits
        Data Width: 64 bits
        Size: 8 GB
        Form Factor: DIMM
        Set: None
        Locator: DIMM 0
        Bank Locator: P0 CHANNEL B
        Type: DDR4
        Type Detail: Synchronous Unbuffered (Unregistered)
        Speed: 3200 MT/s
        Manufacturer: Kingston
        Serial Number: 10049BD4
        Asset Tag: Not Specified
        Part Number: KHX3200C16D4/8GX    
        Rank: 1
        Configured Memory Speed: 3200 MT/s
        Minimum Voltage: 1.2 V
        Maximum Voltage: 1.2 V
        Configured Voltage: 1.2 V

Handle 0x0021, DMI type 17, 40 bytes
Memory Device
        Array Handle: 0x0010
        Error Information Handle: 0x0020
        Total Width: 64 bits
        Data Width: 64 bits
        Size: 8 GB
        Form Factor: DIMM
        Set: None
        Locator: DIMM 1
        Bank Locator: P0 CHANNEL B
        Type: DDR4
        Type Detail: Synchronous Unbuffered (Unregistered)
        Speed: 3200 MT/s
        Manufacturer: Kingston
        Serial Number: 0B04E0D4
        Asset Tag: Not Specified
        Part Number: KHX3200C16D4/8GX    
        Rank: 1
        Configured Memory Speed: 3200 MT/s
        Minimum Voltage: 1.2 V
        Maximum Voltage: 1.2 V
        Configured Voltage: 1.2 V```

cyring, 2021-06-20:
3950X is my favorite Processor and my current development workstation, paired with a **dual memory kit**

This kit of four 8GB DIMM is a very instructive case as the IMC here shows different size and geometry results. 

The issue is now to fix decoding of 4 DIMMs layout.

giganode, 2021-06-20:
Thank you btw @cyring !!!

Keep up your good work!

cyring, 2021-06-22:
Hello,

To debug this 4 sticks setup, can you run my following user-space code.
1. Download
[zencli.c](https://github.com/cyring/Tips/blob/master/C/zencli.c)
2. Compile
```
cc zencli.c -o zencli
```
3. Execute UMC (as root)
```
zencli umc
```
---
```
Data Fabric: scanning UMC @ BAR[0x00050000] : 0 1 2 3 4 5 6 7 for 2 Channels

CHA[0] CHIP[0:0] @ 0x00050000[0x00000000] Disable, Rank=0
CHA[0] MASK[0:0] @ 0x00050020[0x00000000]
CHA[0] CHIP[0:1] @ 0x00050010[0x00000000] Disable, Rank=0
CHA[0] MASK[0:1] @ 0x00050028[0x00000000]
CHA[0] CHIP[1:0] @ 0x00050004[0x00000000] Disable, Rank=0
CHA[0] MASK[1:0] @ 0x00050020[0x00000000]
CHA[0] CHIP[1:1] @ 0x00050014[0x00000000] Disable, Rank=0
CHA[0] MASK[1:1] @ 0x00050028[0x00000000]
CHA[0] CHIP[2:0] @ 0x00050008[0x00000001] Enable, Rank=2
CHA[0] MASK[2:0] @ 0x00050024[0x03fffdfe] ChipSize[8388608]
CHA[0] CHIP[2:1] @ 0x00050018[0x00000000] Disable, Rank=0
CHA[0] MASK[2:1] @ 0x0005002c[0x00000000]
CHA[0] CHIP[3:0] @ 0x0005000c[0x00000201] Enable, Rank=2
CHA[0] MASK[3:0] @ 0x00050024[0x03fffdfe] ChipSize[8388608]
CHA[0] CHIP[3:1] @ 0x00050018[0x00000000] Disable, Rank=0
CHA[0] MASK[3:1] @ 0x0005002c[0x00000000]

DIMM Size[16777216 KB] [16384 MB]

CHA[1] CHIP[0:0] @ 0x00150000[0x00000000] Disable, Rank=0
CHA[1] MASK[0:0] @ 0x00150020[0x00000000]
CHA[1] CHIP[0:1] @ 0x00150010[0x00000000] Disable, Rank=0
CHA[1] MASK[0:1] @ 0x00150028[0x00000000]
CHA[1] CHIP[1:0] @ 0x00150004[0x00000000] Disable, Rank=0
CHA[1] MASK[1:0] @ 0x00150020[0x00000000]
CHA[1] CHIP[1:1] @ 0x00150014[0x00000000] Disable, Rank=0
CHA[1] MASK[1:1] @ 0x00150028[0x00000000]
CHA[1] CHIP[2:0] @ 0x00150008[0x00000001] Enable, Rank=2
CHA[1] MASK[2:0] @ 0x00150024[0x03fffdfe] ChipSize[8388608]
CHA[1] CHIP[2:1] @ 0x00150018[0x00000000] Disable, Rank=0
CHA[1] MASK[2:1] @ 0x0015002c[0x00000000]
CHA[1] CHIP[3:0] @ 0x0015000c[0x00000201] Enable, Rank=2
CHA[1] MASK[3:0] @ 0x00150024[0x03fffdfe] ChipSize[8388608]
CHA[1] CHIP[3:1] @ 0x00150018[0x00000000] Disable, Rank=0
CHA[1] MASK[3:1] @ 0x0015002c[0x00000000]

DIMM Size[16777216 KB] [16384 MB]
```
---
( _above is an example of my 3950X with two DIMMs in dual channel mode_ )

cyring, 2021-06-29:
@ich777 @giganode EVRY1

Please I need your help with the above request https://github.com/cyring/CoreFreq/issues/258#issuecomment-865681047 to debug the 4 x DIMMs issue on Zen/UMC

Thank you

ich777, 2021-06-29:
@cyring sorry for the delay.
Will compile it today evening and send it out to @giganode so that he van test it.

giganode, 2021-06-29:
Hey @cyring... 
I'm sorry, I think I sorted out the last notification mail. Thought it was spam :(

Here is the output:

zencli umc

Data Fabric: scanning UMC @ BAR[0x00050000] : 0 1 2 3 4 5 6 7 for 2 Channels

CHA[0] CHIP[0:0] @ 0x00050000[0x00000001] Enable, Rank=2
CHA[0] MASK[0:0] @ 0x00050020[0x03fffdfe] ChipSize[8388608]
CHA[0] CHIP[0:1] @ 0x00050010[0x00000000] Disable, Rank=0
CHA[0] MASK[0:1] @ 0x00050028[0x00000000]
CHA[0] CHIP[1:0] @ 0x00050004[0x00000000] Disable, Rank=2
CHA[0] MASK[1:0] @ 0x00050020[0x03fffdfe]
CHA[0] CHIP[1:1] @ 0x00050014[0x00000000] Disable, Rank=0
CHA[0] MASK[1:1] @ 0x00050028[0x00000000]
CHA[0] CHIP[2:0] @ 0x00050008[0x00000201] Enable, Rank=2
CHA[0] MASK[2:0] @ 0x00050024[0x03fffdfe] ChipSize[8388608]
CHA[0] CHIP[2:1] @ 0x00050018[0x00000000] Disable, Rank=0
CHA[0] MASK[2:1] @ 0x0005002c[0x00000000]
CHA[0] CHIP[3:0] @ 0x0005000c[0x00000000] Disable, Rank=2
CHA[0] MASK[3:0] @ 0x00050024[0x03fffdfe]
CHA[0] CHIP[3:1] @ 0x00050018[0x00000000] Disable, Rank=0
CHA[0] MASK[3:1] @ 0x0005002c[0x00000000]

DIMM Size[16777216 KB] [16384 MB]

CHA[1] CHIP[0:0] @ 0x00150000[0x00000001] Enable, Rank=2
CHA[1] MASK[0:0] @ 0x00150020[0x03fffdfe] ChipSize[8388608]
CHA[1] CHIP[0:1] @ 0x00150010[0x00000000] Disable, Rank=0
CHA[1] MASK[0:1] @ 0x00150028[0x00000000]
CHA[1] CHIP[1:0] @ 0x00150004[0x00000000] Disable, Rank=2
CHA[1] MASK[1:0] @ 0x00150020[0x03fffdfe]
CHA[1] CHIP[1:1] @ 0x00150014[0x00000000] Disable, Rank=0
CHA[1] MASK[1:1] @ 0x00150028[0x00000000]
CHA[1] CHIP[2:0] @ 0x00150008[0x00000201] Enable, Rank=2
CHA[1] MASK[2:0] @ 0x00150024[0x03fffdfe] ChipSize[8388608]
CHA[1] CHIP[2:1] @ 0x00150018[0x00000000] Disable, Rank=0
CHA[1] MASK[2:1] @ 0x0015002c[0x00000000]
CHA[1] CHIP[3:0] @ 0x0015000c[0x00000000] Disable, Rank=2
CHA[1] MASK[3:0] @ 0x00150024[0x03fffdfe]
CHA[1] CHIP[3:1] @ 0x00150018[0x00000000] Disable, Rank=0
CHA[1] MASK[3:1] @ 0x0015002c[0x00000000]

DIMM Size[16777216 KB] [16384 MB]

cyring, 2021-06-29:
@giganode Thanks a lot !
Can you tell about your DIMM setup : how much do you have of them physically ?
`dmidecode -t memory` will be fine

giganode, 2021-06-30:
@cyring no problem!!! Does the provided data help you?

cyring, 2021-06-30:
> @cyring no problem!!! Does the provided data help you?

Yes, thank you.

I'm now not sure about an obvious discriminant between the 4 DIMM and the 2 DIMM dumps !
In both cases, we have two enabled chips per DIMM

The difference with your DIMM case is the following:
```
CHA[1] CHIP[1:0] @ 0x00150004[0x00000000] Disable, Rank=2
CHA[1] MASK[1:0] @ 0x00150020[0x03fffdfe]
```
where bit position zero is clear thus chip is disabled but we however have a size of `0x03fffdfe`

I met that case with a 2700X where ignoring that zero bit lead to a wrong DIMM size.

cyring, 2021-06-30:
EDIT: I will refactor the whole algorithm. I think bits are probably CHIP/DiMM **populated**

For example: addr[val]  `0x00150008[0x00000201]` in binary `0000 0010 0000 0001`

One populated slot in each group of 8

Has Zen architecture a max of 8 DIMMs per channel ? Like EPYC.

itspngu, 2021-07-29:
Hey, saw this at random and figured I'd be useful for once and leave some data. 3950X on X570 Aorus Pro, 4x8GB Samsung B-Dies (F4-3200C15D-16GVK)

```shell_session
# zencli umc

Data Fabric: scanning UMC @ BAR[0x00050000] : 0 1 2 3 4 5 6 7 for 2 Channels

CHA[0] CHIP[0:0] @ 0x00050000[0x00000001] Enable, Rank=2
CHA[0] MASK[0:0] @ 0x00050020[0x03fffdfe] ChipSize[8388608]
CHA[0] CHIP[0:1] @ 0x00050010[0x00000000] Disable, Rank=0
CHA[0] MASK[0:1] @ 0x00050028[0x00000000]
CHA[0] CHIP[1:0] @ 0x00050004[0x00000000] Disable, Rank=2
CHA[0] MASK[1:0] @ 0x00050020[0x03fffdfe]
CHA[0] CHIP[1:1] @ 0x00050014[0x00000000] Disable, Rank=0
CHA[0] MASK[1:1] @ 0x00050028[0x00000000]
CHA[0] CHIP[2:0] @ 0x00050008[0x00000201] Enable, Rank=2
CHA[0] MASK[2:0] @ 0x00050024[0x03fffdfe] ChipSize[8388608]
CHA[0] CHIP[2:1] @ 0x00050018[0x00000000] Disable, Rank=0
CHA[0] MASK[2:1] @ 0x0005002c[0x00000000]
CHA[0] CHIP[3:0] @ 0x0005000c[0x00000000] Disable, Rank=2
CHA[0] MASK[3:0] @ 0x00050024[0x03fffdfe]
CHA[0] CHIP[3:1] @ 0x00050018[0x00000000] Disable, Rank=0
CHA[0] MASK[3:1] @ 0x0005002c[0x00000000]

DIMM Size[16777216 KB] [16384 MB]

CHA[1] CHIP[0:0] @ 0x00150000[0x00000001] Enable, Rank=2
CHA[1] MASK[0:0] @ 0x00150020[0x03fffdfe] ChipSize[8388608]
CHA[1] CHIP[0:1] @ 0x00150010[0x00000000] Disable, Rank=0
CHA[1] MASK[0:1] @ 0x00150028[0x00000000]
CHA[1] CHIP[1:0] @ 0x00150004[0x00000000] Disable, Rank=2
CHA[1] MASK[1:0] @ 0x00150020[0x03fffdfe]
CHA[1] CHIP[1:1] @ 0x00150014[0x00000000] Disable, Rank=0
CHA[1] MASK[1:1] @ 0x00150028[0x00000000]
CHA[1] CHIP[2:0] @ 0x00150008[0x00000201] Enable, Rank=2
CHA[1] MASK[2:0] @ 0x00150024[0x03fffdfe] ChipSize[8388608]
CHA[1] CHIP[2:1] @ 0x00150018[0x00000000] Disable, Rank=0
CHA[1] MASK[2:1] @ 0x0015002c[0x00000000]
CHA[1] CHIP[3:0] @ 0x0015000c[0x00000000] Disable, Rank=2
CHA[1] MASK[3:0] @ 0x00150024[0x03fffdfe]
CHA[1] CHIP[3:1] @ 0x00150018[0x00000000] Disable, Rank=0
CHA[1] MASK[3:1] @ 0x0015002c[0x00000000]

DIMM Size[16777216 KB] [16384 MB]
```

```shell_session
dmidecode -t memory
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.
# SMBIOS implementations newer than version 3.2.0 are not
# fully supported by this version of dmidecode.

Handle 0x000B, DMI type 16, 23 bytes
Physical Memory Array
	Location: System Board Or Motherboard
	Use: System Memory
	Error Correction Type: None
	Maximum Capacity: 128 GB
	Error Information Handle: 0x000A
	Number Of Devices: 4

Handle 0x0013, DMI type 17, 92 bytes
Memory Device
	Array Handle: 0x000B
	Error Information Handle: 0x0012
	Total Width: 64 bits
	Data Width: 64 bits
	Size: 8 GB
[... 108 lines cut ...]
	Bank Locator: P0 CHANNEL B
	Type: DDR4
	Type Detail: Synchronous Unbuffered (Unregistered)
	Speed: 3200 MT/s
	Manufacturer: Unknown
	Serial Number: 00000000
	Asset Tag: Not Specified
	Part Number: F4-3200C15-8GVK
	Rank: 1
	Configured Memory Speed: 3200 MT/s
	Minimum Voltage: 1.2 V
	Maximum Voltage: 1.2 V
	Configured Voltage: 1.2 V
	Memory Technology: DRAM
	Memory Operating Mode Capability: Volatile memory
	Firmware Version: Unknown
	Module Manufacturer ID: Bank 5, Hex 0xCD
	Module Product ID: Unknown
	Memory Subsystem Controller Manufacturer ID: Unknown
	Memory Subsystem Controller Product ID: Unknown
	Non-Volatile Size: None
	Volatile Size: 8 GB
	Cache Size: None
	Logical Size: None
```

cyring, 2021-07-29:
@itspngu Thanks.
What do you get from CoreFreq IMC output ?

cyring, 2021-09-06:
Released with question above.

closed 2021-09-06 by cyring