cyring/CoreFreq #259: Pentium J3710 on ASRock J3710-ITX

ich777, 2021-06-20:
[image]
[image]
[image]
[image]
[image]
[image]

[lscpu.txt](https://github.com/cyring/CoreFreq/files/6682990/lscpu.txt)
[lspci.txt](https://github.com/cyring/CoreFreq/files/6682991/lspci.txt)
[motherboard.txt](https://github.com/cyring/CoreFreq/files/6682992/motherboard.txt)

`dmidecode -t memory`:

```# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 2.8 present.

Handle 0x000B, DMI type 16, 23 bytes
Physical Memory Array
        Location: System Board Or Motherboard
        Use: System Memory
        Error Correction Type: None
        Maximum Capacity: 8 GB
        Error Information Handle: Not Provided
        Number Of Devices: 2

Handle 0x000D, DMI type 17, 40 bytes
Memory Device
        Array Handle: 0x000B
        Error Information Handle: Not Provided
        Total Width: 64 bits
        Data Width: 64 bits
        Size: 8 GB
        Form Factor: DIMM
        Set: None
        Locator: A1_DIMM0
        Bank Locator: A1_BANK0
        Type: DDR3
        Type Detail: Unknown
        Speed: 1600 MT/s
        Manufacturer: Kingston        
        Serial Number: 13062223  
        Asset Tag: A1_AssetTagNum0
        Part Number: 99U5428-018.A00LF 
        Rank: 2
        Configured Memory Speed: 1600 MT/s
        Minimum Voltage: 1.35 V
        Maximum Voltage: 1.5 V
        Configured Voltage: 1.35 V

Handle 0x000F, DMI type 17, 40 bytes
Memory Device
        Array Handle: 0x000B
        Error Information Handle: Not Provided
        Total Width: Unknown
        Data Width: 64 bits
        Size: No Module Installed
        Form Factor: DIMM
        Set: None
        Locator: A1_DIMM1
        Bank Locator: A1_BANK1
        Type: Unknown
        Type Detail: Unknown
        Speed: Unknown
        Manufacturer: A1_Manufacturer1
        Serial Number: A1_SerNum1
        Asset Tag: A1_AssetTagNum1
        Part Number: Array1_PartNumber1
        Rank: Unknown
        Configured Memory Speed: Unknown
        Minimum Voltage: Unknown
        Maximum Voltage: Unknown
        Configured Voltage: Unknown```

cyring, 2021-06-20:
Thank you.

CoreFreq enhancements here are:
* IMC decoder
* Uncore counter
* Turbo boost not capable [ark](https://ark.intel.com/content/www/us/en/ark/products/91532/intel-pentium-processor-j3710-2m-cache-up-to-2-64-ghz.html)

cyring, 2021-06-22:
# Vcore
* Stressing a single or all Core(s), voltage should fall within this range:
[image]
[image]
* CoreFreq integrated stress tool can help to verify if its formula is well calibrated ?
  If not, please let me know about your lowest and highest values.
# Power and TjMax
* TDP is expected to be at most `6W` and Thermal junction at `90°C`
[image]
* Values should match in the window "Power, Current & Thermal" ( or press shortcut [`w`] )

cyring, 2021-07-23:
Changes have been brought regarding https://github.com/cyring/CoreFreq/issues/259#issuecomment-864608848
Feel free to come back when you have a chance to try the latest development version.

closed 2021-07-23 by cyring