cyring/CoreFreq #260: AMD Athlon 200GE on Gigabyte AX370M-DS3H-CF

ich777, 2021-06-21:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

[lspci.txt](https://github.com/cyring/CoreFreq/files/6687665/lspci.txt)
[motherboard.txt](https://github.com/cyring/CoreFreq/files/6687667/motherboard.txt)
[lscpu.txt](https://github.com/cyring/CoreFreq/files/6687666/lscpu.txt)

`dmidecode -t memory`:
```# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.2.1 present.

Handle 0x0009, DMI type 16, 23 bytes
Physical Memory Array
        Location: System Board Or Motherboard
        Use: System Memory
        Error Correction Type: None
        Maximum Capacity: 256 GB
        Error Information Handle: 0x0008
        Number Of Devices: 4

Handle 0x0010, DMI type 17, 40 bytes
Memory Device
        Array Handle: 0x0009
        Error Information Handle: 0x000F
        Total Width: Unknown
        Data Width: Unknown
        Size: No Module Installed
        Form Factor: Unknown
        Set: None
        Locator: DIMM 0
        Bank Locator: P0 CHANNEL A
        Type: Unknown
        Type Detail: Unknown
        Speed: Unknown
        Manufacturer: Unknown
        Serial Number: Unknown
        Asset Tag: Not Specified
        Part Number: Unknown
        Rank: Unknown
        Configured Memory Speed: Unknown
        Minimum Voltage: Unknown
        Maximum Voltage: Unknown
        Configured Voltage: Unknown

Handle 0x0012, DMI type 17, 40 bytes
Memory Device
        Array Handle: 0x0009
        Error Information Handle: 0x0011
        Total Width: 64 bits
        Data Width: 64 bits
        Size: 8 GB
        Form Factor: DIMM
        Set: None
        Locator: DIMM 1
        Bank Locator: P0 CHANNEL A
        Type: DDR4
        Type Detail: Synchronous Unbuffered (Unregistered)
        Speed: 2666 MT/s
        Manufacturer: Unknown
        Serial Number: 00000000
        Asset Tag: Not Specified
        Part Number: CMK16GX4M2A2666C16  
        Rank: 1
        Configured Memory Speed: 2666 MT/s
        Minimum Voltage: 1.2 V
        Maximum Voltage: 1.2 V
        Configured Voltage: 1.2 V

Handle 0x0015, DMI type 17, 40 bytes
Memory Device
        Array Handle: 0x0009
        Error Information Handle: 0x0014
        Total Width: Unknown
        Data Width: Unknown
        Size: No Module Installed
        Form Factor: Unknown
        Set: None
        Locator: DIMM 0
        Bank Locator: P0 CHANNEL B
        Type: Unknown
        Type Detail: Unknown
        Speed: Unknown
        Manufacturer: Unknown
        Serial Number: Unknown
        Asset Tag: Not Specified
        Part Number: Unknown
        Rank: Unknown
        Configured Memory Speed: Unknown
        Minimum Voltage: Unknown
        Maximum Voltage: Unknown
        Configured Voltage: Unknown

Handle 0x0017, DMI type 17, 40 bytes
Memory Device
        Array Handle: 0x0009
        Error Information Handle: 0x0016
        Total Width: 64 bits
        Data Width: 64 bits
        Size: 8 GB
        Form Factor: DIMM
        Set: None
        Locator: DIMM 1
        Bank Locator: P0 CHANNEL B
        Type: DDR4
        Type Detail: Synchronous Unbuffered (Unregistered)
        Speed: 2666 MT/s
        Manufacturer: Unknown
        Serial Number: 00000000
        Asset Tag: Not Specified
        Part Number: CMK16GX4M2A2666C16  
        Rank: 1
        Configured Memory Speed: 2666 MT/s
        Minimum Voltage: 1.2 V
        Maximum Voltage: 1.2 V
        Configured Voltage: 1.2 V```

cyring, 2021-06-21:
Nice.

About the instructions counter `IRPerf`, we met similar issue #206 with Zen/Whitehaven that it's part of Processor erratas.

cyring, 2021-06-22:
Hello,

* Please, no other driver should be run in parallel of CoreFreq ([Software incompatibilities](https://github.com/cyring/CoreFreq/wiki/Software-incompatibilities-and-workarounds#k10temp-and-zenpower-kernel-modules))
```
00:18.3 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Device 24: Function 3 [1022:15eb]
	Kernel driver in use: k10temp
	Kernel modules: k10temp
```
* Unload k10temp or add it in the kernel boot blacklist modules list
```
modprobe.blacklist=k10temp
```

cyring, 2021-07-23:
Instructions counter is an errata of AMD Whitehaven.
Unfortunately anything else I can do.

closed 2021-07-23 by cyring