cyring/CoreFreq #265: AMD PRO A6-8570 R5, 8 COMPUTE CORES 2C+6G

cyring, 2021-06-26:
@svmlegacy
Very instructive, especially the enumeration of P-states.
**Thank you**. 

I can add temperature and other remaining features. Are you ok for testings ?

CyrIng

svmlegacy, 2021-06-26:
@cyring This machine is currently on my test bed, so I can test any changes immediately.

svmlegacy, 2021-06-26:
[AMD PRO A6-8570 lspci -nn](https://gist.github.com/svmlegacy/b1fbd46164393ffb69ac6a4e668ae5d8)

cyring, 2021-06-27:
Hi,

Can you pull and try latest `develop` branch which contains this commit aee2a52be60f15a427b36c96709f5bb12258f739 for temperature sensor.

Please don't let any other SMU drivers run in parallel. Unload `k10temp`

CyrIng

svmlegacy, 2021-06-27:
[Looks like we have a temp sensor!](https://gist.github.com/svmlegacy/26c0b9a3e1fe3739c082412c50cc8815)

This is Tctl, correct? Not actual package temp? (Though 19 would be pretty close.)

cyring, 2021-06-27:
> [Looks like we have a temp sensor!](https://gist.github.com/svmlegacy/26c0b9a3e1fe3739c082412c50cc8815)
> 
> This is Tctl, correct? Not actual package temp? (Though 19 would be pretty close.)

Great!

Can't really tell from here, but stressing Core(s) may give us a hint about the max temp is about  what ? die or package ?

Value is coming from SMU and I don't see how to specify a query per CPU. I'm relying on your observations to define what kind of sensor it is.

By the end if we just have one sensor then the temperature default scope will be set to "Package" in source code.

svmlegacy, 2021-06-27:
> Great!
>
> Can't really tell from here, but stressing Core(s) may give us a hint about the max temp is about what ? die or package ?
>
> Value is coming from SMU and I don't see how to specify a query per [...]

AMD's Family 15h has a rather strange temperature sensor in my experience. The sensor here actually is one of the most sensible values I've seen from it. This is a pretty low power chip, but I'm seeing roughly ambient temperature at idle, and ~ 40 C max temp under Prime95 loading. Locking up the fan, I'm seeing throttle at 94 C, again a very sensible value.

I don't believe Family 15h has a sensor for each individual core at any rate. I've never seen a tool report that.

cyring, 2021-06-27:
So I will show as a Package scope in next UI commit.

cyring, 2021-06-27:
I had programmed the `C1E` state query and the ability to alter it but, because based on K8 register, I left it as Experimental:

https://github.com/cyring/CoreFreq/blob/dbc6ebcfad9cfad11047da17e775806c70a41470/corefreqk.c#L9623

From the UI could you enable the `Settings > Experimental` mode (or using the driver start-up option of same name)

Next toggle ON and OFF the C1E from the `Performance Monitoring` window.

Thank you.

svmlegacy, 2021-06-27:
[image]
C1E does not seem to be able to toggle "ON" for this system. This is the latest develop build.

cyring, 2021-06-27:
> [image]
> C1E does not seem to be able to toggle "ON" for this system. This is the latest develop build.

Thanks for trying.

cyring, 2021-06-28:
Hello,

Latest commits in `develop` bring the following features for all AMD families:
1. the Watchdog state which can be toggle ON/OFF (in `Technologies` window)
2. the I/O C-State Base Address (in `Performance Monitoring` window)
  when you register _CoreFreq_ as the kernel idle handler to do I/O WAIT

The default temperature scope is now `Package`

cyring, 2021-09-06:
Released.

closed 2021-09-06 by cyring