cyring/CoreFreq #274: 2x Intel E5-2620 v3

jorgelaranjo, 2021-09-03:
Not sure if related to #264 but getting this error:

```
corefreq-cli 
Daemon connection error code 1
Version 1.87.1: 'Operation not permitted' @ line 18070
```

System: Dual Xeon E5-2620 v3 144GB RAM, ArchLinux kernel 5.13.13-arch-1

```
$uname -r
5.13.13-arch1-1
```

`$ lscpu`
```
Architecture: x86_64
CPU op-mode(s): 32-bit, 64-bit
Address sizes: 46 bits physical, 48 bits virtual
Byte Order: Little Endian
CPU(s): 24
On-line CPU(s) list: 0-23
Vendor ID: GenuineIntel
Model name: Intel(R) Xeon(R) CPU E5-2620 v3 @ 2.40GHz
CPU family: 6
Model: 63
Thread(s) per core: 2
Core(s) per socket: 6
Socket(s): 2
Stepping: 2
CPU max MHz: 2400.0000
CPU min MHz: 1200.0000
BogoMIPS: 4790.27
Flags: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid dca sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm cpuid_fault epb invpcid_single pti intel_ppin ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid cqm xsaveopt cqm_llc cqm_occup_llc dtherm arat pln pts md_clear flush_l1d
Virtualization: VT-x
L1d cache: 384 KiB (12 instances)
L1i cache: 384 KiB (12 instances)
L2 cache: 3 MiB (12 instances)
L3 cache: 30 MiB (2 instances)
NUMA node(s): 2
NUMA node0 CPU(s): 0-5,12-17
NUMA node1 CPU(s): 6-11,18-23
Vulnerability Itlb multihit: KVM: Mitigation: Split huge pages
Vulnerability L1tf: Mitigation; PTE Inversion; VMX conditional cache flushes, SMT vulnerable
Vulnerability Mds: Mitigation; Clear CPU buffers; SMT vulnerable
Vulnerability Meltdown: Mitigation; PTI
Vulnerability Spec store bypass: Mitigation; Speculative Store Bypass disabled via prctl and seccomp
Vulnerability Spectre v1: Mitigation; usercopy/swapgs barriers and __user pointer sanitization
Vulnerability Spectre v2: Mitigation; Full generic retpoline, IBPB conditional, IBRS_FW, STIBP conditional, RSB filling
Vulnerability Srbds: Not affected
Vulnerability Tsx async abort: Not affected
```
```
systemctl status corefreqd 
* corefreqd.service - CoreFreq Daemon
     Loaded: loaded (/usr/lib/systemd/system/corefreqd.service; disabled; vendor preset: disabled)
     Active: active (running) since Fri 2021-09-03 16:15:20 BST; 12min ago
   Main PID: 243999 (corefreqd-pmgr)
      Tasks: 52 (limit: 173910)
     Memory: 2.8M
        CPU: 2.205s
     CGroup: /system.slice/corefreqd.service
             |-243999 corefreqd -q
             `-244000 corefreqd -q

Sep 03 16:15:20 workstation systemd[1]: Started CoreFreq Daemon.
```

cyring, 2021-09-03:
Hello.
Do you have crash or running trace of ? 
1. Driver `corefreqk.ko` in kernel log
2. Daemon in debug mode `corefreqd -d`

cyring, 2021-09-03:
CLI stopped at a wrong version:
https://github.com/cyring/CoreFreq/blob/8006b76630235fe1ffa3d78aaf171fd9598c228c/corefreq-cli.c#L18070

Do you confirm CoreFreq source code is not mixed up ?
Like a previous CoreFreq Daemon or Driver is still resident in memory ?
I suggest to fully unload them; remove and install package again.

jorgelaranjo, 2021-09-04:
1.87 and 1.86 from AUR were in conflict. 
Removed and installed version 1.86 for kernel 5.13.13 and it works now.
[image]

closed 2021-09-04 by jorgelaranjo

cyring, 2021-09-04:
> 1.87 and 1.86 from AUR were in conflict.
> Removed and installed version 1.86 for kernel 5.13.13 and it works now.
> [image]

Not sure to understand : is the screenshot showing 1.86 or 1.87 ?

jorgelaranjo, 2021-09-05:
```
$corefreq-cli -v
1.87
```