cyring/CoreFreq #275: Missing temperature data in Cezanne

PJVol, 2021-09-13:
Here is -s log
vol@vol-H81M-S2VP:~/ryzen_smu/CoreFreq$ cat corefreq-s.log 
Processor                               [AMD Ryzen 7 5700G with Radeon Graphics]
|- Architecture                                                       [AMD Zen3]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0a50000c]
|- Signature                                                           [  AF_50]
|- Stepping                                                            [      0]
|- Online CPU                                                          [ 16/ 16]
|- Base Clock                                                          [ 99.314]
|- Frequency            (MHz)                      Ratio                        
                 Max   3773.93                    <  38 >                       
|- Factory                                                             [100.000]
                       3800                       [  38 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   1688.02                    <  17 >                       
|- Turbo Boost                                                         [   LOCK]
                 XFR   3773.93                    [  38 ]                       
                 CPB   3773.93                    [  38 ]                       
                  1C   1688.34                    <  17 >                       
                  2C   1390.40                    <  14 >                       
|- Uncore                                                              [   LOCK]
|- TDP                                                           Level [  0:0  ]
   |- Programmable                                                     [   LOCK]
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N] 
|- AVX512-VBMI  [N] AVX512-VBMI2 [N]  AVX512-VNMI [N]  AVX512-ALG [N] 
|- AVX512-VPOP  [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [N]  BMI1/BMI2 [Y/Y]         CLWB [Y] CLFLUSH/O [Y/Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [N]        UMIP [N] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Capable]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast-String Operation                                Fast-Strings   [Capable]
|- Fused Multiply Add                                     FMA | FMA4   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Instruction Based Sampling                                    IBS   [Capable]
|- Instruction INVLPGB                                       INVLPGB   [Missing]
|- Long Mode 64 bits                                       IA64 | LM   [Capable]
|- LightWeight Profiling                                         LWP   [Missing]
|- Memory Bandwidth Enforcement                                  MBE   [Capable]
|- Machine-Check Architecture                                    MCA   [Capable]
|- Instruction MCOMMIT                                       MCOMMIT   [Missing]
|- Memory Protection Extensions                                  MPX   [Missing]
|- Model Specific Registers                                      MSR   [Capable]
|- Memory Type Range Registers                                  MTRR   [Capable]
|- No-Execute Page Protection                                     NX   [Capable]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Capable]
|- Physical Address Extension                                    PAE   [Capable]
[... 79 lines cut ...]
|- Package Thermal Management                                    PTM   [Missing]
|- Thermal Monitor 1                                             TTP   [Capable]
|- Thermal Monitor 2                                             HTC   [Capable]
|- Thermal Design Power                                          TDP   [Missing]
   |- Minimum Power                                              Min   [Missing]
   |- Maximum Power                                              Max   [  128 W]
|- Thermal Design Power                                      Package   [Disable]
   |- Power Limit                                                PL1   [Missing]
   |- Power Limit                                                PL2   [Missing]
|- Thermal Design Power                                         Core   [Disable]
   |- Power Limit                                                PL1   [Missing]
|- Thermal Design Power                                       Uncore   [Disable]
   |- Power Limit                                                PL1   [Missing]
|- Thermal Design Power                                         DRAM   [Disable]
   |- Power Limit                                                PL1   [Missing]
|- Thermal Design Power                                     Platform   [Disable]
   |- Power Limit                                                PL1   [Missing]
   |- Power Limit                                                PL2   [Missing]
|- Package Power Tracking                                        PPT   [Missing]
|- Electrical Design Current                                     EDC   [Missing]
|- Thermal Design Current                                        TDC   [Missing]
|- Units                                                                        
   |- Power                                               watt   [  0.125000000]
   |- Energy                                             joule   [  0.000015259]
   |- Window                                            second   [  0.000976562]

cyring, 2021-09-13:
Okay, first thing to change is the CPUID signature in file:
https://github.com/cyring/CoreFreq/blob/ab278fc9fb01924397c7f19657fdf0772880dbed/corefreqk.h#L1490

with this:
```
#define _AMD_Zen3_CZN	{.ExtFamily=0xa, .Family=0xF, .ExtModel=0x5, .Model=0x0}
```

Next rebuild and restart all.

PJVol, 2021-09-13:
Rebuit, but temp still missing, 0, zero. How to attach pictures here? Is it possible without 3rd party image hosting?

cyring, 2021-09-13:
> Rebuit, but temp still missing, 0, zero. How to attach pictures here? Is it possible without 3rd party image hosting?
> Added picture in a forum where users learn some AI (for free) what the difference between truck and plane :)

* You can now switch to the `develop` branch where the CPUID fix is provided

* I have checked Temperature code which should be processed the same as Vermeer (family 19h)
  However it may have change for previous Zen2 register or a **new** register. 
  So let's try the followings in the `develop` branch.
1. In function `Query_AMD_Family_17h` at line:
https://github.com/cyring/CoreFreq/blob/3b70181da8f00a202901b2e434869d2bff669c53/corefreqk.c#L6291
1. Comment the `case AMD_Zen3_CZN:` ( which will route Cesanne to the default case )
```
/*	case AMD_Zen3_CZN:	*/
```
3. Rebuild & reload all

* To add a picture, click in the bottom line [`Attach files by dragging & dropping`]
  Next browse to your picture file and click `Previous` to test.
  Same for any source code and command outputs, just surround them with triple anti-quote

PJVol, 2021-09-14:
[image]
Temp is here, but is it Tctl? Who is SMBus Master 0, Nuvoton? Who can request temp via SB-TSI ? Does it poll amd's SBI?

cyring, 2021-09-14:
> [image]
> Temp is here, but is it Tctl? Who is SMBus Master 0, Nuvoton? Who can request temp via SB-TSI ? Does it poll amd's SBI?

I have to back in AMD PPR specs to confirm you if SB-TSI or not ?

But my code is getting sensor value through an address register of the SMU.

Are you finally getting temperature with or without changing code in `corefreqk.c` ?

PJVol, 2021-09-14:
After commenting out Cezanne "case"

cyring, 2021-09-14:
> After commenting out Cezanne "case"

Thanks, I will reflect this in the next commit

closed 2021-09-14 by PJVol

cyring, 2021-09-14:
Based on latest `develop` branch, could you now change this line:
https://github.com/cyring/CoreFreq/blob/ee14fd3ba1a95c8bcdbe67a3953670a4e0f5b215/corefreqk.c#L6316
with:
```
PUBLIC(RO(Proc))->Features.HSMP_Capable = 1;
```

This will try to read bunch of SMU registers like the Power Limits `PL1` `PL2` and `PPT` `EDC` `TDC` like my Matisse below:

[image]

cyring, 2021-09-15:
> [image]
> Temp is here, but is it Tctl? Who is SMBus Master 0, Nuvoton? Who can request temp via SB-TSI ? Does it poll amd's SBI?

What is specify about offset register `0x00059800`
https://github.com/cyring/CoreFreq/blob/c8f2a0e4215ecb5c27eea5d459422e43898e0178/amdmsr.h#L133

> Provides the current control temperature (T ctl) after the slew-rate controls have been applied.
> CUR_TEMP. Reset: 000h. Provides current control temperature.

Have a look into the AMD  Processor Programming
Reference
Repo link in CoreFreq wiki