cyring/CoreFreq #276: Crash after calling corefreq-cli

ich777, 2021-09-13:
Hi, got a report about a crash after issuing the command `corefreq-cli`.
The system is a Intel Xeon E5-2630L running on a Supermicro - X9DRL-3F/iF maybe you can tell what caused this.
I link the forum entry with more information here too: [Click](https://forums.unraid.net/topic/109314-plugin-corefreq/?do=findComment&comment=1034326)

Let me know if you need more information!

cyring, 2021-09-13:
If the PMU version is greater than 3 then the Uncore is programmable

For **testings** a fix is available in branch `develop` of `CoreFreq v1.87.3`

cyring, 2021-09-15:
Please show me if fix is solving the crash ?

Observe that SNB/EP Xeon(s) are therefore losing the Uncore  counter.

ich777, 2021-09-15:
The user doesn't responded yet on the forums, will keep you updated...

cyring, 2021-09-25:
OK, fix is waiting in development branch

closed 2021-09-25 by cyring