cyring/CoreFreq #277: DKMS module build fail on tk-pds kernel

puskapet, 2021-09-14:
Hi, DKMS module works for 5.14.3-arch-1-1 but fails for 5.14.3-198-tkg-pds (patched gaming kernel).
I build the kernel with everything default in the MAKEPKG except the `zen` optimization option.

I have Ryzen 5 2600.

He is the log from paru 

> `:: Running post-transaction hooks...`
> `(1/3) Reloading system manager configuration...`
> `(2/3) Arming ConditionNeedsUpdate...`
> `(3/3) Install DKMS modules`
> `==> dkms install --no-depmod -m corefreq -v 1.87.2.r4.g4220eac -k 5.14.3-arch1-1`
> `==> dkms install --no-depmod -m corefreq -v 1.87.2.r4.g4220eac -k 5.14.3-198-tkg-pds`
> `Error! Bad return status for module build on kernel: 5.14.3-198-tkg-pds (x86_64)`
> `Consult /var/lib/dkms/corefreq/1.87.2.r4.g4220eac/build/make.log for more information.`
> `==> Warning, "dkms install --no-depmod -m corefreq -v 1.87.2.r4.g4220eac -k 5.14.3-198-tkg-pds" returned 10`
> `==> depmod 5.14.3-198-tkg-pds`
> `==> depmod 5.14.3-arch1-1`

and here is _/var/lib/dkms/corefreq/1.87.2.r4.g4220eac/build/make.log_

> DKMS make.log for corefreq-1.87.2.r4.g4220eac for kernel 5.14.3-198-tkg-pds (x86_64)
> Ut 14. september 2021, 12:22:12 CEST
> make: Entering directory '/usr/src/corefreq-1.87.2.r4.g4220eac'
> cc  -Wall -Wfatal-errors corefreqd.c corefreqm.c \
>   -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
>   -o corefreqd -lpthread -lm -lrt
> cc  -Wall -Wfatal-errors \
>   corefreq-cli.c corefreq-ui.c corefreq-cli-rsc.c \
>   corefreq-cli-json.c corefreq-cli-extra.c \
>   -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0  \
>   -o corefreq-cli -lm -lrt
> make -j1 -C /usr/lib/modules/5.14.3-198-tkg-pds/build M=/var/lib/dkms/corefreq/1.87.2.r4.g4220eac/build modules
> make[1]: Entering directory '/usr/src/corefreq-1.87.2.r4.g4220eac'
> make[1]: warning: -j1 forced in submake: resetting jobserver mode.
>   CC [M]  /var/lib/dkms/corefreq/1.87.2.r4.g4220eac/build/corefreqk.o
> /var/lib/dkms/corefreq/1.87.2.r4.g4220eac/build/corefreqk.c: In function ‘Sys_DumpTask’:
> /var/lib/dkms/corefreq/1.87.2.r4.g4220eac/build/corefreqk.c:10012:71: error: ‘struct task_struct’ has no member named ‘wake_cpu’; did you mean ‘wake_q’?
> 10012 |                 SysGate->taskList[cnt].wake_cpu = (short int) thread->wake_cpu;
>       |                                                                       ^~~~~~~~
>       |                                                                       wake_q
> compilation terminated due to -Wfatal-errors.
> make[2]: *** [scripts/Makefile.build:271: /var/lib/dkms/corefreq/1.87.2.r4.g4220eac/build/corefreqk.o] Error 1
> make[1]: *** [Makefile:1851: /var/lib/dkms/corefreq/1.87.2.r4.g4220eac/build] Error 2
> make[1]: Leaving directory '/usr/lib/modules/5.14.3-198-tkg-pds/build'
> make: *** [Makefile:76: all] Error 2
> make: Leaving directory '/usr/src/corefreq-1.87.2.r4.g4220eac'

cyring, 2021-09-14:
Hello

It looks like the same build we encountered with `CONFIG_SCHED_BMQ`

https://github.com/cyring/CoreFreq/blob/ab278fc9fb01924397c7f19657fdf0772880dbed/corefreqk.c#L10014

1. You have to find, among your kernel `CONFIG_*`, one directive which is specific to `tkg` kernel build
2. Add this directive to the condition statement like:
```
#if defined(CONFIG_SCHED_BMQ) \
 || defined(CONFIG_YOUR_TKG)
```
3. Rebuild and reload all
4. And let me know about the name of this directive I will provide in next version

puskapet, 2021-09-14:
okay I added 
` || defined(CONFIG_SCHED_PDS)`
and rebuilt without errors and now it works.
[image]
What is interesting, this issue started from 5.14, previously everything was working on 5.13 and earlier.

closed 2021-09-14 by puskapet

cyring, 2021-09-14:
> okay I added
> ` || defined(CONFIG_SCHED_PDS)`
> and rebuilt without errors and now it works.
> [image]
> What is interesting, this issue started from 5.14, previously everything was working on 5.13 and earlier.

Great! I'll make use of `CONFIG_SCHED_PDS`

Thank you for this.