cyring/CoreFreq #279: build on AMD Rome w/CentOS8 4.18.0-240

fleers, 2021-09-26:
# make
cc  -Wall -Wfatal-errors -pthread -c corefreqd.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreqd.o
cc  -Wall -Wfatal-errors -c corefreqm.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreqm.o
cc  -Wall -Wfatal-errors corefreqd.c corefreqm.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreqd -lpthread -lm -lrt
cc  -Wall -Wfatal-errors -c corefreq-cli.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0  \
  -o corefreq-cli.o
cc  -Wall -Wfatal-errors -c corefreq-ui.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreq-ui.o
cc  -Wall -Wfatal-errors -c corefreq-cli-rsc.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0  \
  -o corefreq-cli-rsc.o
cc  -Wall -Wfatal-errors -c corefreq-cli-json.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreq-cli-json.o
cc  -Wall -Wfatal-errors -c corefreq-cli-extra.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreq-cli-extra.o
cc  -Wall -Wfatal-errors \
  corefreq-cli.c corefreq-ui.c corefreq-cli-rsc.c \
  corefreq-cli-json.c corefreq-cli-extra.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0  \
  -o corefreq-cli -lm -lrt
make -j1 -C /lib/modules/4.18.0-240.el8.x86_64/build M=/tmp/CoreFreq modules
make[1]: Entering directory '/usr/src/kernels/4.18.0-240.el8.x86_64'
  CC [M]  /tmp/CoreFreq/corefreqk.o
/tmp/CoreFreq/corefreqk.c:358:26: error: initialization of ‘int (*)(struct cpufreq_policy_data *)’ from incompatible pointer type ‘int (*)(struct cpufreq_policy *)’ [-Werror=incompatible-pointer-types]
  /*MANDATORY*/ .verify = CoreFreqK_Policy_Verify,
                          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:322: /tmp/CoreFreq/corefreqk.o] Error 1
make[1]: *** [Makefile:1544: _module_/tmp/CoreFreq] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-240.el8.x86_64'
make: *** [Makefile:76: all] Error 2

# cat /etc/redhat-release
CentOS Linux release 8.3.2011

# uname -a
Linux gb 4.18.0-240.el8.x86_64 #1 SMP Fri Sep 25 19:48:47 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

# lscpu
Architecture:        x86_64
CPU op-mode(s):      32-bit, 64-bit
Byte Order:          Little Endian
CPU(s):              128
On-line CPU(s) list: 0-127
Thread(s) per core:  1
Core(s) per socket:  64
Socket(s):           2
NUMA node(s):        32
Vendor ID:           AuthenticAMD
CPU family:          23
Model:               49
Model name:          AMD EPYC 7742 64-Core Processor
Stepping:            0
CPU MHz:             3072.430
CPU max MHz:         2250.0000
CPU min MHz:         1500.0000
BogoMIPS:            4500.17
Virtualization:      AMD-V
L1d cache:           32K
L1i cache:           32K
L2 cache:            512K
L3 cache:            16384K
NUMA node0 CPU(s):   0-3
NUMA node1 CPU(s):   4-7
NUMA node2 CPU(s):   8-11
NUMA node3 CPU(s):   12-15
NUMA node4 CPU(s):   16-19
NUMA node5 CPU(s):   20-23
NUMA node6 CPU(s):   24-27
NUMA node7 CPU(s):   28-31
NUMA node8 CPU(s):   32-35
NUMA node9 CPU(s):   36-39
NUMA node10 CPU(s):  40-43
NUMA node11 CPU(s):  44-47
NUMA node12 CPU(s):  48-51
NUMA node13 CPU(s):  52-55
NUMA node14 CPU(s):  56-59
NUMA node15 CPU(s):  60-63
NUMA node16 CPU(s):  64-67
NUMA node17 CPU(s):  68-71
NUMA node18 CPU(s):  72-75
NUMA node19 CPU(s):  76-79
NUMA node20 CPU(s):  80-83
NUMA node21 CPU(s):  84-87
NUMA node22 CPU(s):  88-91
NUMA node23 CPU(s):  92-95
NUMA node24 CPU(s):  96-99
NUMA node25 CPU(s):  100-103
NUMA node26 CPU(s):  104-107
NUMA node27 CPU(s):  108-111
NUMA node28 CPU(s):  112-115
NUMA node29 CPU(s):  116-119
NUMA node30 CPU(s):  120-123
NUMA node31 CPU(s):  124-127
Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 x2apic movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs skinit wdt tce topoext perfctr_core perfctr_nb bpext perfctr_llc mwaitx cpb cat_l3 cdp_l3 hw_pstate sme ssbd mba sev ibrs ibpb stibp vmmcall fsgsbase bmi1 avx2 smep bmi2 cqm rdt_a rdseed adx smap clflushopt clwb sha_ni xsaveopt xsavec xgetbv1 xsaves cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local clzero irperf xsaveerptr wbnoinvd arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_vmsave_vmload vgif umip rdpid overflow_recov succor smca

cyring, 2021-09-26:
Hello,

Can you print the following please:
```
zgrep CONFIG_CPU_FREQ /proc/config.gz
```

cyring, 2021-09-26:
Forget the above request. I have virtualized CentOS 8 and encountered the same issue.

A fix is available in the [development branch](https://github.com/cyring/CoreFreq/tree/develop)
Here is the result into the VM

[image]

When you will fetch the `develop` branch, be sure :
* not to mix source code with any previous downloads
* CoreFreq' Daemon and Driver are fully unload from memory before starting up again

This fix is part of CoreFreq version `1.87.4`

Waiting for your confirmation before merging fix into `master`

fleers, 2021-09-26:
Thanks.  The compilation error has changed to this:

```
# make
cc  -Wall -Wfatal-errors -pthread -c corefreqd.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreqd.o
cc  -Wall -Wfatal-errors -c corefreqm.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreqm.o
cc  -Wall -Wfatal-errors corefreqd.c corefreqm.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreqd -lpthread -lm -lrt
cc  -Wall -Wfatal-errors -c corefreq-cli.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0  \
  -o corefreq-cli.o
cc  -Wall -Wfatal-errors -c corefreq-ui.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreq-ui.o
cc  -Wall -Wfatal-errors -c corefreq-cli-rsc.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0  \
  -o corefreq-cli-rsc.o
cc  -Wall -Wfatal-errors -c corefreq-cli-json.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreq-cli-json.o
cc  -Wall -Wfatal-errors -c corefreq-cli-extra.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreq-cli-extra.o
cc  -Wall -Wfatal-errors \
  corefreq-cli.c corefreq-ui.c corefreq-cli-rsc.c \
  corefreq-cli-json.c corefreq-cli-extra.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0  \
  -o corefreq-cli -lm -lrt
make -j1 -C /lib/modules/4.18.0-240.el8.x86_64/build M=/tmp/CoreFreq modules
make[1]: Entering directory '/usr/src/kernels/4.18.0-240.el8.x86_64'
  CC [M]  /tmp/CoreFreq/corefreqk.o
/tmp/CoreFreq/corefreqk.c:361:17: error: initialization of ‘int (*)(int)’ from incompatible pointer type ‘int (*)(struct cpufreq_policy *, int)’ [-Werror=incompatible-pointer-types]
    .set_boost = CoreFreqK_SetBoost
                 ^~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:322: /tmp/CoreFreq/corefreqk.o] Error 1
make[1]: *** [Makefile:1544: _module_/tmp/CoreFreq] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-240.el8.x86_64'
make: *** [Makefile:76: all] Error 2
```

cyring, 2021-09-26:
Can you replace the following lines with the code changed for each;

https://github.com/cyring/CoreFreq/blob/6e1d7d414017a24cce69edb24c5586324d0c1fa4/corefreqk.c#L15624
```
#if LINUX_VERSION_CODE >= KERNEL_VERSION(5, 8, 0) && (REDHAT != 8)
```

https://github.com/cyring/CoreFreq/blob/6e1d7d414017a24cce69edb24c5586324d0c1fa4/corefreqk.h#L5362
```
#if LINUX_VERSION_CODE >= KERNEL_VERSION(5, 8, 0) && (REDHAT != 8)
```

Next fully rebuild `make clean all`

cyring, 2021-09-26:
**EDIT** My Minor is `4`, your is `3`
Try the bellow lines instead.

https://github.com/cyring/CoreFreq/blob/6e1d7d414017a24cce69edb24c5586324d0c1fa4/corefreqk.c#L15624
```
#if LINUX_VERSION_CODE >= KERNEL_VERSION(5, 8, 0) \
  || ((REDHAT == 8) && (RHEL_MINOR > 3))
```

https://github.com/cyring/CoreFreq/blob/6e1d7d414017a24cce69edb24c5586324d0c1fa4/corefreqk.h#L5362
```
#if LINUX_VERSION_CODE >= KERNEL_VERSION(5, 8, 0) \
  || ((REDHAT == 8) && (RHEL_MINOR > 3))
```

fleers, 2021-09-26:
```
# diff -u corefreqk.c.orig corefreqk.c
--- corefreqk.c.orig    2021-09-26 17:42:38.405889600 -0600
+++ corefreqk.c 2021-09-26 17:44:38.486764252 -0600
@@ -15621,7 +15621,7 @@
     }
 }

-#if LINUX_VERSION_CODE >= KERNEL_VERSION(5, 8, 0) || (REDHAT == 8)
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(5, 8, 0) || ((REDHAT == 8) && (RHEL_MINOR > 3))
 static int CoreFreqK_SetBoost(struct cpufreq_policy *policy, int state)
 {
        UNUSED(policy);
```

```
# diff -u corefreqk.h.orig corefreqk.h
--- corefreqk.h.orig    2021-09-26 17:42:51.556985407 -0600
+++ corefreqk.h 2021-09-26 17:45:04.314952338 -0600
@@ -5359,7 +5359,7 @@
 #endif
 static int CoreFreqK_SetPolicy(struct cpufreq_policy*) ;
 static int CoreFreqK_Bios_Limit(int, unsigned int*) ;
-#if LINUX_VERSION_CODE >= KERNEL_VERSION(5, 8, 0) || (REDHAT == 8)
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(5, 8, 0) || ((REDHAT == 8) && (RHEL_MINOR > 3))
 static int CoreFreqK_SetBoost(struct cpufreq_policy*, int) ;
 #else
 static int CoreFreqK_SetBoost(int) ;
```

```
# make clean && make
rm -f corefreqd corefreq-cli
make -j1 -C /lib/modules/4.18.0-240.el8.x86_64/build M=/tmp/CoreFreq clean
make[1]: Entering directory '/usr/src/kernels/4.18.0-240.el8.x86_64'
  CLEAN   /tmp/CoreFreq/.tmp_versions
make[1]: Leaving directory '/usr/src/kernels/4.18.0-240.el8.x86_64'
cc  -Wall -Wfatal-errors -pthread -c corefreqd.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreqd.o
cc  -Wall -Wfatal-errors -c corefreqm.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreqm.o
cc  -Wall -Wfatal-errors corefreqd.c corefreqm.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreqd -lpthread -lm -lrt
cc  -Wall -Wfatal-errors -c corefreq-cli.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0  \
  -o corefreq-cli.o
cc  -Wall -Wfatal-errors -c corefreq-ui.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreq-ui.o
cc  -Wall -Wfatal-errors -c corefreq-cli-rsc.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0  \
  -o corefreq-cli-rsc.o
cc  -Wall -Wfatal-errors -c corefreq-cli-json.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreq-cli-json.o
cc  -Wall -Wfatal-errors -c corefreq-cli-extra.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreq-cli-extra.o
cc  -Wall -Wfatal-errors \
  corefreq-cli.c corefreq-ui.c corefreq-cli-rsc.c \
  corefreq-cli-json.c corefreq-cli-extra.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0  \
  -o corefreq-cli -lm -lrt
make -j1 -C /lib/modules/4.18.0-240.el8.x86_64/build M=/tmp/CoreFreq modules
make[1]: Entering directory '/usr/src/kernels/4.18.0-240.el8.x86_64'
  CC [M]  /tmp/CoreFreq/corefreqk.o
/tmp/CoreFreq/corefreqk.c: In function ‘CoreFreqK_IdleDriver_Init’:
/tmp/CoreFreq/corefreqk.c:15260:18: error: assignment to ‘void (*)(struct cpuidle_device *, struct cpuidle_driver *, int)’ from incompatible pointer type ‘int (*)(struct cpuidle_device *, struct cpuidle_driver *, int)’ [-Werror=incompatible-pointer-types]
   ].enter_s2idle = CoreFreqK_Alt_S2_MWAIT_Handler;
                  ^
compilation terminated due to -Wfatal-errors.
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:322: /tmp/CoreFreq/corefreqk.o] Error 1
make[1]: *** [Makefile:1544: _module_/tmp/CoreFreq] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-240.el8.x86_64'
make: *** [Makefile:76: all] Error 2
```

cyring, 2021-09-27:
Can you post a link to the ISO of your current CentOS version ?

fleers, 2021-09-27:
8.3 ISOs can be found [here](https://vault.centos.org/8.3.2011/isos/x86_64/)
I'm using the [dvd1 ISO](https://vault.centos.org/8.3.2011/isos/x86_64/CentOS-8.3.2011-x86_64-dvd1.iso)

cyring, 2021-09-27:
> 8.3 ISOs can be found [here](https://vault.centos.org/8.3.2011/isos/x86_64/)
> I'm using the [dvd1 ISO](https://vault.centos.org/8.3.2011/isos/x86_64/CentOS-8.3.2011-x86_64-dvd1.iso)

I'm downloading CentOS-8.3.2011-x86_64-minimal.iso because that server link is too slow.
Let's hope I will find enough packages for kernel development ...

cyring, 2021-09-27:
> Notes from the [CentOS-8.3.2011-x86_64-minimal.iso](https://vault.centos.org/8.3.2011/isos/x86_64/CentOS-8.3.2011-x86_64-minimal.iso)
* `dnf install perl-interpreter gcc make elfutils-libelf-devel`
* `rpm -i https://vault.centos.org/8.3.2011/BaseOS/x86_64/os/Packages/`[kernel-devel-4.18.0-240.el8.x86_64.rpm](https://vault.centos.org/8.3.2011/BaseOS/x86_64/os/Packages/kernel-devel-4.18.0-240.el8.x86_64.rpm)
* `rpm -i https://vault.centos.org/8.3.2011/BaseOS/x86_64/os/Packages/`[kernel-headers-4.18.0-240.el8.x86_64.rpm](https://vault.centos.org/8.3.2011/BaseOS/x86_64/os/Packages/kernel-headers-4.18.0-240.el8.x86_64.rpm)

cyring, 2021-09-27:
Please try latest commit in `develop` branch.

fleers, 2021-09-27:
It builds successfully now!
Thanks for the quick fix :-)

cyring, 2021-09-27:
> It builds successfully now!
> Thanks for the quick fix :-)

That's great !

In future commit, I have to fix build combinations among CentOS/RHEL 8 Minor 3 vs 4; do no regression testings with CentOS 7; and probably other Redhat flavours.
So I'll come back to you for a final pre-release testing.

In Wiki-Support there is a list of processors supported by CoreFreq:
https://github.com/cyring/CoreFreq/wiki/CPU-support
Feel free to add a page with your  screenshots and Cli output, especially `corefreq-cli -s`
Your data will help me a lot to find the remaining work.
Thank you.

cyring, 2021-10-01:
Hello,

In `develop` branch is now available a fix to compile with CentOS RHEL 8.3 and 8.4 (and still CentOS 7)

Can you give it a try ?

Your screenshots and CLI outputs are also welcomed.

cyring, 2021-10-12:
> Issues found:
> 
> * Architecture codename should be Rome
> 
> [`[Zen2/Castle Peak]`](https://gist.github.com/fleers/a161e770b186c6410d229a38837d2685#file-gistfile1-txt-L37)
> 
> * HSMP does not give `PPT`
> 
> [`PPT [ 4 W]`](https://gist.github.com/fleers/a161e770b186c6410d229a38837d2685#file-gistfile1-txt-L204)

According to the CPU support page in [Wiki](https://github.com/cyring/CoreFreq/wiki/CPU-support), bellow processors have the same CPUID signature
* EPYC 7742 CPUID converted to hexa model `31`
```
CPU family:          23
Model:               49
Model name:          AMD EPYC 7742 64-Core Processor
Stepping:            0
```
* EPYC 7502P has a signature `8F_31` Stepping 0
* Threadripper 3970X has a signature `8F_31` Stepping 0

Consequently HSMP was not trigerred b/c brand `AMD EPYC 7502P 32-Core Processor` was out of the Castle Peak list.
https://github.com/cyring/CoreFreq/blob/5127cc858078fbac640a443bfa9d4dabd552c5e0/corefreqk.h#L4551

I have no evidence that `8F_30` Stepping 0 specifies EPYC/Rome
https://github.com/cyring/CoreFreq/blob/5127cc858078fbac640a443bfa9d4dabd552c5e0/corefreqk.h#L1484

cyring, 2021-10-15:
> OK, here's an [updated cli output](https://gist.github.com/fleers/143efa91cb924a9813968d81c53ad7ff)

Thank you very much.
Let me know if you wish to provide more EPYC tests ?

fleers, 2021-10-18:
I had to switch to another machine - it's single socket.

Here's the [cli output](<script src="https://gist.github.com/fleers/8b00efb27e2d2cab0f35f6f972ade485.js"></script>) after SMT on

L1/L2 off and ON - 

[image]

[image]

Temp. with turbo on 4 cores -

[image]

Voltage with random turbo - 

[image]

cyring, 2021-11-01:
RHEL are part of the new version. Closing the issue.

closed 2021-11-01 by cyring