cyring/CoreFreq #281: clang compile issues

thor2002ro, 2021-09-30:
the register here is an issue 

https://github.com/cyring/CoreFreq/blob/8d4ec519a8fcc565c2c4e7440e337ea4bce6e365/corefreq-cli.c#L16545

`corefreq-cli.c:16550:23: fatal error: address of register variable requested
        LayerAt(layer, attr, col[1]     , row) = eventAttr[_hot][0];
                             ^~~
`

From Wikipedia:
> In C (but not C++ where the keyword is essentially ignored) the location of a variable declared with register cannot be accessed, but the sizeof operator can be applied. Aside from this limitation, register is essentially meaningless in modern compilers due to optimization which will place variables in a register if appropriate regardless of whether the hint is given.

cyring, 2021-09-30:
> the register here is an issue
>
> https://github.com/cyring/CoreFreq/blob/8d4ec519a8fcc565c2c4e7440e337ea4bce6e365/corefreq-cli.c#L16545
>
> `corefreq-cli.c:16550:23: fatal error: address of register [...]

I do agree. Will remove that in next commit.

closed 2021-09-30 by thor2002ro

cyring, 2021-10-01:
Hello,
Code fix is available in `develop` branch.