cyring/CoreFreq #302: [AMD/EPYC/Rome] Wrong Registers reading on second socket

cyring, 2021-12-05:
* Temperature
[image]
[image]

cyring, 2021-12-06:
https://github.com/cyring/CoreFreq/blob/478eee81930e1c339f13787a17d7d0ffe2231e2d/corefreqk.c#L11223
* Need to verify the CCD topology.

cyring, 2021-12-11:
* The CPU topology issue

[image]
[image]
[image]
[image]

cyring, 2021-12-11:
Still part of the `develop` branch, the fix is showing off in this [gist](https://gist.github.com/cyring/31090bde7ff0c276db08f4b4dd8d0915)

jowa2021, 2021-12-16:
Hello sir, this is recent from AMD system, suspect that my kernel having keys disabled or somthing is causing the issue....
after this point, kernel panic and reboot.

[image]

cyring, 2021-12-16:
> Hello sir, this is recent from AMD system, suspect that my kernel having keys disabled or somthing is causing the issue.... after this point, kernel panic and reboot.
> 
> [image]

I don't see in your issue something about _CoreFreq_ !
But if signed kernel drivers is the purpose, you'll find various readings in kernel documentation.
My `Makefile` does not sign, you have to do it by your own.
The easier and most common way to get ride of a secured boot is to set some BIOS features:
* Disable safe boot
* Enable CSM (with UEFI support)

jowa2021, 2021-12-24:
[image]
hello, can you help to figure out what RDL column is please ?

cyring, 2021-12-24:
> hello, can you help to figure out what RDL column is please ?

[overclock.net](https://www.overclock.net/threads/tphyrdl-changed.1794471/post-28878129)

[image]
[image]

cyring, 2021-12-29:
* Updates from the EPYC/Rome since reverting commit  f1e9a8aad7c48fb5faacde08e1fe691a55fab349
* Topology fix
[image]
* Sensors
[image]
[image]
* Temperature per CCD
[image]
* Memory Controller
  DIMM size remain an issue; other timings are OK
[image]
* Performance Monitoring
[image]
* Power, Current & Thermal
  PPT, others need a review against specs
[image]
* Like all Ryzen, MWAIT a bit toasty, I/O is preferred (See Vcore & Temp in bottom, right)
[image]
[image]
* Switching OFF/ON cache prefetchers
[image]
[image]

closed 2022-01-19 by cyring