cyring/CoreFreq #303: Incorrect memory XMP frequency

mxw39, 2021-12-06:
Thanks for making this awesome tool. I'm truly impressed by the number of functionalities offered and the clean UI which presents them all (with resizable and repositionable floating windows even)!

The Memory Controller window doesn't appear to offer the correct DRAM speed. I have mine on XMP and BIOS shows 3600 MHz (MT/s). In CoreFreq it shows the default CPU-supported DRAM speed 2667 MHz.

Can you help me understand why CoreFreq gives a different speed?

P.S. CAS Latency seems correct. I didn't closely check other timing readings.

cyring, 2021-12-06:
Thank you for using _CoreFreq_.
Can you provide some data like : 
```
corefreq-cli -s -m -M
```

mxw39, 2021-12-06:
Certainly.

```
% corefreq-cli -s -m -M
Processor                             [Intel(R) Core(TM) i9-9900K CPU @ 3.60GHz]
|- Architecture                                                  [Coffee Lake/R]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [0x000000ea]
|- Signature                                                           [  06_9E]
|- Stepping                                                            [     13]
|- Online CPU                                                          [ 16/ 16]
|- Base Clock                                                          [100.000]
|- Frequency            (MHz)                      Ratio                        
                 Min    799.99                    <   8 >                       
                 Max   3599.95                    <  36 >                       
|- Factory                                                             [100.000]
                       3600                       [  36 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   4999.94                    <  50 >                       
   |- HWP                                                                       
                 Min   4999.94                    <  50 >                       
                 Max   4999.94                    <  50 >                       
                 TGT      AUTO                    <   0 >                       
|- Turbo Boost                                                         [ UNLOCK]
                  1C   4999.94                    <  50 >                       
                  2C   4999.94                    <  50 >                       
                  3C   4999.94                    <  50 >                       
                  4C   4999.94                    <  50 >                       
                  5C   4999.94                    <  50 >                       
                  6C   4999.94                    <  50 >                       
                  7C   4999.94                    <  50 >                       
                  8C   4999.94                    <  50 >                       
|- Uncore                                                              [ UNLOCK]
                 Min    799.99                    <   8 >                       
                 Max   4699.94                    <  47 >                       
|- TDP                                                           Level [  0:3  ]
   |- Programmable                                                     [ UNLOCK]
   |- Configuration                                                    [   LOCK]
   |- Turbo Activation                                                 [ UNLOCK]
             Nominal   3599.95                    [  36 ]                       
               Turbo      AUTO                    <   0 >                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N] 
|- AVX512-VBMI  [N] AVX512-VBMI2 [N]  AVX512-VNMI [N]  AVX512-ALG [N] 
|- AVX512-VPOP  [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [N]  BMI1/BMI2 [Y/Y]         CLWB [N] CLFLUSH/O [Y/Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/N] MON/MWAITX [Y/N]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [N]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/N]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]          SGX [Y]       RDPID [N] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Missing]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Capable]
|- CPL Qualified Debug Store                                  DS-CPL   [Capable]
|- 64-Bit Debug Store                                         DTES64   [Capable]
|- Fast-String Operation                                Fast-Strings   [Capable]
[... 168 lines cut ...]
014:  0   13     6      1    32768  8     32768  8    262144  4  16777216 16 i
015:  0   15     7      1    32768  8     32768  8    262144  4  16777216 16 i
[... cut ...]
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          32768                    
       #1                                                                  
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          32768                    
       #1
```

cyring, 2021-12-07:
Can you print or debug (`gdb`) the `DMFC` value at the following `switch` line:

https://github.com/cyring/CoreFreq/blob/478eee81930e1c339f13787a17d7d0ffe2231e2d/corefreqd.c#L3937

For instance, if `DMFC` is equal to zero then Controller Speed is set to `2667`. You can replace it with your BIOS frequency.

mxw39, 2021-12-07:
Breaking at the `switch` line shows DMFC is `0b1`. It leads to the code that assigns 2667 to the control speed.

But I doubt replacing the frequency assigned here is getting to the bottom of it. The memory is overclocked in BIOS, which is where the 3600 MHz originates. Is it possible to read the running clock speed of the memory from BIOS or some CPU registers at all?

cyring, 2021-12-07:
> Breaking at the `switch` line shows DMFC is `0b1`. It leads to the code that assigns 2667 to the control speed.
>
> But I doubt replacing the frequency assigned here is getting to the bottom of it. The [...]

The [i9900K](https://ark.intel.com/content/www/fr/fr/ark/products/186605/intel-core-i99900k-processor-16m-cache-up-to-5-00-ghz.html) is part of the 9th generation.
From my [Wiki/Doocumentation](https://github.com/cyring/CoreFreq/wiki/Documentation), the link "Intel® Core™ Processors Technical Resources" sends us to the [Intel datasheets](https://www.intel.com/content/www/us/en/products/docs/processors/core/core-technical-resources.html) where we download/preview the "9th/8th Generation Intel® Core™ Processor Family" datasheet volume two.

* DMFC_DDR4 is a 3 bits encoding but no frequencies table found so far.

[image]

cyring, 2021-12-07:
_EDIT_:
Some registers later published for TGL, like the `DATA_RATE_GEAR1` field (may help)

https://github.com/cyring/CoreFreq/blob/478eee81930e1c339f13787a17d7d0ffe2231e2d/intelmsr.h#L3498

At this location ...
https://github.com/cyring/CoreFreq/blob/478eee81930e1c339f13787a17d7d0ffe2231e2d/corefreqd.c#L3932
... can you print the full hexa value of the `Cap_C` register
```
printf("%x\n", RO(Proc)->Uncore.Bus.SKL_Cap_C.value);

mxw39, 2021-12-08:
The value gets read twice on `corefreqd` startup. I get `0x2c000` for the `SKL_Cap_C.value` both times.

cyring, 2021-12-08:
Found this one in vol 2 datasheet:

> 7.91 PCU_CR_MC_BIOS_REQ_0_0_0_MCHBAR_PCU—
Offset 5E00h

> This register allows BIOS to request Memory Controller clock frequency.

> The encoding of this field is the 133/266 MHz multiplier for 
DCLK/QCLK

cyring, 2021-12-08:
To peek the above register, in `corefreqk.c` driver code, add at the beginning of  function `Query_SKL_IMC`:

https://github.com/cyring/CoreFreq/blob/478eee81930e1c339f13787a17d7d0ffe2231e2d/corefreqk.c#L4104
```c
void Query_SKL_IMC(void __iomem *mchmap, unsigned short mc)
{	/*Source: 6th & 7th Generation Intel® Processor for S-Platforms Vol 2*/
	unsigned int MC_BIOS_REQ = 0;
	unsigned short cha;

	MC_BIOS_REQ = readl(mchmap+0x5e00);
	printk("MC_BIOS_REQ[%x]", MC_BIOS_REQ);
```

mxw39, 2021-12-09:
Thanks for the pointer to the datasheet.

I get `0x10d` from `mchmap+0x5e00`. This calculation would put the frequency at 13 * 266.67 MHz == 3466 MHz, which differs from the BIOS settings. There are DRAMs marketed at 3466 MHz but the sticks I have aren't marketed at 3466 MHz. They are at 3600 MHz.

But I did find section 7.81 -- System Agent performance status. The lowest 8 bits are the DDR QCLK reference base and the ratio. I get `0x2c000a1b` from `mchmap+0x5918`. This puts reference base == 0 (133 MHz) and the ratio 27, which gets the final frequency to be 3591 MHz, within error margin of 3600.

Do you think it makes sense to calculate DRAM frequency in this manner?

cyring, 2021-12-09:
This is excellent 👍

And we should understand the following meaning from `QCLK_REFERENCE == 0`:

DRAM = `(133333333 x 27) / 1000000`

DRAM = `3600` MHz

* The best way to validate our formula would be to match against other BIOS frequencies.
   Can you try in BIOS at least 2 other DRAM OC frequencies ?

* Not sure if the offset `0x5918` is available across the Skylake and superior families ?
   I will develop that specific formula in a Coffee Lake function (until some Users are showing up and validate with other Skylake derivatives)

cyring, 2021-12-09:
**EDIT**: In addition to the above requests, can you also try **_standard_** DRAM frequency in BIOS ?
I mean non-overclocked frequency.
And check the resulted ratios in offset `0x5918`

This will help us to define a _fallback_ strategy; if the current `switch-cases` should be considered when offset `0x5918` is not meaningful.

mxw39, 2021-12-09:
That sounds like a plan. Thanks Cyril!

I read 7th gen datasheet and `0x5918` is still valid.

I have a put-together 7700K system and a spare 6700K CPU (hopefully still alive). Let me run several DRAM frequency with my current 9900K to determine if the reference * ratio formula is always valid. Then I'll test the same on the 7700K system.

cyring, 2021-12-11:
@mxw39 : You can now pull the [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch to test the DRAM frequency.

Remarks:
1. Make sure to clean any previous builds.
```
make clean all
```
2. This new code applies to `SKYLAKE, KABYLAKE, COFFEELAKE, WHISKEYLAKE, COMETLAKE, ICELAKE` (but not `COMETLAKE/U` and `COMETLAKE/H`)

   Any **non-regression tests** are warmly welcomed from anyone.

mxw39, 2021-12-11:
Here are the results on the 9900K system. You can see a pattern that the base and ratio gets lower by 1 decrement respectively in `SA_PERF_STAT`. `MC_BIOS_REQ` does not quite make sense in this result set. Perhaps my BIOS doesn't implement this interface correctly. At one point it was `0x0` for the lower meaningful 4 bits.

I tried all settings from 3200 - 3600, and the default, non-OC settings.

```
BIOS freq   MC_BIOS_REQ     SA_PERF_STAT
3600        0x10d           0x2c000a1b
3500        0x131           0x2c000aa3
3467        0xd             0x2c000a1a
3400        0x31            0x2c000aa2
3333        0x10c           0x2c000a19
3000        0x130           0x2c000aa1
3200        0xc             0x2c000a18

2667        0xa             0x2c000a14
2600        0x1d            0x2c000a9a
2600(auto)  0xa             0x2c000a14
```

2600(auto) is the BIOS auto frequency when all OC settings are removed. But clearly it is running 2667 because the register values are the same as when I set 2667 manually. This is perhaps a BIOS bug. CoreFreq `develop` branch calculates the correct value.

The 7700K system test will come later. I'll get my hands on it next week.

cyring, 2021-12-11:
The ratio is not 4 but 7 bits wide.
See this header file, based on specs:
https://github.com/cyring/CoreFreq/blob/4e02ecbf709bd25c3e64bf28641212f600a1f277/intelmsr.h#L3359
Can you please show the Memory Controller output of _CoreFreq_

cyring, 2021-12-11:
A remaining issue of temperature sensors permutation along the topology may be fixed by the following maths in CCD numbering at line:
https://github.com/cyring/CoreFreq/blob/d83712122824a2cbc303715c9c571f8a868d5fc4/corefreqk.c#L1996
... to replace with:
```c
Core->T.Cluster.CCD = (Core->T.CoreID << factor) >> 3;
```

I will try this change next time I can rent such server ...

cyring, 2021-12-12:
Sorry, my last answer was for EPYC.
Can you however try the latest development from branch `develop`  ?

mxw39, 2021-12-17:
Sorry this reply took a bit long.

I wasn't clear on my wording when I said decrement. I meant as I go down the BIOS defined frequency choices, the base alternates between 100 and 133. When grouping all 100 based ratio together, we find the ratio goes down by 1. The same for 133 based ratio.

```
BIOS freq   MC_BIOS_REQ     SA_PERF_STAT    BASE    RATIO
3600        0x10d           0x2c000a1b      133     27
3500        0x131           0x2c000aa3      100     35
3467        0xd             0x2c000a1a      133     26
3400        0x31            0x2c000aa2      100     34
3333        0x10c           0x2c000a19      133     25
3000        0x130           0x2c000aa1      100     33
3200        0xc             0x2c000a18      133     24

2667        0xa             0x2c000a14      133     20
2600        0x1d            0x2c000a9a      100     26
2600(auto)  0xa             0x2c000a14      133     20
```

With the current BIOS 3600 MHz settings, I get

```
% ./corefreq-cli -M
                            Cannon Point  [3E30]                           
Controller #0                                                Dual Channel  
 Bus Rate  8000 MT/s      Bus Speed 7999 MT/s          DDR4 Speed 3599 MHz 
                                                                           
 Cha    CL  RCD   RP  RAS  RRD  RFC   WR RTPr WTPr  FAW  B2B  CWL CMD  REFI
  #0    16   18   18   38    9  631   24   12   44   38    0   16  2T 14040
  #1    16   18   18   38    9  631   24   12   44   38    0   16  2T 14040
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4    6    6        12   12   12   12        36   27   10   10
  #1     7    4    6    6        12   12   12   12        36   27   10   10
      sgWW dgWW drWW ddWW                                         CKE   ECC
  #0     7    4   10   10                                           4    0 
  #1     7    4   10   10                                           4    0 
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          32768                    
       #1                                                                  
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          32768                    
       #1                                                                  
```

cyring, 2021-12-17:
* Hard to find a discriminant whether the low 4 bits of MC_BIOS_REQ are significant or not; which frequency algorithm should be employed ?

[image]

* According to the datasheet, the 4-bits value has to be either `0`
 or within `[3, 4, 5, 6, 7, 8]`

BIOS freq | MC_BIOS_REQ| MC_BIOS_REQ `AND 0b1111` | SA_PERF_STAT | BASE | RATIO
--------------|-----------------------|--------------------|------------------------|---------|----------
3600        |0x10d|0xd           |0x2c000a1b      |133     |27
3500        |0x131|0x1           |0x2c000aa3      |100     |35
3467        |0xd|0xd             |0x2c000a1a      |133     |26
3400        |0x31|0x1            |0x2c000aa2      |100     |34
3333        |0x10c|0xc           |0x2c000a19      |133     |25
3000        |0x130|0x0           |0x2c000aa1      |100     |33
3200        |0xc|0xc             |0x2c000a18      |133     |24

BIOS freq | MC_BIOS_REQ| MC_BIOS_REQ `AND 0b1111` | SA_PERF_STAT | BASE | RATIO
--------------|-----------------------|--------------------|------------------------|----------|----------
2667        |0xa|0xa             |0x2c000a14      |133     |20
2600        |0x1d|0xd            |0x2c000a9a      |100     |26
2600(auto)  |0xa|0xa             |0x2c000a14      |133     |20

* In addition to that, I don't have any confirmation if those registers, MC_BIOS_REQ (`0x5e00`) and SA_PERF_STAT(`0x5918`), are available in Skylake Desktop, Mobile, Server architectures and deratives.

cyring, 2021-12-17:
Those are the problematic cases:

BIOS freq | MC_BIOS_REQ| MC_BIOS_REQ `AND 0b1111` | SA_PERF_STAT | BASE | RATIO
--------------|-----------------------|--------------------|------------------------|---------|----------
3000        |0x130|0x0           |0x2c000aa1      |100     |33
2600 (auto)  |0xa|0xa             |0x2c000a14      |133     |20

mxw39, 2021-12-18:
Sorry, the BIOS freq == 3000 line is my typo. It should have been 3300.

I set my memory to 3300 again to confirm the value in MC_BIOS_REQ is correct. It is indeed `0x130` which means `MC_BIOS_REQ & 0b1111 == 0x0` and makes no sense. 

SA_PERF_STAT is however correct and `develop` version of corefreq at `1cccd64351bf34043f2fb8efa4e5abfdf4c9c4e3` gives this output

```
% ./corefreq-cli -M
                            Cannon Point  [3E30]                           
Controller #0                                                Dual Channel  
 Bus Rate  8000 MT/s      Bus Speed 8000 MT/s          DDR4 Speed 3300 MHz 
                                                                           
 Cha    CL  RCD   RP  RAS  RRD  RFC   WR RTPr WTPr  FAW  B2B  CWL CMD  REFI
  #0    16   18   18   38    9  631   24   12   42   38    0   14  2T 12998
  #1    16   18   18   38    9  631   24   12   42   38    0   14  2T 12998
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4    6    6        12   12   12   12        32   25   10   10
  #1     7    4    6    6        12   12   12   12        32   25   10   10
      sgWW dgWW drWW ddWW                                         CKE   ECC
  #0     7    4   10   10                                           4    0 
  #1     7    4   10   10                                           4    0 
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          32768                    
       #1                                                                  
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    2     65536      1024          32768                    
       #1                                                                  
```

With `MC_BIOS_REQ & 0b1111 == 0x0` I find it hard to believe the value is at all meaningful. Should SA_PERF_STAT(`0x5918`) be available I think it is advisable to use it. It gives the same result as the BIOS at all times. 

The only exception is the auto settings with all overclocking disabled. However, I have demonstrated the registers we are interested in are the same as those of 2677. So I am convinced that the memory is really running at 2677 despite BIOS showing 2600 when set to auto. It's not very hard to believe the BIOS isn't interfacing with MC_BIOS_REQ properly. A buggy BIOS may also explain why MC_BIOS_REQ is not making sense and why 2600 (auto) is really 2677.

About Skylake and derivative: I have gotten the 7700K system and should have some results for you today or tomorrow. This should help with our confidence SA_PERF_STAT(`0x5918`) is available on 7th Gen desktop at least.

cyring, 2021-12-18:
@mxw39 well done. I think issue is almost fixed. 
Let's see with SKL non regression tests.

mxw39, 2021-12-20:
Here are the results from 7700K. `SA_PERF_STAT (0x5918)` still exists and the formula is correct.

```
BIOS freq   MC_BIOS_REQ     SA_PERF_STAT    BASE    RATIO   corefreq-cli -M
3200(XMP)   0xc             0x29000a18      133     24      3199
3100        0x11f           0x29000a9f      100     31      3100
3000        0x1f            0x29000a9e      100     30      3000
2933        0xb             0x29000a16      133     23      2933
2900        0x11e           0x29000a9d      100     29      2900
2800        0x1e            0x29000a9c      100     28      2800
2133(auto)  0x8             0x29000a10      133     16      2133
```

cyring, 2021-12-20:
Looks like this is the right register to decode XMP frequencies.

In Wiki there is all Skylake tested so far, starting from 6600K :
https://github.com/cyring/CoreFreq/wiki/CPU-support

The remaining work is to avoid a crash where SA_PERF_STAT is denied: what kind of capability bit we can query ?

mxw39, 2021-12-20:
That's a very good question. I looked at the 8th Gen datasheet but didn't find anything useful other than the default value will be `0x0` if not valid. I don't know yet what will happen if `0x5918` is read on an unsupported arch.

The current fallback logic based on `QCLK == 0` should be good. When the register defaults to `0x0`, QCLK will be as well.

cyring, 2021-12-20:
>

`SKL_IMC()` is the entry function.
All PCI `DID` associated with this function in the following lists will reach the `0x5918`

https://github.com/cyring/CoreFreq/blob/d253207ee4fc5ce0ebe107b47a306e6b4468d95b/corefreqk.h#L2213

https://github.com/cyring/CoreFreq/blob/d253207ee4fc5ce0ebe107b47a306e6b4468d95b/corefreqk.h#L2250

cyring, 2022-01-13:
Hello, 

I'll be thankful if you can make a gist page with a collection of _CoreFreq_ screenshots and outputs of your `i9-9900K`, using the fresh `master` version.

This page will join the Wiki [CPU support](https://github.com/cyring/CoreFreq/wiki/CPU-support)

Merci

mxw39, 2022-01-18:
Certainly. I'll let you know when it is ready!

closed 2022-01-24 by cyring