cyring/CoreFreq #309: [WiP] Wanted: Cannon Lake, Ice Lake, Tiger Lake, Comet Lake, Rocket Lake, Alder Lake

cyring, 2022-01-01:
### Discussed in https://github.com/cyring/CoreFreq/discussions/238

<div type='discussions-op-text'>

<sup>Originally posted by **cyring** May 20, 2021</sup>
I miss testings reports from all processors, post Coffee lake generation.

Signatures are installed into _CoreFreq_ with the assumption they are using the same KBL, CFL registers. 

Thanks for any contribution.
</div>

cyring, 2022-01-01:
@svmlegacy Hello

In `develop` branch you are getting code to decode the IMC of Cannon Lake.

If results are wrong, please replace successively `SKL_IMC` by `RKL_IMC` , `TGL_IMC` at the following line:
https://github.com/cyring/CoreFreq/blob/f549cdea48e263fb76fdb72c9b67bcaa9af92c31/corefreqk.h#L2361

cyring, 2022-01-11:
* Based on the Intel SDM, bellow are the registers I would to test.

  Feel free to join or dump if you have such 12th gen. processors 😉

* MSRs Introduced in 12th Generation Intel® Core™ Processors Supporting Performance Hybrid Architecture
  CPUID of Alder Lake: `06_97H`, `06_9AH`, and `06_BFH`

[image]

ich777, 2022-02-15:
Will provide my Alder Lake test results ASAP.

CPU: i7-12700k
Motherboard: ASUS ROG STRIX Z690-G GAMING WIFI
RAM: Corsair Vengance DDR5 4800MHz 2x16GB

cyring, 2022-02-15:
> Will provide my Alder Lake test results ASAP, do you need anything special, my system specs are:
> 
> CPU: i7-12700k Motherboard: ASUS ROG STRIX Z690-G GAMING WIFI RAM: Corsair Vengance DDR5 4800MHz 2x16GB

Hello,

Thanks a lot for your help, you can see this [page](https://gist.github.com/cyring/6ed0f017ed196fd385934866ce47ce40) which will give a global overview of what  is working or not.
If you also notice some errors in the kernel log, linked to the CoreFreq driver; please post them here.
If features settings reported from the CLI don't match your BIOS values, such Memory Controller timings and DIMM size topology; also various technologies states; please let me know.

Best regards,
CyrIng

cyring, 2022-02-15:
Let's continue with Alder Lake in #323

closed 2022-02-15 by cyring