cyring/CoreFreq #313: Memory Controller data is not available on Intel Gold 6230N

wwyf, 2022-01-16:
The problem the same as #160, though my CPU is Intel Gold 6230N.

Problems:

1. The Memory Controller window is greyed out in the CLI.
2. Running `corefreq-cli -M` gives no output.

This is some output of various commands about environment, may be they are useful.

```
$ uname -r
3.10.0-957.el7.x86_64

$ cat /etc/os-release 
NAME="CentOS Linux"
VERSION="7 (Core)"
ID="centos"
ID_LIKE="rhel fedora"
VERSION_ID="7"
PRETTY_NAME="CentOS Linux 7 (Core)"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:centos:centos:7"
HOME_URL="https://www.centos.org/"
BUG_REPORT_URL="https://bugs.centos.org/"

CENTOS_MANTISBT_PROJECT="CentOS-7"
CENTOS_MANTISBT_PROJECT_VERSION="7"
REDHAT_SUPPORT_PRODUCT="centos"
REDHAT_SUPPORT_PRODUCT_VERSION="7"

$ lscpu
Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                40
On-line CPU(s) list:   0-39
Thread(s) per core:    2
Core(s) per socket:    20
Socket(s):             1
NUMA node(s):          1
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 85
Model name:            Intel(R) Xeon(R) Gold 6230N CPU @ 2.30GHz
Stepping:              7
CPU MHz:               2300.000
BogoMIPS:              4600.00
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              1024K
L3 cache:              28160K
NUMA node0 CPU(s):     0-39
Flags:                 fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid dca sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch epb cat_l3 cdp_l3 intel_ppin intel_pt ssbd mba ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm cqm mpx rdt_a avx512f avx512dq rdseed adx smap clflushopt clwb avx512cd avx512bw avx512vl xsaveopt xsavec xgetbv1 cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local dtherm ida arat pln pts pku ospke avx512_vnni spec_ctrl intel_stibp flush_l1d arch_capabilities
```

(`-u` make output too long, so I upload the output as file)
[corefreq-cli -B -k -u -M.txt](https://github.com/cyring/CoreFreq/files/7876483/corefreq-cli.-B.-k.-u.-M.txt)

```
$ ./corefreq-cli -B -k -M                                                                                              
[ 0] Dell Inc.                                                                  
[ 1] 2.5.4                                                                      
[ 2] 01/13/2020                                                                 
[ 3] Dell Inc.                                                                  
[ 4] PowerEdge R740                                                             
[ 5]                                                                            
[ 6] 4---D--                                                                    
[ 7]                                                                            
[ 8]                                                                            
[... 263 lines cut ...]
00:11.5 SATA controller [0106]: Intel Corporation C620 Series Chipset Family SSATA Controller [AHCI mode] [8086:a1d2] (rev 09)
00:14.0 USB controller [0c03]: Intel Corporation C620 Series Chipset Family USB 3.0 xHCI Controller [8086:a1af] (rev 09)
00:16.0 Communication controller [0780]: Intel Corporation C620 Series Chipset Family MEI Controller #1 [8086:a1ba] (rev 09)
00:16.1 Communication controller [0780]: Intel Corporation C620 Series Chipset Family MEI Controller #2 [8086:a1bb] (rev 09)
00:16.4 Communication controller [0780]: Intel Corporation C620 Series Chipset Family MEI Controller #3 [8086:a1be] (rev 09)
00:17.0 SATA controller [0106]: Intel Corporation C620 Series Chipset Family SATA Controller [AHCI mode] [8086:a182] (rev 09)
00:1f.0 ISA bridge [0601]: Intel Corporation C621 Series Chipset LPC/eSPI Controller [8086:a1c1] (rev 09)
00:1f.2 Memory controller [0580]: Intel Corporation C620 Series Chipset Family Power Management Controller [8086:a1a1] (rev 09)
00:1f.5 Serial bus controller [0c80]: Intel Corporation C620 Series Chipset Family SPI Controller [8086:a1a4] (rev 09)
03:00.0 VGA compatible controller [0300]: Matrox Electronics Systems Ltd. Integrated Matrox G200eW3 Graphics Controller [102b:0536] (rev 04)
18:00.0 Ethernet controller [0200]: Broadcom Inc. and subsidiaries BCM57412 NetXtreme-E 10Gb RDMA Ethernet Controller [14e4:16d6] (rev 01)
18:00.1 Ethernet controller [0200]: Broadcom Inc. and subsidiaries BCM57412 NetXtreme-E 10Gb RDMA Ethernet Controller [14e4:16d6] (rev 01)
64:08.0 System peripheral [0880]: Intel Corporation Sky Lake-E Integrated Memory Controller [8086:2066] (rev 07)
64:09.0 System peripheral [0880]: Intel Corporation Sky Lake-E Integrated Memory Controller [8086:2066] (rev 07)
64:0a.0 System peripheral [0880]: Intel Corporation Sky Lake-E Integrated Memory Controller [8086:2040] (rev 07)
64:0a.1 System peripheral [0880]: Intel Corporation Sky Lake-E Integrated Memory Controller [8086:2041] (rev 07)
64:0a.2 System peripheral [0880]: Intel Corporation Sky Lake-E Integrated Memory Controller [8086:2042] (rev 07)
64:0a.3 System peripheral [0880]: Intel Corporation Sky Lake-E Integrated Memory Controller [8086:2043] (rev 07)
64:0a.4 System peripheral [0880]: Intel Corporation Sky Lake-E Integrated Memory Controller [8086:2044] (rev 07)
64:0c.0 System peripheral [0880]: Intel Corporation Sky Lake-E Integrated Memory Controller [8086:2040] (rev 07)
64:0c.1 System peripheral [0880]: Intel Corporation Sky Lake-E Integrated Memory Controller [8086:2041] (rev 07)
64:0c.2 System peripheral [0880]: Intel Corporation Sky Lake-E Integrated Memory Controller [8086:2042] (rev 07)
64:0c.3 System peripheral [0880]: Intel Corporation Sky Lake-E Integrated Memory Controller [8086:2043] (rev 07)
64:0c.4 System peripheral [0880]: Intel Corporation Sky Lake-E Integrated Memory Controller [8086:2044] (rev 07)
```

cyring, 2022-01-16:
According to Xeon scalable [volume 2](https://www.intel.com/content/www/us/en/processors/xeon/xeon-technical-resources.html) , the IMC could be on PCI `BUS:2` 

One way is like function `Jasper_Forest_IMC()` ...
https://github.com/cyring/CoreFreq/blob/6fcba8da6d8a2de75a248c040b14f3cb75869cd3/corefreqk.c#L4513
... where any memory controller is scanned on all CPU for a **MEMBAR** present on PCI Device and PCI Function identifiers

But MEMBAR appears NDA! Timings are not specified in vol 2.

What do you suggest to experiment and program IMC codes on your hardware ?

wwyf, 2022-01-16:
Thanks for your quick reply!
It sounds like this issue can not be easily fixed by finding some key parameters in the document.

emmmmm... I also have no idea right now.

cyring, 2022-01-16:
> Linux:                                                                          
> |- Release                                               [3.10.0-957.el7.x86_64]

If you boot an updated kernel we _may_ read more `lspci` identifiers
I suggest my [live](https://github.com/cyring/CoreFreq/wiki/Live-CD) ISO which is Linux `5.15` based. This image will automatically compile/load/start _CoreFreq_
You can switch then to any VT for commands for PCI dump.
_CoreFreq_ source code is within `/opt/CoreFreq` if you are ready to experiment a new IMC implementation ?

wwyf, 2022-01-16:
I can't boot this image since this server is in the locked server room.
But there is another server using kernel `5.10.78` with the same hardware.
Let me try that server.

wwyf, 2022-01-16:
```
$ ./corefreq-cli -B -k -M  
[ 0] Dell Inc.                                                                  
[ 1] 2.5.4                                                                      
[ 2] 01/13/2020                                                                 
[ 3] Dell Inc.                                                                  
[ 4] PowerEdge R740                                                             
[ 5]                                                                            
[ 6] 4---D--                                                                    
[ 7] SKU=NotProvided;ModelName=PowerEdge R740                                   
[ 8] PowerEdge                                                                  
[ 9] Dell Inc.                                                                  
[10] 0WXD1Y                                                                     
[11] A00                                                                        
[12] .---5---.---V---0---0---                                                   
Linux:                                                                          
|- Release                                                            [5.10.78+]
|- Version                                  [#2 SMP Sun Nov 7 22:48:58 CST 2021]
|- Machine                                                              [x86_64]
Memory:                                                                         
|- Total RAM                                                        196325896 KB
|- Shared RAM                                                           13388 KB
|- Free RAM                                                          65936176 KB
|- Buffer RAM                                                           23376 KB
|- Total High                                                               0 KB
|- Free High                                                                0 KB
CPU-Freq driver                                               [         Missing]
Governor                                                      [         Missing]
CPU-Idle driver                                               [      intel_idle]
|- Idle Limit                                                 [              C6]
   |- State        POLL      C1     C1E      C6                                 
   |-           CPUIDLE MWAIT 0 MWAIT 0 MWAIT 0                                 
   |- Power          -1       0       0       0                                 
   |- Latency         0       2      10     133                                 
   |- Residency       0       2      20     600                                 
                            GenuineIntel  [   0]  
```

```
$ ./corefreq-cli -s
Processor                            [Intel(R) Xeon(R) Gold 6230N CPU @ 2.30GHz]
|- PPIN#                                                      [75e862aaaa16b77b]
|- Architecture                                                 [Cascade Lake/X]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [0x0500002c]
|- Signature                                                           [  06_55]
|- Stepping                                                            [      7]
|- Online CPU                                                          [ 40/ 40]
|- Base Clock                                                          [ 99.760]
|- Frequency            (MHz)                      Ratio                        
                 Min    997.66                    <  10 >                       
                 Max   2294.61                    <  23 >                       
|- Factory                                                             [100.000]
                       2300                       [  23 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   3491.80                    <  35 >                       
|- Turbo Boost                                                         [ UNLOCK]
                  1C   3491.80                    <  35 >                       
                  2C   3292.26                    <  33 >                       
                  3C   3192.50                    <  32 >                       
                  4C   3192.50                    <  32 >                       
                  5C   3092.73                    <  31 >                       
                  6C   2893.20                    <  29 >                       
                  7C   2893.20                    <  29 >                       
                  8C   2893.20                    <  29 >                       
                  9C    199.53                    <   2 >                       
                 10C    399.06                    <   4 >                       
                 12C   1197.19                    <  12 >                       
                 13C   1596.25                    <  16 >                       
[... 172 lines cut ...]
|- Thermal Design Power                                      Package   < Enable>
   |- Power Limit (1 sec)                                        PL1   <  125 W>
   |- Power Limit (1 sec)                                        PL2   <  150 W>
[... cut ...]
|- Package Thermal Point                                                        
   |- DTS Threshold #1                                     Threshold   [Missing]
   |- DTS Threshold #2                                     Threshold   [Missing]
|- Units                                                                        
   |- Power                                               watt   [  0.125000000]
   |- Energy                                             joule   [  0.000061035]
   |- Window                                            second   [  0.000976562]
```

cyring, 2022-01-16:
EDIT:
```
lspci -nn | grep -e "Controller" -e "Memory"
```
is probably masking the bus `2`

Memory BAR is more often remapped from PCI root

wwyf, 2022-01-16:
I should run the command `lspci -nn` ?

```
$ lspci -nn
00:00.0 Host bridge [0600]: Intel Corporation Sky Lake-E DMI3 Registers [8086:2020] (rev 07)
00:04.0 System peripheral [0880]: Intel Corporation Sky Lake-E CBDMA Registers [8086:2021] (rev 07)
00:04.1 System peripheral [0880]: Intel Corporation Sky Lake-E CBDMA Registers [8086:2021] (rev 07)
00:04.2 System peripheral [0880]: Intel Corporation Sky Lake-E CBDMA Registers [8086:2021] (rev 07)
00:04.3 System peripheral [0880]: Intel Corporation Sky Lake-E CBDMA Registers [8086:2021] (rev 07)
00:04.4 System peripheral [0880]: Intel Corporation Sky Lake-E CBDMA Registers [8086:2021] (rev 07)
00:04.5 System peripheral [0880]: Intel Corporation Sky Lake-E CBDMA Registers [8086:2021] (rev 07)
00:04.6 System peripheral [0880]: Intel Corporation Sky Lake-E CBDMA Registers [8086:2021] (rev 07)
00:04.7 System peripheral [0880]: Intel Corporation Sky Lake-E CBDMA Registers [8086:2021] (rev 07)
00:05.0 System peripheral [0880]: Intel Corporation Sky Lake-E MM/Vt-d Configuration Registers [8086:2024] (rev 07)
00:05.2 System peripheral [0880]: Intel Corporation Sky Lake-E RAS [8086:2025] (rev 07)
00:05.4 PIC [0800]: Intel Corporation Sky Lake-E IOAPIC [8086:2026] (rev 07)
00:08.0 System peripheral [0880]: Intel Corporation Sky Lake-E Ubox Registers [8086:2014] (rev 07)
00:08.1 Performance counters [1101]: Intel Corporation Sky Lake-E Ubox Registers [8086:2015] (rev 07)
00:08.2 System peripheral [0880]: Intel Corporation Sky Lake-E Ubox Registers [8086:2016] (rev 07)
00:11.0 Unassigned class [ff00]: Intel Corporation C620 Series Chipset Family MROM 0 [8086:a1ec] (rev 09)
00:11.5 SATA controller [0106]: Intel Corporation C620 Series Chipset Family SSATA Controller [AHCI mode] [8086:a1d2] (rev 09)
00:14.0 USB controller [0c03]: Intel Corporation C620 Series Chipset Family USB 3.0 xHCI Controller [8086:a1af] (rev 09)
00:14.2 Signal processing controller [1180]: Intel Corporation C620 Series Chipset Family Thermal Subsystem [8086:a1b1] (rev 09)
00:16.0 Communication controller [0780]: Intel Corporation C620 Series Chipset Family MEI Controller #1 [8086:a1ba] (rev 09)
00:16.1 Communication controller [0780]: Intel Corporation C620 Series Chipset Family MEI Controller #2 [8086:a1bb] (rev 09)
00:16.4 Communication controller [0780]: Intel Corporation C620 Series Chipset Family MEI Controller #3 [8086:a1be] (rev 09)
00:17.0 SATA controller [0106]: Intel Corporation C620 Series Chipset Family SATA Controller [AHCI mode] [8086:a182] (rev 09)
00:1c.0 PCI bridge [0604]: Intel Corporation C620 Series Chipset Family PCI Express Root Port #1 [8086:a190] (rev f9)
00:1c.4 PCI bridge [0604]: Intel Corporation C620 Series Chipset Family PCI Express Root Port #5 [8086:a194] (rev f9)
00:1f.0 ISA bridge [0601]: Intel Corporation C621 Series Chipset LPC/eSPI Controller [8086:a1c1] (rev 09)
00:1f.2 Memory controller [0580]: Intel Corporation C620 Series Chipset Family Power Management Controller [8086:a1a1] (rev 09)
00:1f.4 SMBus [0c05]: Intel Corporation C620 Series Chipset Family SMBus [8086:a1a3] (rev 09)
00:1f.5 Serial bus controller [0c80]: Intel Corporation C620 Series Chipset Family SPI Controller [8086:a1a4] (rev 09)
01:00.0 Ethernet controller [0200]: Broadcom Inc. and subsidiaries NetXtreme BCM5720 2-port Gigabit Ethernet PCIe [14e4:165f]
01:00.1 Ethernet controller [0200]: Broadcom Inc. and subsidiaries NetXtreme BCM5720 2-port Gigabit Ethernet PCIe [14e4:165f]
02:00.0 PCI bridge [0604]: PLDA PCI Express Bridge [1556:be00] (rev 02)
03:00.0 VGA compatible controller [0300]: Matrox Electronics Systems Ltd. Integrated Matrox G200eW3 Graphics Controller [102b:0536] (rev 04)
16:00.0 PCI bridge [0604]: Intel Corporation Sky Lake-E PCI Express Root Port A [8086:2030] (rev 07)
16:02.0 PCI bridge [0604]: Intel Corporation Sky Lake-E PCI Express Root Port C [8086:2032] (rev 07)
16:05.0 System peripheral [0880]: Intel Corporation Sky Lake-E VT-d [8086:2034] (rev 07)
16:05.2 System peripheral [0880]: Intel Corporation Sky Lake-E RAS Configuration Registers [8086:2035] (rev 07)
16:05.4 PIC [0800]: Intel Corporation Sky Lake-E IOxAPIC Configuration Registers [8086:2036] (rev 07)
16:08.0 System peripheral [0880]: Intel Corporation Sky Lake-E CHA Registers [8086:208d] (rev 07)
16:08.1 System peripheral [0880]: Intel Corporation Sky Lake-E CHA Registers [8086:208d] (rev 07)
16:08.2 System peripheral [0880]: Intel Corporation Sky Lake-E CHA Registers [8086:208d] (rev 07)
16:08.3 System peripheral [0880]: Intel Corporation Sky Lake-E CHA Registers [8086:208d] (rev 07)
16:08.4 System peripheral [0880]: Intel Corporation Sky Lake-E CHA Registers [8086:208d] (rev 07)
16:08.5 System peripheral [0880]: Intel Corporation Sky Lake-E CHA Registers [8086:208d] (rev 07)
16:08.6 System peripheral [0880]: Intel Corporation Sky Lake-E CHA Registers [8086:208d] (rev 07)
16:08.7 System peripheral [0880]: Intel Corporation Sky Lake-E CHA Registers [8086:208d] (rev 07)
16:09.0 System peripheral [0880]: Intel Corporation Sky Lake-E CHA Registers [8086:208d] (rev 07)
16:09.1 System peripheral [0880]: Intel Corporation Sky Lake-E CHA Registers [8086:208d] (rev 07)
16:09.2 System peripheral [0880]: Intel Corporation Sky Lake-E CHA Registers [8086:208d] (rev 07)
16:09.3 System peripheral [0880]: Intel Corporation Sky Lake-E CHA Registers [8086:208d] (rev 07)
16:09.4 System peripheral [0880]: Intel Corporation Sky Lake-E CHA Registers [8086:208d] (rev 07)
[... cut ...]
b2:16.4 System peripheral [0880]: Intel Corporation Sky Lake-E M2PCI Registers [8086:2018] (rev 07)
b2:16.5 Performance counters [1101]: Intel Corporation Sky Lake-E DDRIO Registers [8086:2088] (rev 07)
b2:17.0 System peripheral [0880]: Intel Corporation Sky Lake-E M2PCI Registers [8086:2018] (rev 07)
b2:17.1 Performance counters [1101]: Intel Corporation Sky Lake-E DDRIO Registers [8086:2088] (rev 07)
b3:00.0 Infiniband controller [0207]: Mellanox Technologies MT27700 Family [ConnectX-4] [15b3:1013]
```

cyring, 2022-01-17:
Quoting the Scalable/X datasheet volume 2:

> For Device 10 and 12 Functions 0-5 for offsets >= 256, PCIe extended configuration 
space are not designed for direct usage by OS or device drivers, and may not be 
accessible directly by OS components such as device drivers. The PCI Capability Pointer 
Register (CAPPTR) is set to a value of 40h. BIOS/firmware and/or BMC can access 
these registers, combine the information obtained with system implementation 
specifics, and if required, make it available to the OS through firmware and/or BMC 
interfaces.

This may sound like a tuff part.

cyring, 2022-01-17:
Refreshing the old and unsolved issue #146 

My notes:
[Source](https://elixir.bootlin.com/linux/latest/source/drivers/edac) codes of interest:
* skx_base.c
* i10nm_base.c
`I10NM_GET_IMC_BAR` base address resolved in function [`i10nm_get_ddr_munits`](https://elixir.bootlin.com/linux/latest/source/drivers/edac/i10nm_base.c#L234)

@wwyf : if you read the kernel debug trace when loading this EDAC driver, you may read the MMIO, next the IMC BAR; strings like:
```
socket# MMIO base

mc# MMIO base
```

wwyf, 2022-01-18:
Got it, I will try and reply the result.

wwyf, 2022-01-18:
This module is already loaded, and I find the log you want.

```
$ lsmod | grep edac
skx_edac               24576  0 
nfit                   77824  2 skx_edac

$ dmesg | grep -C 2 edac
[   25.828843] XFS (dm-2): Ending clean mount
[   25.828899] xfs filesystem being mounted at /home supports timestamps until 2038 (0x7fffffff)
[   26.014371] EDAC MC0: Giving out device to module skx_edac controller Skylake Socket#0 IMC#0: DEV 0000:64:0a.0 (INTERRUPT)
[   26.014435] EDAC MC1: Giving out device to module skx_edac controller Skylake Socket#0 IMC#1: DEV 0000:64:0c.0 (INTERRUPT)
[   26.131185] intel_rapl_common: Found RAPL domain package
[   26.131193] intel_rapl_common: Found RAPL domain dram
```

cyring, 2022-11-09:
IMC Timings still not found, but DRAM Power could be improved.

This register is not implemented yet in _CoreFreq_
[image]

Can you pull the [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch, refresh the whole CLI output `corefreq-cli -s -n -m -n -B -n -k`

... and also provide the `Sensors > Power` view, with some CPU load in parallel; I'm interested in the `RAM` energy consumed (left-bottom screen)
[image]

wwyf, 2022-11-12:
Got it. I will have a try.

cyring, 2022-11-12:
> Got it. I will have a try.

Be aware `Skylake/X` is crashing with `master` branch, and I believe your `Cascade Lake/X` is having the same registers set.
Please check #374 about that issue.