cyring/CoreFreq #319: Supermicro A1SAi with an Atom C2750 support

rbronco21, 2022-01-27:
Is the Supermicro A1SAi with an Atom C2750 supported? I get a blank screen without a prompt in SSH when running corefreq-cli. I get a blinking cursor when I run corefreq-cli -c. When I ctrl-c, I get this:
[image]
I posted on the unRAID forum and was directed here.

rbronco21, 2022-01-28:
This is going to be beyond me. If you are interested in the answer, I can work through it with you, but I'm fine if it's not worth the work. I believe I can boot the live cd through IPMI the next time my server is down, and I can run whatever commands you want and report back. Here is the relevant log and the results of corefreqd -d.

Jan 27 00:06:40 unRAID kernel: corefreqk: loading out-of-tree module taints kernel.
Jan 27 00:06:40 unRAID kernel: general protection fault: 0000 [#1] SMP PTI
Jan 27 00:06:40 unRAID kernel: CPU: 0 PID: 28661 Comm: modprobe Tainted: G           O      5.10.28-Unraid #1
Jan 27 00:06:40 unRAID kernel: Hardware name: Supermicro A1SAi/A1SAi, BIOS 2.2 11/22/2019
Jan 27 00:06:40 unRAID kernel: RIP: 0010:Intel_PowerInterface+0x26/0x3c [corefreqk]
Jan 27 00:06:40 unRAID kernel: Code: 33 00 00 c3 b9 06 06 00 00 0f 32 48 8b 35 1c 19 03 00 48 c1 e2 20 89 c0 48 09 c2 48 8b 0e 48 89 91 00 33 00 00 b9 14 06 00 00 <0f> 32 48 8b 0e 48 c1 e2 20 89 c0 48 09 c2 48 89 91 30 33 00 00 c3
Jan 27 00:06:40 unRAID kernel: RSP: 0018:ffffc9000856fad8 EFLAGS: 00010206
Jan 27 00:06:40 unRAID kernel: RAX: 00000000000a1003 RBX: ffffc9000856fbe8 RCX: 0000000000000614
Jan 27 00:06:40 unRAID kernel: RDX: 00000000000a1003 RSI: ffff8882f59cc9c0 RDI: ffff88810016d940
Jan 27 00:06:40 unRAID kernel: RBP: ffffc9000856fba8 R08: 0000000000000007 R09: 0000000000000007
Jan 27 00:06:40 unRAID kernel: R10: 0000000000000000 R11: 0000000000000000 R12: ffff8882f59cc9c0
Jan 27 00:06:40 unRAID kernel: R13: ffff8882f5e6358c R14: 0000000000000008 R15: ffffc9000856fe80
Jan 27 00:06:40 unRAID kernel: FS:  000014a17d4f7740(0000) GS:ffff88846fc00000(0000) knlGS:0000000000000000
Jan 27 00:06:40 unRAID kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jan 27 00:06:40 unRAID kernel: CR2: 000014a17cbfb000 CR3: 000000024b86c000 CR4: 00000000001006f0
Jan 27 00:06:40 unRAID kernel: Call Trace:
Jan 27 00:06:40 unRAID kernel: Controller_Init+0x8b/0x3b4 [corefreqk]
Jan 27 00:06:40 unRAID kernel: CoreFreqK_Ignition_Level_Up+0x32c/0x489 [corefreqk]
Jan 27 00:06:40 unRAID kernel: ? 0xffffffffa0660000
Jan 27 00:06:40 unRAID kernel: CoreFreqK_StartUp+0x4c/0xd7 [corefreqk]
Jan 27 00:06:40 unRAID kernel: ? CoreFreqK_Alloc_Public_Level_Up+0x5c/0x5c [corefreqk]
Jan 27 00:06:40 unRAID kernel: ? Query_Features+0x5cb/0x5cb [corefreqk]
Jan 27 00:06:40 unRAID kernel: ? CoreFreqK_Make_Device_Level_Up+0x35/0x35 [corefreqk]
Jan 27 00:06:40 unRAID kernel: ? CoreFreqK_Create_Device_Level_Up+0x54/0x54 [corefreqk]
Jan 27 00:06:40 unRAID kernel: ? CoreFreqK_Register_NMI+0x1bd/0x1bd [corefreqk]
Jan 27 00:06:40 unRAID kernel: ? CoreFreqK_Alloc_Private_Level_Up+0x3d/0x3d [corefreqk]
Jan 27 00:06:40 unRAID kernel: ? CoreFreqK_Alloc_Processor_RO_Level_Up+0x4f/0x4f [corefreqk]
Jan 27 00:06:40 unRAID kernel: ? CoreFreqK_Alloc_Processor_RW_Level_Up+0x47/0x47 [corefreqk]
Jan 27 00:06:40 unRAID kernel: ? CoreFreqK_ResetChip+0x1e/0x1e [corefreqk]
Jan 27 00:06:40 unRAID kernel: ? Compute_Interval+0x93/0x93 [corefreqk]
Jan 27 00:06:40 unRAID kernel: ? CoreFreqK_Alloc_Private_Cache_Level_Up+0x32/0x32 [corefreqk]
Jan 27 00:06:40 unRAID kernel: ? CoreFreqK_ProbePCI+0x8e/0x8e [corefreqk]
Jan 27 00:06:40 unRAID kernel: ? Define_CPUID+0x24/0x24 [corefreqk]
Jan 27 00:06:40 unRAID kernel: ? SMBIOS_Collect+0x1c2/0x1c2 [corefreqk]
Jan 27 00:06:40 unRAID kernel: ? Controller_Stop+0xe1/0xe1 [corefreqk]
Jan 27 00:06:40 unRAID kernel: CoreFreqK_Init+0x6/0x1000 [corefreqk]
Jan 27 00:06:40 unRAID kernel: do_one_initcall+0x71/0x162
Jan 27 00:06:40 unRAID kernel: ? do_init_module+0x19/0x1eb
Jan 27 00:06:40 unRAID kernel: ? kmem_cache_alloc+0x108/0x130
Jan 27 00:06:40 unRAID kernel: do_init_module+0x51/0x1eb
Jan 27 00:06:40 unRAID kernel: load_module+0x1b18/0x20cf
Jan 27 00:06:40 unRAID kernel: ? map_kernel_range_noflush+0xdf/0x255
Jan 27 00:06:40 unRAID kernel: ? __do_sys_init_module+0xc4/0x105
Jan 27 00:06:40 unRAID kernel: ? _cond_resched+0x1b/0x1e
Jan 27 00:06:40 unRAID kernel: __do_sys_init_module+0xc4/0x105
Jan 27 00:06:40 unRAID kernel: do_syscall_64+0x5d/0x6a
Jan 27 00:06:40 unRAID kernel: entry_SYSCALL_64_after_hwframe+0x44/0xa9
Jan 27 00:06:40 unRAID kernel: RIP: 0033:0x14a17d63a09a
Jan 27 00:06:40 unRAID kernel: Code: 48 8b 0d f9 7d 0c 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 49 89 ca b8 af 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d c6 7d 0c 00 f7 d8 64 89 01 48
Jan 27 00:06:40 unRAID kernel: RSP: 002b:00007ffcc315aac8 EFLAGS: 00000202 ORIG_RAX: 00000000000000af
Jan 27 00:06:40 unRAID kernel: RAX: ffffffffffffffda RBX: 0000000000427fe0 RCX: 000014a17d63a09a
Jan 27 00:06:40 unRAID kernel: RDX: 000000000041c368 RSI: 00000000000812e0 RDI: 000014a17cb7a010
Jan 27 00:06:40 unRAID kernel: RBP: 000014a17cb7a010 R08: 000000000042701a R09: 0000000000000001
Jan 27 00:06:40 unRAID kernel: R10: 0000000000427010 R11: 0000000000000202 R12: 000000000041c368
Jan 27 00:06:40 unRAID kernel: R13: 0000000000000000 R14: 000000000042f6a0 R15: 0000000000427fe0
Jan 27 00:06:40 unRAID kernel: Modules linked in: corefreqk(O+) xt_nat xt_tcpudp veth xt_conntrack xt_MASQUERADE nf_conntrack_netlink nfnetlink xt_addrtype iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 br_netfilter xfs dm_crypt dm_mod dax md_mod jc42 ipmi_devintf i915 video iosf_mbi drm_kms_helper drm backlight intel_gtt agpgart syscopyarea sysfillrect sysimgblt fb_sys_fops ip6table_filter ip6_tables iptable_filter ip_tables x_tables bonding igb i2c_algo_bit ipmi_ssif intel [... cut ...]
Jan 27 00:06:40 unRAID kernel: R10: 0000000000000000 R11: 0000000000000000 R12: ffff8882f59cc9c0
Jan 27 00:06:40 unRAID kernel: R13: ffff8882f5e6358c R14: 0000000000000008 R15: ffffc9000856fe80
Jan 27 00:06:40 unRAID kernel: FS:  000014a17d4f7740(0000) GS:ffff88846fc00000(0000) knlGS:0000000000000000
Jan 27 00:06:40 unRAID kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jan 27 00:06:40 unRAID kernel: CR2: 000014a17cbfb000 CR3: 000000024b86c000 CR4: 00000000001006f0

root@unRAID:~# corefreqd -d
Driver connection error code 16
/dev/corefreqk: 'Device or resource busy' @ line 7788

cyring, 2022-01-28:
Found the issue:
https://github.com/cyring/CoreFreq/blob/6fcba8da6d8a2de75a248c040b14f3cb75869cd3/corefreqk.c#L3345

MSR_PKG_POWER_INFO(0x614) is not part of Avoton

I will provide a fix in `develop` branch for your testings.

cyring, 2022-01-29:
@rbronco21 : Hello

I'm providing a hotfix for Avoton architecture.

For your testings, you have the choice between:
* cloning the branch `Hotfix_Avoton`
* trying image [`archlinux-corefreq-hotfix-avn.iso`](https://www.cyring.fr/live/archlinux-corefreq-hotfix-avn.iso)

cyring, 2022-02-04:
Can you plz give a try to the fix ?

rbronco21, 2022-02-05:
I really wanted to get it compiled, but it was beyond me. I ran the live cd in an unRAID VM and it looks better than before. Is there anything specific you want me to try in this VM? Is the VM an acceptable test?

[image]

This is as far as I got with compiling. I wouldn't mind a hint if you have an idea. I got the packages installed to get this far, but I wouldn't be surprised if I missed something.

cd /tmp
git clone -b Hotfix_Avoton --single-branch https://github.com/cyring/CoreFreq.git
cd CoreFreq
make

.
.
.
make -j1 -C /lib/modules/5.10.28-Unraid/build M=/tmp/CoreFreq modules
make[1]: Entering directory '/usr/src/linux-5.10.28-Unraid'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/src/linux-5.10.28-Unraid'
make: *** [Makefile:76: all] Error 2

cyring, 2022-02-05:
Thanks for giving a test.
In fact, _CoreFreq_ is aware about virtualization, it then won't probe the CPU very deep.
Only a bare-metal run will validate a good architecture support.

You already have the ISO: you now have to write it on an USB key and ask you BIOS to boot it.
There are many ways to flash the ISO; like `dd` within Linux; Rufus for Windows. 
See
https://github.com/cyring/CoreFreq/wiki/Live-CD

As far I can see in your building log above, you are missing the Kernel development stack: headers on so on.
unRAID is a distribution for Production; this might explain those missing files.
That's why the ISO image is filling all the prerequisites.

rbronco21, 2022-02-05:
Yeah, I just need time to shut down and reboot my server. Glad I got a good server mb so I can just mount the iso from my PC!
https://support.us.ovhcloud.com/hc/en-us/articles/360000108630-How-to-Install-an-OS-with-IPMI

cyring, 2022-02-05:
> Yeah, I just need time to shut down and reboot my server. Glad I got a good server mb so I can just mount the iso from my PC! https://support.us.ovhcloud.com/hc/en-us/articles/360000108630-How-to-Install-an-OS-with-IPMI

I'm also testing some architectures through the [Scaleway](https://www.scaleway.com/en/bare-metal-servers/) infrastructure.
Unfortunately AVOTON is not part of their servers.

Once again, thanks a lot for your contribution to the project.
Can't wait to see your results.

Regards,
CyrIng

ich777, 2022-02-05:
@rbronco21 compiled the files for you and built a unRAID plugin package, but it will only work when you are on unRAID v6.9.2 (Kernel v5.10.28), I will send you a message on the unRAID Forums shortly.

@cyring please redirect such users directly to me when some things should be compiled on unRAID since this can be a little bit complicated if you don't have the right setup.

cyring, 2022-02-05:
> @rbronco21 compiled the files for you and built a unRAID plugin package, but it will only work when you are on unRAID v6.9.2 (Kernel v5.10.28), I will send you a message on the unRAID Forums [...]

@ich777 sure I will. Thank you.

rbronco21, 2022-02-08:
[image]

cyring, 2022-02-08:
> [image]

Lovely!

Are you happy with those results ?

rbronco21, 2022-02-08:
I was looking for a better way to check frequency and temps so I only use the basic functionality, but it did that fine.

cyring, 2022-03-24:
Issue appears solved

closed 2022-03-24 by cyring