cyring/CoreFreq #321: Problems with launch on i7-10700

InvaderDolan, 2022-02-06:
Hi there! First of all, I really like CoreFreq, it's the only way to check RAM timings. I appreciate it.
I have successfully installed the CoreFreq on 4 different machines with R9 5950X (3 x Gigabyte B550 Aorus Master + 1 x ASUS ROG X570-E), on 5.11 kernel. And they're working like a charm.
But I have a problem on machine with i7-10700 (MSI MAG Tomahawk B450) with 5.13 kernel.
Here is the output when I `make`:
```
invader@invader-tomahawk:~/CoreFreq$ make
cc  -Wall -Wfatal-errors -pthread -c corefreqd.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreqd.o
cc  -Wall -Wfatal-errors -c corefreqm.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreqm.o
cc  -Wall -Wfatal-errors corefreqd.c corefreqm.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreqd -lpthread -lm -lrt
cc  -Wall -Wfatal-errors -c corefreq-cli.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0  \
  -o corefreq-cli.o
cc  -Wall -Wfatal-errors -c corefreq-ui.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreq-ui.o
cc  -Wall -Wfatal-errors -c corefreq-cli-rsc.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0  \
  -o corefreq-cli-rsc.o
cc  -Wall -Wfatal-errors -c corefreq-cli-json.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreq-cli-json.o
cc  -Wall -Wfatal-errors -c corefreq-cli-extra.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0 \
  -o corefreq-cli-extra.o
cc  -Wall -Wfatal-errors \
  corefreq-cli.c corefreq-ui.c corefreq-cli-rsc.c \
  corefreq-cli-json.c corefreq-cli-extra.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=5250000000 -D UBENCH=0  \
  -o corefreq-cli -lm -lrt
make -j1 -C /lib/modules/5.13.0-28-generic/build M=/home/invader/CoreFreq modules
make[1]: Entering directory '/usr/src/linux-headers-5.13.0-28-generic'
  CC [M]  /home/invader/CoreFreq/corefreqk.o
  MODPOST /home/invader/CoreFreq/Module.symvers
  CC [M]  /home/invader/CoreFreq/corefreqk.mod.o
  LD [M]  /home/invader/CoreFreq/corefreqk.ko
  BTF [M] /home/invader/CoreFreq/corefreqk.ko
Skipping BTF generation for /home/invader/CoreFreq/corefreqk.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.13.0-28-generic'
```
After this build, I do `sudo insmod corefreqk.ko`, `sudo ./corefreqd` and then `./corefreq-cli`, after that nothing happens, just insertion point blinks. 
I see the problem: `Skipping BTF generation for /home/invader/CoreFreq/corefreqk.ko due to unavailability of vmlinux` but couldn't fix it after googling.
I am pretty sure that the CoreFreq will work on 5.11 kernel because one of the machines on 5950X didn't work on 5.13 as well. But how can I fix it without changing kernel?

Thank you for your attention.

cyring, 2022-02-06:
Hello,

Thanks a lot for using _CoreFreq_
I've also no clue about this message, and after Googling found this [answer](https://forum.proxmox.com/threads/kernel-module-not-found-when-compile-skipping-btf-generation.100974/)  which use `dkms`

But `i7-10700` of Comet Lake may also not enough be implemented.
I just have received two of this architecture. See i5-10400 and i5-10600 in [Wiki](https://github.com/cyring/CoreFreq/wiki/CPU-support)
Can you check the kernel log for any driver error or crash ? and post the callflow and registers trap if any ?

cyring, 2022-02-12:
Are you struggling with the issue ?
Is Secured Boot the root cause ?

closed 2022-02-17 by cyring