cyring/CoreFreq #323: Intel Core i7-12700K on ASUS ROG STRIX Z690-G GAMING WIFI

ich777, 2022-02-15:
Just to let you know when I try to load the module on Alder Lake:

```
Feb 15 09:23:38 Tower kernel: corefreqk: loading out-of-tree module taints kernel.
Feb 15 09:23:38 Tower kernel: CoreFreq(4:5): Processor [ 06_97] Architecture [Alder Lake] SMT [20/20]
Feb 15 09:23:38 Tower kernel: general protection fault: 0000 [#1] SMP NOPTI
Feb 15 09:23:38 Tower kernel: CPU: 4 PID: 4089 Comm: modprobe Tainted: G           O      5.10.28-Unraid #1
Feb 15 09:23:38 Tower kernel: Hardware name: ASUS System Product Name/ROG STRIX Z690-G GAMING WIFI, BIOS 0811 12/16/2021
Feb 15 09:23:38 Tower kernel: RIP: 0010:Start_Uncore_Skylake+0x18/0x8f [corefreqk]
Feb 15 09:23:38 Tower kernel: Code: 01 00 00 b9 03 07 00 00 48 89 c2 48 c1 ea 20 0f 30 c3 48 8b 35 fd ca 03 00 48 8b 3e 80 bf 20 02 00 00 02 76 7b b9 01 0e 00 00 <0f> 32 48 c1 e2 20 89 c0 48 09 c2 48 89 d0 48 89 97 58 01 00 00 48
Feb 15 09:23:38 Tower kernel: RSP: 0018:ffffc90002c03ac0 EFLAGS: 00010006
Feb 15 09:23:38 Tower kernel: RAX: ffffffffa00aba14 RBX: ffff8881507b8000 RCX: 0000000000000e01
Feb 15 09:23:38 Tower kernel: RDX: 000003e1c4a12f52 RSI: ffff88810281a600 RDI: ffff8881507ac000
Feb 15 09:23:38 Tower kernel: RBP: 0000000000000004 R08: ffffff00ffffffff R09: 0000000000000049
Feb 15 09:23:38 Tower kernel: R10: 000005dc9a318481 R11: 000000168535cc8e R12: 00000000000000bc
Feb 15 09:23:38 Tower kernel: R13: 0000000000000000 R14: 0000000282d2bb61 R15: 000001f852bd9912
Feb 15 09:23:38 Tower kernel: FS:  000014f63bd77740(0000) GS:ffff88885f900000(0000) knlGS:0000000000000000
Feb 15 09:23:38 Tower kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Feb 15 09:23:38 Tower kernel: CR2: 000014f63b47b000 CR3: 0000000150292006 CR4: 0000000000770ee0
Feb 15 09:23:38 Tower kernel: PKRU: 55555554
Feb 15 09:23:38 Tower kernel: Call Trace:
Feb 15 09:23:38 Tower kernel: Start_Skylake+0x160/0x288 [corefreqk]
Feb 15 09:23:38 Tower kernel: ? irq_work_queue+0x13/0x19
Feb 15 09:23:38 Tower kernel: generic_exec_single+0x39/0x8b
Feb 15 09:23:38 Tower kernel: smp_call_function_single+0xb6/0xe5
Feb 15 09:23:38 Tower kernel: Controller_Start+0xbe/0xe1 [corefreqk]
Feb 15 09:23:38 Tower kernel: CoreFreqK_Ignition_Level_Up+0x3f7/0x489 [corefreqk]
Feb 15 09:23:38 Tower kernel: ? 0xffffffffa0081000
Feb 15 09:23:38 Tower kernel: CoreFreqK_StartUp+0x4c/0xd7 [corefreqk]
Feb 15 09:23:38 Tower kernel: ? CoreFreqK_Alloc_Public_Level_Up+0x5c/0x5c [corefreqk]
Feb 15 09:23:38 Tower kernel: ? Query_Features+0x5cb/0x5cb [corefreqk]
Feb 15 09:23:38 Tower kernel: ? CoreFreqK_Make_Device_Level_Up+0x35/0x35 [corefreqk]
Feb 15 09:23:38 Tower kernel: ? CoreFreqK_Create_Device_Level_Up+0x54/0x54 [corefreqk]
Feb 15 09:23:38 Tower kernel: ? CoreFreqK_Register_NMI+0x1bd/0x1bd [corefreqk]
Feb 15 09:23:38 Tower kernel: ? CoreFreqK_Alloc_Private_Level_Up+0x3d/0x3d [corefreqk]
Feb 15 09:23:38 Tower kernel: ? CoreFreqK_Alloc_Processor_RO_Level_Up+0x4f/0x4f [corefreqk]
Feb 15 09:23:38 Tower kernel: ? CoreFreqK_Alloc_Processor_RW_Level_Up+0x47/0x47 [corefreqk]
Feb 15 09:23:38 Tower kernel: ? CoreFreqK_ResetChip+0x1e/0x1e [corefreqk]
Feb 15 09:23:38 Tower kernel: ? Compute_Interval+0x93/0x93 [corefreqk]
Feb 15 09:23:38 Tower kernel: ? CoreFreqK_Alloc_Private_Cache_Level_Up+0x32/0x32 [corefreqk]
Feb 15 09:23:38 Tower kernel: ? CoreFreqK_ProbePCI+0x8e/0x8e [corefreqk]
Feb 15 09:23:38 Tower kernel: ? Define_CPUID+0x24/0x24 [corefreqk]
Feb 15 09:23:38 Tower kernel: ? SMBIOS_Collect+0x1c2/0x1c2 [corefreqk]
Feb 15 09:23:38 Tower kernel: ? Controller_Stop+0xe1/0xe1 [corefreqk]
Feb 15 09:23:38 Tower kernel: CoreFreqK_Init+0x6/0x1000 [corefreqk]
Feb 15 09:23:38 Tower kernel: do_one_initcall+0x71/0x162
Feb 15 09:23:38 Tower kernel: ? do_init_module+0x19/0x1eb
Feb 15 09:23:38 Tower kernel: ? kmem_cache_alloc+0x108/0x130
Feb 15 09:23:38 Tower kernel: do_init_module+0x51/0x1eb
Feb 15 09:23:38 Tower kernel: load_module+0x1b18/0x20cf
Feb 15 09:23:38 Tower kernel: ? map_kernel_range_noflush+0xdf/0x255
Feb 15 09:23:38 Tower kernel: ? __do_sys_init_module+0xc4/0x105
Feb 15 09:23:38 Tower kernel: ? _cond_resched+0x1b/0x1e
Feb 15 09:23:38 Tower kernel: __do_sys_init_module+0xc4/0x105
Feb 15 09:23:38 Tower kernel: do_syscall_64+0x5d/0x6a
Feb 15 09:23:38 Tower kernel: entry_SYSCALL_64_after_hwframe+0x44/0xa9
Feb 15 09:23:38 Tower kernel: RIP: 0033:0x14f63beba09a
Feb 15 09:23:38 Tower kernel: Code: 48 8b 0d f9 7d 0c 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 49 89 ca b8 af 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d c6 7d 0c 00 f7 d8 64 89 01 48
Feb 15 09:23:38 Tower kernel: RSP: 002b:00007ffd6791bfe8 EFLAGS: 00000206 ORIG_RAX: 00000000000000af
Feb 15 09:23:38 Tower kernel: RAX: ffffffffffffffda RBX: 0000000000427fe0 RCX: 000014f63beba09a
Feb 15 09:23:38 Tower kernel: RDX: 000000000041c368 RSI: 00000000000812f8 RDI: 000014f63b3fa010
Feb 15 09:23:38 Tower kernel: RBP: 000014f63b3fa010 R08: 000000000042701a R09: 0000000000000001
Feb 15 09:23:38 Tower kernel: R10: 0000000000427010 R11: 0000000000000206 R12: 000000000041c368
Feb 15 09:23:38 Tower kernel: R13: 0000000000000000 R14: 000000000042f6a0 R15: 0000000000427fe0
Feb 15 09:23:38 Tower kernel: Modules linked in: corefreqk(O+) md_mod ip6table_filter ip6_tables iptable_filter ip_tables x_tables bonding sr_mod cdrom wmi_bmof mlx4_en x86_pkg_temp_thermal intel_powerclamp coretemp crct10dif_pclmul crc32_pclmul crc32c_intel ghash_clmulni_intel aesni_intel crypto_simd cryptd glue_helper btusb btrtl btbcm btintel i2c_i801 ahci i2c_smbus mlx4_core i2c_core bluetooth libahci ecdh_generic ecc nvme input_leds led_class nvme_core fan thermal wmi video backlight acpi_pad button
Feb 15 09:23:38 Tower kernel: ---[ end trace d601d0ea2d38f7c0 ]---
Feb 15 09:23:38 Tower kernel: RIP: 0010:Start_Uncore_Skylake+0x18/0x8f [corefreqk]
Feb 15 09:23:38 Tower kernel: Code: 01 00 00 b9 03 07 00 00 48 89 c2 48 c1 ea 20 0f 30 c3 48 8b 35 fd ca 03 00 48 8b 3e 80 bf 20 02 00 00 02 76 7b b9 01 0e 00 00 <0f> 32 48 c1 e2 20 89 c0 48 09 c2 48 89 d0 48 89 97 58 01 00 00 48
Feb 15 09:23:38 Tower kernel: RSP: 0018:ffffc90002c03ac0 EFLAGS: 00010006
Feb 15 09:23:38 Tower kernel: RAX: ffffffffa00aba14 RBX: ffff8881507b8000 RCX: 0000000000000e01
Feb 15 09:23:38 Tower kernel: RDX: 000003e1c4a12f52 RSI: ffff88810281a600 RDI: ffff8881507ac000
Feb 15 09:23:38 Tower kernel: RBP: 0000000000000004 R08: ffffff00ffffffff R09: 0000000000000049
Feb 15 09:23:38 Tower kernel: R10: 000005dc9a318481 R11: 000000168535cc8e R12: 00000000000000bc
Feb 15 09:23:38 Tower kernel: R13: 0000000000000000 R14: 0000000282d2bb61 R15: 000001f852bd9912
Feb 15 09:23:38 Tower kernel: FS:  000014f63bd77740(0000) GS:ffff88885f900000(0000) knlGS:0000000000000000
Feb 15 09:23:38 Tower kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Feb 15 09:23:38 Tower kernel: CR2: 000014f63b47b000 CR3: 0000000150292006 CR4: 0000000000770ee0
Feb 15 09:23:38 Tower kernel: PKRU: 55555554
```
[lscpu.txt](https://github.com/cyring/CoreFreq/files/8073288/lscpu.txt)
[motherboard.txt](https://github.com/cyring/CoreFreq/files/8073290/motherboard.txt)

Please let me know if you need more information about the system or anything else.

cyring, 2022-02-15:
This MSR register (stored in `RCX`):

https://github.com/cyring/CoreFreq/blob/b9fa2c12f8b8c51a3399a7b5db1d01317f7fca61/intelmsr.h#L105

You can comment the body of this function:
https://github.com/cyring/CoreFreq/blob/b9fa2c12f8b8c51a3399a7b5db1d01317f7fca61/corefreqk.c#L13755
As below:
```c
static void Start_Uncore_Skylake(void *arg)
{
	UNUSED(arg);
/*
	Uncore_Counters_Set(SKL);
*/
}
```
Rebuild and load driver again.
(save your files before shooting)

ich777, 2022-02-15:
@cyring thank you for the quick response. Will do that hopefully tomorrow, got a pretty bad cold... :/

cyring, 2022-02-15:
> MSR_SKL_UNCORE_PERF_GLOBAL_CTRL

You will also have to:
1. comment the _stopping_ function at:
https://github.com/cyring/CoreFreq/blob/b9fa2c12f8b8c51a3399a7b5db1d01317f7fca61/corefreqk.c#L13762

... with:
```c
static void Stop_Uncore_Skylake(void *arg)
{
	UNUSED(arg);
/*
	Uncore_Counters_Clear(SKL);
*/
}
```
2. remove the Uncore counters at:
https://github.com/cyring/CoreFreq/blob/b9fa2c12f8b8c51a3399a7b5db1d01317f7fca61/corefreqk.c#L10882

... with:
```c
#define PKG_Counters_Skylake(Core, T)					\
({									\
    RDTSCP_COUNTERx4(PUBLIC(RO(Proc))->Counter[T].PTSC,			\
		MSR_PKG_C2_RESIDENCY, PUBLIC(RO(Proc))->Counter[T].PC02,\
		MSR_PKG_C3_RESIDENCY, PUBLIC(RO(Proc))->Counter[T].PC03,\
		MSR_PKG_C6_RESIDENCY, PUBLIC(RO(Proc))->Counter[T].PC06,\
		MSR_PKG_C7_RESIDENCY, PUBLIC(RO(Proc))->Counter[T].PC07);\
})
```

**Take care of yourself**

cyring, 2022-02-16:
# Programming notes
>  Table 2-49. Uncore PMU MSRs Supported by 12th Generation Intel® Core™ Processors
## `MSR_UNC_PERF_GLOBAL_CTRL`
[image]

ich777, 2022-02-16:
Hope this is what you are after:
[image]
[image]
[image]
[image]
[image]
[image]

`dmidecode -t memory`:

```
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.
# SMBIOS implementations newer than version 3.3.0 are not
# fully supported by this version of dmidecode.

Handle 0x0045, DMI type 16, 23 bytes
Physical Memory Array
	Location: System Board Or Motherboard
	Use: System Memory
	Error Correction Type: None
	Maximum Capacity: 64 GB
	Error Information Handle: Not Provided
	Number Of Devices: 4

Handle 0x0046, DMI type 17, 92 bytes
Memory Device
	Array Handle: 0x0045
	Error Information Handle: Not Provided
	Total Width: Unknown
	Data Width: Unknown
	Size: No Module Installed
	Form Factor: Unknown
	Set: None
	Locator: Controller0-DIMM0
	Bank Locator: BANK 0
	Type: Unknown
	Type Detail: None
	Speed: Unknown
	Manufacturer: Not Specified
	Serial Number: Not Specified
	Asset Tag: Not Specified
	Part Number: Not Specified
	Rank: Unknown
	Configured Memory Speed: Unknown
	Minimum Voltage: Unknown
	Maximum Voltage: Unknown
	Configured Voltage: Unknown
	Memory Technology: <OUT OF SPEC>
	Memory Operating Mode Capability: None
	Firmware Version: Not Specified
	Module Manufacturer ID: Unknown
	Module Product ID: Unknown
	Memory Subsystem Controller Manufacturer ID: Unknown
	Memory Subsystem Controller Product ID: Unknown
	Non-Volatile Size: None
	Volatile Size: None
	Cache Size: None
	Logical Size: None

Handle 0x0047, DMI type 17, 92 bytes
Memory Device
	Array Handle: 0x0045
	Error Information Handle: Not Provided
	Total Width: 64 bits
	Data Width: 64 bits
	Size: 16 GB
	Form Factor: DIMM
	Set: None
	Locator: Controller0-DIMM1
	Bank Locator: BANK 0
	Type: <OUT OF SPEC>
	Type Detail: Synchronous
	Speed: 4800 MT/s
	Manufacturer: Corsair
	Serial Number: 00000000
	Asset Tag: 9876543210
	Part Number: CMK32GX5M2A4800C40  
	Rank: 1
	Configured Memory Speed: 4800 MT/s
	Minimum Voltage: 1.1 V
	Maximum Voltage: 1.1 V
	Configured Voltage: 1.1 V
	Memory Technology: DRAM
	Memory Operating Mode Capability: Volatile memory
	Firmware Version: Not Specified
	Module Manufacturer ID: Bank 3, Hex 0x9E
	Module Product ID: Unknown
	Memory Subsystem Controller Manufacturer ID: Unknown
	Memory Subsystem Controller Product ID: Unknown
	Non-Volatile Size: None
	Volatile Size: 16 GB
	Cache Size: None
	Logical Size: None

Handle 0x0048, DMI type 17, 92 bytes
Memory Device
	Array Handle: 0x0045
	Error Information Handle: Not Provided
	Total Width: Unknown
	Data Width: Unknown
	Size: No Module Installed
	Form Factor: Unknown
	Set: None
	Locator: Controller1-DIMM0
	Bank Locator: BANK 0
	Type: Unknown
	Type Detail: None
	Speed: Unknown
	Manufacturer: Not Specified
	Serial Number: Not Specified
	Asset Tag: Not Specified
	Part Number: Not Specified
	Rank: Unknown
	Configured Memory Speed: Unknown
	Minimum Voltage: Unknown
	Maximum Voltage: Unknown
	Configured Voltage: Unknown
	Memory Technology: <OUT OF SPEC>
	Memory Operating Mode Capability: Volatile memory
	Firmware Version: Not Specified
	Module Manufacturer ID: Unknown
	Module Product ID: Unknown
	Memory Subsystem Controller Manufacturer ID: Unknown
	Memory Subsystem Controller Product ID: Unknown
	Non-Volatile Size: None
	Volatile Size: None
	Cache Size: None
	Logical Size: None

Handle 0x0049, DMI type 17, 92 bytes
Memory Device
	Array Handle: 0x0045
	Error Information Handle: Not Provided
	Total Width: 64 bits
	Data Width: 64 bits
	Size: 16 GB
	Form Factor: DIMM
	Set: None
	Locator: Controller1-DIMM1
	Bank Locator: BANK 0
	Type: <OUT OF SPEC>
	Type Detail: Synchronous
	Speed: 4800 MT/s
	Manufacturer: Corsair
	Serial Number: 00000000
	Asset Tag: 9876543210
	Part Number: CMK32GX5M2A4800C40  
	Rank: 1
	Configured Memory Speed: 4800 MT/s
	Minimum Voltage: 1.1 V
	Maximum Voltage: 1.1 V
	Configured Voltage: 1.1 V
	Memory Technology: DRAM
	Memory Operating Mode Capability: Volatile memory
	Firmware Version: Not Specified
	Module Manufacturer ID: Bank 3, Hex 0x9E
	Module Product ID: Unknown
	Memory Subsystem Controller Manufacturer ID: Unknown
	Memory Subsystem Controller Product ID: Unknown
	Non-Volatile Size: None
	Volatile Size: 16 GB
	Cache Size: None
	Logical Size: None
```

If you need anything else let me know.

cyring, 2022-02-16:
> Hope this is what you are after:
> ...
> If you need anything else let me know.

Oh Oh ! welcome to Alder Lake.

So Uncore counters are the missing part to complete this architecture. 

Can you test those registers that way ?
```sh
## New ADL MSR_UNC_PERF_GLOBAL_CTRL
rdmsr -aX 0x2ff0

## MSR_NHM_UNCORE_PERF_FIXED_CTR0
rdmsr -aX 0x00000394

## MSR_SNB_UNCORE_PERF_FIXED_CTR0 or MSR_SKL_UNCORE_PERF_FIXED_CTR0
rdmsr -aX  0x00000395

## MSR_SNB_EP_UNCORE_PERF_FIXED_CTR0
rdmsr -aX 0x00000c09

## MSR_HSW_EP_UNCORE_PERF_FIXED_CTR0
rdmsr -aX 0x00000704
```
_Remark_: you may need:
1. `msr.allow_writes=on` in your boot command line
2. `modprobe msr`
3. MSR-Tools package

ich777, 2022-02-16:
Sure thing!

`rdmsr -aX 0x2ff0`:
```
CPU 0: 20000000
CPU 1: 20000000
CPU 2: 20000000
CPU 3: 20000000
CPU 4: 20000000
CPU 5: 20000000
CPU 6: 20000000
CPU 7: 20000000
CPU 8: 20000000
CPU 9: 20000000
CPU 10: 20000000
CPU 11: 20000000
CPU 12: 20000000
CPU 13: 20000000
CPU 14: 20000000
CPU 15: 20000000
CPU 16: 20000000
CPU 17: 20000000
CPU 18: 20000000
CPU 19: 20000000
```

`rdmsr -aX 0x00000394`:
`rdmsr: CPU 0 cannot read MSR 0x00000394`

`rdmsr -aX  0x00000395`:
`rdmsr: CPU 0 cannot read MSR 0x00000395`

`rdmsr -aX 0x00000c09`:
`rdmsr: CPU 0 cannot read MSR 0x00000c09`

`rdmsr -aX 0x00000704`:
`rdmsr: CPU 0 cannot read MSR 0x00000704`

cyring, 2022-02-16:
> Sure thing!
>
> `rdmsr -aX 0x2ff0`:
>
> ```
> CPU 0: 20000000
> CPU 1: 20000000
> CPU 2: 20000000
> CPU 3: 20000000
> CPU 4: 20000000
> CPU 5: 20000000
> CPU 6: 20000000
> CPU 7: 20000000
> CPU 8: 20000000
> CPU 9: 20000000 [...]

Great, we now have this new register to enable/disable counting but no register found yet to read count value.

Do you also have some means to screenshot your BIOS Memory Controller for its current settings:
* Bus and DRAM frequencies
* DIMM topology & size
* Memory timings

cyring, 2022-02-16:
Here what we need:
[image]

Those registers should be safely read, even if some return zero so far.
```sh
##  MSR_UNC_PERF_FIXED_CTRL
rdmsr -aX 0x2FDE

## MSR_UNC_PERF_FIXED_CTR
rdmsr -aX 0x2FDF

## MSR_UNC_PERF_GLOBAL_STATUS
rdmsr -aX 0x2FF2
```

ich777, 2022-02-16:
I hope that's all you need:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ich777, 2022-02-16:
`rdmsr -aX 0x2FDE`:
```
CPU 0: 0
CPU 1: 0
CPU 2: 0
CPU 3: 0
CPU 4: 0
CPU 5: 0
CPU 6: 0
CPU 7: 0
CPU 8: 0
CPU 9: 0
CPU 10: 0
CPU 11: 0
CPU 12: 0
CPU 13: 0
CPU 14: 0
CPU 15: 0
CPU 16: 0
CPU 17: 0
CPU 18: 0
CPU 19: 0

```
`rdmsr -aX 0x2FDF`:
```
CPU 0: 0
CPU 1: 0
CPU 2: 0
CPU 3: 0
CPU 4: 0
CPU 5: 0
CPU 6: 0
CPU 7: 0
CPU 8: 0
CPU 9: 0
CPU 10: 0
CPU 11: 0
CPU 12: 0
CPU 13: 0
CPU 14: 0
CPU 15: 0
CPU 16: 0
CPU 17: 0
CPU 18: 0
CPU 19: 0

```
`rdmsr -aX 0x2FF2`:
```
CPU 0: 0
CPU 1: 0
CPU 2: 0
CPU 3: 0
CPU 4: 0
CPU 5: 0
CPU 6: 0
CPU 7: 0
CPU 8: 0
CPU 9: 0
CPU 10: 0
CPU 11: 0
CPU 12: 0
CPU 13: 0
CPU 14: 0
CPU 15: 0
CPU 16: 0
CPU 17: 0
CPU 18: 0
CPU 19: 0
```

cyring, 2022-02-16:
Yes we are on track.
I have enough materials to start programming some ADL specifics. They'll be part of the `develop` branch.

By the way, sounds like a powerful motherboard. Congrats.

ich777, 2022-02-16:
Nice, keep me updated and I will compile the develop branch when everything is ready.

> By the way, sounds like a powerful motherboard. Congrats.

Thank you, buts adly enough my motherboard from my previous Skylake system died otherwise I hadn't buyed that monster... 😆 
I was not that far off to build a system with 10th gen but I pulled the trigger on the 12th gen.

cyring, 2022-02-16:
Can you please give a try to the [`Hotfix_AlderLake`](https://github.com/cyring/CoreFreq/tree/Hotfix_AlderLake) branch.
This branch is just made to avoid the ADL crash and will be merged in `master` as version `1.89.4`

ich777, 2022-02-16:
Got a instant panic with the Hotfix_AlderLake branch.

```
Feb 16 10:52:00 Tower kernel: corefreqk: loading out-of-tree module taints kernel.
Feb 16 10:52:00 Tower kernel: CoreFreq(8:9): Processor [ 06_97] Architecture [Alder Lake] SMT [20/20]
Feb 16 10:52:00 Tower kernel: general protection fault, maybe for address 0x0: 0000 [#1] SMP NOPTI
Feb 16 10:52:00 Tower kernel: CPU: 8 PID: 1969 Comm: modprobe Tainted: G           O      5.15.16-Unraid #1
Feb 16 10:52:00 Tower kernel: Hardware name: ASUS System Product Name/ROG STRIX Z690-G GAMING WIFI, BIOS 0811 12/16/2021
Feb 16 10:52:00 Tower kernel: RIP: 0010:Start_Skylake+0x1cf/0x28d [corefreqk]
Feb 16 10:52:00 Tower kernel: Code: f9 03 00 00 0f 32 48 c1 e2 20 48 09 d0 49 89 c5 48 c7 c1 fa 03 00 00 0f 32 48 c1 e2 20 48 09 d0 49 89 c6 48 c7 c1 95 03 00 00 <0f> 32 48 c1 e2 20 48 09 d0 49 89 c7 4c 89 16 4c 89 5e 08 4c 89 66
Feb 16 10:52:00 Tower kernel: RSP: 0018:ffffc9000128bbf8 EFLAGS: 00010046
Feb 16 10:52:00 Tower kernel: RAX: 0000000000000000 RBX: ffff888101138000 RCX: 0000000000000395
Feb 16 10:52:00 Tower kernel: RDX: 0000000000000000 RSI: ffff888156170000 RDI: 0000000000000000
Feb 16 10:52:00 Tower kernel: RBP: 0000000000000008 R08: ffffff00ffffffff R09: 0000000000000049
Feb 16 10:52:00 Tower kernel: R10: 0000004a55c6b565 R11: 0000000d1e184272 R12: 0000000000000000
Feb 16 10:52:00 Tower kernel: R13: 0000000000000000 R14: 0000000000000000 R15: 00000021bdf24cb4
Feb 16 10:52:00 Tower kernel: FS:  00001524467ca740(0000) GS:ffff88885fa00000(0000) knlGS:0000000000000000
Feb 16 10:52:00 Tower kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Feb 16 10:52:00 Tower kernel: CR2: 000055be9bef1048 CR3: 0000000104e6c004 CR4: 0000000000770ee0
Feb 16 10:52:00 Tower kernel: PKRU: 55555554
Feb 16 10:52:00 Tower kernel: Call Trace:
Feb 16 10:52:00 Tower kernel: <TASK>
Feb 16 10:52:00 Tower kernel: ? Start_Skylake_X+0x2aa/0x2aa [corefreqk]
Feb 16 10:52:00 Tower kernel: ? Start_Skylake_X+0x2aa/0x2aa [corefreqk]
Feb 16 10:52:00 Tower kernel: generic_exec_single+0x3d/0x9c
Feb 16 10:52:00 Tower kernel: smp_call_function_single+0xc2/0xf7
Feb 16 10:52:00 Tower kernel: Controller_Start+0xc3/0xe6 [corefreqk]
Feb 16 10:52:00 Tower kernel: CoreFreqK_Ignition_Level_Up+0x3fc/0x48e [corefreqk]
Feb 16 10:52:00 Tower kernel: ? 0xffffffffa0062000
Feb 16 10:52:00 Tower kernel: CoreFreqK_StartUp+0x4e/0xdc [corefreqk]
Feb 16 10:52:00 Tower kernel: ? CoreFreqK_Alloc_Public_Level_Up+0x61/0x61 [corefreqk]
Feb 16 10:52:00 Tower kernel: ? Query_Features+0x5d0/0x5d0 [corefreqk]
Feb 16 10:52:00 Tower kernel: ? CoreFreqK_Make_Device_Level_Up+0x3a/0x3a [corefreqk]
Feb 16 10:52:00 Tower kernel: ? CoreFreqK_Create_Device_Level_Up+0x59/0x59 [corefreqk]
Feb 16 10:52:00 Tower kernel: ? CoreFreqK_Register_NMI+0x1c2/0x1c2 [corefreqk]
Feb 16 10:52:00 Tower kernel: ? CoreFreqK_Alloc_Private_Level_Up+0x42/0x42 [corefreqk]
Feb 16 10:52:00 Tower kernel: ? CoreFreqK_Alloc_Processor_RO_Level_Up+0x4d/0x4d [corefreqk]
Feb 16 10:52:00 Tower kernel: ? CoreFreqK_Alloc_Processor_RW_Level_Up+0x51/0x51 [corefreqk]
Feb 16 10:52:00 Tower kernel: ? CoreFreqK_ResetChip+0x23/0x23 [corefreqk]
Feb 16 10:52:00 Tower kernel: ? Compute_Interval+0x98/0x98 [corefreqk]
Feb 16 10:52:00 Tower kernel: ? CoreFreqK_Alloc_Private_Cache_Level_Up+0x37/0x37 [corefreqk]
Feb 16 10:52:00 Tower kernel: ? CoreFreqK_ProbePCI+0x93/0x93 [corefreqk]
Feb 16 10:52:00 Tower kernel: ? Define_CPUID+0x29/0x29 [corefreqk]
Feb 16 10:52:00 Tower kernel: ? SMBIOS_Collect+0x1c7/0x1c7 [corefreqk]
Feb 16 10:52:00 Tower kernel: ? Controller_Stop+0xe6/0xe6 [corefreqk]
Feb 16 10:52:00 Tower kernel: CoreFreqK_Init+0xb/0x1000 [corefreqk]
Feb 16 10:52:00 Tower kernel: do_one_initcall+0x75/0x178
Feb 16 10:52:00 Tower kernel: ? do_init_module+0x23/0x218
Feb 16 10:52:00 Tower kernel: ? kmem_cache_alloc_trace+0x11f/0x146
Feb 16 10:52:00 Tower kernel: do_init_module+0x5b/0x218
Feb 16 10:52:00 Tower kernel: __do_sys_init_module+0xb6/0xf5
Feb 16 10:52:00 Tower kernel: do_syscall_64+0x80/0xa5
Feb 16 10:52:00 Tower kernel: entry_SYSCALL_64_after_hwframe+0x44/0xae
Feb 16 10:52:00 Tower kernel: RIP: 0033:0x15244690112a
Feb 16 10:52:00 Tower kernel: Code: 48 8b 0d 41 8d 0c 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 49 89 ca b8 af 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 0e 8d 0c 00 f7 d8 64 89 01 48
Feb 16 10:52:00 Tower kernel: RSP: 002b:00007ffc0828ac68 EFLAGS: 00000202 ORIG_RAX: 00000000000000af
Feb 16 10:52:00 Tower kernel: RAX: ffffffffffffffda RBX: 00000000004290e0 RCX: 000015244690112a
Feb 16 10:52:00 Tower kernel: RDX: 000000000041d268 RSI: 000000000008c148 RDI: 0000152445e3f010
Feb 16 10:52:00 Tower kernel: RBP: 0000152445e3f010 R08: 0000000000000007 R09: 0000000000429050
Feb 16 10:52:00 Tower kernel: R10: 0000000000000002 R11: 0000000000000202 R12: 000000000041d268
Feb 16 10:52:00 Tower kernel: R13: 0000000000000000 R14: 0000000000430560 R15: 00000000004290e0
Feb 16 10:52:00 Tower kernel: </TASK>
Feb 16 10:52:00 Tower kernel: Modules linked in: corefreqk(O+) ip6table_filter ip6_tables iptable_filter ip_tables x_tables bonding sr_mod cdrom wmi_bmof x86_pkg_temp_thermal intel_powerclamp coretemp crct10dif_pclmul crc32_pclmul crc32c_intel ghash_clmulni_intel aesni_intel crypto_simd cryptd rapl intel_cstate intel_uncore mlx4_en i2c_i801 i2c_smbus i2c_core mlx4_core btusb btrtl btbcm btintel bluetooth ecdh_generic ecc input_leds led_class nvme ahci libahci nvme_core thermal fan wmi video backlight tpm_crb tpm_tis tp [... cut ...]
Feb 16 10:52:00 Tower kernel: R10: 0000004a55c6b565 R11: 0000000d1e184272 R12: 0000000000000000
Feb 16 10:52:00 Tower kernel: R13: 0000000000000000 R14: 0000000000000000 R15: 00000021bdf24cb4
Feb 16 10:52:00 Tower kernel: FS:  00001524467ca740(0000) GS:ffff88885fa00000(0000) knlGS:0000000000000000
Feb 16 10:52:00 Tower kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Feb 16 10:52:00 Tower kernel: CR2: 000055be9bef1048 CR3: 0000000104e6c004 CR4: 0000000000770ee0
Feb 16 10:52:00 Tower kernel: PKRU: 55555554
```

cyring, 2022-02-16:
> Got a instant panic with the Hotfix_AlderLake branch.
> 
> ```
> Feb 16 10:52:00 Tower kernel: corefreqk: loading out-of-tree module taints kernel.
> Feb 16 10:52:00 Tower kernel: CoreFreq(8:9): Processor [ 06_97] Architecture [Alder Lake] SMT [20/20]
> ...
> Feb 16 10:52:00 Tower kernel: RSP: 0018:ffffc9000128bbf8 EFLAGS: 00010046
> Feb 16 10:52:00 Tower kernel: RAX: 0000000000000000 RBX: ffff888101138000 RCX: 0000000000000395
> Feb 16 10:52:00 Tower kernel: RDX: 0000000000000000 RSI: ffff888156170000 RDI: 0000000000000000
> Feb 16 10:52:00 Tower kernel: RBP: 0000000000000008 R08: ffffff00ffffffff R09: 0000000000000049
> Feb 16 10:52:00 Tower kernel: R10: 0000004a55c6b565 R11: 0000000d1e184272 R12: 0000000000000000
> Feb 16 10:52:00 Tower kernel: R13: 0000000000000000 R14: 0000000000000000 R15: 00000021bdf24cb4
> Feb 16 10:52:00 Tower kernel: FS:  00001524467ca740(0000) GS:ffff88885fa00000(0000) knlGS:0000000000000000
> Feb 16 10:52:00 Tower kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
> Feb 16 10:52:00 Tower kernel: CR2: 000055be9bef1048 CR3: 0000000104e6c004 CR4: 0000000000770ee0
> ...
> ```

Can you rebuild and try latest commit of the same branch ?
If OK, can you then unload and restart _CoreFreq_ to make sure it is stable ?

Thank you

ich777, 2022-02-17:
> If OK, can you then unload and restart _CoreFreq_ to make sure it is stable ?

Sure thing, seems to be working as intended:
```
root@Tower:~# lsmod | grep corefreqk
corefreqk             327680  0
root@Tower:~# modprobe -r corefreqk
root@Tower:~# lsmod | grep corefreq
root@Tower:~# modprobe corefreqk
root@Tower:~# lsmod | grep corefreq
corefreqk             327680  0
```

> Can you rebuild and try latest commit of the same branch ?

Seems to be also working as intended:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ich777, 2022-02-17:
@cyring one short off topic question is it still required to pass over: `MSR_CORE_PERF_UC=MSR_CORE_PERF_FIXED_CTR1 MSR_CORE_PERF_URC=MSR_CORE_PERF_FIXED_CTR2` to the make command for Intel or is this now deprecated?

cyring, 2022-02-17:
Thank you, I will add the `i7-12700K` to the Wiki

All changes are now merged into `master`; archive [`1.89.4`](https://github.com/cyring/CoreFreq/releases/tag/1.89.4) is ready for packaging purposes.

ISO [images](https://www.cyring.fr/) are also updated accordingly.

Next steps in `develop` branch will consist in:
* activating the Uncore counters
* add/remove ADL Core & Package specific C-states
* debugging the Memory Controller results facing the BIOS settings
* testing/enhancing the Power Limiters, Clock Limiters, technologies ...
* research on hybrid architecture: C-states, frequencies, topology tags and any supported data which differs between the P-Core and E-Core ...

See above screenshot: the capability of the `HWP` highest frequency is `6.4 GHz` ; really !

ich777, 2022-02-17:
As always, thank you @cyring!

Already compiled the custom packages for unRAID, hopefully I will get you a 12600k too, already contacted a member on the unRAID forums. :)

cyring, 2022-02-20:
@ich777 Hello

For your testings are available the Uncore Counters in the dedicated branch [`develop_AlderLake_uncore`](https://github.com/cyring/CoreFreq/tree/develop_AlderLake_uncore)

Once, fully rebuild, reload and start, you switch `corefreq-cli` to the `Package cycles` view where `UNCORE:` should count, depending on some load.

Thank you

cyring, 2022-02-20:
# MSR_MC6_RESIDENCY_COUNTER

This register is found on Silvermont and Airmont but this time it is also specified for the 12th architecture Alder Lake.

I imagine it exists because of the E-Cores which are somehow a heritage of Atom micro-arch.

Whatever, let's try to read this register:
```sh
rdmsr -aX 0x664
```

Depending on some load, I'm expecting one of those results:
1. values on E-Cores only; P-Cores remain zero
2. values on both P and E-Cores

# Core C3 and C1 Residency Counters
Those registers are specified for the 8th Cannon Lake micro-architecture  

**experimental**
```sh
## MSR_CORE_C3_RESIDENCY
rdmsr -aX 0x662

## MSR_CORE_C1_RESIDENCY
rdmsr -aX 0x660
```

cyring, 2022-02-20:
# Hybrid - Stability test

Referring to [screenshot](https://user-images.githubusercontent.com/11563789/154823107-72d62958-78b2-418a-8ae8-1d101f6646ff.png), CPU `[0-15]` are part of the P-cores whereas CPU `[16-19]` are the E-Cores.  

I would like to bind the Driver collect and Daemon aggregation to a couple of E-Cores:

```
insmod corefreqk.ko ServiceProcessor=18
```
Next, start the Daemon and the Client:

Q: does `corefreq-cli` highlight the CPU `18` and `19` in the first column ?
Q: are you getting `L1` and `L2` cache size specific to the E-Core ?
Q: is there a gap between the estimated Base Clocks of P and E ?
Q: please screenshot the main view and post the output of `corefreq-cli -s -n -m`

Q: are you able to change the Turbo Boost ratio `1C`  from `Processor` window ?
  * for the whole processor (P and E Cores) ?
  * per CPU number (either P or E) ?
  _(you may have to stress the single Core during the change to observe the effect)_

cyring, 2022-02-20:
# Programming notes
## [Datasheet, Volume 2](https://cdrdv2.intel.com/v1/dl/getContent/655259)
## [Errata](https://cdrdv2.intel.com/v1/dl/getContent/682436)

> ## ADL026 - Reading The PPERF MSR May Not Return Correct Values
> Under complex micro-architectural conditions, RDMSR instruction to Productive 
Performance (MSR_PPERF) MSR (Offset 64eh) may not return correct values in the 
upper 32 bits (EDX register) if Core C6 is enabled.

> ## ADL010 - No `#GP` Will be Signaled When Setting `MSR_MISC_PWR_MGMT.ENABLE_SDC` if `MSR_MISC_PWR_MGMT.LOCK` is Set
>

ich777, 2022-02-20:
> Once, fully rebuild, reload and start, you switch `corefreq-cli` to the `Package cycles` view where `UNCORE:` should count, depending on some load.

Yes it counts up or down depending on the load:
[image]

```
root@Tower:/# rdmsr -aX 0x664
CPU 0: 0
CPU 1: 0
CPU 2: 0
CPU 3: 0
CPU 4: 0
CPU 5: 0
CPU 6: 0
CPU 7: 0
CPU 8: 0
CPU 9: 0
CPU 10: 0
CPU 11: 0
CPU 12: 0
CPU 13: 0
CPU 14: 0
CPU 15: 0
CPU 16: 11D84C6C11E
CPU 17: 11D84C6C11E
CPU 18: 11D84C6C11E
CPU 19: 11D84C6C11E
```

```
root@Tower:/# rdmsr -aX 0x662
rdmsr: CPU 0 cannot read MSR 0x00000662
root@Tower:/# rdmsr -aX 0x660
CPU 0: AD920F882
CPU 1: AD920FAB6
CPU 2: 29E27B6C12
CPU 3: 29E27B6E46
CPU 4: 3048CAE36A
CPU 5: 3048CAE36A
CPU 6: 2EC5093E9E
CPU 7: 2EC5093E9E
CPU 8: 2D83D2CF0E
CPU 9: 2D83D2CF0E
CPU 10: 295014FC50
CPU 11: 295014FC50
CPU 12: 292BE78AFE
CPU 13: 292BE78AFE
CPU 14: 27CD9BFA9E
CPU 15: 27CD9BFA9E
CPU 16: 27033BACA8
CPU 17: 259C125224
CPU 18: 2678FC6A32
CPU 19: 25AC3B3688
```

> Q: are you getting `L1` and `L2` cache size specific to the E-Core ?

Yes, those are reported correct.

> Q: is there a gap between the estimated Base Clocks of P and E ?

I don't know exactly what you mean but when I stress the whole CPU all cores run at max turbo (P: 4,9Ghz | E: 3,8Ghz)

> Q: does `corefreq-cli` highlight the CPU `18` and `19` in the first column ?

It highlights cores 000 and 001 if I get that right:
[image]

> Q: are you able to change the Turbo Boost ratio `1C` from `Processor` window ?

[... 248 lines cut ...]
   |- Window                                            second   [  0.000976562]

CPU Pkg  Apic  Core/Thread  Caches      (w)rite-Back (i)nclusive              
 #   ID   ID    ID     ID  L1-Inst Way  L1-Data Way      L2  Way      L3  Way 
000:BSP    0     0      0    32768  8     49152 12   1310720 10  26214400 10  
001:  0    1     0      1    32768  8     49152 12   1310720 10  26214400 10  
002:  0    8     4      0    32768  8     49152 12   1310720 10  26214400 10  
003:  0    9     4      1    32768  8     49152 12   1310720 10  26214400 10  
004:  0   16     8      0    32768  8     49152 12   1310720 10  26214400 10  
005:  0   17     8      1    32768  8     49152 12   1310720 10  26214400 10  
006:  0   24    12      0    32768  8     49152 12   1310720 10  26214400 10  
007:  0   25    12      1    32768  8     49152 12   1310720 10  26214400 10  
008:  0   32    16      0    32768  8     49152 12   1310720 10  26214400 10  
009:  0   33    16      1    32768  8     49152 12   1310720 10  26214400 10  
010:  0   40    20      0    32768  8     49152 12   1310720 10  26214400 10  
011:  0   41    20      1    32768  8     49152 12   1310720 10  26214400 10  
012:  0   48    24      0    32768  8     49152 12   1310720 10  26214400 10  
013:  0   49    24      1    32768  8     49152 12   1310720 10  26214400 10  
014:  0   56    28      0    32768  8     49152 12   1310720 10  26214400 10  
015:  0   57    28      1    32768  8     49152 12   1310720 10  26214400 10  
016:  0   72    36      0    65536  8     32768  8   2097152 16  26214400 10  
017:  0   74    37      0    65536  8     32768  8   2097152 16  26214400 10  
018:  0   76    38      0    65536  8     32768  8   2097152 16  26214400 10  
019:  0   78    39      0    65536  8     32768  8   2097152 16  26214400 10 
```

cyring, 2022-02-20:
Happy to see Uncore counter. Thank you.

MC6 and C1 can also be safely added to the ADL monitoring loop.

A bit surprise by the `corefreqk.ko` seek which decided to _move_ the binding to P-Core rather than E-Core. I wonder what would happen in a disabled SMT BIOS setup.

Now reading datasheets to solve various IMC results; but we're dealing with DDR5 here...

cyring, 2022-02-20:
@ich777 For your testings, C1 and MC6 are now committed into branch `develop_AlderLake_uncore`

1. C1 C-state counter might differ from previous code where it was a computed value rather than a hardware register.  

2. MC6 may show up in the `Package` view iff the driver is bound to one of the E-Cores. But that's something we could not manage to do, above, because the driver is assigning the `Service Processor` to a couple of {physical, logical} Cores as a preference. (load distribution reason)
  The _easiest_ way is to `HotPlug` the CPU associated to P-Core.  (UI Menu)
  Driver will fallback to E-Core by the end.
  Remark: CPU `0` can not be disabled b/c it is the `BSP` but its SMT can.

cyring, 2022-02-20:
Another interesting test is about discrete Vcore among P and E Cores:
1. switch to the `Voltage` view
2. from `Settings` change the `Voltage scope` to `Thread`
3. in `Tools` `>` `Turbo < Select CPU >` apply load to one P-Core and to one E-Core
4. observe if the max voltage differs significantly on both Cores ?

[image]
_Remark: if you don't need the UI bar charts_ `make NO_UPPER=1`

Depending of the result, I will define a new voltage scope for Hybrid processor.

ich777, 2022-02-21:
> A bit surprise by the `corefreqk.ko` seek which decided to _move_ the binding to P-Core rather than E-Core. I wonder what would happen in a disabled SMT BIOS setup.

There you go, I've disabled SMT, I also choose CPU 10 and it is highlighted as you can see (keep in mind this is with the newer version, don't know if you changed anything):
[image]

> 4\. observe if the max voltage differs significantly on both Cores ?

From my perspective yes:
[image]

cyring, 2022-02-22:
> There you go, I've disabled SMT, I also choose CPU 10 and it is highlighted as you can see (keep in mind this is with the newer version, don't know if you changed anything): [image]

Yes it now includes Core C-state `C1` and Package C-state `MC6`.
The later can be seen in the UI view `Package Cycles`

> > 4. observe if the max voltage differs significantly on both Cores ?
> 
> From my perspective yes: [image]

If voltage accuracy is in question ?then there is impact on code where two Vcore formulas should be applied. 
BIOS screenshots show that ASUS is displaying P and E voltage, certainly for a good reason that I've not found in datasheets yet.

cyring, 2022-02-22:
# Hybrid
_Source ref: `325462-sdm-vol-1-2abcd-3abcd.pdf`_

* `CPUID.07H.0H:EDX[15]` 
>If 1, the processor is identified as a hybrid part.
>
>Additionally, on hybrid parts (CPUID.07H.0H:EDX[15]=1), software must consult the native model ID and core type from the Hybrid Information Enumeration Leaf.

* `CPUID.1AH.0H:EAX` 
>Enumerates the native model ID and core type.
Bits 31-24: Core type
10H: Reserved
20H: Intel Atom®
30H: Reserved
40H: Intel® Core™
>
>Bits 23-0: Native model ID of the core. The core-type and native mode ID can be used to uniquely identify the microarchitecture of the core. This native model ID is not unique across core types, and not related to the model ID reported in CPUID leaf 01H, and does not identify the SOC.

* CPUID dump

  Based on latest `develop` branch, can you please dump CPUID using `corefreq-cli -u` and  print only those relevant lines for all CPU ?  

_example_
```
CPU #0   function         EAX          EBX          ECX          EDX
|- 00000000:00000000    00000010     68747541     444d4163     69746e65
   |- Largest Standard Function=00000010
|- 80000000:00000000    80000020     68747541     444d4163     69746e65
   |- Largest Extended Function=80000020
|- 00000007:00000000    00000000     219c91a9     00400004     00000000
|- 0000001a:00000000    00000000     00000000     00000000     00000000

...

CPU #31  function         EAX          EBX          ECX          EDX
|- 00000000:00000000    00000010     68747541     444d4163     69746e65
   |- Largest Standard Function=00000010
|- 80000000:00000000    80000020     68747541     444d4163     69746e65
   |- Largest Extended Function=80000020
|- 00000007:00000000    00000000     219c91a9     00400004     00000000
|- 0000001a:00000000    00000000     00000000     00000000     00000000
```

ich777, 2022-02-22:
> * Based on latest `develop` branch, can you please dump CPUID using `corefreq-cli -u` and  print only those relevant lines for all CPU ?

Sure:
```
CPU #0   function         EAX          EBX          ECX          EDX            
|- 00000000:00000000    00000020     756e6547     6c65746e     49656e69         
   |- Largest Standard Function=00000020                                        
|- 80000000:00000000    80000008     00000000     00000000     00000000         
   |- Largest Extended Function=80000008                                        
|- 00000001:00000000    00090672     00800800     7ffafbff     bfebfbff         
|- 00000002:00000000    00feff01     000000f0     00000000     00000000         
...
CPU #1   function         EAX          EBX          ECX          EDX            
|- 00000000:00000000    00000020     756e6547     6c65746e     49656e69         
   |- Largest Standard Function=00000020                                        
|- 80000000:00000000    80000008     00000000     00000000     00000000         
   |- Largest Extended Function=80000008                                        
|- 00000001:00000000    00090672     01800800     7ffafbff     bfebfbff         
|- 00000002:00000000    00feff01     000000f0     00000000     00000000         
...
CPU #2   function         EAX          EBX          ECX          EDX            
|- 00000000:00000000    00000020     756e6547     6c65746e     49656e69         
   |- Largest Standard Function=00000020                                        
|- 80000000:00000000    80000008     00000000     00000000     00000000         
   |- Largest Extended Function=80000008                                        
|- 00000001:00000000    00090672     08800800     7ffafbff     bfebfbff         
|- 00000002:00000000    00feff01     000000f0     00000000     00000000         
...
CPU #3   function         EAX          EBX          ECX          EDX            
|- 00000000:00000000    00000020     756e6547     6c65746e     49656e69         
   |- Largest Standard Function=00000020                                        
|- 80000000:00000000    80000008     00000000     00000000     00000000         
   |- Largest Extended Function=80000008                                        
|- 00000001:00000000    00090672     09800800     7ffafbff     bfebfbff         
|- 00000002:00000000    00feff01     000000f0     00000000     00000000         
...
CPU #4   function         EAX          EBX          ECX          EDX            
|- 00000000:00000000    00000020     756e6547     6c65746e     49656e69         
   |- Largest Standard Function=00000020                                        
|- 80000000:00000000    80000008     00000000     00000000     00000000         
   |- Largest Extended Function=80000008                                        
|- 00000001:00000000    00090672     10800800     7ffafbff     bfebfbff         
|- 00000002:00000000    00feff01     000000f0     00000000     00000000         
...
CPU #5   function         EAX          EBX          ECX          EDX            
|- 00000000:00000000    00000020     756e6547     6c65746e     49656e69         
   |- Largest Standard Function=00000020                                        
|- 80000000:00000000    80000008     00000000     00000000     00000000         
   |- Largest Extended Function=80000008                                        
|- 00000001:00000000    00090672     11800800     7ffafbff     bfebfbff         
|- 00000002:00000000    00feff01     000000f0     00000000     00000000         
...
CPU #6   function         EAX          EBX          ECX          EDX            
|- 00000000:00000000    00000020     756e6547     6c65746e     49656e69         
   |- Largest Standard Function=00000020                                        
|- 80000000:00000000    80000008     00000000     00000000     00000000         
   |- Largest Extended Function=80000008                                        
|- 00000001:00000000    00090672     18800800     7ffafbff     bfebfbff         
|- 00000002:00000000    00feff01     000000f0     00000000     00000000         
...
CPU #7   function         EAX          EBX          ECX          EDX            
|- 00000000:00000000    00000020     756e6547     6c65746e     49656e69         
   |- Largest Standard Function=00000020                                        
|- 80000000:00000000    80000008     00000000     00000000     00000000         
   |- Largest Extended Function=80000008                                        
|- 00000001:00000000    00090672     19800800     7ffafbff     bfebfbff         
|- 00000002:00000000    00feff01     000000f0     00000000     00000000         
...
CPU #8   function         EAX          EBX          ECX          EDX            
|- 00000000:00000000    00000020     756e6547     6c65746e     49656e69         
[... 71 lines cut ...]
|- 00000000:00000000    00000020     756e6547     6c65746e     49656e69         
   |- Largest Standard Function=00000020                                        
|- 80000000:00000000    80000008     00000000     00000000     00000000         
   |- Largest Extended Function=80000008                                        
|- 00000001:00000000    00090672     4a800800     7ffafbff     bfebfbff         
|- 00000002:00000000    00feff01     00000000     00000000     00000000         
...
CPU #18  function         EAX          EBX          ECX          EDX            
[... cut ...]
   |- Largest Standard Function=00000020                                        
|- 80000000:00000000    80000008     00000000     00000000     00000000         
   |- Largest Extended Function=80000008                                        
|- 00000001:00000000    00090672     4e800800     7ffafbff     bfebfbff         
|- 00000002:00000000    00feff01     00000000     00000000     00000000         
```
I've also attached the full output:
[corefreq-cli -u.txt](https://github.com/cyring/CoreFreq/files/8115898/corefreq-cli.-u.txt)

cyring, 2022-02-22:
@ich777 : Thank you

# Hybrid based on [topology](https://gist.github.com/cyring/ef3a34321b00ecfb6251b61a4290ee94?permalink_comment_id=4072375#gistcomment-4072375)

## P-Core
```
CPU Pkg  Apic  Core/Thread
000:BSP    0     0      0
```
```
CPU #0   function         EAX          EBX          ECX          EDX            
|- 00000000:00000000    00000020     756e6547     6c65746e     49656e69         
   |- Largest Standard Function=00000020                                        
|- 80000000:00000000    80000008     00000000     00000000     00000000         
   |- Largest Extended Function=80000008                                        
|- 00000007:00000000    00000001     239ca7eb     98c027bc     fc1cc410         
|- 0000001a:00000000    40000001     00000000     00000000     00000000         
```

```
CPU Pkg  Apic  Core/Thread
001:  0    1     0      1
```
```
CPU #1   function         EAX          EBX          ECX          EDX            
|- 00000000:00000000    00000020     756e6547     6c65746e     49656e69         
   |- Largest Standard Function=00000020                                        
|- 80000000:00000000    80000008     00000000     00000000     00000000         
   |- Largest Extended Function=80000008                                        
|- 00000007:00000000    00000001     239ca7eb     98c027bc     fc1cc410         
|- 0000001a:00000000    40000001     00000000     00000000     00000000         
```

## E-Core
```
CPU Pkg  Apic  Core/Thread
016:  0   72    36      0
```
```
CPU #16  function         EAX          EBX          ECX          EDX            
|- 00000000:00000000    00000020     756e6547     6c65746e     49656e69         
   |- Largest Standard Function=00000020                                        
|- 80000000:00000000    80000008     00000000     00000000     00000000         
   |- Largest Extended Function=80000008                                        
|- 00000007:00000000    00000001     239ca7eb     98c027bc     fc1cc410         
|- 0000001a:00000000    20000001     00000000     00000000     00000000         
```
```
CPU Pkg  Apic  Core/Thread
019:  0   78    39      0
```
```
CPU #19  function         EAX          EBX          ECX          EDX            
|- 00000000:00000000    00000020     756e6547     6c65746e     49656e69         
   |- Largest Standard Function=00000020                                        
|- 80000000:00000000    80000008     00000000     00000000     00000000         
   |- Largest Extended Function=80000008                                        
|- 00000007:00000000    00000001     239ca7eb     98c027bc     fc1cc410         
|- 0000001a:00000000    20000001     00000000     00000000     00000000         
```

## Hybrid Decoding

CPU |`EDX[15]`|`EAX[31-24]`|Core type         |Model ID
-------|-------------|------------------|--------------------|-------------
`#0`  |TRUE      |`0x40`           |`Intel® Core™`|`0x1`
`#1`  |TRUE      |`0x40`           |`Intel® Core™`|`0x1`
`#16`|TRUE      |`0x20`           |`Intel Atom®`   |`0x1`
`#19`|TRUE      |`0x20`           |`Intel Atom®`   |`0x1`

ich777, 2022-02-22:
```
rdmsr -a -d 0x3f9
CPU 0: 0
CPU 1: 0
CPU 2: 0
CPU 3: 0
CPU 4: 0
CPU 5: 0
CPU 6: 0
CPU 7: 0
CPU 8: 0
CPU 9: 0
CPU 10: 0
CPU 11: 0
CPU 12: 0
CPU 13: 0
CPU 14: 0
CPU 15: 0
CPU 16: 0
CPU 17: 0
CPU 18: 0
CPU 19: 0
```

```
rdmsr -a -d 0x3fa
CPU 0: 0
CPU 1: 0
CPU 2: 0
CPU 3: 0
CPU 4: 0
CPU 5: 0
CPU 6: 0
CPU 7: 0
CPU 8: 0
CPU 9: 0
CPU 10: 0
CPU 11: 0
CPU 12: 0
CPU 13: 0
CPU 14: 0
CPU 15: 0
CPU 16: 0
CPU 17: 0
CPU 18: 0
CPU 19: 0
```

```
rdmsr -p 16 -d 0x3f9
0
```

```
rdmsr -p 19 -d 0x3f9
0
```

```
rdmsr -p 16 -d 0x3fa
0
```

```
rdmsr -p 19 -d 0x3fa
0
```

```
rdmsr -a -d 0x632
CPU 0: 0
CPU 1: 0
CPU 2: 0
CPU 3: 0
CPU 4: 0
CPU 5: 0
CPU 6: 0
CPU 7: 0
CPU 8: 0
CPU 9: 0
CPU 10: 0
CPU 11: 0
CPU 12: 0
CPU 13: 0
CPU 14: 0
CPU 15: 0
CPU 16: 0
CPU 17: 0
CPU 18: 0
CPU 19: 0
```

```
rdmsr -p 16 -d 0x632
0
```

```
rdmsr -p 19 -d 0x632
0
```

I've run all the commands... I realized to late that some of them are not needed because the first succeeded... :rofl:

cyring, 2022-02-22:
> I've run all the commands... I realized to late that some of them are not needed because the first succeeded... 🤣

They don't crash, value is zero but I forgot to mention that system needs to idle to have a chance to catch those C-states.

Not sure if I have seen `MC06` in action ?

cyring, 2022-02-22:
> > They don't crash, value is zero but I forgot to mention that system needs to idle to have a chance to catch those C-states.
> 
> The system was actually idling at that time because I only boot up unRAID when you want me to test something, this is actually my main PC.

Thanks a lot for what you are doing.

In the latest `develop` commit, you will get the CPU Topology enhancements:
* Hybrid ID
* Hybrid Arch

Those are so far available through the JSON export.
```
corefreq-cli -j > corefreq.json
```
I'm using [jsoneditoronline.org](https://jsoneditoronline.org/) to preview data.

_example_
[image]

With your `i7-12700K` you should get data in coherency with the table [Hybrid Decoding](https://github.com/cyring/CoreFreq/issues/323#issuecomment-1047715383).

Please note: JSON output is decimal whereas the above table is shown in hexadecimal.

cyring, 2022-02-23:
@ich777 : UI Topology window change available

[image]

This an example where I've force the Hybrid bit for rendering tests.
One CPU is also disabled to check color alignment.

With your Alder Lake, you should read `P` or `E` followed by the Hybrid model ID, per Core

ich777, 2022-02-23:
@cyring currently at work, compiling the new version ASAP and I will report back.

ich777, 2022-02-23:
> With your `i7-12700K` you should get data in coherency with the table [Hybrid Decoding](https://github.com/cyring/CoreFreq/issues/323#issuecomment-1047715383).

This looks right:
[image]
(also included the output file if you need it: [corefreq.json.txt](https://github.com/cyring/CoreFreq/files/8125522/corefreq.json.txt))

> With your Alder Lake, you should read `P` or `E` followed by the Hybrid model ID, per Core

This seems also right:
[image]

cyring, 2022-02-23:
> This seems also right: 

Beautiful ! thank you

And for the understanding of the JSON output, here is how to read the output:

* CPU `#0` is a Pcore
[image]
* CPU `#19` is a Ecore
[image]

cyring, 2022-02-25:
@SimonFair , @ich777 

For your Alder Lake, you will find two enhancements in the `develop` branch:

1. A new Service Processor is selected among Ecores
  In the UI, its CPU number is displayed in highlighted green
  (the cyan ones remain for Pcores)

2. The Core and Package C-states are now monitoring as a function of the architecture (Alder Lake or Atom)
* Core C-states are displayed the `Frequency` and `Idle C-States` views
* Package C-states are shown is this view
[image]
`PC04`, `PC10` and `MC06` are sourced from the Ecore
The other `PCx` `TSC` and `UNCORE` are sourced from Pcore

Please post your screenshots to let me verify results: `Frequency` and `Package` views with no load.

ich777, 2022-02-25:
Built the new version and sent it over to @SimonFair through the forums already, will test it hopefully tomorrow.

SimonFair, 2022-02-25:
For i5-12600k

[image]
[image]

cyring, 2022-02-26:
> For i5-12600k

Great! 
So we now have `MC06` showing up.

ich777, 2022-02-26:
> Please post your screenshots to let me verify results: `Frequency` and `Package` views with no load.

Looking good:

[image]

[image]

cyring, 2022-02-26:
* Below is a list of architectural registers  for Alder Lake support tests.
  Can you please report your results in the placeholder `<Report results here>`
  Thank you
----
# Power Registers

## MSR_DRAM_POWER_LIMIT
```sh
## MSR_DRAM_POWER_LIMIT
rdmsr -aX 0x618
```
```sh
<Report results here>
````

## MSR_DRAM_ENERGY_STATUS
```sh
## MSR_DRAM_ENERGY_STATUS
rdmsr -aX 0x61b
```
```sh
<Report results here>
````

## MSR_PLATFORM_POWER_LIMIT
```sh
## MSR_PLATFORM_POWER_LIMIT
rdmsr -aX 0x65c
```
```sh
<Report results here>
````

## MSR_PLATFORM_ENERGY_STATUS
```sh
## MSR_PLATFORM_ENERGY_STATUS
rdmsr -aX 0x64d
```
```sh
<Report results here>
````

## MSR_AVN_PKG_POWER_INFO
```sh
## MSR_AVN_PKG_POWER_INFO
rdmsr -aX 0x66e
```
```sh
<Report results here>
````

# C-state Registers
## MSR_PKG_C8_RESIDENCY
```sh
## MSR_PKG_C8_RESIDENCY
rdmsr -aX 0x630
```
```sh
<Report results here>
````

## MSR_PKG_C9_RESIDENCY
```sh
## MSR_PKG_C9_RESIDENCY
rdmsr -aX 0x631
```
```sh
<Report results here>
````

## MSR_CC6_DEMOTION_POLICY_CONFIG
```sh
## MSR_CC6_DEMOTION_POLICY_CONFIG
rdmsr -aX 0x668
```
```sh
<Report results here>
````

## MSR_MC6_DEMOTION_POLICY_CONFIG
```sh
## MSR_MC6_DEMOTION_POLICY_CONFIG
rdmsr -aX 0x669
```
```sh
<Report results here>
````

# Thermal Registers
## MSR_THERM2_CTL
```sh
## MSR_THERM2_CTL
rdmsr -aX 0x19d
```
```sh
<Report results here>
````

ich777, 2022-02-27:
Attached the results:

> * Below is a list of architectural registers  for Alder Lake support tests.
>       Can you please report your results in the placeholder `<Report results here>`
>       Thank you
> 
> 
> # Power Registers
> ## MSR_DRAM_POWER_LIMIT
> 
> ```shell
> ## MSR_DRAM_POWER_LIMIT
> rdmsr -aX 0x618
> ```
> 
> ```shell
> CPU 0: 0
> CPU 1: 0
> CPU 2: 0
> CPU 3: 0
> CPU 4: 0
> CPU 5: 0
> CPU 6: 0
> CPU 7: 0
> CPU 8: 0
> CPU 9: 0
> CPU 10: 0
> CPU 11: 0
> CPU 12: 0
> CPU 13: 0
> CPU 14: 0
> CPU 15: 0
> CPU 16: 0
> CPU 17: 0
> CPU 18: 0
> CPU 19: 0
> ```
> 
> ## MSR_DRAM_ENERGY_STATUS
> 
> ```shell
> ## MSR_DRAM_ENERGY_STATUS
> rdmsr -aX 0x61b
> ```
> 
> ```shell
> CPU 0: 0
> CPU 1: 0
> CPU 2: 0
> CPU 3: 0
> CPU 4: 0
> CPU 5: 0
> CPU 6: 0
> CPU 7: 0
> CPU 8: 0
> CPU 9: 0
> CPU 10: 0
> CPU 11: 0
> CPU 12: 0
> CPU 13: 0
> CPU 14: 0
> CPU 15: 0
> CPU 16: 0
> CPU 17: 0
> CPU 18: 0
> CPU 19: 0
> ```
> 
> ## MSR_PLATFORM_POWER_LIMIT
> 
> ```shell
> ## MSR_PLATFORM_POWER_LIMIT
> rdmsr -aX 0x65c
> ```
> 
> ```shell
> CPU 0: DD0000
> CPU 1: DD0000
> CPU 2: DD0000
> CPU 3: DD0000
> CPU 4: DD0000
> CPU 5: DD0000
> CPU 6: DD0000
> CPU 7: DD0000
> CPU 8: DD0000
> CPU 9: DD0000
> CPU 10: DD0000
> CPU 11: DD0000
> CPU 12: DD0000
> CPU 13: DD0000
> CPU 14: DD0000
> CPU 15: DD0000
> CPU 16: DD0000
> CPU 17: DD0000
> CPU 18: DD0000
> CPU 19: DD0000
> ```
> 
> ## MSR_PLATFORM_ENERGY_STATUS
> 
> ```shell
> ## MSR_PLATFORM_ENERGY_STATUS
> rdmsr -aX 0x64d
> ```
> 
> ```shell
> CPU 0: 0
> CPU 1: 0
> CPU 2: 0
> CPU 3: 0
> CPU 4: 0
> CPU 5: 0
> CPU 6: 0
> CPU 7: 0
> CPU 8: 0
> CPU 9: 0
> CPU 10: 0
> CPU 11: 0
> CPU 12: 0
> CPU 13: 0
> CPU 14: 0
> CPU 15: 0
> CPU 16: 0
> CPU 17: 0
> CPU 18: 0
> CPU 19: 0
> ```
> 
> ## MSR_AVN_PKG_POWER_INFO
> 
> ```shell
> ## MSR_AVN_PKG_POWER_INFO
> rdmsr -aX 0x66e
> ```
> 
> ```shell
> rdmsr: CPU 0 cannot read MSR 0x0000066e
> ```
> 
> # C-state Registers
> ## MSR_PKG_C8_RESIDENCY
> 
> ```shell
> ## MSR_PKG_C8_RESIDENCY
> rdmsr -aX 0x630
> ```
> 
> ```shell
> CPU 0: 0
> CPU 1: 0
> CPU 2: 0
> CPU 3: 0
> CPU 4: 0
> CPU 5: 0
> CPU 6: 0
> CPU 7: 0
> CPU 8: 0
> CPU 9: 0
> CPU 10: 0
> CPU 11: 0
> CPU 12: 0
> CPU 13: 0
> CPU 14: 0
> CPU 15: 0
> CPU 16: 0
> CPU 17: 0
> CPU 18: 0
> CPU 19: 0
> ```
> 
> ## MSR_PKG_C9_RESIDENCY
> 
> ```shell
> ## MSR_PKG_C9_RESIDENCY
> rdmsr -aX 0x631
> ```
> 
> ```shell
> CPU 0: 0
> CPU 1: 0
> CPU 2: 0
> CPU 3: 0
> CPU 4: 0
> CPU 5: 0
> CPU 6: 0
> CPU 7: 0
> CPU 8: 0
> CPU 9: 0
> CPU 10: 0
> CPU 11: 0
> CPU 12: 0
> CPU 13: 0
> CPU 14: 0
> CPU 15: 0
> CPU 16: 0
> CPU 17: 0
> CPU 18: 0
> CPU 19: 0
> ```
> 
> ## MSR_CC6_DEMOTION_POLICY_CONFIG
> 
> ```shell
> ## MSR_CC6_DEMOTION_POLICY_CONFIG
> rdmsr -aX 0x668
> ```
> 
> ```shell
> rdmsr: CPU 0 cannot read MSR 0x00000668
> ```
> 
> ## MSR_MC6_DEMOTION_POLICY_CONFIG
> 
> ```shell
> ## MSR_MC6_DEMOTION_POLICY_CONFIG
> rdmsr -aX 0x669
> ```
> 
> ```shell
> rdmsr: CPU 0 cannot read MSR 0x00000669
> ```
> 
> # Thermal Registers
> ## MSR_THERM2_CTL
> 
> ```shell
> ## MSR_THERM2_CTL
> rdmsr -aX 0x19d
> ```
> 
> ```shell
> CPU 0: 0
> CPU 1: 0
> CPU 2: 0
> CPU 3: 0
> CPU 4: 0
> CPU 5: 0
> CPU 6: 0
> CPU 7: 0
> CPU 8: 0
> CPU 9: 0
> CPU 10: 0
> CPU 11: 0
> CPU 12: 0
> CPU 13: 0
> CPU 14: 0
> CPU 15: 0
> CPU 16: 0
> CPU 17: 0
> CPU 18: 0
> CPU 19: 0
> ```

cyring, 2022-02-27:
@ich777 , @SimonFair 

Based on those [results](https://github.com/cyring/CoreFreq/issues/323#issuecomment-1053321148), the `develop` branch has a new commit:

* Voltage view is now per Core
* Power view may display `RAM` and `PLATFORM`
* `TM` in footer could be green
* Package view could show `PC08` and `PC09`

ich777, 2022-02-28:
> Voltage view is now per Core

[image]

> `TM` in footer could be green

Yes, displayed like in the picture above.

> Package view could show `PC08` and `PC09`

[image]

> **EDIT** Limits for DRAM and PLATFORM in `Power, Current & Thermal`

[image]

cyring, 2022-02-28:
@ich777 thank you for your answers
That `13` in bottom left of terminal, is it a UI glitch ?

ich777, 2022-02-28:
> That `13` in bottom left of terminal, is it a UI glitch ?

TBH didn't noticed that, made the screenshots when I woke up... :D
Will try to reproduce this when I'm home for lunch and report back.

cyring, 2022-02-28:
> > That `13` in bottom left of terminal, is it a UI glitch ?
> 
> TBH didn't noticed that, made the screenshots when I woke up... :D Will try to reproduce this when I'm home for lunch and report back.

It may be linked to an unsual high frequency ratio of `63` observed among `HWP` frequencies, and which is much above than the currently default `6175000000 Hz` found in the `Makefile`

You can rebuild with a higher value, in Hertz like: 
```
make MAX_FREQ_HZ=6475000000 clean all
```

The CLI ratios aggregation will take into account a _rounded_ max ratio of `64` which may avoid various computing glitches.

If that works, I will change the default value during next `develop` commit.

ich777, 2022-02-28:
> If that works, I will change the default value during next `develop` commit.

At first I can't reproduce this again but then I got it... (without compiling it with the above options.
Go to `CPUID Hexa Dump` -> press `ESC` -> go to something else like `Freqency`

Now I got:
[image]

cyring, 2022-03-01:
> At first I can't reproduce this again but then I got it... (without compiling it with the above options. Go to `CPUID Hexa Dump` -> press `ESC` -> go to something else like `Freqency`
> 
> Now I got `16`: ...
> 
> **EDIT:** With the newly compiled version with the options from above everything works as expected: ...

I have pushed two commits in `develop`:
1. float conversion debugged in the UI
2. raise the max builtin frequency to `6575000000 Hz`

cyring, 2022-03-02:
@ich777 @SimonFair & Other Alder Lake Users : Hello

Commit f830971dfd1fb83f734d9ef134bc3d83d4f28570 you will get new code for DDR5

Thank you for your IMC tests. I will appreciate a screenshot or output of `Memory Controller`

BugReporterZ, 2022-03-02:
Not sure if it can help but here is the output of `Memory Controller` from my i7-12700K. My motherboard is using DDR4 memory. The DDR4 speed shown however seems wrong. My modules ([Corsair CMK32GX4M2D3600C18](https://www.corsair.com/us/en/Categories/Products/Memory/VENGEANCE-LPX/p/CMK32GX4M2D3600C18)) should be running at 3600 MHz.

[image]
[image]

cyring, 2022-03-02:
@BugReporterZ Thanks, it helps a lot.
Are you using the branch `develop` of CoreFreq ?

ich777, 2022-03-02:
> Thank you for your IMC tests. I will appreciate a screenshot or output of `Memory Controller`

Sure:
[image]

cyring, 2022-03-02:
> > Thank you for your IMC tests. I will appreciate a screenshot or output of `Memory Controller`
> 
> Sure (sent the new build over to @SimonFair on the Unraid forums): 

Thanks.
Can not see it but we're DIMMs detected as DDR5 or DDR4 ?

BugReporterZ, 2022-03-02:
@cyring 
No, I used `master`. I just switched to `develop` and now indeed the reported data is correct, although some memory timings are still missing.

[image]

[49 more replies]

closed 2022-03-24 by cyring