cyring/CoreFreq #325: AUR Install with Pamac fails on Manjaro

Ragebone, 2022-02-22:
Hi,
i was failing to install corefreq from the AUR in Pamac UI with the error-message.
>target not found: corefreq

I got this message even when attempting to build all 3, 2 or only 1 of the needed packages and independent of the package itself.
Until i suddenly didn't get this error anymore.
 
On a previous attempt on a different Manjaro Installation, i had this issue as well.
Here, installing the client and with that all 3 as dependencies automatically, caused the same error. 
But manually installing the dkms, then the service, then the client, worked in that case.
That did not work in this instance.
I wasn't even able to edit the build-files.
Trying that caused the same error and abort.

The only thing i did differently before it worked is i chose just the dkms package to build, then looked and clicked into the displayed build-file.
and then it just build. 

Any ideas whats going on here?
I have none.

cyring, 2022-02-22:
Hello

Adding @inglor in the loop for any help.
To my knowledge the AUR packages target the vanilla ArchLinux.

`dkms` may have solved the issue because the kernel development dependencies are properly installed by `dkms`

Linux headers, gcc, make and glibc are the main prerequisites to build _CoreFreq_
```sh
pacman -Q linux-headers gcc make glibc
```
```sh
linux-headers 5.16.10.arch1-1
gcc 11.2.0-3
make 4.3-3
glibc 2.35-2
```
and `dkms` is pulling most of them but optionally the Headers
```sh
pacman -Si dkms
```
```sh
Repository      : extra
Name            : dkms
Version         : 3.0.3-1
Description     : Dynamic Kernel Modules System
Architecture    : any
URL             : https://github.com/dell/dkms
Licenses        : GPL2
Groups          : None
Provides        : None
Depends On      : coreutils  bash  gawk  sed  kmod  gcc  make  patch
Optional Deps   : linux-headers: build modules against the Arch kernel
```
I would bet the Manjaro Linux Headers were in trouble ?

It will be helpful if you post the link to the Manjaro's CoreFreq setup.

inglor, 2022-02-22:
>  To my knowledge the AUR packages target the vanilla ArchLinux.

Correct. The only supported build of AUR packages is [Building_in_a_clean_chroot]. I think you might have better luck trying pamac's support channels (which I don't know).

[Building_in_a_clean_chroot]: https://wiki.archlinux.org/title/DeveloperWiki:Building_in_a_clean_chroot

cyring, 2022-02-22:
> Correct. The only supported build of AUR packages is [Building_in_a_clean_chroot](https://wiki.archlinux.org/title/DeveloperWiki:Building_in_a_clean_chroot). I think you might have better luck [...]

Thanks @inglor for your answer.

@Ragebone : feel free however to request `pamac` team to join to this thread. Will provide any help to improve the Manjaro package of _CoreFreq_

Ragebone, 2022-02-23:
i am seriously confused by this issue because i can't make any sense of it.
I just tried it on my own WS and was unable to get it to build. 

[image]
[image]
[image]
[image]
[image]

Just choosing corefreq-dkms fails the same way, until it suddenly doesn't.
And from there, it builds fine.

cyring, 2022-02-24:
@Ragebone Hi

What are those MISSING coming from ?
Files mentioned on Error lines are not part of my program. Why do they appear as some kind of dependencies!

What about installing CoreFreq the ArchLinux way ?
You could just grab the _CoreFreq_ PKGBUILD from AUR and do `makepkg -si`

Ragebone, 2022-02-25:
I think those missing error messages are caused by the missing kernel headers.
But i'm not sure what is responsible for the print.

Looking at the buildfile, they aren't a declared dependency.
And about the version of the packages, there are two groups of packages.
The installed non git labeled as 1.89.4-2 where as the -git packages are declared 1.88.1r2... .
So i specifically picked the non git because of that.

inglor, 2022-02-25:
@Ragebone have you read the wiki for DKMS here: https://wiki.archlinux.org/title/Dynamic_Kernel_Module_Support ?

cyring, 2022-02-25:
> I think those missing error messages are caused by the missing kernel headers. But i'm not sure what is responsible for the print.
>
> Looking at the buildfile, they aren't a declared dependency. And [...]

Why don't you give up with **that** GUI installer ?

Just do a plain AUR install from a terminal and enjoy _CoreFreq_

cyring, 2022-02-28:
@Ragebone : [here](https://forum.manjaro.org/t/cant-boot-previously-working-install/104119/9) some Manjaro Users are attempting _CoreFreq_ with one of this shiny new Alder Lake Mobile processors: **`i9-12900H`**
Could you ask them to join this issue for testings ?
I need such hardware to mprove code.

cyring, 2022-03-05:
Alder Lake done

closed 2022-03-05 by cyring