cyring/CoreFreq #340: Problem compiling

vincenzogianfelice, 2022-05-07:
```
make -j1 -C /lib/modules/4.9.0-18-amd64/build M=/opt/CoreFreq modules
make[1]: ingresso nella directory "/usr/src/linux-headers-4.9.0-18-amd64"
  CC [M]  /opt/CoreFreq/corefreqk.o
In file included from /opt/CoreFreq/corefreqk.c:51:0:
/opt/CoreFreq/corefreqk.c: In function ‘AMD_Zen_UMC’:
/opt/CoreFreq/corefreqk.h:707:37: error: ‘struct amd_northbridge’ has no member named ‘root’
     if ((root = node_to_amd_nb(node)->root) != NULL)   \
                                     ^
/opt/CoreFreq/corefreqk.h:736:7: note: in expansion of macro ‘AMD_SMN_RW’
   if (AMD_SMN_RW( amd_pci_dev_to_node_id(UMC_device), \
       ^~~~~~~~~~
/opt/CoreFreq/corefreqk.h:785:3: note: in expansion of macro ‘AMD_SMN_Read’
   AMD_SMN_Read(SMN_Register, SMN_Address, UMC_device); \
   ^~~~~~~~~~~~
/opt/CoreFreq/corefreqk.c:5077:2: note: in expansion of macro ‘Core_AMD_SMN_Read’
  Core_AMD_SMN_Read(
  ^~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
```

Debian 9(x64)

cyring, 2022-05-07:
Fix is available in the `develop` branch

Here's my compilation test based on a Debian 9 VM
Not tried bare-metal. Please let me know.

[image]

cyring, 2022-05-09:
Is the fix for you ?

vincenzogianfelice, 2022-05-11:
Partial...

```
[132936.702608] ------------[ cut here ]------------
[132936.702612] WARNING: CPU: 0 PID: 0 at /build/linux-9cDhkD/linux-4.9.303/drivers/pci/search.c:271 pci_get_dev_by_id+0x78/0x90
[132936.702612] Modules linked in: corefreqk(O) xt_conntrack xt_iprange xt_multiport xt_tcpudp xt_mac ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_mangle iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack arptable_filter arp_tables iptable_filter intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel evdev kvm irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel intel_cstate intel_uncore intel_rapl_perf pcspkr serio_raw fuse iTCO_wdt iTCO_vendor_support i915 drm_kms_helper drm mei_me i2c_algo_bit sg mei shpchp wmi video acpi_pad button ip_tables x_tables autofs4 ext4 crc16 jbd2 fscrypto ecb mbcache raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c crc32c_generic raid1 multipath linear raid0 md_mod sd_mod uas
[132936.702642]  usb_storage ahci libahci crc32c_intel aesni_intel aes_x86_64 glue_helper lrw gf128mul ablk_helper cryptd psmouse i2c_i801 xhci_pci i2c_smbus libata xhci_hcd r8169 mii scsi_mod usbcore usb_common fan thermal
[132936.702653] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G           O    4.9.0-18-amd64 #1 Debian 4.9.303-1
[132936.702654] Hardware name: Gigabyte Technology Co., Ltd. H110M-S2H/H110M-S2H-CF, BIOS F22 06/12/2017
[132936.702655]  0000000000000000 ffffffff8fc14ba7 0000000000000000 0000000000000000
[132936.702656]  ffffffff8f67aa4b ffff9a83bec03ef0 0000000000000000 0000000000000000
[132936.702658]  00000000ffffffed ffffffff8fff4dc8 000078e7bcee847e ffffffff8f988758
[132936.702660] Call Trace:
[132936.702661]  <IRQ> 
[132936.702664]  [<ffffffff8fc14ba7>] ? dump_stack+0x66/0x81
[132936.702666]  [<ffffffff8f67aa4b>] ? __warn+0xcb/0xf0
[132936.702667]  [<ffffffff8f988758>] ? pci_get_dev_by_id+0x78/0x90
[132936.702669]  [<ffffffff8f98881d>] ? pci_get_device+0x4d/0x70
[132936.702675]  [<ffffffffc078985f>] ? CoreFreqK_ProbePCI+0x4f/0xa0 [corefreqk]
[132936.702680]  [<ffffffffc07996a8>] ? Intel_Watchdog+0x58/0x80 [corefreqk]
[132936.702684]  [<ffffffffc07896e0>] ? CoreFreqK_Register_ClockSource+0x170/0x170 [corefreqk]
[132936.702688]  [<ffffffffc079fa81>] ? PerCore_Skylake_Query+0x271/0x290 [corefreqk]
[132936.702692]  [<ffffffffc07a4ef6>] ? Start_Skylake+0x76/0x2d0 [corefreqk]
[132936.702696]  [<ffffffffc07a4e80>] ? Start_Skylake_X+0x2f0/0x2f0 [corefreqk]
[132936.702698]  [<ffffffff8f6ffd02>] ? flush_smp_call_function_queue+0x62/0x150
[132936.702700]  [<ffffffff8fc273b1>] ? smp_call_function_single_interrupt+0x31/0x40
[132936.702701]  [<ffffffff8fc262fe>] ? call_function_single_interrupt+0x9e/0xb0
[132936.702702]  <EOI> 
[132936.702704]  [<ffffffff8fae02b2>] ? cpuidle_enter_state+0xa2/0x2d0
[132936.702706]  [<ffffffff8f6beb14>] ? cpu_startup_entry+0x154/0x240
[132936.702708]  [<ffffffff90343f10>] ? start_kernel+0x44f/0x472
[132936.702709]  [<ffffffff90343120>] ? early_idt_handler_array+0x120/0x120
[132936.702711]  [<ffffffff903433a9>] ? x86_64_start_kernel+0x14c/0x170
[132936.702711] ---[ end trace d2a84bc6100b278e ]---
```
(after load module)...

cyring, 2022-05-11:
> Partial...
>
> ```
> [132936.702608] ------------[ cut here ]------------
> [132936.702612] WARNING: CPU: 0 PID: 0 at /build/linux-9cDhkD/linux-4.9.303/drivers/pci/search.c:271 pci_get_dev_by_id+0x78/0x90 [...]

Need more detail:
```
uname -a
lscpu
lspci -nn
```

cyring, 2022-05-12:
Don't understand issue yet.
Have other Skylake ar
https://github.com/cyring/CoreFreq/wiki/CPU-support

Was it bare metal run or virtualized ?

cyring, 2022-05-12:
Could you comment or empty the whole body of function `Intel_Watchdog`

https://github.com/cyring/CoreFreq/blob/f2aff9895ce77c4c1a08e5b2619c02be0e9bba73/corefreqk.c#L7937

Something like
```c
void Intel_Watchdog(CORE_RO *Core)
{
/*
	struct pci_device_id PCI_WDT_ids[] = {
		{
		PCI_VDEVICE(INTEL, DID_INTEL_ICH10_LPC),
		.driver_data = (kernel_ulong_t) ICH_TCO
		},
		{0, }
	};
	UNUSED(Core);

	CoreFreqK_ProbePCI(PCI_WDT_ids, NULL, NULL);
*/
}
```

Next rebuild and reload _CoreFreq_
```sh
make clean all
```

cyring, 2022-05-13:
EDIT: or without touching any soure code, you can unload the `iTCO_wdt` kernel module prior loading `corefreqk.ko`

cyring, 2022-05-13:
In this commit 166c30558759ce949a6fbaaa68617f3836c68d5c iTCO will be tested for reservation.
Can you please give a try to the branch ?

vincenzogianfelice, 2022-05-15:
Hi, i have tested branch **develop** (with fix in code)...and everything works fine!. I suppose it is a problem with the watchdog, but i have disabled in grub. Why does this happen? Anyway, i hope you will fix.

Ps. unfortunately lately I find myself not often at home ... I answer as soon as I can, sorry

cyring, 2022-05-15:
@vincenzogianfelice Thank you for your test.

Issue should be linked to `nmi_watchdog`

Problem I had documented into [Wiki/Software-incompatibilities-and-workarounds](https://github.com/cyring/CoreFreq/wiki/Software-incompatibilities-and-workarounds#nmi_watchdog)

closed 2022-06-13 by cyring