cyring/CoreFreq #343: sporadic system freeze - kernel 5.15+

lhommev, 2022-06-07:
Hi,

First, i'd like to say i really like the tool. Congrats for the achievements!

I experience system freeze (required hard reset) using corefreq with kernels 5.15 onwards.
hardware: zen3  ryzen 5950x,  x570 motherboard, linux ubuntu 22.
The issue doesn't look like it s related and it took me months to narrow it down to this software. I lost some hairs..

Symptoms:
* Freeze sometimes when opening a firefox tab. Some sites are more likely to trigger it. The more embedded videos, the more probable is the trigger.
* Freeze sometimes when opening photo editing software
* Freeze sometimes when opening videos
* Freeze 50% times after system wakeup
The freeze happens every 10-20 minutes in my regular workflow.

What is NOT having effect:
* swapping hardware
* disabling power states in bios
* disabling power states in linux command line
* actually any firmware setting except SMT
* forcing CPU frequency
* system load

What is doing something:
* Disabling SMT (simultaneous multithreading) in BIOS -> 100% stable
* Running linux 20.04 , kernels up to 5.11 -> 100% stable
* Running without corefreqk loaded -> 100% stable

It sounds like some very complicated corner case.

lhommev, 2022-06-07:
Yes k10temp is loaded.

cyring, 2022-06-07:
> Yes k10temp is loaded.

Please remove/unload it and use _CoreFreq_ with a full SMU control.

For your information, the SMU component is providing various registers: temperature, voltage, power and so on.

SMU offers a select register and a data register: 
1. Sofware writes a targetted register into the SMU select port 
2. and then reads/writes its value through the SMU data port.

If many SW are selecting/reading/writing simultaneously then the SMU gets lost and freezes. 
It may takes a while for such conflict to happen.

The solution provided by Linux kernel is to protect the SMU accesses through a mutex.
_CoreFreq_ does not use this solution because its CPU monitoring runs in an interrupt context where no blocking call (mutex) is acceptable.

See Wiki/[k10temp-and-zenpower-kernel-modules](https://github.com/cyring/CoreFreq/wiki/Software-incompatibilities-and-workarounds#k10temp-and-zenpower-kernel-modules)

So _CoreFreq_ has to drive the AMD/Zen/SMU; exclusively.

lhommev, 2022-06-07:
I see thanks. It makes sense. As an engineer, i see this incompatibility as quite hazardous. IMO, Corefreq should not bypass the mutex, assuming the users took care about the software compatibility issues.
I mean, most user wont know.

Does corefreq perform write operations to the SMU registers?

cyring, 2022-06-13:
Hello,

If you find the `master` branch stable, including the SMU conflict recommendations, then feel free to close the issue ?

I will also appreciate a `5950X` report. 
As an example, my `3950X` [page](https://gist.github.com/cyring/3d038a537cfae0383602823a94136ab6)

Your report will then be added to the [CPU support](https://github.com/cyring/CoreFreq/wiki/CPU-support) list.

Thank you,
CyrIng

lhommev, 2022-06-14:
ok. I ll update the CPU list.

regards,
Vincent