cyring/CoreFreq #346: [Intel/Haswell/Desktop] Core i5-4570 IMC Timings and DIMM geometry

cyring, 2022-06-13:
# `i5-4570`
## Timings
Setting | _CoreFreq_ | BIOS
----------|------------------|--------
`tRRD` | `5` | `4`
`tWR` | N/A  | `10`
`tFAW` | N/A  | `20`
`tRTP` | N/A  | `5`
`tWTP` | N/A  | `21`
`drWR` | `4` | `5`
`tCKE` | N/A | `4`

## Geometry
```
( 8 x Bank x Rank x Rows x Cols ) / ( 1024 x 1024 )
```
```
( 8 x 8 x 2 x 16384 x 1024 ) / ( 1024 x 1024) = 2048 MB vs 8192 MB
```

_Originally posted by @cyring in https://github.com/cyring/CoreFreq/discussions/342#discussioncomment-2934706_

cyring, 2022-09-18:
@svmlegacy and any Broadwell, Haswell owner:

I need to dump Register at MCHBAR address `0x4c00`
(_This register is undocumented, so we are reversing in this [header file](https://github.com/cyring/CoreFreq/blob/2fe50f3d859b479b3b54f178d2a4939019663308/intelmsr.h#L3269)_)

Based on `develop` branch, in function `HSW_IMC()`, add the following `printf` to dump the register value.

https://github.com/cyring/CoreFreq/blob/2fe50f3d859b479b3b54f178d2a4939019663308/corefreqd.c#L3824

```C
    for (cha = 0; cha < RO(Shm)->Uncore.MC[mc].ChannelCount; cha++)
    {
	printf( "Cha(%d) REG4C00[%x]\n", cha,
		RO(Proc)->Uncore.MC[mc].Channel[cha].HSW.REG4C00.value );

```

Please post the output of `corefreqd` here, bellow:

cyring, 2022-10-08:
@svmlegacy : For IMC testing, here's a Haswell/Broadwell development archive.

[CoreFreq_develop.tar.gz](https://github.com/cyring/CoreFreq/files/9739853/CoreFreq_develop.tar.gz)

cyring, 2022-10-10:
@svmlegacy Hello,

Changes are now tracked through a dedicated Haswell, Broadwell branch

You just have to pull and build [develop_HSW_BDW](https://github.com/cyring/CoreFreq/tree/develop_HSW_BDW) to test the IMC

Thank you

svmlegacy, 2022-10-10:
Intel Core i5-4570

Develop branch:
[image]

develop_HSW_BDW branch:
[image]

cyring, 2022-10-11:
> develop_HSW_BDW branch: 

Thank you.

Can you pull and test `develop_HSW_BDW` in which I'm trying to better estimate the DIMM slot count.

svmlegacy, 2022-10-11:
[image]
Motherboard is a Gigabyte GA-Z97-HD3P Rev. 1.0 Four DIMM slots available, slots "3" & "4" populated (Closest and 3rd closest to CPU socket)
[image]

cyring, 2022-10-12:
> Motherboard is a Gigabyte GA-Z97-HD3P Rev. 1.0 Four DIMM slots available, slots "3" & "4" populated (Closest and 3rd closest to CPU socket) 

If dual channel set, topology looks OK but what is wrong is the number of rows: it should be `65536` instead of `16384`

cyring, 2022-10-12:
@svmlegacy Can you please pull and test the [`develop`](https://github.com/cyring/CoreFreq/commit/c2aa1ccfd9e43c59b3c5dcb0ad5fac858874e040) branch ?

DIMM rows count is now computed using a formula made for Skylake.

cyring, 2022-10-13:
> @svmlegacy Can you please pull and test the [`develop`](https://github.com/cyring/CoreFreq/commit/c2aa1ccfd9e43c59b3c5dcb0ad5fac858874e040) branch ?
> 
> DIMM rows count is now computed using a formula made for Skylake.

@svmlegacy could you please test Haswell IMC from latest `develop` ?

svmlegacy, 2022-10-13:
Fresh pull from Develop:

[image]

cyring, 2022-10-14:
@svmlegacy Can you please pull and try `develop` again ?

Commit attempts to fix the DIMM banks. 
I think it should be `8` for `CM3X8GA2133C9D8`
It can be confirmed if you run `dmidecode -t memory`

svmlegacy, 2022-10-14:
> @svmlegacy Can you please pull and try `develop` again ?
> 
> Commit attempts to fix the DIMM banks. I think it should be `8` for `CM3X8GA2133C9D8` It can be confirmed if you run `dmidecode -t memory`

```
$ ./corefreq-cli -M
                            Lynx Point-M  [ C00]                           
Controller #0                                                Dual Channel  
 Bus Rate  1333 MHz       Bus Speed 1333 MHz           DDR3 Speed 1333 MHz 
                                                                           
 Cha    CL  RCD   RP  RAS  RRD  RFC   WR RTPr WTPr  FAW  B2B  CWL CMD  REFI
  #0     9    9    9   24    5  174    0    0    0    0    0    7  1T  5200
  #1     9    9    9   24    5  174    0    0    0    0    0    7  1T  5200
      ddWR drWR srWR ddRW drRW srRW ddRR drRR srRR ddWW drWW srWW CKE   ECC
  #0     5    4   18    9    9    9    6    6    4    7    7    4   0    0 
  #1     5    4   18    9    9    9    6    6    4    7    7    4   0    0 
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0     8    2     65536      1024           8192  CM3X8GA2133C9D8   
       #1                                                                  
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0     8    2     65536      1024           8192  CM3X8GA2133C9D8   
       #1                                                                  
```

svmlegacy, 2022-10-17:
Adding my Xeon E3-1270L v4 (Broadwell LGA1150) before I switch back to 1155...

```
$ ./corefreq-cli -M
                           X99/Wellsburg  [1618]                           
Controller #0                                                Dual Channel  
 Bus Rate  2133 MHz       Bus Speed 2128 MHz           DDR3 Speed 2133 MHz 
                                                                           
 Cha    CL  RCD   RP  RAS  RRD  RFC   WR RTPr WTPr  FAW  B2B  CWL CMD  REFI
  #0     9    9    9   24    5  174    0    0    0    0    0    7  1T  5200
  #1     9    9    9   24    5  174    0    0    0    0    0    7  1T  5200
      ddWR drWR srWR ddRW drRW srRW ddRR drRR srRR ddWW drWW srWW CKE   ECC
  #0     5    4   18    9    9    9    6    6    4    7    7    4   0    0 
  #1     5    4   18    9    9    9    6    6    4    7    7    4   0    0 
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0     8    2     65536      1024           8192  CM3X8GA2133C9D8   
       #1                                                                  
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0     8    2     65536      1024           8192  CM3X8GA2133C9D8   
       #1                                                                  
```

cyring, 2022-10-17:
> Adding my Xeon E3-1270L v4 (Broadwell LGA1150) before I switch back to 1155...
>
> ```
> $ ./corefreq-cli -M
>                            X99/Wellsburg  [1618]
> Controller [...]

On that Broadwell, `CM3X8GA2133C9D8` DIMMs are this time at `2133 MHz`
Do you confirm it matches BIOS setup ?

svmlegacy, 2022-10-18:
> On that Broadwell, `CM3X8GA2133C9D8` DIMMs are this time at `2133 MHz`
> Do you confirm it matches BIOS setup ?

BIOS lists memory speed and timings as 1333 MHz, 9-9-9

cyring, 2022-10-18:
> > On that Broadwell, `CM3X8GA2133C9D8` DIMMs are this time at `2133 MHz`
> > Do you confirm it matches BIOS setup ?
> 
> BIOS lists memory speed and timings as 1333 MHz, 9-9-9

Haswell & Broadwell cases are decoded by same code.
I think that if speed differs, using same DIMM, means both platform registers are not of same bit values.

cyring, 2022-10-31:
Remaining actions are linked to unspecified Timings registers.