cyring/CoreFreq #351: MSI Tomahawk Z690 - Memory controller shows RAM is in single channel mode

rushvora, 2022-07-04:
I need to double check this in Windows (with CPU-Z Or HWInfo), will update this issue with findings.
The memory controller screen shows the RAM is in single channel. It should be dual channel, unless for some reason the motherboard is running them in single channel mode.

[image]

Output of `corefreq-cli -s -n -m -n-M` as asked here - https://twitter.com/cyring_labs/status/1543862290579169283

Let me know if there is any more information y'all need.
```
Processor                                 [12th Gen Intel(R) Core(TM) i9-12900K]
|- Architecture                                                     [Alder Lake]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [0x0000001f]
|- Signature                                                           [  06_97]
|- Stepping                                                            [      2]
|- Online CPU                                                          [ 24/ 24]
|- Base Clock                                                          [ 99.598]
|- Frequency            (MHz)                      Ratio                        
                 Min    796.74                    <   8 >                       
                 Max   3186.96                    <  32 >                       
|- Factory                                                             [100.000]
                       3200                       [  32 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT    995.92                    <  10 >                       
   |- HWP                                                                       
                 Min    995.92                    <  10 >                       
                 Max   6473.51                    <  65 >                       
                 TGT      AUTO                    <   0 >                       
|- Turbo Boost                                                         [ UNLOCK]
                  1C   5178.80                    <  52 >                       
                  2C   5079.21                    <  51 >                       
                  3C   4979.62                    <  50 >                       
                  4C   4979.62                    <  50 >                       
                  5C   4880.03                    <  49 >                       
                  6C   4880.03                    <  49 >                       
                  7C   4880.03                    <  49 >                       
                  8C   4880.03                    <  49 >                       
|- Hybrid                                                              [ UNLOCK]
                  1C   3884.43                    <  39 >                       
                  2C   3884.43                    <  39 >                       
                  3C   3884.43                    <  39 >                       
                  4C   3884.43                    <  39 >                       
                  5C   3685.23                    <  37 >                       
                  6C   3685.23                    <  37 >                       
                  7C   3685.23                    <  37 >                       
                  8C   3685.23                    <  37 >                       
|- Uncore                                                              [ UNLOCK]
                 Min    796.81                    <   8 >                       
                 Max   5179.24                    <  52 >                       
|- TDP                                                           Level [  0:3  ]
   |- Programmable                                                     [ UNLOCK]
   |- Configuration                                                    [   LOCK]
   |- Turbo Activation                                                 [ UNLOCK]
             Nominal   3186.96                    [  32 ]                       
               Turbo      AUTO                    <   0 >                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N] 
|- AVX512-VBMI  [N] AVX512-VBMI2 [N]  AVX512-VNNI [N]  AVX512-ALG [N] 
|- AVX512-VPOP  [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [N] AVX-VNNI-VEX [Y]      MOVDIRI [Y]   MOVDIR64B [Y] 
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/N] MON/MWAITX [Y/N]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
[... 193 lines cut ...]
000:BSP    0  P   1   0  0   32768  8     49152 12   1310720 10  31457280 12  
001:  0    1  P   1   0  1   32768  8     49152 12   1310720 10  31457280 12  
002:  0    8  P   1   4  0   32768  8     49152 12   1310720 10  31457280 12  
003:  0    9  P   1   4  1   32768  8     49152 12   1310720 10  31457280 12  
004:  0   16  P   1   8  0   32768  8     49152 12   1310720 10  31457280 12  
005:  0   17  P   1   8  1   32768  8     49152 12   1310720 10  31457280 12  
006:  0   24  P   1  12  0   32768  8     49152 12   1310720 10  31457280 12  
007:  0   25  P   1  12  1   32768  8     49152 12   1310720 10  31457280 12  
008:  0   32  P   1  16  0   32768  8     49152 12   1310720 10  31457280 12  
009:  0   33  P   1  16  1   32768  8     49152 12   1310720 10  31457280 12  
010:  0   40  P   1  20  0   32768  8     49152 12   1310720 10  31457280 12  
011:  0   41  P   1  20  1   32768  8     49152 12   1310720 10  31457280 12  
012:  0   48  P   1  24  0   32768  8     49152 12   1310720 10  31457280 12  
013:  0   49  P   1  24  1   32768  8     49152 12   1310720 10  31457280 12  
014:  0   56  P   1  28  0   32768  8     49152 12   1310720 10  31457280 12  
015:  0   57  P   1  28  1   32768  8     49152 12   1310720 10  31457280 12  
016:  0   64  E   1  32  0   65536  8     32768  8   2097152 16  31457280 12  
017:  0   66  E   1  33  0   65536  8     32768  8   2097152 16  31457280 12  
018:  0   68  E   1  34  0   65536  8     32768  8   2097152 16  31457280 12  
019:  0   70  E   1  35  0   65536  8     32768  8   2097152 16  31457280 12  
020:  0   72  E   1  36  0   65536  8     32768  8   2097152 16  31457280 12  
021:  0   74  E   1  37  0   65536  8     32768  8   2097152 16  31457280 12  
022:  0   76  E   1  38  0   65536  8     32768  8   2097152 16  31457280 12  
023:  0   78  E   1  39  0   65536  8     32768  8   2097152 16  31457280 12  
```

cyring, 2022-07-04:
Thank you for output.
Indeed Windows collections may help to have a better understanding of IMC topology.
Hybrid comes with additional features that make previous SNB-SKL decoders questionable

cyring, 2022-08-03:
Hello,

Can you pull the [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch and try again the IMC ?

```sh
cd tmp
git clone -b develop https://github.com/cyring/CoreFreq
cd CoreFreq
make clean all
```
```sh
# as root
insmod corefreqk.ko
./corefreqd
# as user
# check development version
./corefreq-cli -v
1.91.5
# dump IMC
./corefreq-cli -M
```

rushvora, 2022-08-04:
Still says Single Channel

```
 13:32:51  ~/Downloads/corefreq/CoreFreq  develop                                                                                              17.2G RAM 
❯ ./corefreq-cli -M
                             Intel Z690  [7A84]                            
Controller #0                                               Single Channel 
 Bus Rate  3600 MHz       Bus Speed 3585 MHz           DDR4 Speed 3585 MHz 
                                                                           
 Cha    CL  RCD   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL CKE  CMD  GEAR
  #0    16   19   19   39    4    9   44   24   12   44   16   7   2T     1
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   15        34   27    7    7
      sgWW dgWW drWW ddWW                     REFI  RFC  XS   XP CPDED  ECC
  #0     7    4   12   12                    14055  630  648   11    4    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    2     65536      1024          16384  F4-3600C16-16GTZNC
                                                                           
Controller #1                                               Single Channel 
 Bus Rate  3600 MHz       Bus Speed 3585 MHz           DDR4 Speed 3585 MHz 
                                                                           
 Cha    CL  RCD   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL CKE  CMD  GEAR
  #0    16   19   19   39    4    9   44   24   12   44   16   7   2T     1
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   15        34   27    7    7
      sgWW dgWW drWW ddWW                     REFI  RFC  XS   XP CPDED  ECC
  #0     7    4   12   12                    14055  630  648   11    4    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    2     65536      1024          16384  F4-3600C16-16GTZNC
 13:32:56  ~/Downloads/corefreq/CoreFreq  develop                                                                                              17.2G RAM 
❯ ./corefreq-cli -v
1.91.5
 13:33:26  ~/Downloads/corefreq/CoreFreq  develop                                                                                              17.3G RAM 
❯ 
```

cyring, 2022-08-04:
@BugReporterZ
Could you try the develop branch with your i7-12700K and post the output of Memory Controller ? 
```
corefreq-cli -M
```

BugReporterZ, 2022-08-04:
@cyring
Just tried that (last commit Sun Jul 31 16:22:48 2022 +0000):

```
./corefreq-cli -M
                             Intel Z690  [7A84]                            
Controller #0                                               Single Channel 
 Bus Rate  4600 MHz       Bus Speed 4612 MHz           DDR4 Speed 3609 MHz 
                                                                           
 Cha    CL  RCD   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL CKE  CMD  GEAR
  #0    18   22   22   42    4    9   40   24   12   46   18   7   2T     1
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   15        36   29    7    7
      sgWW dgWW drWW ddWW                     REFI  RFC  XS   XP CPDED  ECC
  #0     7    4   12   12                    14055  630  648   11    4    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    2     65536      1024          16384  CMK32GX4M2D3600C18
                                                                           
Controller #1                                               Single Channel 
 Bus Rate  4600 MHz       Bus Speed 4612 MHz           DDR4 Speed 3609 MHz 
                                                                           
 Cha    CL  RCD   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL CKE  CMD  GEAR
  #0    18   22   22   42    4    9   40   24   12   46   18   7   2T     1
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   15        36   29    7    7
      sgWW dgWW drWW ddWW                     REFI  RFC  XS   XP CPDED  ECC
  #0     7    4   12   12                    14055  630  648   11    4    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    2     65536      1024          16384  CMK32GX4M2D3600C18
```

cyring, 2022-08-04:
@rushvora @BugReporterZ 

For your IMC testings is attached this archive which attempts to decode the channel mode from `DDPCD` of `ADL_CAPID_A` register. (more to read in [655259_003.pdf](https://www.intel.com/content/www/us/en/products/docs/processors/core/core-technical-resources.html))

[CoreFreq_develop.tar.gz](https://github.com/cyring/CoreFreq/files/9260874/CoreFreq_develop.tar.gz)

The kernel log will also print data values of these capabilities
```
"CoreFreq:ADL_Cap_A[%x]\n"
"CoreFreq:ADL_Cap_B[%x]\n"
"CoreFreq:ADL_Cap_C[%x]\n"
"CoreFreq:ADL_Cap_E[%x]\n"
```

rushvora, 2022-08-04:
The files within the tar you provided seems to work, it says Dual Channel now!

```
 02:10:08  ~/Downloads/corefreq-2/CoreFreq                                                                                                        16G RAM 
❯ ./corefreq-cli -v
1.91.5
 02:10:13  ~/Downloads/corefreq-2/CoreFreq                                                                                                        16G RAM 
❯ ./corefreq-cli -M
                             Intel Z690  [7A84]                            
Controller #0                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3585 MHz           DDR4 Speed 3585 MHz 
                                                                           
 Cha    CL  RCD   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL CKE  CMD  GEAR
  #0    16   19   19   39    4    9   44   24   12   44   16   7   2T     1
  #1     5    8    8   28    4    4   16   14    6   24    6   4   1T     1
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   15        34   27    7    7
  #1     4    4    4    4         4    4    4    4         4    4    4    4
      sgWW dgWW drWW ddWW                     REFI  RFC  XS   XP CPDED  ECC
  #0     7    4   12   12                    14055  630  648   11    4    0
  #1     4    4    4    4                     4100  180    0    4    1    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    2     65536      1024          16384  F4-3600C16-16GTZNC
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1                                                                  
                                                                           
Controller #1                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3585 MHz           DDR4 Speed 3585 MHz 
                                                                           
 Cha    CL  RCD   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL CKE  CMD  GEAR
  #0    16   19   19   39    4    9   44   24   12   44   16   7   2T     1
  #1     5    8    8   28    4    4   16   14    6   24    6   4   1T     1
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   15        34   27    7    7
  #1     4    4    4    4         4    4    4    4         4    4    4    4
      sgWW dgWW drWW ddWW                     REFI  RFC  XS   XP CPDED  ECC
  #0     7    4   12   12                    14055  630  648   11    4    0
  #1     4    4    4    4                     4100  180    0    4    1    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    2     65536      1024          16384  F4-3600C16-16GTZNC
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1                                           
       ```

rushvora, 2022-08-04:
> I need to find a way to program directly on your hardware to discover the missing parts...

I'll try and set up the env that's safe for me to allow you to SSH into my machine. I have a spare drive on which I can set up a fresh install of your preferred distro. I need to prevent/disable the other connected drives, which is the cumbersome part.

If there's any guide or something I can follow as an alternative to the above for you to SSH in, let me know (unless you've already figured it out, going by my previous comment).

cyring, 2022-08-04:
> The files within the tar you provided seems to work, it says Dual Channel now!
>
> ```
>  02:10:08  ~/Downloads/corefreq-2/CoreFreq [...]

Great progress, thank you.

Can you also post the CoreFreq traces from the kernel log (`dmesg`)

Verifying the channels count will also mean to detect the opposite case:
Currently Dual, we need to test Single.
I think if you just boot with only one DIMM, CoreFreq should print Single.

rushvora, 2022-08-04:
> Can you also post the CoreFreq traces from the kernel log (`dmesg`)

```
[39144.288416] CoreFreq(0:1:23): Processor [ 06_97] Architecture [Alder Lake] SMT [24/24]
[39144.288432] CoreFreq:ADL_Cap_A[62002028]
               CoreFreq:ADL_Cap_B[b0200000]
               CoreFreq:ADL_Cap_C[c6402380]
               CoreFreq:ADL_Cap_E[1e1640]
[39144.288458] resource sanity check: requesting [mem 0xfedc0000-0xfedcffff], which spans more than pnp 00:03 [mem 0xfedc0000-0xfedc7fff]
[39144.288459] caller Router+0xd6/0x120 [corefreqk] mapping multiple BARs
```

> Verifying the channels count will also mean to detect the opposite case: Currently Dual, we need to test Single. I think if you just boot with only one DIMM, CoreFreq should print Single.

Will try this over the weekend.

cyring, 2022-08-04:
> ```
> [39144.288416] CoreFreq(0:1:23): Processor [ 06_97] Architecture [Alder Lake] SMT [24/24]
> [39144.288432] CoreFreq:ADL_Cap_A[62002028]
>                CoreFreq:ADL_Cap_B[b0200000] [...]

Cool. Also don't forget to post CoreFreq kernel log when in single mode. 
I'll then make a bits difference between Capabilities in Dual and Single configs.

cyring, 2022-08-06:
Fix, and other enhancements, are now available in the [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch.

cyring, 2022-08-08:
@rushvora Hello,

Can you tell about Single channel test ?

closed 2022-08-10 by cyring

rushvora, 2022-08-11:
@cyring My apologies, I didn't get the time I expected to have to test this out over the weekend. I'll try to get it done as soon as I can. I will be out of town this weekend (long weekend), I'll try to get it done sometime next week.

I believe removing/putting back the RAM stick will reset my BIOS configuration, so I want to do it when I have the time to record all the screens and the settings, since I have BIOS level CPU undervolt set up that's been stable all this while.

rushvora, 2023-01-17:
@cyring Firstly, apologies (again!) for not doing the single channel test… I'm exceptionally lazy to redo the BIOS settings for the RAM and CPU :see_no_evil:

The output you've requested.

```
Processor                                 [12th Gen Intel(R) Core(TM) i9-12900K]
|- Architecture                                                     [Alder Lake]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [0x00000026]
|- Signature                                                           [  06_97]
|- Stepping                                                            [      2]
|- Online CPU                                                          [ 24/ 24]
|- Base Clock                                                          [ 99.598]
|- Frequency            (MHz)                      Ratio                        
                 Min    796.79                    <   8 >                       
                 Max   3187.14                    <  32 >                       
|- Factory                                                             [100.000]
                       3200                       [  32 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   1095.58                    <  11 >                       
   |- HWP                                                                       
                 Min   1095.58                    <  11 >                       
                 Max   6473.89                    <  65 >                       
                 TGT      AUTO                    <   0 >                       
|- Turbo Boost                                                         [ UNLOCK]
                  1C   5179.11                    <  52 >                       
                  2C   5079.51                    <  51 >                       
                  3C   4979.91                    <  50 >                       
                  4C   4979.91                    <  50 >                       
                  5C   4880.32                    <  49 >                       
                  6C   4880.32                    <  49 >                       
                  7C   4880.32                    <  49 >                       
                  8C   4880.32                    <  49 >                       
|- Hybrid                                                              [ UNLOCK]
                  1C   3884.36                    <  39 >                       
                  2C   3884.36                    <  39 >                       
                  3C   3884.36                    <  39 >                       
                  4C   3884.36                    <  39 >                       
                  5C   3685.17                    <  37 >                       
                  6C   3685.17                    <  37 >                       
                  7C   3685.17                    <  37 >                       
                  8C   3685.17                    <  37 >                       
|- Uncore                                                              [ UNLOCK]
                 Min    796.79                    <   8 >                       
                 Max   5179.15                    <  52 >                       
|- TDP                                                           Level [  0:3  ]
   |- Programmable                                                     [ UNLOCK]
   |- Configuration                                                    [   LOCK]
   |- Turbo Activation                                                 [ UNLOCK]
             Nominal   3187.14                    [  32 ]                       
               Turbo      AUTO                    <   0 >                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N] 
|- AVX512-VBMI  [N] AVX512-VBMI2 [N]  AVX512-VNNI [N]  AVX512-ALG [N] 
|- AVX512-VPOP  [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [N] AVX-VNNI-VEX [Y]      MOVDIRI [Y]   MOVDIR64B [Y] 
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- ENQCMD       [N]         GFNI [Y]        OSPKE [Y]     WAITPKG [Y] 
|- MMX/Ext    [Y/N] MON/MWAITX [Y/N]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/N]      SSE4.2 [Y] 
|- SERIALIZE    [Y]      SYSCALL [Y]        RDPID [Y]         SGX [N] 
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y] 
[... 341 lines cut ...]
       #0                                                                  
       #1                                                                  
[... cut ...]
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    2     65536      1024          16384  F4-3600C16-16GTZNC
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1                                               
```

rushvora, 2023-01-24:
@cyring 

```
00:00.0 Host bridge [0600]: Intel Corporation 12th Gen Core Processor Host Bridge/DRAM Registers [8086:4660] (rev 02)
00:01.0 PCI bridge [0604]: Intel Corporation 12th Gen Core Processor PCI Express x16 Controller #1 [8086:460d] (rev 02)
00:02.0 Display controller [0380]: Intel Corporation AlderLake-S GT1 [8086:4680] (rev 0c)
00:06.0 PCI bridge [0604]: Intel Corporation 12th Gen Core Processor PCI Express x4 Controller #0 [8086:464d] (rev 02)
00:08.0 System peripheral [0880]: Intel Corporation 12th Gen Core Processor Gaussian & Neural Accelerator [8086:464f] (rev 02)
00:14.0 USB controller [0c03]: Intel Corporation Alder Lake-S PCH USB 3.2 Gen 2x2 XHCI Controller [8086:7ae0] (rev 11)
00:14.2 RAM memory [0500]: Intel Corporation Alder Lake-S PCH Shared SRAM [8086:7aa7] (rev 11)
00:14.3 Network controller [0280]: Intel Corporation Alder Lake-S PCH CNVi WiFi [8086:7af0] (rev 11)
00:16.0 Communication controller [0780]: Intel Corporation Alder Lake-S PCH HECI Controller #1 [8086:7ae8] (rev 11)
00:17.0 SATA controller [0106]: Intel Corporation Alder Lake-S PCH SATA Controller [AHCI Mode] [8086:7ae2] (rev 11)
00:1a.0 PCI bridge [0604]: Intel Corporation Device [8086:7ac8] (rev 11)
00:1c.0 PCI bridge [0604]: Intel Corporation Alder Lake-S PCH PCI Express Root Port #1 [8086:7ab8] (rev 11)
00:1c.3 PCI bridge [0604]: Intel Corporation Device [8086:7abb] (rev 11)
00:1d.0 PCI bridge [0604]: Intel Corporation Alder Lake-S PCH PCI Express Root Port #9 [8086:7ab0] (rev 11)
00:1f.0 ISA bridge [0601]: Intel Corporation Z690 Chipset LPC/eSPI Controller [8086:7a84] (rev 11)
00:1f.3 Audio device [0403]: Intel Corporation Alder Lake-S HD Audio Controller [8086:7ad0] (rev 11)
00:1f.4 SMBus [0c05]: Intel Corporation Alder Lake-S PCH SMBus Controller [8086:7aa3] (rev 11)
00:1f.5 Serial bus controller [0c80]: Intel Corporation Alder Lake-S PCH SPI Controller [8086:7aa4] (rev 11)
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GA102 [GeForce RTX 3080] [10de:2206] (rev a1)
01:00.1 Audio device [0403]: NVIDIA Corporation GA102 High Definition Audio Controller [10de:1aef] (rev a1)
02:00.0 Non-Volatile memory controller [0108]: Samsung Electronics Co Ltd NVMe SSD Controller SM981/PM981/PM983 [144d:a808]
03:00.0 Non-Volatile memory controller [0108]: SK hynix Gold P31/PC711 NVMe Solid State Drive [1c5c:174a]
05:00.0 Ethernet controller [0200]: Intel Corporation Ethernet Controller I225-V [8086:15f3] (rev 03)
06:00.0 Non-Volatile memory controller [0108]: SK hynix Gold P31/PC711 NVMe Solid State Drive [1c5c:174a]
```

rushvora, 2023-01-31:
> Based on `master` can you please post the output `corefreq-cli -M`

```
 18:16:08  ~/Code/Github/CoreFreq  master                                                                                                      10.4G RAM 
❯ ./corefreq-cli -M
                             Intel Z690  [7A84]                            
Controller #0                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3585 MHz           DDR4 Speed 3585 MHz 
                                                                           
 Cha    CL  RCD   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL CKE  CMD  GEAR
  #0    16   19   19   39    4    9   44   24   12   44   16   7   2T     1
  #1     0    0    0    0    0    0    0    0    0    0    0   0   1T     1
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   15        34   27    7    7
  #1     0    0    0    0         0    0    0    0         0    0    0    0
      sgWW dgWW drWW ddWW                     REFI  RFC  XS   XP CPDED  ECC
  #0     7    4   12   12                    14055  630  648   11    4    0
  #1     0    0    0    0                        0    0    0    0    0    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    2     65536      1024          16384  F4-3600C16-16GTZNC
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1                                                                  
                                                                           
Controller #1                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3585 MHz           DDR4 Speed 3585 MHz 
                                                                           
 Cha    CL  RCD   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL CKE  CMD  GEAR
  #0    16   19   19   39    4    9   44   24   12   44   16   7   2T     1
  #1     0    0    0    0    0    0    0    0    0    0    0   0   1T     1
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   15        34   27    7    7
  #1     0    0    0    0         0    0    0    0         0    0    0    0
      sgWW dgWW drWW ddWW                     REFI  RFC  XS   XP CPDED  ECC
  #0     7    4   12   12                    14055  630  648   11    4    0
  #1     0    0    0    0                        0    0    0    0    0    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    2     65536      1024          16384  F4-3600C16-16GTZNC
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1     
       ```

rushvora, 2023-01-31:
> @rushvora Hello, Could you show me the Memory Controller output using latest `develop` branch ? `RCDw` is the new timing I would like to check.

Here you go

```
 18:18:59  ~/Code/Github/CoreFreq  develop                                                                                                     10.5G RAM 
❯ ./corefreq-cli -v   
1.95.2
 18:19:06  ~/Code/Github/CoreFreq  develop                                                                                                     10.5G RAM 
❯ ./corefreq-cli -M
                             Intel Z690  [7A84]                            
Controller #0                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3585 MHz           DDR4 Speed 3585 MHz 
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    16   19   19   19   39    4    9   44   24   12   44   16    7   2T
  #1     0    0    0    0    0    0    0    0    0    0    0    0    0   1T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   15        34   27    7    7
  #1     0    0    0    0         0    0    0    0         0    0    0    0
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0     7    4   12   12               14055  630  648   11    4    1    0
  #1     0    0    0    0                   0    0    0    0    0    1    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    2     65536      1024          16384  F4-3600C16-16GTZNC
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1                                                                  
                                                                           
Controller #1                                                Dual Channel  
 Bus Rate  3600 MHz       Bus Speed 3585 MHz           DDR4 Speed 3585 MHz 
                                                                           
 Cha    CL RCDr RCDw   RP  RAS RRDs RRDl  FAW   WR RTPr WTPr  CWL  CKE  CMD
  #0    16   19   19   19   39    4    9   44   24   12   44   16    7   2T
  #1     0    0    0    0    0    0    0    0    0    0    0    0    0   1T
      sgRR dgRR drRR ddRR      sgRW dgRW drRW ddRW      sgWR dgWR drWR ddWR
  #0     7    4   12   12        14   15   15   15        34   27    7    7
  #1     0    0    0    0         0    0    0    0         0    0    0    0
      sgWW dgWW drWW ddWW                REFI  RFC  XS   XP CPDED GEAR  ECC
  #0     7    4   12   12               14055  630  648   11    4    1    0
  #1     0    0    0    0                   0    0    0    0    0    1    0
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    2     65536      1024          16384  F4-3600C16-16GTZNC
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1                                            
      
```