cyring/CoreFreq #352: CoreFreq gets stuck at 100% CPU load when kernel module is not active

Betaminos, 2022-07-28:
Hello there,

I seem to have stumbled across one or two bugs.
My device is based around an Intel 1165G7 and I am running Arch Linux, up to date as of now, with the default kernel (5.18.14-arch1-1.1).
I have installed CoreFreq via yay and have selected the non-git versions:
`yay corefreq-client corefreq-server corefreq-dkms
`

When I do the insmod, I only get a "Killed" as reply.
However, startig corefreqd seems to work just fine, as it starts and stays up:
> [beta@oxp ~]$ sudo systemctl status corefreqd
> ● corefreqd.service - CoreFreq Daemon
>      Loaded: loaded (/usr/lib/systemd/system/corefreqd.service; disabled; preset: disabled)
>      Active: active (running) since Thu 2022-07-28 18:01:35 +08; 1min 43s ago
>    Main PID: 11602 (corefreqd-pmgr)
>       Tasks: 12 (limit: 18904)
>      Memory: 948.0K
>         CPU: 36ms
>      CGroup: /system.slice/corefreqd.service
>              ├─11602 corefreqd -q
>              └─11603 corefreqd -q
> 
> Jul 28 18:01:35 oxp systemd[1]: Started CoreFreq Daemon.

However, upon starting "corefreq-cli", the terminal freezes, in the sense of no longer accepting any input - no [CTRL]+[C] or the likes do anything.
Checking on the process with htop showsthat the process is eating a full core, pegging it at 100% indefinitely.
I need to forcefully kill corefreq-cli via a SIGTERM to get rid of it.

I would therefore like to propose that corefreq-cli gets a kind of timeout, if it is not able to start up or connect to the kernel module.
Alternatively, (going on my gut feeling, sorry if I am completely wrong with this) the 100% CPU load seems to hint at the program getting stuck in a loop and trying to do the same thing indefinitely.
If this is the case, getting it to stop with an error message after a set amount of failed attempts would be able to help the user in narrowing down what is happening.

Please let me know if I can provide any additional information that can be of help here.

Thank you for your time.

cyring, 2022-07-28:
Hello,

Driver `corefreqk.ko` may have crashed for some reasons.

I will need any trace of driver crash from the kernel log (`dmesg`)

and the _CoreFreq_ version (`corefreq-cli -v`)

cyring, 2022-07-29:
I can't reproduce your issue with `[11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80`

[image]

Try running _CoreFreq_ manually without any package you'll have to remove before the bellow procedure:
1. `git clone https://github.com/cyring/CoreFreq.git`
2. `cd CoreFreq`
3. `make DELAY_TSC=1 clean all`

Open Terminal as root to start Driver then Daemon
```
insmod corefreqk.ko
./corefreqd -d
```

Open Terminal as a User to start Client
```
./corefreq-cli 
```

Betaminos, 2022-07-29:
I have uninstalled my previous CoreFreq installation, cloned the git and built it using your commands.
Afterwards, I did a reboot, logged in and tried to insert the module:

> [beta@oxp CoreFreq]$ sudo insmod corefreqk.ko
> [sudo] password for beta: 
> Killed

I have dumped my dmesg into a log file and have attached it:
[dmesg.log](https://github.com/cyring/CoreFreq/files/9217085/dmesg.log)

Please let me know if I can provide anything else.

cyring, 2022-07-29:
`SMBIOS_Decoder` appears to fail on your BIOS

I'll ask to comment that function call in `corefreqk.c` driver source code; rebuild and test again.

cyring, 2022-07-29:
Go to this line: 
https://github.com/cyring/CoreFreq/blob/a485cecf002c950c97955e81d20d87059115f7f2/corefreqk.c#L21586

and comment function as below:

```c

	/*	Copy various SMBIOS data [version 3.2]			*/
	SMBIOS_Collect();
/*
	SMBIOS_Decoder();
*/
```

then rebuild and test

Betaminos, 2022-07-29:
Commenting out line 21586 and rebuilding CoreFreq does work around the crash.
Is there any help I can provide related to narrowing it down further or making the application more resistant towards this error?

cyring, 2022-07-29:
> Commenting out line 21586 and rebuilding CoreFreq does successfully work around the crash- i.e. it loads and runs fine. Is there any help I can provide related to narrowing it down further or making the application more resistant towards this error?

Thanks a lot for your help.

I have to simulate/fake your setup to reproduce the issue.

According to kernel log, somewhere `strlen` fails and I presume this happens around those lines:
https://github.com/cyring/CoreFreq/blob/a485cecf002c950c97955e81d20d87059115f7f2/corefreqk.c#L20974

cyring, 2022-07-29:
Can you try this attached version ?
[CoreFreq_develop.tar.gz](https://github.com/cyring/CoreFreq/files/9218519/CoreFreq_develop.tar.gz)

Betaminos, 2022-07-29:
> Can you try this attached version ? [CoreFreq_develop.tar.gz](https://github.com/cyring/CoreFreq/files/9218519/CoreFreq_develop.tar.gz)

This version compiles, inserts and runs seemingly fine as well.

cyring, 2022-07-29:
Great  !

Are you also decoding the DIMM manufacturer and identifiers ?
Like `G Skill Intl` and `F4-3600C16-16GTZN` bellow
[image]
Also printed in IMC window

Betaminos, 2022-07-29:
I am not sure about this.
The SMBIOS lists this:
[image]

Memory Controller
[image]

However, this is a Chinese handheld laptop with soldered RAM, so it could very well be that this information is not present in the device.
Due to this, I would assume that the output is as good as it gets, but I will gladly test whatever you need to improve your program.

Thank you for the great support!

cyring, 2022-07-29:
> I am not sure about this. The SMBIOS lists this: [image]
>
> Memory Controller [image]
>
> However, this is a Chinese handheld laptop with soldered RAM, so it could very well be that this information is [...]

OK, I see, we are getting DIMM Manufacturer but not the IDs which are now substitute with an empty string.

Thank you for helping troubleshooting this issue
Time to release bug fix

cyring, 2022-07-30:
_CoreFreq_ ArchLinux [packages](https://aur.archlinux.org/packages?O=0&K=corefreq) have been updated, I think you can now install from the manager (and close this issue if everything is OK)

closed 2022-07-31 by Betaminos