cyring/CoreFreq #354: Ryzen Threadripper 3960X - memory topology not shown

Chlorophytus, 2022-07-30:
I tried both the git and the regular AUR packages but the memory topology is not shown.

Using ASRock TRX40 Creator with beta BIOS 1.83. I have quad channel memory configuration, it should show 3600MHz on all 4 DIMMs.

The CastlePeakPI AGESA is 1.0.0.7

[image]

cyring, 2022-07-30:
Hello,

Can you post (Markdown formatted) the output of `corefreq-cli -s -m -M -v`

cyring, 2022-07-30:
Also edit the driver source file at:
https://github.com/cyring/CoreFreq/blob/20426b669d2da01f93cab8dd208ac3c476333aa7/corefreqk.c#L5870

and replace with `2` the umc_max to query as below:

```c
static PCI_CALLBACK AMD_DataFabric_Starship(struct pci_dev *pdev)
{
	return AMD_17h_DataFabric(pdev, 2);
}
```

Rebuild (`make clean all`) , reload and test UMC again

Chlorophytus, 2022-07-30:
> Hello,
> 
> Can you post (Markdown formatted) the output of `corefreq-cli -s -m -M -v`

```
accelshark@Funbox ~ % corefreq-cli -s -m -M -v
Processor                       [AMD Ryzen Threadripper 3960X 24-Core Processor]
|- Architecture                                               [Zen2/Castle Peak]
|- Vendor ID                                                      [AuthenticAMD]
|- Firmware                                                         [ 36.34.0-2]
|- Microcode                                                        [0x08301055]
|- Signature                                                           [  8F_31]
|- Stepping                                                            [      0]
|- Online CPU                                                          [ 48/ 48]
|- Base Clock                                                          [100.000]
|- Frequency            (MHz)                      Ratio                        
                 Min   2200.04                    <  22 >                       
                 Max   3800.06                    <  38 >                       
|- Factory                                                             [100.000]
                       3800                       [  38 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   3800.06                    <  38 >                       
|- Turbo Boost                                                         [ UNLOCK]
                 XFR   4600.07                    [  46 ]                       
                 CPB   4500.07                    [  45 ]                       
                  1C   2800.05                    <  28 >                       
                  2C   2200.04                    <  22 >                       
|- Uncore                                                              [   LOCK]
|- TDP                                                           Level [  0:0  ]
   |- Programmable                                                     [   LOCK]
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N] 
|- AVX512-VBMI  [N] AVX512-VBMI2 [N]  AVX512-VNNI [N]  AVX512-ALG [N] 
|- AVX512-VPOP  [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [N] AVX-VNNI-VEX [N]    AVX-FP128 [N]   AVX-FP256 [Y] 
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [N]        UMIP [N] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast Short REP MOVSB                                         FSRM   [Missing]
|- Fast-String Operation                                        ERMS   [Missing]
|- Fused Multiply Add                                     FMA | FMA4   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Hardware P-state control                                      HwP   [Capable]
|- Instruction Based Sampling                                    IBS   [Capable]
|- Instruction INVLPGB                                       INVLPGB   [Missing]
|- Instruction INVPCID                                       INVPCID   [Missing]
[... 226 lines cut ...]
	-i <#>	Monitor Instructions
	-s	Print System Information
	-j	Print System Information (json-encoded)
	-M	Print Memory Controller
	-R	Print System Registers
	-m	Print Topology
	-u	Print CPUID
	-B	Print SMBIOS
	-k	Print Kernel
	-n	New line
	-h	Print out this message
	-v	Print the version number

Exit status:
	0	SUCCESS		Successful execution
	3	CMD_SYNTAX	Command syntax error
	4	SHM_FILE	Shared memory file error
	5	SHM_MMAP	Shared memory mapping error
	6	PERM_ERR	Execution not permitted
	7	MEM_ERR		Memory operation error
	8	EXEC_ERR	General execution error
	9	SYS_CALL	System call error

Report bugs to labs[at]cyring.fr
```

Chlorophytus, 2022-07-30:
> Also edit the driver source file at:
>
> https://github.com/cyring/CoreFreq/blob/20426b669d2da01f93cab8dd208ac3c476333aa7/corefreqk.c#L5870
>
> and replace with `2` the umc_max to query as below:
>
> ```c [...]

The UMC is still unrecognized and reads dual channel.

cyring, 2022-07-30:
If `2` does not give better results, try `4`

Chlorophytus, 2022-07-30:
> If `2` does not give better results, try `4`

I tried compiling it outside of the PKGBUILD. Neither `2` nor `4` works.

Chlorophytus, 2022-07-30:
Does this have anything to do with the problem, saw it on `dmesg`.

```
[  144.162745] CoreFreq: AMD_17h_DataFabric() Break UMC(1) probing @ PCI(0x0:0x0:0xc8)
```

cyring, 2022-07-30:
> Does this have anything to do with the problem, saw it on `dmesg`.
> 
> ```
> [  144.162745] CoreFreq: AMD_17h_DataFabric() Break UMC(1) probing @ PCI(0x0:0x0:0xc8)
> ```

I need to confirm the Data Fabric PCI Id.
Can you post the output of `lspci -nn`

Chlorophytus, 2022-07-30:
> > Does this have anything to do with the problem, saw it on `dmesg`.
> > ```
> > [  144.162745] CoreFreq: AMD_17h_DataFabric() Break UMC(1) probing @ PCI(0x0:0x0:0xc8)
> > ```
> 
> I need to confirm the Data Fabric PCI Id. Can you post the output of `lspci -nn`

```00:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Root Complex [1022:1480]
00:00.2 IOMMU [0806]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse IOMMU [1022:1481]
00:01.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
00:01.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse GPP Bridge [1022:1483]
00:02.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
00:03.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
00:04.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
00:05.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
00:07.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
00:07.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Internal PCIe GPP Bridge 0 to bus[E:B] [1022:1484]
00:08.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
00:08.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Internal PCIe GPP Bridge 0 to bus[E:B] [1022:1484]
00:14.0 SMBus [0c05]: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller [1022:790b] (rev 61)
00:14.3 ISA bridge [0601]: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge [1022:790e] (rev 51)
00:18.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship Device 24; Function 0 [1022:1490]
00:18.1 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship Device 24; Function 1 [1022:1491]
00:18.2 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship Device 24; Function 2 [1022:1492]
00:18.3 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship Device 24; Function 3 [1022:1493]
00:18.4 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship Device 24; Function 4 [1022:1494]
00:18.5 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship Device 24; Function 5 [1022:1495]
00:18.6 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship Device 24; Function 6 [1022:1496]
00:18.7 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship Device 24; Function 7 [1022:1497]
01:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 XL Upstream Port of PCI Express Switch [1002:1478] (rev c0)
02:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 XL Downstream Port of PCI Express Switch [1002:1479]
03:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Navi 21 [Radeon RX 6800/6800 XT / 6900 XT] [1002:73bf] (rev c0)
03:00.1 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] Navi 21/23 HDMI/DP Audio Controller [1002:ab28]
03:00.2 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD/ATI] Device [1002:73a6]
03:00.3 Serial bus controller [0c80]: Advanced Micro Devices, Inc. [AMD/ATI] Navi 21 USB [1002:73a4]
04:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Function [1022:148a]
05:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Reserved SPP [1022:1485]
05:00.3 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Starship USB 3.0 Host Controller [1022:148c]
20:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Root Complex [1022:1480]
20:00.2 IOMMU [0806]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse IOMMU [1022:1481]
20:01.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
20:02.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
20:03.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
20:04.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
20:05.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
20:07.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
20:07.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Internal PCIe GPP Bridge 0 to bus[E:B] [1022:1484]
20:08.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
20:08.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Internal PCIe GPP Bridge 0 to bus[E:B] [1022:1484]
21:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Function [1022:148a]
22:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Reserved SPP [1022:1485]
22:00.1 Encryption controller [1080]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Cryptographic Coprocessor PSPCPP [1022:1486]
[... cut ...]
60:08.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
60:08.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Internal PCIe GPP Bridge 0 to bus[E:B] [1022:1484]
61:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Function [1022:148a]
62:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Reserved SPP [1022:1485]```

cyring, 2022-07-30:
Can you compile and run zencli to dump UMC

https://github.com/cyring/Tips/blob/master/C/zencli.c

```sh
cc zencli.c -o zencli

# run as root
zencli umc
```

Please post the output.

Chlorophytus, 2022-07-30:
```
Data Fabric: scanning UMC @ BAR[0x00050000] : 0 1 2 3 4 5 6 7 for 4 Channels

CHA[0] CHIP[0:0] @ 0x00250000[0x00000000] Disable, Rank=0
CHA[0] MASK[0:0] @ 0x00250020[0x00000000]
CHA[0] CHIP[0:1] @ 0x00250010[0x00000000] Disable, Rank=0
CHA[0] MASK[0:1] @ 0x00250028[0x00000000]
CHA[0] CHIP[1:0] @ 0x00250004[0x00000000] Disable, Rank=0
CHA[0] MASK[1:0] @ 0x00250020[0x00000000]
CHA[0] CHIP[1:1] @ 0x00250014[0x00000000] Disable, Rank=0
CHA[0] MASK[1:1] @ 0x00250028[0x00000000]
CHA[0] CHIP[2:0] @ 0x00250008[0x00000001] Enable, Rank=2
CHA[0] MASK[2:0] @ 0x00250024[0x07fffdfe] ChipSize[16777216]
CHA[0] CHIP[2:1] @ 0x00250018[0x00000000] Disable, Rank=0
CHA[0] MASK[2:1] @ 0x0025002c[0x00000000]
CHA[0] CHIP[3:0] @ 0x0025000c[0x00000201] Enable, Rank=2
CHA[0] MASK[3:0] @ 0x00250024[0x07fffdfe] ChipSize[16777216]
CHA[0] CHIP[3:1] @ 0x00250018[0x00000000] Disable, Rank=0
CHA[0] MASK[3:1] @ 0x0025002c[0x00000000]

DIMM Size[33554432 KB] [32768 MB]

CHA[1] CHIP[0:0] @ 0x00350000[0x00000000] Disable, Rank=0
CHA[1] MASK[0:0] @ 0x00350020[0x00000000]
CHA[1] CHIP[0:1] @ 0x00350010[0x00000000] Disable, Rank=0
CHA[1] MASK[0:1] @ 0x00350028[0x00000000]
CHA[1] CHIP[1:0] @ 0x00350004[0x00000000] Disable, Rank=0
CHA[1] MASK[1:0] @ 0x00350020[0x00000000]
CHA[1] CHIP[1:1] @ 0x00350014[0x00000000] Disable, Rank=0
CHA[1] MASK[1:1] @ 0x00350028[0x00000000]
CHA[1] CHIP[2:0] @ 0x00350008[0x00000001] Enable, Rank=2
CHA[1] MASK[2:0] @ 0x00350024[0x07fffdfe] ChipSize[16777216]
CHA[1] CHIP[2:1] @ 0x00350018[0x00000000] Disable, Rank=0
CHA[1] MASK[2:1] @ 0x0035002c[0x00000000]
CHA[1] CHIP[3:0] @ 0x0035000c[0x00000201] Enable, Rank=2
CHA[1] MASK[3:0] @ 0x00350024[0x07fffdfe] ChipSize[16777216]
CHA[1] CHIP[3:1] @ 0x00350018[0x00000000] Disable, Rank=0
CHA[1] MASK[3:1] @ 0x0035002c[0x00000000]

DIMM Size[33554432 KB] [32768 MB]

CHA[2] CHIP[0:0] @ 0x00450000[0x00000000] Disable, Rank=0
CHA[2] MASK[0:0] @ 0x00450020[0x00000000]
CHA[2] CHIP[0:1] @ 0x00450010[0x00000000] Disable, Rank=0
CHA[2] MASK[0:1] @ 0x00450028[0x00000000]
CHA[2] CHIP[1:0] @ 0x00450004[0x00000000] Disable, Rank=0
CHA[2] MASK[1:0] @ 0x00450020[0x00000000]
CHA[2] CHIP[1:1] @ 0x00450014[0x00000000] Disable, Rank=0
CHA[2] MASK[1:1] @ 0x00450028[0x00000000]
CHA[2] CHIP[2:0] @ 0x00450008[0x00000001] Enable, Rank=2
CHA[2] MASK[2:0] @ 0x00450024[0x07fffdfe] ChipSize[16777216]
CHA[2] CHIP[2:1] @ 0x00450018[0x00000000] Disable, Rank=0
CHA[2] MASK[2:1] @ 0x0045002c[0x00000000]
CHA[2] CHIP[3:0] @ 0x0045000c[0x00000201] Enable, Rank=2
CHA[2] MASK[3:0] @ 0x00450024[0x07fffdfe] ChipSize[16777216]
CHA[2] CHIP[3:1] @ 0x00450018[0x00000000] Disable, Rank=0
CHA[2] MASK[3:1] @ 0x0045002c[0x00000000]

DIMM Size[33554432 KB] [32768 MB]

CHA[3] CHIP[0:0] @ 0x00550000[0x00000000] Disable, Rank=0
CHA[3] MASK[0:0] @ 0x00550020[0x00000000]
CHA[3] CHIP[0:1] @ 0x00550010[0x00000000] Disable, Rank=0
CHA[3] MASK[0:1] @ 0x00550028[0x00000000]
CHA[3] CHIP[1:0] @ 0x00550004[0x00000000] Disable, Rank=0
CHA[3] MASK[1:0] @ 0x00550020[0x00000000]
CHA[3] CHIP[1:1] @ 0x00550014[0x00000000] Disable, Rank=0
CHA[3] MASK[1:1] @ 0x00550028[0x00000000]
CHA[3] CHIP[2:0] @ 0x00550008[0x00000001] Enable, Rank=2
CHA[3] MASK[2:0] @ 0x00550024[0x07fffdfe] ChipSize[16777216]
CHA[3] CHIP[2:1] @ 0x00550018[0x00000000] Disable, Rank=0
CHA[3] MASK[2:1] @ 0x0055002c[0x00000000]
CHA[3] CHIP[3:0] @ 0x0055000c[0x00000201] Enable, Rank=2
CHA[3] MASK[3:0] @ 0x00550024[0x07fffdfe] ChipSize[16777216]
CHA[3] CHIP[3:1] @ 0x00550018[0x00000000] Disable, Rank=0
CHA[3] MASK[3:1] @ 0x0055002c[0x00000000]

DIMM Size[33554432 KB] [32768 MB]

```

cyring, 2022-07-30:
Do you have 4 DIMMs, 32GB each ?

Chlorophytus, 2022-07-30:
> Do you have 4 DIMMs, 32GB each ?

Yes

cyring, 2022-07-31:
Hello,

Can you try this attached version ?

[CoreFreq_develop.tar.gz](https://github.com/cyring/CoreFreq/files/9228185/CoreFreq_develop.tar.gz)

Chlorophytus, 2022-07-31:
> Hello,
> 
> Can you try this attached version ?
> 
> [CoreFreq_develop.tar.gz](https://github.com/cyring/CoreFreq/files/9228185/CoreFreq_develop.tar.gz)

[image]

That fixes it partially. Channels 0 and 1 are not detected.

cyring, 2022-07-31:
Hello,

Can you test this one:

[CoreFreq_develop.tar.gz](https://github.com/cyring/CoreFreq/files/9228977/CoreFreq_develop.tar.gz)

Chlorophytus, 2022-07-31:
Thanks, that fixed it.

cyring, 2022-07-31:
> Thanks, that fixed it.

Great, can you please show me the output of `corefreq-cli -M`

Chlorophytus, 2022-07-31:
> > Thanks, that fixed it.
> 
> Great, can you please show me the output of `corefreq-cli -M`

```
accelshark@Funbox ~/Downloads/CoreFreq % ./corefreq-cli -M
                              Zen UMC  [1490]                              
Controller #0                                                Quad Channel  
 Bus Rate  1800 MHz       Bus Speed 1799 MHz           DDR4 Speed 3599 MT/s
                                                                           
 Cha   CL  RCDr RCDw  RP  RAS   RC  RRDs RRDl FAW  WTRs WTRl  WR  clRR clWW
  #0   18   22   22   22   42   82    6    9   38    5   14   26    5    5 
  #1   18   22   22   22   42   82    6    9   38    5   14   26    5    5 
  #2   18   22   22   22   42   82    6    9   38    5   14   26    5    5 
  #3   18   22   22   22   42   82    6    9   38    5   14   26    5    5 
      CWL  RTP RdWr WrRd scWW sdWW ddWW scRR sdRR ddRR drRR drWW drWR drRRD
  #0   18   14    7    3    1    7    6    1    5    4    0    0    0    0 
  #1   18   14    8    3    1    7    6    1    5    4    0    0    0    0 
  #2   18   14    7    3    1    7    6    1    5    4    0    0    0    0 
  #3   18   14    8    3    1    7    6    1    5    4    0    0    0    0 
      REFI RFC1 RFC2 RFC4 RCPB RPPB  BGS:Alt  Ban  Page  CKE  CMD  GDM  ECC
  #0 14029  312  192  132   0    0   OFF  ON  R1W1   0    0   1T    ON   0 
  #1 14029  312  192  132   0    0   OFF  ON  R1W1   0    0   1T    ON   0 
  #2 14029  312  192  132   0    0   OFF  ON  R1W1   0    0   1T    ON   0 
  #3 14029  312  192  132   0    0   OFF  ON  R1W1   0    0   1T    ON   0 
      MRD:PDA   MOD:PDA  WRMPR STAG PDM RDDATA WRD  WRL  RDL  XS   XP CPDED
  #0    8  18    27  27    24  255 0:F:0   13   2   13   26 1008   11    4 
  #1    8  18    27  27    24  255 0:F:0   13   2   13   26 1008   11    4 
  #2    8  18    27  27    24  255 0:F:0   13   2   13   26 1008   11    4 
  #3    8  18    27  27    24  255 0:F:0   13   2   13   26 1008   11    4 
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    2    131072      1024          32768                    
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    2    131072      1024          32768                    
 DIMM Geometry for channel #2                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    2    131072      1024          32768                    
 DIMM Geometry for channel #3                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    2    131072      1024          32768 
```

This seems correct to me, the motherboard has 8 DIMM slots and only 4 are occupied.

cyring, 2022-07-31:
Released in [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch

If you have any other Ryzen, Threadripper, EPYC processors, let me know if you have a chance to test them with this branch ?

Outputs are saved in 3960X wiki [page](https://gist.github.com/cyring/131f2c549fd6c8a8274e4e9e6c15015a?permalink_comment_id=4251034#gistcomment-4251034)

closed 2022-08-09 by cyring