cyring/CoreFreq #355: Values in sub-windows are not refreshed

Betaminos, 2022-07-31:
Hello there,

thank you for this great program! :)
I have come across another minor bug that I would like to point out.

When opening sub-windows, e.g. Technologies, CoreFreq will show the values:
[image]
The values as they are shown are partially stored as a value of 4005e in MSR 0x1FC:
> sudo rdmsr 0x1fc
> 4005e

Switching the values like so (as an example), i.e. the lower 3 ONs are now OFFs:
[image]
causes the MSR to change to 1c005e:
> sudo rdmsr 0x1fc
> 1c005e

The bug happens, when I manually set the MSR 0x1FC back to 4005e:
> sudo wrmsr 0x1fc 0x4005e

The values take effect / are written correctly, as can be seen via:
> sudo rdmsr 0x1fc
> 4005e

The bug, in my opinion, is that CoreFreq still shows:
[image]
(I understand that reading the various MSRs once per second, or whatever refresh interval is set, would cause issues.)
This display does not change, however, even when the Technologies window is closed and opened back up.
I am only able to get CoreFreq to refresh the values, when I change one value stored inside the same MSR.
For example, changing Turbo Boost back to ON causes the refresh only for fields where the value is stored in the same MSR as Turbo Boost.

My proposal is to refresh / read all the values of MSRs contained in a window, each time the window is opened.
A full refresh while the window is open seems excessive to me, but having an up to date status of the values when re-opening the window would be great.

Thank you for your time and consideration.

PS: I have started messing about with the MSRs as I would like to have some settings applied on boot and have not found a way for CoreFreq to do so.
Due to this, I am changing the MSRs via a script on boot and have tried to check them via CoreFreq when I came across this bug.

cyring, 2022-07-31:
You are right that bits within _CoreFreq_ are **not** updated when registers are **externally** altered.

This was an **implementation choice** in favor of less _CoreFreq_ CPU overhead. 
The main purpose of _CoreFreq_ is CPU monitoring and I'm chasing any wasted cycle.

_CoreFreq_ UI is an all in one tool where you are supposed to tweak bits straight from it, and not externally (`wrmsr`, `setpci`, and so on).

Meanwhile, a special shortcut exists to request a full refresh: press the **star** key `*`

However if some indicators like `EEO`, `R2H`, despite a refresh request, are not reflected then I'll debug those cases.

cyring, 2022-07-31:
* `EEO` and `R2H` use case

Nominal is ON
[image]
After disabling both
[image]
Re-enabling
[image]
[image]
Final state
[image]

* Probably those features are missing or not implemented for your processor ?
Please post the output of `corefreq-cli -s -n -k`

cyring, 2022-07-31:
* FYI, `EEO` and `R2H` are set in `MSR_IA32_POWER_CTL`

https://github.com/cyring/CoreFreq/blob/20426b669d2da01f93cab8dd208ac3c476333aa7/corefreqk.c#L3331

https://github.com/cyring/CoreFreq/blob/20426b669d2da01f93cab8dd208ac3c476333aa7/intelmsr.h#L86

Betaminos, 2022-07-31:
> This was an **implementation choice** in favor of less _CoreFreq_ CPU overhead. The main purpose of _CoreFreq_ is CPU monitoring and I'm chasing any wasted cycle.
>
> _CoreFreq_ UI is an all in one [...]

I see.
My scenario is the external change of MSRs, e.g. through a script or embedded controller, unknown to the user.
Due to this, I would have preferred if opening a window causes a refresh of all values contained in that window.
After all, if a user is actively switching windows, then CoreFreq is not running idly in the background.
Due to this, I would be fine with it using a few more CPU cycles when opening a window, if these CPU cycles go towards refreshing the content.

> Meanwhile, a special shortcut exists to request a full refresh: press the **star** key `*`
The star key does not seem to do anything on my keyboard, but your description sounds exactly like the thing I am expecting opening a window to do.

cyring, 2022-07-31:
Are these issues still your processor `i7-1165G7` of Alder Lake architecture ?

If not, please post the output of `corefreq-cli -s -n -k`

cyring, 2022-07-31:
> I see. My scenario is the external change of MSRs, e.g. through a script or embedded controller, unknown to the user. Due to this, I would have preferred if opening a window causes a refresh of all values contained in that window. After all, if a user is actively switching windows inside CoreFreq, then CoreFreq is not running idly in the background. Due to this, I would be fine with it using a few more CPU cycles when opening a window, if these CPU cycles go towards refreshing the content.

In fact _CoreFreq_ model is not a pull model but a push model.

List of registers are collected by driver. CLI has no control over the chosen list of MSRs, CSRs, and other low level means.

Which means that Driver is continually collecting whatever the state of Deamon, CLI and/or GUI. 
In short, it's a broadcast model

What I understand from your needs seems a code refactoring to me. But I tends to stabilize _CoreFreq_: bug fix only.