cyring/CoreFreq #356: [WANTED] AMD/Zen3: Cezanne, Rembrandt, Milan, Chagall

cyring, 2022-08-02:
Hello

The targeted processors have a CPUID signature among the following:

Architecture |CPUID|Codename|Support
-----------------|----------|--------------|-----------
`[Zen3/Cezanne]`|`AF_50h`| Ryzen Vermeer|DONE
`[EPYC/Milan]`|`AF_01h`| Genesis[GN]|UNKNOWN
`[EPYC/Milan-X]`|`AF_01h`| EPYC|UNKNOWN
`[Zen3/Chagall]`|`AF_08h`| Ryzen Threadripper|UNKNOWN
`[Zen3/Badami]`|`AF_30h`|[BA] - 7 nm - SVR|UNKNOWN
`[Zen3+ Rembrandt]`|`AF_44h`| Ryzen[RMB]|UNKNOWN

Can anyone give a try to the [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch of _CoreFreq_ next version `1.91.5`  and post the output of the Client (`corefreq-cli -s -n -m -n -M`)

One may have to start the driver in Experimental mode to claim an [UMC collect](https://github.com/cyring/CoreFreq/blob/33f10208cf0c8383a7e3e6ff9c01023339202815/corefreqk.c#L5909).

```
insmod corefreqk.ko Experimental=1
```

Thanks for any help

nhattu1986, 2022-08-06:
Hi, 

This is the output of the program on my Ryzen 7 5800H 
[AF_50h.txt](https://github.com/cyring/CoreFreq/files/9275567/AF_50h.txt)

cyring, 2022-08-06:
> Hi,
> 
> This is the output of the program on my Ryzen 7 5800H

Thank you very much.

zencli is decoding 2 DIMMs 4GB each. Is this what your setup has ?

nhattu1986, 2022-08-07:
> > Hi,
> > This is the output of the program on my Ryzen 7 5800H
> 
> Thank you very much.
> 
> zencli is decoding 2 DIMMs 4GB each. Is this what your setup has ?

no, my machine is configured with 2x8GB module
[dmidecode.txt](https://github.com/cyring/CoreFreq/files/9276324/dmidecode.txt)

cyring, 2022-08-07:
> > > Hi,
> > > This is the output of the program on my Ryzen 7 5800H
> > 
> > 
> > Thank you very much.
> > zencli is decoding 2 DIMMs 4GB each. Is this what your setup has ?
> 
> no, my machine is configured with 2x8GB module [dmidecode.txt](https://github.com/cyring/CoreFreq/files/9276324/dmidecode.txt)

Can you pull, build and run the **latest** [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch ?
_Last commit is removing the Experimental mode to query the UMC Memory Controller_

The output I would like to examine:
```
 corefreq-cli -M -n -B -n -k -n -C 1
```

nhattu1986, 2022-08-07:
> Can you pull, build and run the **latest** [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch ? _Last commit is removing the Experimental mode to query the UMC Memory Controller_ [...]

Hi
This is the output using the latest develop branch build

```
                            AuthenticAMD  [   0]                           
Controller #0                                                    Disabled  

[ 0] American Megatrends International, LLC.                                    
[ 1] E158LAMS.108                                                               
[ 2] 04/07/2022                                                                 
[ 3] Micro-Star International Co., Ltd.                                         
[ 4] Alpha 15 B5EEK                                                             
[ 5] REV:1.0                                                                    
[ 6] 9---5---2---Z---0---5                                                      
[ 7] 158L.1                                                                     
[ 8] Alpha                                                                      
[ 9] Micro-Star International Co., Ltd.                                         
[10] MS-158L                                                                    
[11] REV:1.0                                                                    
[12] B---0---4---8-                                                             
[13] Number Of Devices:2\Maximum Capacity:33554432 bytes                        
[14] DIMM 0\P0 CHANNEL A                                                        
[15] DIMM 0\P0 CHANNEL B                                                        
[16]                                                                            
[17]                                                                            
[18] Samsung                                                                    
[19] Samsung                                                                    
[20]                                                                            
[21]                                                                            
[22] M471A1K43EB1-CWE                                                           
[23] M471A1K43EB1-CWE                                                           
[24]                                                                            
[25]                                                                            

Linux:                                                                          
|- Release                                                             [5.18.16]
|- Version                  [#1 SMP PREEMPT_DYNAMIC Wed Aug 3 07:13:22 CDT 2022]
|- Machine                                                              [x86_64]
Memory:                                                                         
|- Total RAM                                                         15743348 KB
|- Shared RAM                                                          193632 KB
|- Free RAM                                                           3455900 KB
|- Buffer RAM                                                          152156 KB
|- Total High                                                               0 KB
|- Free High                                                                0 KB
Clock Source                                                  <             tsc>
CPU-Freq driver                                               [    acpi-cpufreq]
Governor                                                      [     performance]
CPU-Idle driver                                               [       acpi_idle]
|- Idle Limit                                                 [              C3]
   |- State        POLL      C1      C2      C3                                 
   |-           CPUIDLE ACPI FF ACPI IO ACPI IO                                 
   |- Power          -1       0       0       0                                 
   |- Latency         0       1      18     350                                 
   |- Residency       0       2      36     700                                 

CPU Freq(MHz) VID  Vcore  TMP(C)    Accumulator       Energy(J)     Power(W)
000   31.90   125  0.7688   47  000000000000010449    0.159439087   0.159439087
001   47.25   122  0.7875   47  000000000000000000    0.000000000   0.000000000
002   51.03   122  0.7875   47  000000000000006615    0.100936890   0.100936890
003   16.07   122  0.7875   47  000000000000000000    0.000000000   0.000000000
004   88.24   121  0.7937   47  000000000000009005    0.137405396   0.137405396
005   17.99   121  0.7937   47  000000000000000000    0.000000000   0.000000000
006   36.43   122  0.7875   47  000000000000011955    0.182418823   0.182418823
007   34.50   122  0.7875   47  000000000000000000    0.000000000   0.000000000
008   46.83   121  0.7937   47  000000000000007135    0.108871460   0.108871460
009   29.55   121  0.7937   47  000000000000000000    0.000000000   0.000000000
010   42.10   122  0.7875   47  000000000000005993    0.091445923   0.091445923
011   24.54   122  0.7875   47  000000000000000000    0.000000000   0.000000000
012   25.71   121  0.7937   47  000000000000007994    0.121978760   0.121978760
013   13.23   121  0.7937   47  000000000000000000    0.000000000   0.000000000
014   54.20   112  0.8500   47  000000000000016453    0.251052856   0.251052856
015  152.56   112  0.8500   47  000000000000000000    0.000000000   0.000000000

             Package       Cores         Uncore        Memory        Platform
Energy(J):  10.459701538   1.457260132   6.430450439   0.000000000   0.000000000
Power(W) :  10.459701538   1.457260132   6.430450439   0.000000000   0.000000000
```

kernel load command:

[image]

[AF_50h_1.txt](https://github.com/cyring/CoreFreq/files/9276625/AF_50h_1.txt)

cyring, 2022-08-07:
@nhattu1986 Thanks. Something I don't understand with this APU which queries no UMC data from driver.

cyring, 2022-08-07:
@nhattu1986

Can you try UMC again but make sure **without** the `Experimental` parameter.

Like:
```
insmod corefreqk.ko
corefreqd
```
```
corefreq-cli -M
```

nhattu1986, 2022-08-07:
> @nhattu1986
> 
> Can you try UMC again but make sure **without** the `Experimental` parameter.
> 
> Like:
> 
> ```
> insmod corefreqk.ko
> corefreqd
> ```
> 
> ```
> corefreq-cli -M
> ```

Hi,

Running the kernel module without parameter still not yield any output

```
nhattu@a15:/tmp/corefreq-develop$ corefreq-cli -M
                            AuthenticAMD  [   0]                           
Controller #0                                                    Disabled  
```

[image]

[image]

cyring, 2022-08-07:
> Hi,
>
> Running the kernel module without parameter still not yield any output
>
> ```
> nhattu@a15:/tmp/corefreq-develop$ corefreq-cli -M
>                             AuthenticAMD  [ [...]

I have to put traces in driver to follow the callflow.

cyring, 2022-08-07:
@nhattu1986 Can you please run the attached archive ?

[CoreFreq_develop.tar.gz](https://github.com/cyring/CoreFreq/files/9277316/CoreFreq_develop.tar.gz)

_To avoid any API conflict, make sure any other versions of CoreFreq is unloaded/uninstalled/removed from your disk. Only this archive has to be `insmod` from its own directory._

Your kernel log should dump some traces as below:
```
CoreFreq: AMD_DataFabric_Cezanne(00000000dcb8cb06)
CoreFreq: AMD_17h_DataFabric(00000000dcb8cb06,1,24)
CoreFreq: AMD_17h_DataFabric(): CtrlCount[1]
CoreFreq: AMD_17h_DataFabric(): ChannelCnt[0}
CoreFreq: AMD_17h_DataFabric(): SDP_CTRL passed
CoreFreq: AMD_17h_DataFabric(): ChannelCnt[1}
CoreFreq: AMD_17h_DataFabric(): SDP_CTRL passed
CoreFreq: AMD_17h_DataFabric(): ChannelCnt[2}
CoreFreq: AMD_17h_DataFabric(): ChannelCnt[3}
CoreFreq: AMD_17h_DataFabric(): ChannelCnt[4}
CoreFreq: AMD_17h_DataFabric(): ChannelCnt[5}
CoreFreq: AMD_17h_DataFabric(): ChannelCnt[6}
CoreFreq: AMD_17h_DataFabric(): ChannelCnt[7}
```

nhattu1986, 2022-08-07:
> @nhattu1986 Can you please run the attached archive ?
>
> [CoreFreq_develop.tar.gz](https://github.com/cyring/CoreFreq/files/9277316/CoreFreq_develop.tar.gz)
>
> _To avoid any API conflict, make sure any [...]

Hi

This is the kernel output:
```
[47012.082458] CoreFreq(2:3:-1): Processor [ AF_50] Architecture [Zen3/Cezanne] SMT [16/16]
[47012.082545] CoreFreq: AMD_DataFabric_Cezanne(00000000339291a7)
[47012.082548] CoreFreq: AMD_17h_DataFabric(00000000339291a7,1,24)
[47012.082552] CoreFreq: AMD_17h_DataFabric(): CtrlCount[1]
[47012.082553] CoreFreq: AMD_17h_DataFabric(): ChannelCnt[0}
[47012.082558] CoreFreq: AMD_17h_DataFabric(): SDP_CTRL passed
[47012.082635] CoreFreq: AMD_17h_DataFabric(): ChannelCnt[1}
[47012.082639] CoreFreq: AMD_17h_DataFabric(): SDP_CTRL passed
[47012.082694] CoreFreq: AMD_17h_DataFabric(): ChannelCnt[2}
[47012.082698] CoreFreq: AMD_17h_DataFabric(): ChannelCnt[3}
[47012.082702] CoreFreq: AMD_17h_DataFabric(): ChannelCnt[4}
[47012.082705] CoreFreq: AMD_17h_DataFabric(): ChannelCnt[5}
[47012.082709] CoreFreq: AMD_17h_DataFabric(): ChannelCnt[6}
[47012.082712] CoreFreq: AMD_17h_DataFabric(): ChannelCnt[7}
```
[image]

cyring, 2022-08-07:
@nhattu1986 Hello, can you pull latest `develop` branch ?
_I have fixed missing Device IDs in Daemon._
I would like the output `corefreq-cli -s -n -M`

nhattu1986, 2022-08-08:
Hi @cyring

I'm building the latest develop branch and got the output:

```
Processor                               [AMD Ryzen 7 5800H with Radeon Graphics]
|- Architecture                                                   [Zen3/Cezanne]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0a50000c]
|- Signature                                                           [  AF_50]
|- Stepping                                                            [      0]
|- Online CPU                                                          [ 16/ 16]
|- Base Clock                                                          [ 99.789]
|- Frequency            (MHz)                      Ratio                        
                 Min   1197.47                    <  12 >                       
                 Max   3193.25                    <  32 >                       
|- Factory                                                             [100.000]
                       3200                       [  32 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   1197.47                    <  12 >                       
|- Turbo Boost                                                         [   LOCK]
                 XFR   4390.72                    [  44 ]                       
                 CPB   4390.72                    [  44 ]                       
                  1C   1297.26                    <  13 >                       
                  2C   1197.47                    <  12 >                       
|- Uncore                                                              [   LOCK]
|- TDP                                                           Level [  0:0  ]
   |- Programmable                                                     [   LOCK]
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N] 
|- AVX512-VBMI  [N] AVX512-VBMI2 [N]  AVX512-VNNI [N]  AVX512-ALG [N] 
|- AVX512-VPOP  [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [N] AVX-VNNI-VEX [N]    AVX-FP128 [N]   AVX-FP256 [Y] 
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [N]        UMIP [N] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Capable]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast Short REP MOVSB                                         FSRM   [Capable]
|- Fast-String Operation                                        ERMS   [Capable]
|- Fused Multiply Add                                     FMA | FMA4   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Hardware P-state control                                      HwP   [Capable]
|- Instruction Based Sampling                                    IBS   [Capable]
|- Instruction INVLPGB                                       INVLPGB   [Missing]
|- Instruction INVPCID                                       INVPCID   [Capable]
|- Long Mode 64 bits                                       IA64 | LM   [Capable]
|- LightWeight Profiling                                         LWP   [Missing]
[... 140 lines cut ...]
                                                                           
 Cha   CL  RCDr RCDw  RP  RAS   RC  RRDs RRDl FAW  WTRs WTRl  WR  clRR clWW
  #0   22   22   22   22   52   74    4    8   34    4   12   24    5    5 
[... cut ...]
       #0    16    1     65536      1024           8192    M471A1K43EB1-CWE
       #1                                                                  
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0    16    1     65536      1024           8192    M471A1K43EB1-CWE
       #1                                                                  
```
 
[AF_50h_2.txt](https://github.com/cyring/CoreFreq/files/9281650/AF_50h_2.txt)

cyring, 2022-08-08:
> Hi @cyring
> 
> I'm building the latest develop branch and got the output:

**Thanks a lot for your contribution.**

Wiki [CPU support](https://github.com/cyring/CoreFreq/wiki/CPU-support) page is updated with [Ryzen 7 5800H](https://gist.github.com/cyring/c93ab792ee9165cfd287d9f71a62de0d)

This development is also available in the _CoreFreq_ ISO image @ [www.cyring.fr](https://www.cyring.fr)

Other features can also be tested with Cezanne:
1. `HSMP enablement` from the `Settings` menu
2. `CPPC` from the `Performance Monitoring` menu