cyring/CoreFreq #359: dual-socket CPU power: Core power is greater than package?

Rimits, 2022-08-10:
Is it normal or a bug that Core Power 279W is greater than Package Power 210W ?

OS:ubuntu 20.04
CoreFreq version: 1.91.5
CPU: epyc 7542 *2
Motherboard: Gigabyte
[image]

cyring, 2022-08-10:
It is not normal and dual EPYC is barely never tested; before your screenshot.

Need to debug: send you code changes; is your dual socket platform available for experimentation ?

cyring, 2022-08-10:
* `Cores` watt is the sum of all **physical** Core threads. Logical SMT Cores are excluded from the sum.  
* `Package` watt is source from a hardware register

Without better specs, I think `Package` refers to one socket

Rimits, 2022-08-10:
Of course, I can do some experiments on that server.
If I was right, the 279W of cores power is accurate, but it's the sum of all the cores of the two sockets, because the BMC shows the current and voltage of VRM_CPU0 are 142A and 0.94V.
The Pacakge power, i don't know, maybe it's showing only the power of the first socket.

cyring, 2022-08-10:
The label `Package0` is a good suggestion.

You can also target the second socket for a `Package` watt reading.
You'll have to bind the CoreFreq Service Processor to a CPU which belongs to second socket.
1. Refer to the `Topology` to find a CPU of second socket. Let's say `64`
2. Reload the driver this way:
```sh
insmod corefreqk.ko ServiceProcessor=64

## Or adapt your modprobe
```
3. Please provides a screenshot of the `Power` view

Remarks:
* I presumed Package0 and Package1 should slightly differ ?
* When not specified, Service Processor is selected according to kernel scheduling
* Service Processor can be assigned to any socket
* If CPU is disabled, Service Processor is auto migrated by driver
* UI is always highlighting the CPU number of the Service Processor and its SMT, (and the Hybrid with Intel 12th gen.)

cyring, 2022-08-10:
> Of course, I can do some experiments on that server. The 279W of cores power could be accurate, because the BMC shows the current and voltage of VRM_CPU0 are 142A and 0.94V. Package watt seems accurate too since the cpu is limited to 225W. Maybe appending a "0" to the `Package` label will fix the misleading situation.

`0.94V` in your readings is during idle case

Computing from my Matisse voltage of `1.48 V` (instant load voltage) and your Rome current of `142A`, I'm getting a consumed power `P = V x I` of `~ 210W`

Rimits, 2022-08-10:
Topology:
```
 ---------------------------------- Topology ----------------------------------      
 CPU Pkg  Apic  Core/Thread  Caches      (w)rite-Back (i)nclusive              
  #   ID   ID CCD CCX ID/ID L1-Inst Way  L1-Data Way      L2  Way      L3  Way 
 000:BSP    0   0  0   0  0      32  8        32  8       512  8 i  131072 16w  
 001:  0    2   0  0   1  0      32  8        32  8       512  8 i  131072 16w 
 ...
 031:  0   62   6  7  31  0      32  8        32  8       512  8 i  131072 16w  
 ...                                          
 ------------------------------------------------------------------------------
```
Reload: `insmod corefreqk.ko ServiceProcessor=32`

Result:
[image]

When Service Processor is 19:
They are slightly different, since CPU2 is cooler and the frequency of it is higher.
[image]

cyring, 2022-08-10:
For your information, below specs of the two `Power` registers:  
[image]
[image]

cyring, 2022-08-10:
Hello,

Can you try this attached version where the Package ID is appended to label.
Should print `0` or `1` depending of your selected Service Processor.

[CoreFreq_develop.tar.gz](https://github.com/cyring/CoreFreq/files/9298270/CoreFreq_develop.tar.gz)

Two UI views are impacted:
* Sensors
* Power

And two CLI outputs are impacted:
* `corefreq-cli -C`
* `corefreq-cli -w`

Rimits, 2022-08-10:
Thx, it's less confusing now.
[image]

closed 2022-08-10 by Rimits

cyring, 2022-08-10:
Meanwhile I have enhanced the `Custom` view
[CoreFreq_develop.tar.gz](https://github.com/cyring/CoreFreq/files/9298553/CoreFreq_develop.tar.gz)

Rimits, 2022-08-10:
well, you do have a fast hand🚴
[image]

cyring, 2022-08-10:
> well, you do have a fast handbicyclist ![截屏_20220810-171014]

Thank you. 

I'll leave those changes in the `develop` branch for a short while; making sure no regression happens on other systems.