cyring/CoreFreq #361: [Intel][IvyBridge] IMC third timings are wrongly decoded

cyring, 2022-09-12:
Memory controller third timings are wrongly decoded on `Core i7-3770K`

[image]

Trying several BIOS values reveals settings are changed in MCHBAR register `0x4008`. Unfortunately specs about this register are reserved.

[image]

Your bit specifications are welcomed -;)

cyring, 2022-09-17:
Results from commit 597c0bbc9c87c9c18cef90771f0f127389d4e548

[image]

cyring, 2022-09-18:
@svmlegacy Hello,

Could you test the `develop` branch with your SandyBridge, IvyBridge and check if third timings are same as BIOS settings ?

svmlegacy, 2022-09-18:
> @svmlegacy Hello,
> 
> Could you test the `develop` branch with your SandyBridge, IvyBridge and check if third timings are same as BIOS settings ?
[2600.txt](https://github.com/cyring/CoreFreq/files/9595035/2600.txt)
[image]
[3770K.txt](https://github.com/cyring/CoreFreq/files/9595036/3770K.txt)
[image]

All of the BIOS settings on this motheboard match, except for the memory speed. Board is running at 1067, but 1333 is reported (coincidentally that is the maximum module speed, non-XMP.)

I'll be trying the 2600 with a second motherboard which I believe has better reporting of third timings in BIOS, but unfortunately it is a Sandy Bridge only board, B2 stepping P67.

svmlegacy, 2022-09-18:
Should have added this picture too. Valid for both processors tested.
[image]

cyring, 2022-09-18:
@svmlegacy Thank you

It's odd, we're not getting the DIMM geometry.
I have tested it also with 3770K same Chipset PCI ID as you can see  above.

I have to debug it but I might slow to answer the next 3 weeks

cyring, 2022-09-19:
@svmlegacy 

I think the missing DIMM topology is due to a regression I have introduced in this commit 977f36efd70276c7f34e663bc0a4975f1510bc4e

This was pleasing my IVB 3770K setup but not all the others tested so far.

If you could test from the previous commit 2994d49e40c86b39017f6c4205612de9d6449a05
your 2600 and 3770K should show the topology again ?

cyring, 2022-09-24:
@svmlegacy Hello,
Can you test with `develop` branch and show the IMC output?
Thank you

svmlegacy, 2022-09-25:
> @svmlegacy Hello, Can you test with `develop` branch and show the IMC output? Thank you
> 
> Remark : there's a geometry regression fix in latest commit.

Hello,

Changes are good on my motherboard for Ivy:
[Develop-3770K-2022-09-25.txt](https://github.com/cyring/CoreFreq/files/9641751/Develop-3770K-2022-09-25.txt)

Still isn't identifying the current clock rate (1067 MHz), but is pulling the maximum module speed in XMP now (2133 MHz).

I'll pull my Haswell board out once we're happy with Ivy & Sandy... Currently in the middle of reorganizing...

cyring, 2022-09-26:
> Still isn't identifying the current clock rate (1067 MHz), but is pulling the maximum module speed in XMP now (2133 MHz).

Is this what you're expecting ?
```
                           Panther Point  [ 150]                           
Controller #0                                                Dual Channel  
 Bus Rate  2133 MT/s       Bus Speed 2128 MT/s           DDR3 Speed 1067 MHz
                                                                           
 Cha    CL  RCD   RP  RAS  RRD  RFC   WR RTPr WTPr  FAW  B2B  CWL CMD  REFI
  #0     7    7    7   20    4  139    8    4    4   16    0    6  1T  4160
  #1     7    7    7   20    4  139    8    4    4   16    0    6  1T  4160
      ddWR drWR srWR ddRW drRW srRW ddRR drRR srRR ddWW drWW srWW CKE   ECC
  #0     1    2    3    3    3    3    2    1    4    3    1    0   3    0 
  #1     1    2    3    3    3    3    2    1    4    3    1    0   3    0 
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0     8    2     32768      1024           8192  CM3X8GA2133C9D8   
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0     8    2     32768      1024           8192  CM3X8GA2133C9D8   
```
To debug, can you add this trace:
https://github.com/cyring/CoreFreq/blob/757fc90d04320a9e63da15c84a5a1783914ade5c/corefreqd.c#L3494
```C
	printf( "DMFC[%x]\n", RO(Proc)->Uncore.Bus.SNB_Cap.DMFC );
	switch (RO(Proc)->Uncore.Bus.SNB_Cap.DMFC) {

cyring, 2022-09-28:
@svmlegacy To make things easier, I'm providing this attached version with the debug traces I need
[CoreFreq_develop.tar.gz](https://github.com/cyring/CoreFreq/files/9665203/CoreFreq_develop.tar.gz)

This version is also making use of a [`DIMM A SELECT`](https://github.com/cyring/CoreFreq/blob/757fc90d04320a9e63da15c84a5a1783914ade5c/intelmsr.h#L3037) bit to map correctly the DIMM slots.

Please build, run and post the output of traces and IMC.

Fyi, here's my IvyBrige results where DIMM A is the _farthest slot_ from processor socket

```
corefreqd -q
[IVB] DMFC[5]
[SNB] FSB_Select[4]
```
```
corefreq-cli -M
                           Panther Point  [ 150]                           
Controller #0                                                Dual Channel  
 Bus Rate  1867 MHz       Bus Speed 1867 MHz           DDR3 Speed 1867 MT/s
                                                                           
 Cha    CL  RCD   RP  RAS  RRD  RFC   WR RTPr WTPr  FAW  B2B  CWL CMD  REFI
  #0     9   10    9   27    8  150   16    8    8   36    0    8  2T  7280
  #1     9   10    9   27    8  150   16    8    8   36    0    8  2T  7280
      ddWR drWR srWR ddRW drRW srRW ddRR drRR srRR ddWW drWW srWW CKE   ECC
  #0     1    1    3    3    3    3    3    1    4    3    1    4   5    0 
  #1     1    1    3    3    3    3    3    1    4    3    1    4   5    0 
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1     8    1     32768      1024           4096  CMZ8GX3M2A1866C9  
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1     8    1     32768      1024           4096  CMZ8GX3M2A1866C9  
```

cyring, 2022-10-01:
@svmlegacy Hello,
Any chance to get results from above request ?
Regards

svmlegacy, 2022-10-02:
I was able to get this today. Looks like good results!
> Is this what you're expecting ?

I'm not sure exactly what the bus rate and bus speed should be, but certainly the DDR3 speed should be 1067 MT/s in my case. While my RAM is capable of 2133, it isn't running at that speed in this case.

```
CoreFreq Daemon 1.92.1  Copyright (C) 2015-2022 CYRIL INGENIERIE
[IVB] DMFC[5]
[SNB] FSB_Select[3]
```

```
corefreq-cli -M
                           Panther Point  [ 150]                           
Controller #0                                                Dual Channel  
 Bus Rate  2133 MHz       Bus Speed 2128 MHz           DDR3 Speed 2133 MT/s
                                                                           
 Cha    CL  RCD   RP  RAS  RRD  RFC   WR RTPr WTPr  FAW  B2B  CWL CMD  REFI
  #0     7    7    7   20    4  139    8    4    4   16    0    6  1T  4160
  #1     7    7    7   20    4  139    8    4    4   16    0    6  1T  4160
      ddWR drWR srWR ddRW drRW srRW ddRR drRR srRR ddWW drWW srWW CKE   ECC
  #0     1    2    3    3    3    3    2    1    4    3    1    0   3    0 
  #1     1    2    3    3    3    3    2    1    4    3    1    0   3    0 
                                                                           
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0     8    2     32768      1024           8192  CM3X8GA2133C9D8   
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0     8    2     32768      1024           8192  CM3X8GA2133C9D8   
```

cyring, 2022-10-02:
> I'm not sure exactly what the bus rate and bus speed should be, but certainly the DDR3 speed should be 1067 MT/s in my case. While my RAM is capable of 2133, it isn't running at that speed at this time.

So let's try [RAM_Select](https://github.com/cyring/CoreFreq/blob/757fc90d04320a9e63da15c84a5a1783914ade5c/intelmsr.h#L2351) first, FSB_Select second.

[image]

Here's the version to test the **current** DDR speed:

[CoreFreq_develop.tar.gz](https://github.com/cyring/CoreFreq/files/9691910/CoreFreq_develop.tar.gz)

cyring, 2022-10-02:
@svmlegacy FYI, in the latest archive, above, there is no trace. This version just focuses on providing you the right DRAM frequency. So only the IMC output will be enough.
Thank you

cyring, 2022-10-03:
@svmlegacy latest code is now available in the development branch. Hope there is no regression on your side and it fixes your current DRAM frequency ?

svmlegacy, 2022-10-03:
Pulled develop up to date as of this afternoon. Still seeing "maximum" DIMM frequency.
[image]

cyring, 2022-10-04:
> Pulled develop up to date as of this afternoon. Still seeing "maximum" DIMM frequency. 

Thank you. 
I'm not finding a specific register for your frequency. Do you have other software which are able to retrieve 1067 ?

cyring, 2022-10-06:
@svmlegacy Hello

Can you pull and test latest `develop` branch: Register [BIOS DDR Mem Clock](https://github.com/cyring/CoreFreq/blob/287b81caeb27fc31a8c4ca3bc034f8c2751ce35b/intelmsr.h#L2815) (`0x5e00`) which is specified in Haswell is apparently also working on Ivy Bridge:

[image]

Remark: I have no SandyBridge, Haswell and Broadwell to test with, in case if you have some ?

svmlegacy, 2022-10-06:
Good news! I am now seeing the current DDR3 speed with the 3770K

$ ./corefreq-cli -M

                           Panther Point  [ 150]                                                                                                                                    
Controller #0                                                Dual Channel                                                                                                           
 Bus Rate  1064 MHz       Bus Speed 1061 MHz           DDR3 Speed 1064 MT/s                                                                                                         
                                                                                                                                                                                    
 Cha    CL  RCD   RP  RAS  RRD  RFC   WR RTPr WTPr  FAW  B2B  CWL CMD  REFI                                                                                                         
  #0     7    7    7   20    4  139    8    4    4   16    0    6  1T  4160                                                                                                         
  #1     7    7    7   20    4  139    8    4    4   16    0    6  1T  4160                                                                                                         
      ddWR drWR srWR ddRW drRW srRW ddRR drRR srRR ddWW drWW srWW CKE   ECC                                                                                                         
  #0     1    2    3    3    3    3    2    1    4    3    1    0   3    0                                                                                                          
  #1     1    2    3    3    3    3    2    1    4    3    1    0   3    0                                                                                                          
                                                                                                                                                                                    
 DIMM Geometry for channel #0                                                                                                                                                       
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                                                                                                                         
       #0     8    2     32768      1024           8192  CM3X8GA2133C9D8                                                                                                            
 DIMM Geometry for channel #1                                                                                                                                                       
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                                                                                                                         
       #0     8    2     32768      1024           8192  CM3X8GA2133C9D8       

Will test Sandy & Haswell soon.

svmlegacy, 2022-10-06:
Also good on Sandy (Intel Core i7-2600):

```
$ ./corefreq-cli -M
                         P67/Cougar Point  [ 100]                                                                                                                                  
Controller #0                                                Dual Channel                                                                                                           
 Bus Rate  1064 MHz       Bus Speed 1061 MHz           DDR3 Speed 1064 MT/s                                                                                                         
                                                                                                                                                                                    
 Cha    CL  RCD   RP  RAS  RRD  RFC   WR RTPr WTPr  FAW  B2B  CWL CMD  REFI                                                                                                         
  #0     7    7    7   20    4  139    8    4    4   16    0    6  1T  4160                                                                                                         
  #1     7    7    7   20    4  139    8    4    4   16    0    6  1T  4160                                                                                                         
      ddWR drWR srWR ddRW drRW srRW ddRR drRR srRR ddWW drWW srWW CKE   ECC                                                                                                         
  #0     1    1    3    3    3    3    2    1    4    3    1    4   3    0                                                                                                          
  #1     1    1    3    3    3    3    2    1    4    3    1    4   3    0                                                                                                          
                                                                                                                                                                                    
 DIMM Geometry for channel #0                                                                                                                                                       
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                                                                                                                         
       #0     8    2     32768      1024           8192  CM3X8GA2133C9D8                                                                                                            
 DIMM Geometry for channel #1                                                                                                                                                       
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                                                                                                                         
       #0     8    2     32768      1024           8192  CM3X8GA2133C9D8                                                                                                            
```

cyring, 2022-10-06:
> Also good on Sandy (Intel Core i7-2600):

Thanks for the answer.

Meanwhile, I have found the DRAM PLL bit which means frequency is now computed based on BIOS FSB, DRAM ratio: `100:133` or `100:100`

Precision is also improved and unit changed to `MHz`.

You should read `DDR3 Speed 1066 MHz`

svmlegacy, 2022-10-08:
```
                          P67/Cougar Point  [ 100]                                                                                                                                  
Controller #0                                                Dual Channel                                                                                                           
 Bus Rate  1066 MHz       Bus Speed 1063 MHz           DDR3 Speed 1066 MHz                                                                                                          
                                                                                                                                                                                    
 Cha    CL  RCD   RP  RAS  RRD  RFC   WR RTPr WTPr  FAW  B2B  CWL CMD  REFI                                                                                                         
  #0     7    7    7   20    4  139    8    4    4   16    0    6  1T  4160                                                                                                         
  #1     7    7    7   20    4  139    8    4    4   16    0    6  1T  4160                                                                                                         
      ddWR drWR srWR ddRW drRW srRW ddRR drRR srRR ddWW drWW srWW CKE   ECC                                                                                                         
  #0     1    1    3    3    3    3    2    1    4    3    1    4   3    0                                                                                                          
  #1     1    1    3    3    3    3    2    1    4    3    1    4   3    0                                                                                                          
                                                                                                                                                                                    
 DIMM Geometry for channel #0                                                                                                                                                       
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                                                                                                                         
       #0     8    2     32768      1024           8192  CM3X8GA2133C9D8                                                                                                            
 DIMM Geometry for channel #1                                                                                                                                                       
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                                                                                                                         
       #0     8    2     32768      1024           8192  CM3X8GA2133C9D8                                                                                                            
```

Changes look good and are valid for my motherboard!

cyring, 2022-10-08:
> Changes look good and are valid for my motherboard!

Thank you very much helping to improve SNB & IVB.
With the new Registers better mastered, I'm now providing same enhancements for HSW & BDW in issue #346

closed 2022-10-08 by cyring

cyring, 2022-10-15:
@svmlegacy I have back-ported IMC code from HSW

Using `develop` branch, you should now read the same geometry on SNB as HSW