cyring/CoreFreq #365: Kernel module compilation error on 6.1.0-rc1

Slaviusz, 2022-10-22:
Since commit 94e371e73fbe44ed338b51739ed2d3fa7b233bf4 I'm receiving an error trying to compile the kernel module.

```
make -j1 -C /lib/modules/6.1.0-rc1-tkg-cfs/build M=/root/CoreFreq modules
make[1]: Entering directory '/usr/src/linux-6.1.0-rc1-tkg-cfs'
  CC [M]  /root/CoreFreq/corefreqk.o
In file included from ./include/vdso/const.h:5,
                 from ./include/linux/const.h:4,
                 from ./include/linux/list.h:9,
                 from ./include/linux/module.h:12,
                 from /root/CoreFreq/corefreqk.c:12:
./include/vdso/bits.h:7:37: error: expected identifier or ‘(’ before numeric constant
    7 | #define BIT(nr)                 (UL(1) << (nr))
      |                                     ^
./include/uapi/linux/const.h:20:26: note: in definition of macro ‘__AC’
   20 | #define __AC(X,Y)       (X##Y)
      |                          ^
./include/uapi/linux/const.h:25:26: note: in expansion of macro ‘_AC’
   25 | #define _UL(x)          (_AC(x, UL))
      |                          ^~~
./include/vdso/const.h:7:26: note: in expansion of macro ‘_UL’
    7 | #define UL(x)           (_UL(x))
      |                          ^~~
./include/vdso/bits.h:7:34: note: in expansion of macro ‘UL’
    7 | #define BIT(nr)                 (UL(1) << (nr))
      |                                  ^~
./arch/x86/include/asm/msr-index.h:1069:41: note: in expansion of macro ‘BIT’
 1069 | #define LEGACY_XAPIC_DISABLED           BIT(0) /*
      |                                         ^~~
/root/CoreFreq/intel_reg.h:510:17: note: in expansion of macro ‘LEGACY_XAPIC_DISABLED’
  510 |                 LEGACY_XAPIC_DISABLED   :  1-0,
      |                 ^~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
make[2]: *** [scripts/Makefile.build:250: /root/CoreFreq/corefreqk.o] Error 1
make[1]: *** [Makefile:1992: /root/CoreFreq] Error 2
make[1]: Leaving directory '/usr/src/linux-6.1.0-rc1-tkg-cfs'
make: *** [Makefile:86: all] Error 2
```

cyring, 2022-10-22:
Thanks, I see the issue is a constant name collision with Kernel header file.

I will boot that Kernel RC to program a fix.

cyring, 2022-10-22:
_CoreFreq_ version `1.92.3` is fixing the issue which I found to also exists on Linux Kernel `6.0.3`

[image]

Slaviusz, 2022-10-22:
Thanks, everything works ok now.

closed 2022-10-22 by Slaviusz