cyring/CoreFreq #366: [Intel/Atom/Airmont] Missing IMC of Pentium(R) CPU N3700

cyring, 2022-10-29:
@svmlegacy Hello,

Refering to the Wiki [page](https://gist.github.com/svmlegacy/f82007b57e26724e0203d7286a9fb125#file-intel-pentium-n3700-L287), `corefreq-cli -m` does not output the IMC data.

Here is an attempt do decode `Airmont` with `SLM/Baytrail` SoC code.

[CoreFreq_develop.tar.gz](https://github.com/cyring/CoreFreq/files/9893308/CoreFreq_develop.tar.gz)

Please feel free to test and post the IMC output of this archive.

svmlegacy, 2022-10-29:
Unfortunately this laptop is a family member's and is not currently available for testing... I'll dig through some of my untested laptops and see if any are Airmont/Braswell like this one.

The laptop which created the dump is an Asus X540SA, with the memory directly attached to the motherboard.

cyring, 2022-10-29:
> Unfortunately this laptop is a family member's and is not currently available for testing... I'll dig through some of my untested laptops and see if any are Airmont/Braswell like this one.
> 
> The laptop which created the dump is an Asus X540SA, with the memory directly attached to the motherboard.

Will it help if I'm proving a bootable ISO including the Airmont branch ?

svmlegacy, 2022-10-29:
I was able to recreate the issue with a Airmont/Cherry Trail CPU, and the new branch seems to be outputting something. Please check for coherency. BIOS does not provide current speeds or timings.

Each branch is in it's own file [here](https://gist.github.com/svmlegacy/deb7288b6dbef976a0002a0ed29ae783)

cyring, 2022-10-29:
> I was able to recreate the issue with a Airmont/Cherry Trail CPU, and the new branch seems to be outputting something. Please check for coherency. BIOS does not provide current speeds or timings.
> 
> [Master Branch](https://gist.github.com/svmlegacy/deb7288b6dbef976a0002a0ed29ae783#file-cherry-trial-master-L250) [Develop Branch](https://gist.github.com/svmlegacy/deb7288b6dbef976a0002a0ed29ae783#file-cherry-trial-develop-L251)

Thank you for those results.

Good you have this `Cherry Trail` platform, according to [ark](https://ark.intel.com/content/www/us/en/ark/products/codename/46629/products-formerly-cherry-trail.html) there are things to solve:
* Burst frequency (ratio)
* Virtualization Technology (VT)

and to fix:
* DIMM size and geometry
  (Kernel reports about 2GB)

cyring, 2022-10-29:
@svmlegacy Hello,

Can you please test those registers ?

```sh
## MSR_IA32_PLATFORM_ID
rdmsr -ax 0x00000017

## MSR_THERM2_CTL
rdmsr -ax 0x0000019d

## MSR_MISC_PWR_MGMT
rdmsr -ax 0x000001aa

## MSR_IA32_THERM_CONTROL
rdmsr -ax 0x0000019a

## MSR_TURBO_RATIO_LIMIT
rdmsr -ax 0x000001ad
```

svmlegacy, 2022-10-31:
I'm not surprised if the core is hotter, the box is passively cooled and warm to touch, while the Braswell chip was actively cooled.

Burst appears to be working, as I do catch the CPU above 1.44 GHz in /proc/cpuinfo

```
## MSR_IA32_PLATFORM_ID
rdmsr -ax 0x00000017

f090041752
f090041752
f090041449
f090041449

## MSR_THERM2_CTL
rdmsr -ax 0x0000019d

62d
62d
62d
62d

## MSR_MISC_PWR_MGMT
rdmsr -ax 0x000001aa

rdmsr: CPU 0 cannot read MSR 0x000001aa

## MSR_IA32_THERM_CONTROL
rdmsr -ax 0x0000019a

0
0
0
0

## MSR_TURBO_RATIO_LIMIT
rdmsr -ax 0x000001ad

0
0
0
0
```

cyring, 2022-11-01:
> Burst appears to be working, as I do catch the CPU above 1.44 GHz in /proc/cpuinfo

Thank you, branch is committed 69e370a51f39c168cb390259da119ccccfbd8ad0 to allow those MSR.
Perhaps Burst frequency ratio will show up ?

cyring, 2022-11-01:
```
|- Clock Modulation                                             ODCM   <Disable>
   |- DutyCycle                                                        [  0.00%]
```
Clock Modulation is now also permitted. You should be able to apply a `DutyCycle` percent with no processor crash. 
Please let me know if it is working safely ?
It is supposed to lower frequency based on the selected percent.

cyring, 2022-11-01:
# Platform codename

Airmont is the common CPUID `06_4C`

Stepping `3` is assigned to both `Atom x5-Z8300` and `Pentium N3700` 

Fortunately we can manage a brands table in _CoreFreq_ driver to split in two platform codenames: `Cherry Trail` and `Braswell`

Code enhancement should also concern the PCI Host bridge id `8086:2280` which differs _only_ in revision number `(rev 22)` vs `(rev 21)` according to `lspci`.

So far driver has no code to query the revision and I wonder if it's worth the effort to provide the true platform codename in the IMC output, rather than a common codename ?

svmlegacy, 2022-11-05:
Just a quick update on this - I'm currently experimenting to get this system stable in Linux. I've been suffering random crashes, especially when the iGPU is loaded. Once it's stable I'll be able to test the duty cycle modulation. I haven't seen it lock up in a tty console, but I can't be certain it is stable. I may have to try alternative distro's to get it, or older kernels.

On platform codename - To me Cherry Trail and Braswell are effectively the same silicon, just with different target markets. I don't think it's worth the effort in splitting the codename without fully understanding if the Host Bridge revisions within each codename are unique. There seems to be some cherry trail chips with rev 20, for example)

cyring, 2022-11-06:
Found these frequency and turbo ratios, and voltage VID registers in kernel. 
Apparently undocumented in SDM.

#define [MSR_ATOM_CORE_RATIOS](https://elixir.bootlin.com/linux/latest/C/ident/MSR_ATOM_CORE_RATIOS)		0x0000066a
#define [MSR_ATOM_CORE_VIDS](https://elixir.bootlin.com/linux/latest/C/ident/MSR_ATOM_CORE_VIDS)		0x0000066b
#define [MSR_ATOM_CORE_TURBO_RATIOS](https://elixir.bootlin.com/linux/latest/C/ident/MSR_ATOM_CORE_TURBO_RATIOS)	0x0000066c
#define [MSR_ATOM_CORE_TURBO_VIDS](https://elixir.bootlin.com/linux/latest/C/ident/MSR_ATOM_CORE_TURBO_VIDS)	0x0000066d

Can you please `rdmsr` these four MSR ?

svmlegacy, 2022-11-06:
I seem to be having a stretch of stability after setting a max c-state of 1, similar to Bay Trail. See this: https://bugzilla.kernel.org/show_bug.cgi?id=109051

```
#define [MSR_ATOM_CORE_RATIOS](https://elixir.bootlin.com/linux/latest/C/ident/MSR_ATOM_CORE_RATIOS) 0x0000066a

# rdmsr -ax 0x0000066a
120602
120602
120602
120602

#define [MSR_ATOM_CORE_VIDS](https://elixir.bootlin.com/linux/latest/C/ident/MSR_ATOM_CORE_VIDS) 0x0000066b

# rdmsr -ax 0x0000066b
442d2d
442d2d
442d2d
442d2d

#define [MSR_ATOM_CORE_TURBO_RATIOS](https://elixir.bootlin.com/linux/latest/C/ident/MSR_ATOM_CORE_TURBO_RATIOS) 0x0000066c

# rdmsr -ax 0x0000066c
14141717
14141717
14141717
14141717

#define [MSR_ATOM_CORE_TURBO_VIDS](https://elixir.bootlin.com/linux/latest/C/ident/MSR_ATOM_CORE_TURBO_VIDS) 0x0000066d

# rdmsr -ax 0x0000066d
49495252
49495252
49495252
49495252

```

cyring, 2022-11-06:
> I seem to be having a stretch of stability after setting a max c-state of 1, similar to Bay Trail. See this: https://bugzilla.kernel.org/show_bug.cgi?id=109051

Well done!
In the past I investigated a `Silvermont/Bay_Trail` crash, issue happened late after resuming from `S3`.

cyring, 2022-11-06:
# MSR_ATOM_CORE_RATIOS
```
# rdmsr -ax 0x0000066a
120602
```
## Frequency
* Min = 6 `--------->` **480 MHz** (6 x 80 MHz)
* Max = 18 (0x12) `->` **1440 MHz**
----
# MSR_ATOM_CORE_VIDS
```
# rdmsr -ax 0x0000066b
442d2d
```
## Base Vcore
* VID = 45 (0x2d) `->` Vcore = **`0.4700V`**
----
# MSR_ATOM_CORE_TURBO_VIDS
```
# rdmsr -ax 0x0000066d
49495252
```
## Turbo Vcore
* VID = 82 (0x52) `->` Vcore = **`0.6550V`**
----
# MSR_ATOM_CORE_TURBO_RATIOS
```
# rdmsr -ax 0x0000066c
14141717
```
## Turbo Boost
* 1C = 23 (0x17) `->` **1840 MHz** (23 x 80 MHz)
* 2C = 23 (0x17) `->` 1840 MHz
* 3C = 20 (0x14) `->` **1600 MHz**
* 4C = 20 (0x14) `->` 1600 MHz
----

What we are getting new are the Turbo Boost frequency ratios.

cyring, 2022-11-06:
Are those four MSR_ATOM writable ?
Can you `wrmsr` them with same value as read ?

cyring, 2022-11-06:
Everything is now part of last commit 8521699cc8f3d5be002abecefd40f705bb2c600a
You should discover the architecture code name and the Turbo Boost ratios.
Feel free to refresh your CLI outputs.

svmlegacy, 2022-11-06:
With my system all four MCR's are not writable.

Refreshed output: https://gist.github.com/svmlegacy/deb7288b6dbef976a0002a0ed29ae783#file-cherry-trail-develop-2022-11-06

cyring, 2022-11-06:
> With my system all four MCR's are not writable.
> 
> Refreshed output: https://gist.github.com/svmlegacy/deb7288b6dbef976a0002a0ed29ae783#file-cherry-trail-develop-2022-11-06

Thank you. Many things I've to process looking at your output.

May be to arrange temperature, you could go down to `C1E` and makes kernel idling with the `HLT` instruction.

See Readme to register _CoreFreq_ as `CPU-Idle` driver. Next select HALT as the route in `[Settings]` , and `C1E` as the limit in `[Kernel]`

cyring, 2022-11-07:
> With my system all four MCR's are not writable.

OK, Commit 5c5b0a7c41dafcd6323b500b9b32044247d891dc to make them as read-only ratios

cyring, 2022-11-07:
I don't get a `C-States Base Address` within the output, can you read this MSR ?

```
rdmsr -ax 0xE4
```

cyring, 2022-11-07:
Keep this
[Intel® Atom™ Z8000 Processor Series Datasheet (Volume 2 of 2)](https://www.intel.la/content/dam/www/public/us/en/documents/datasheets/atom-z8000-datasheet-vol-2.pdf)

cyring, 2022-11-07:
Can you please dump some registers ?
About this source code [line](https://github.com/cyring/CoreFreq/blob/c48d6ef53c63df464a3cce9512ee0e0c8c4d169c/corefreqd.c#L3199) in function `SLM_PTR`, add debugging code as below:

```c
/* Error Correcting Code */
	TIMING(mc, cha).ECC = \
			  RO(Proc)->Uncore.MC[mc].SLM.BIOS_CFG.EFF_ECC_EN
			| RO(Proc)->Uncore.MC[mc].SLM.BIOS_CFG.ECC_EN;
/* Debugging Code */
	printf( "IMC(%d:%d)\tDRP[%x]\tBIOS[%x]\n", mc, cha,
		RO(Proc)->Uncore.MC[mc].SLM.DRP.value,
		RO(Proc)->Uncore.MC[mc].SLM.BIOS_CFG.value );
		
```
and post the traces which should output from daemon `corefreqd`

cyring, 2022-11-10:
> Keep this [Intel® Atom™ Z8000 Processor Series Datasheet (Volume 2 of 2)](https://www.intel.la/content/dam/www/public/us/en/documents/datasheets/atom-z8000-datasheet-vol-2.pdf)

@svmlegacy What you will notice in specs is that IMC registers differ from those of Silvermont/BYT; some bits are shifted of just a few position.
Now I don't confirm previous Timings without a reference coming from your BIOS, memtest, or SPD. 
The Z8000's IMC decoder has to be programmed.

cyring, 2022-11-23:
@svmlegacy Hello,
I would like to merge current developments into master; but it would be nice to finalize Airmont IMC before.
Is your platform still available ?

svmlegacy, 2022-11-24:
@cyring 

Thanks for your patience... Life was just being life for a bit. Should be able to catch back up this weekend!

svmlegacy, 2022-11-29:
> @svmlegacy I don't get a `C-States Base Address` within the output, can you read this MSR ?
> 
> ```
> rdmsr -ax 0xE4
> ```
> 
> _EDIT:_ It may also be a BIOS option.

Result:
```
# rdmsr -ax 0xE4
20000
20000
20000
20000
```

> @svmlegacy Can you please dump some registers ? About this source code [line](https://github.com/cyring/CoreFreq/blob/c48d6ef53c63df464a3cce9512ee0e0c8c4d169c/corefreqd.c#L3199) in function `SLM_PTR`, add debugging code as below:
> 
> ```c
> /* Error Correcting Code */
> 	TIMING(mc, cha).ECC = \
> 			  RO(Proc)->Uncore.MC[mc].SLM.BIOS_CFG.EFF_ECC_EN
> 			| RO(Proc)->Uncore.MC[mc].SLM.BIOS_CFG.ECC_EN;
> /* Debugging Code */
> 	printf( "IMC(%d:%d)\tDRP[%x]\tDTR0[%x]\tBIOS[%x]\n", mc, cha,
> 		RO(Proc)->Uncore.MC[mc].SLM.DRP.value,
> 		RO(Proc)->Uncore.MC[mc].SLM.DTR0.value,
> 		RO(Proc)->Uncore.MC[mc].SLM.BIOS_CFG.value );
> 		
> ```
> 
> and post the traces which should output from daemon `corefreqd`

Result:
```
# ./corefreqd
CoreFreq Daemon 1.93.0  Copyright (C) 2015-2022 CYRIL INGENIERIE
IMC(0:0)	DRP[58091]	DTR0[1344c630]	BIOS[468f0010]
```

> @svmlegacy What you will notice in specs is that IMC registers differ from those of Silvermont/BYT; some bits are shifted of just a few position. Now I don't confirm previous Timings without a [...]

Unfortunately the BIOS is very limited and does not provide the information. The memory is soldered directly onto the mainboard, as well. I'll see about getting Memtest running on this machine and posting the results.

cyring, 2022-12-02:
@svmlegacy Fyi commit ff64902ede51b63996391edfe9b85060fb6f2380 adds specs of Z8000 IMC registers