cyring/CoreFreq #373: [Ubuntu 16.04] fail to build kernel module

wjc404, 2022-11-12:
I encountered a compilation error when trying to build the kernel module with Ubuntu 16.04. However the build was successful on another computer with Ubuntu 20.04.
[image]
It seems that the change of acpi cppc interface should account for the difference.
[image]

cyring, 2022-11-12:
> I encountered a compilation error when trying to build the kernel module with Ubuntu 16.04. 

Please pull and try the [hotfix_cppc_acpi](https://github.com/cyring/CoreFreq/tree/hotfix_cppc_acpi) branch

wjc404, 2022-11-12:
Thanks:)   What about the 4th parameter "unsigned short CPB" of function CPPC_AMD_Zen_ScaleRatio, does it have a default value ?

cyring, 2022-11-12:
> Thanks:) What about the 4th parameter "unsigned short CPB" of function CPPC_AMD_Zen_ScaleRatio, does it have a default value ?

Please pull the branch for this fix. Thank you.

wjc404, 2022-11-12:
Now the compilation passes. But the terminal hangs during module insertion. The CPU architecture of my computer is skylake-X. Thank again for looking at this issue. I will roll back to older version.

cyring, 2022-11-12:
> Now the compilation passes. But the terminal hangs during module insertion. The CPU architecture of my computer is skylake-X. Thank again for looking at this issue. I will roll back to older version.

Don't you have any kernel log to show me ?
Since which version of _CoreFreq_ your SKL/X is crashing ?

wjc404, 2022-11-12:
Well, I found nothing related in /var/log/syslog (actually the logging is stopped until I press the restart button). 

The version v1.90 RC1(commit a154015) works fine without crash.

closed 2022-11-15 by cyring