cyring/CoreFreq #374: [Skylake-X] Terminal hangs during module insertion

cyring, 2022-11-12:
> Now the compilation passes. But the terminal hangs during module insertion. The CPU architecture of my computer is skylake-X. Thank again for looking at this issue. I will roll back to older version.

_Originally posted by @wjc404 in https://github.com/cyring/CoreFreq/issues/373#issuecomment-1312450134_
      
> Well, I found nothing related in /var/log/syslog (actually the logging was stopped until I press the restart button).

> The version v1.90 RC1(commit https://github.com/cyring/CoreFreq/commit/a1540153123db1b2614dcc2d8cddede1be3a42cb) works fine without crash.

cyring, 2022-11-12:
@wjc404 Hello

Can you try to read the following MSR on your Skylake/X ?

```sh
## MSR_SKL_CORE_PERF_LIMIT_REASONS
rdmsr -ax 0x0000064f

## MSR_GRAPHICS_PERF_LIMIT_REASONS
rdmsr -ax 0x000006b0

## MSR_RING_PERF_LIMIT_REASONS
rdmsr -ax 0x000006b1
```

Thank you

wjc404, 2022-11-12:
| request | result |
| --- | --- |
| 0x0000064f | 0xe1848080 |
| 0x000006b0 | cannot read |
| 0x000006b1 | 0x0 |

cyring, 2022-11-12:
@wjc404 Thank you for the MSR tests. 

* Can you first pull and test branch [`hotfix_Skylake_X`](https://github.com/cyring/CoreFreq/tree/hotfix_Skylake_X) ?

* If branch does not crash, can you then allow `MSR_RING_PERF_LIMIT_REASONS` by replacing comments in the following lines:

https://github.com/cyring/CoreFreq/blob/c761e4657bd29d37fcb73ef07eb20c6a458b05d2/corefreqk.c#L11818
```c
/*TODO(Unsolved)
	GraphicsPerfLimitReasons(Core);
*/
	RingPerfLimitReasons(Core);
```

https://github.com/cyring/CoreFreq/blob/c761e4657bd29d37fcb73ef07eb20c6a458b05d2/corefreqk.c#L16726
```c
/*TODO(Unsolved)
		Monitor_GraphicsPerfLimitReasons(PUBLIC(RO(Proc)));
*/
		Monitor_RingPerfLimitReasons(PUBLIC(RO(Proc)));
```

wjc404, 2022-11-13:
> * Can you first pull and test branch [`hotfix_Skylake_X`](https://github.com/cyring/CoreFreq/tree/hotfix_Skylake_X) ?

Succeeded, nothing wrong with corefreq-cli.

> * If branch does not crash, can you then allow `MSR_RING_PERF_LIMIT_REASONS` by replacing comments in the following lines: [...]
Also succeeded.

cyring, 2022-11-13:
@wjc404 Thank you for your contribution.

Do you mind to run the current development branch [`develop`](https://github.com/cyring/CoreFreq/tree/develop) for non regression tests ?

Any CLI outputs and screenshots are greatly appreciated. Such as [CPU-support](https://github.com/cyring/CoreFreq/wiki/CPU-support)

wjc404, 2022-11-13:
Well, maybe there's no need to call `Monitor_GraphicsPerfLimitReasons` and `GraphicsPerfLimitReasons` for skylake_X processors, since they don't have built-in graphics.

cyring, 2022-11-13:
> Well, maybe there's no need to call `Monitor_GraphicsPerfLimitReasons` and `GraphicsPerfLimitReasons` for skylake-X processors, since they don't have built-in graphics.

Exactly. 
And I'm looking for an iGPU discriminant to jump around or not `GraphicsPerfLimitReasons` in code.

CSR registers for capabilities may tell if a Graphic is present or not:

https://github.com/cyring/CoreFreq/blob/35b504f3a26c6e2a1d18fd1a6f197e3881a3027b/intel_reg.h#L4394

But this required to test various `Skylake-{X}` and successors for the validity of such bit.

> I will test after this fix is taken into the target branch.

Are we closing the issue ?