cyring/CoreFreq #378: Ryzen 9 7950X

cyring, 2022-11-28:
[image]

* Temperature is zero. 
* Vcore is too low. 
* Architecture is Zen4

cyring, 2022-11-29:
* Using `master 1.92.4`

[image]

cyring, 2022-11-30:
```
$ lspci -nn
00:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14d8]
00:00.2 IOMMU [0806]: Advanced Micro Devices, Inc. [AMD] Device [1022:14d9]
00:01.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14da]
00:01.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14db]
00:02.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14da]
00:02.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14db]
00:02.2 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14db]
00:03.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14da]
00:04.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14da]
00:08.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14da]
00:08.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14dd]
00:08.3 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14dd]
00:14.0 SMBus [0c05]: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller [1022:790b] (rev 71)
00:14.3 ISA bridge [0601]: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge [1022:790e] (rev 51)
00:18.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e0]
00:18.1 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e1]
00:18.2 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e2]
00:18.3 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e3]
00:18.4 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e4]
00:18.5 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e5]
00:18.6 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e6]
00:18.7 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e7]
01:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 XL Upstream Port of PCI Express Switch [1002:1478] (rev c1)
02:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 XL Downstream Port of PCI Express Switch [1002:1479]
03:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Navi 21 [Radeon RX 6800/6800 XT / 6900 XT] [1002:73bf] (rev c1)
03:00.1 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] Navi 21/23 HDMI/DP Audio Controller [1002:ab28]
03:00.2 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD/ATI] Device [1002:73a6]
03:00.3 Serial bus controller [0c80]: Advanced Micro Devices, Inc. [AMD/ATI] Navi 21 USB [1002:73a4]
04:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f4] (rev 01)
05:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
05:04.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
05:05.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
05:06.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
05:07.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
05:08.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
05:09.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
05:0a.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
05:0b.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
05:0c.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
05:0d.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
06:00.0 Non-Volatile memory controller [0108]: Sandisk Corp WD PC SN810 / Black SN850 NVMe SSD [15b7:5011] (rev 01)
0b:00.0 SATA controller [0106]: ASMedia Technology Inc. ASM1062 Serial ATA Controller [1b21:0612] (rev 02)
0d:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller [10ec:8125] (rev 05)
0e:00.0 Network controller [0280]: MEDIATEK Corp. MT7922 802.11ax PCI Express Wireless Network Adapter [14c3:0616]
0f:00.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f7] (rev 01)
10:00.0 SATA controller [0106]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f6] (rev 01)
11:00.0 Non-Volatile memory controller [0108]: Samsung Electronics Co Ltd NVMe SSD Controller SM961/PM961/SM963 [144d:a804]
12:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Device [1022:14de] (rev c1)
12:00.2 Encryption controller [1080]: Advanced Micro Devices, Inc. [AMD] VanGogh PSP/CCP [1022:1649]
12:00.3 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:15b6]
12:00.4 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:15b7]
12:00.6 Audio device [0403]: Advanced Micro Devices, Inc. [AMD] Family 17h/19h HD Audio Controller [1022:15e3]
13:00.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:15b8]
```

cyring, 2022-12-01:
```
$ corefreq-cli -s -n -m -n -V 1 -n -W 1 -n -c 1 -n -i 1 -n -B -n -k -n -M
```
```
Processor                                  [AMD Ryzen 9 7950X 16-Core Processor]
|- Architecture                                                   [Zen4/Raphael]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0a601203]
|- Signature                                                           [  AF_61]
|- Stepping                                                            [      2]
|- Online CPU                                                          [ 32/ 32]
|- Base Clock                                                          [ 99.998]
|- Frequency            (MHz)                      Ratio                        
                 Min   2999.95                    <  30 >                       
                 Max   4499.93                    <  45 >                       
|- Factory                                                             [100.000]
                       4500                       [  45 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   2999.95                    <  30 >                       
|- Turbo Boost                                                         [ UNLOCK]
                 XFR   5799.91                    [  58 ]                       
                 CPB   5699.91                    [  57 ]                       
                  1C   2999.95                    <  30 >                       
|- Uncore                                                              [   LOCK]
                 CLK    899.99                    [   9 ]                       
                 MEM   2799.96                    [  28 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [Y]    AVX512-DQ [Y]  AVX512-IFMA [Y]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [Y]    AVX512-BW [Y]   AVX512-VL [Y] 
|- AVX512-VBMI  [Y] AVX512-VBMI2 [Y]  AVX512-VNNI [Y]  AVX512-ALG [Y] 
|- AVX512-VPOP  [Y] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [Y] AVX-VNNI-VEX [N]    AVX-FP128 [N]   AVX-FP256 [Y] 
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [Y]        UMIP [Y] 
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Advanced Virtual Interrupt Controller                        AVIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- Clear Zero Instruction                                     CLZERO   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Capable]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast Short REP MOVSB                                         FSRM   [Capable]
|- Fast-String Operation                                        ERMS   [Capable]
|- Fused Multiply Add                                     FMA | FMA4   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Hardware P-state control                                      HwP   [Capable]
|- Instruction Based Sampling                                    IBS   [Capable]
|- Instruction INVLPGB                                       INVLPGB   [Missing]
|- Instruction INVPCID                                       INVPCID   [Capable]
|- Long Mode 64 bits                                       IA64 | LM   [Capable]
[... 412 lines cut ...]
Controller #0                                                Dual Channel  
 Bus Rate   933 MHz       Bus Speed  933 MHz           DDR5 Speed 1399 MT/s
[... cut ...]
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    1     32768      1024           4096    F5-6000J3238F16G
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    1     32768      1024           4096    F5-6000J3238F16G
```

cyring, 2022-12-02:
Need a 7000 owner to test this version:

[CoreFreq_develop.tar.gz](https://github.com/cyring/CoreFreq/files/10137130/CoreFreq_develop.tar.gz)

Buid,run, and post the output of `corefreq-cli -s -n -C 1 -n -M`

cyring, 2022-12-04:
Getting the `MEMCLK` may be decoded from other encoding since Zen4

https://github.com/cyring/CoreFreq/blob/aa3d63efd3de6e02cda1112aad406182e22bc805/amd_reg.h#L1571

For example, Zen2 case
```
zencli smu 0x50200
```
```
[0x00050200] READ(smu) = 0x00001937 (6455)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 1001 0011 0111
```

Using [zencli](https://github.com/cyring/Tips/blob/master/C/zencli.c) need someone to read SMU at address 0x50200

Jon0, 2022-12-17:
7950x :

```
# zencli smu 0x50200
[0x00050200] READ(smu) = 0x80050bb8 (2147814328)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1000 0000 0000 0101 0000 1011 1011 1000
```

```
Processor                                  [AMD Ryzen 9 7950X 16-Core Processor]
|- Architecture                                                   [Zen4/Raphael]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0a601203]
|- Signature                                                           [  AF_61]
|- Stepping                                                            [      2]
|- Online CPU                                                          [ 32/ 32]
|- Base Clock                                                          [ 99.999]
|- Frequency            (MHz)                      Ratio                        
                 Min   2999.99                    <  30 >                       
                 Max   4499.98                    <  45 >                       
|- Factory                                                             [100.000]
                       4500                       [  45 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   4499.98                    <  45 >                       
|- Turbo Boost                                                         [ UNLOCK]
                 XFR   5799.97                    [  58 ]                       
                 CPB   5699.97                    [  57 ]                       
                  1C   2999.99                    <  30 >                       
|- Uncore                                                              [   LOCK]
                 CLK   1799.99                    [  18 ]                       
                 MEM   5599.98                    [  56 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [Y]    AVX512-DQ [Y]  AVX512-IFMA [Y]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [Y]    AVX512-BW [Y]   AVX512-VL [Y] 
|- AVX512-VBMI  [Y] AVX512-VBMI2 [Y]  AVX512-VNNI [Y]  AVX512-ALG [Y] 
|- AVX512-VPOP  [Y] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [Y] AVX-VNNI-VEX [N]    AVX-FP128 [N]   AVX-FP256 [Y] 
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [Y]        UMIP [Y] 
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Advanced Virtual Interrupt Controller                        AVIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- Clear Zero Instruction                                     CLZERO   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Capable]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast Short REP MOVSB                                         FSRM   [Capable]
|- Fast-String Operation                                        ERMS   [Capable]
|- Fused Multiply Add                                     FMA | FMA4   [Capable]
[... 230 lines cut ...]
                                                                           
 Cha   CL  RCDr RCDw  RP  RAS   RC  RRDs RRDl FAW  WTRs WTRl  WR  clRR clWW
  #0   40   40   40   40   77  146    8   15   32    8   30   90    8   23 
  #1   40   40   40   40   77  146    8   15   32    8   30   90    8   23 
      CWL  RTP RdWr WrRd scWW sdWW ddWW scRR sdRR ddRR drRR drWW drWR drRRD
  #0   38   23   21    8    1   15   15    1    8    8    0    0    0    0 
[... cut ...]
       #0                                                                  
       #1    16    1     32768      1024           4096  CMK32GX5M2B6000C40
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    1     32768      1024           4096  CMK32GX5M2B6000C40
```

Is it possible to find the DIMM temperatures somewhere?

Jon0, 2022-12-17:
I had some issues with the develop branch when starting the daemon - compile works and insmod reported success:
```
sudo corefreqd
Driver connection error code 11
/dev/corefreqk: 'Resource temporarily unavailable' @ line 8791
```

```
sudo dmesg | grep CoreFreq
[   96.571073] CoreFreq: PCI_HSMP_Mailbox(1) Timeout
[   96.700603] CoreFreq(9:25:-1): Processor [ AF_61] Architecture [Zen4/Raphael] SMT [32/32]
```

Tested with stock arch kernel:
```
uname -a
Linux sapphire 6.0.12-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 08 Dec 2022 11:03:38 +0000 x86_64 GNU/Linux
```

Jon0, 2022-12-18:
I found the issue was I ran `corefreqd` which was installed on my system via AUR, and not `./corefreqd` the one I just built...

Here is the screenshots:
[image]
[image]
[image]

Jon0, 2022-12-19:
Frequency is not quite right, I tested a few settings in bios:

Actual Bios Setting -> CoreFreq Display Freq
6000 -> 5600 (screenshot)
5600 -> 4800
5200 -> 4000

Don't see Data Fabric in bios unless its hidden somewhere...

Jon0, 2022-12-19:
Yep heres another when I set 2000M/T in bios (lowest it can go)

```
zencli smu 0x50200
[0x00050200] READ(smu) = 0x000503e8 (328680)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0101 0000 0011 1110 1000
```

I don't have a windows install available right now to check other software, but the above formula does look correct

KeithMyers, 2022-12-20:
corefreq-cli -s -n -C 1 -n -M
Processor                                  [AMD Ryzen 9 7950X 16-Core Processor]
|- Architecture                                                   [Zen4/Raphael]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0a601203]
|- Signature                                                           [  AF_61]
|- Stepping                                                            [      2]
|- Online CPU                                                          [ 32/ 32]
|- Base Clock                                                          [100.005]
|- Frequency            (MHz)                      Ratio                        
                 Min   3000.15                    <  30 >                       
                 Max   4500.23                    <  45 >                       
|- Factory                                                             [100.000]
                       4500                       [  45 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   4500.23                    <  45 >                       
   |- CPPC                                                                      
                 Min   3700.19                    <  37 >                       
                 Max   1900.10                    <  19 >                       
                 TGT   3700.19                    <  37 >                       
|- Turbo Boost                                                         [ UNLOCK]
                 XFR   5800.29                    [  58 ]                       
                 CPB   5700.29                    [  57 ]                       
                  1C   3000.15                    <  30 >                       
|- Uncore                                                              [   LOCK]
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [Y]    AVX512-DQ [Y]  AVX512-IFMA [Y]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [Y]    AVX512-BW [Y]   AVX512-VL [Y] 
|- AVX512-VBMI  [Y] AVX512-VBMI2 [Y]  AVX512-VNNI [Y]  AVX512-ALG [Y] 
|- AVX512-VPOP  [Y] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [Y] AVX-VNNI-VEX [N]    AVX-FP128 [N]   AVX-FP256 [Y] 
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [Y]        UMIP [Y] 
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Advanced Virtual Interrupt Controller                        AVIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- Clear Zero Instruction                                     CLZERO   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Capable]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast Short REP MOVSB                                         FSRM   [Capable]
|- Fast-String Operation                                        ERMS   [Capable]
|- Fused Multiply Add                                     FMA | FMA4   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Hardware P-state control                                      HwP   [Capable]
|- Instruction Based Sampling                                    IBS   [Capable]
|- Instruction INVLPGB                                       INVLPGB   [Missing]
|- Instruction INVPCID                                       INVPCID   [Capable]
|- Long Mode 64 bits                                       IA64 | LM   [Capable]
|- LightWeight Profiling                                         LWP   [Missing]
[... 197 lines cut ...]
[... cut ...]
029 4921.33   206  1.2875   81  000000000000000000    0.000000000   0.000000000
030 4698.44   206  1.2875   81  000000000000000000    0.000000000   0.000000000
031 4454.73   206  1.2875   81  000000000000000000    0.000000000   0.000000000

             Package[0]    Cores         Uncore        Memory        Platform
Energy(J): 208.999008179 197.947296143   0.000000000   0.000000000   0.000000000
Power(W) : 208.999008179 197.947296143   0.000000000   0.000000000   0.000000000

                              Zen UMC  [14E0]                              
Controller #0

KeithMyers, 2022-12-20:
Sure. Here you go.

  corefreq-cli -s -n -C 1 -n -M
Processor                                  [AMD Ryzen 9 7950X 16-Core Processor]
|- Architecture                                                   [Zen4/Raphael]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0a601203]
|- Signature                                                           [  AF_61]
|- Stepping                                                            [      2]
|- Online CPU                                                          [ 32/ 32]
|- Base Clock                                                          [ 99.997]
|- Frequency            (MHz)                      Ratio                        
                 Min   2999.94                    <  30 >                       
                 Max   4499.92                    <  45 >                       
|- Factory                                                             [100.000]
                       4500                       [  45 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   4499.92                    <  45 >                       
   |- CPPC                                                                      
                 Min   3699.93                    <  37 >                       
                 Max   1899.96                    <  19 >                       
                 TGT   2099.96                    <  21 >                       
|- Turbo Boost                                                         [ UNLOCK]
                 XFR   5799.89                    [  58 ]                       
                 CPB   5699.89                    [  57 ]                       
                  1C   2999.94                    <  30 >                       
|- Uncore                                                              [   LOCK]
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [Y]    AVX512-DQ [Y]  AVX512-IFMA [Y]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [Y]    AVX512-BW [Y]   AVX512-VL [Y] 
|- AVX512-VBMI  [Y] AVX512-VBMI2 [Y]  AVX512-VNNI [Y]  AVX512-ALG [Y] 
|- AVX512-VPOP  [Y] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [Y] AVX-VNNI-VEX [N]    AVX-FP128 [N]   AVX-FP256 [Y] 
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [Y]        UMIP [Y] 
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Advanced Virtual Interrupt Controller                        AVIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- Clear Zero Instruction                                     CLZERO   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Capable]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast Short REP MOVSB                                         FSRM   [Capable]
|- Fast-String Operation                                        ERMS   [Capable]
|- Fused Multiply Add                                     FMA | FMA4   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Hardware P-state control                                      HwP   [Capable]
|- Instruction Based Sampling                                    IBS   [Capable]
|- Instruction INVLPGB                                       INVLPGB   [Missing]
|- Instruction INVPCID                                       INVPCID   [Capable]
[... 202 lines cut ...]
017 5172.89   206  1.2875   92  000000000000000000    0.000000000   0.000000000
[... cut ...]
030 3582.44   206  1.2875   81  000000000000000000    0.000000000   0.000000000
031 4033.24   206  1.2875   81  000000000000000000    0.000000000   0.000000000

             Package[0]    Cores         Uncore        Memory        Platform
Energy(J): 208.384918213 200.727081299   0.000000000   0.000000000   0.000000000
Power(W) : 208.384918213 200.727081299   0.000000000   0.000000000   0.000000000

                              Zen UMC  [14E0]                              
Controller #0                                                    Disabled

corefreq-cli -v
1.94.0

KeithMyers, 2022-12-20:
So is the primary terminal view supposed to show voltage?  All mine shows is 0.0 in the bottom right corner.

[image]

KeithMyers, 2022-12-20:
> > So is the primary terminal view supposed to show voltage? All mine shows is 0.0 in the bottom right corner.
> 
> This voltage aggregation is not programmed yet for Zen4 Meanwhile you can switch to the `Voltage` view from menu.
> 
> About the UMC something is odd, it has been working with @Jon0 Not sure if it's a code regression but I don't find any.
> 
> @Jon0 : can you please pull latest `develop` branch and post the UMC ?

I guess I need to go reread the docs. 

What menu??  I don't see any menu or prompts for keystrokes to pull up a menu from the cli view.  I know you can use a command parameter to show voltages. But haven't figured out how to do anything in the cli terminal "top"- like view.

KeithMyers, 2022-12-20:
> @KeithMyers : can you build as below and tell if UMC is showing up ? ... then reload _CoreFreq_ from the working directory.
> 
> ```shell
> make ARCH_PMC=UMC clean all
> ```

Maybe doesn't work because of all the build errors?

root@Pipsqueek:/home/keith/Downloads/CoreFreq-develop# make ARCH_PMC=UMC clean all
rm -f corefreqd corefreq-cli
make -j1 -C /lib/modules/5.15.0-56-generic/build M=/home/keith/Downloads/CoreFreq-develop clean
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-56-generic'
  CLEAN   /home/keith/Downloads/CoreFreq-develop/Module.symvers
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-56-generic'
cc  -Wall -Wfatal-errors -pthread -c corefreqd.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=6575000000 -D UBENCH=0 -D DELAY_TSC=1 -D ARCH_PMC=UMC \
  -o corefreqd.o
cc  -Wall -Wfatal-errors -c corefreqm.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=6575000000 -D UBENCH=0 -D DELAY_TSC=1 -D ARCH_PMC=UMC \
  -o corefreqm.o
cc  -Wall -Wfatal-errors corefreqd.c corefreqm.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=6575000000 -D UBENCH=0 -D DELAY_TSC=1 -D ARCH_PMC=UMC \
  -o corefreqd -lpthread -lm -lrt
cc  -Wall -Wfatal-errors -c corefreq-cli.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=6575000000 -D UBENCH=0 -D DELAY_TSC=1 -D ARCH_PMC=UMC  \
  -o corefreq-cli.o
cc  -Wall -Wfatal-errors -c corefreq-ui.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=6575000000 -D UBENCH=0 -D DELAY_TSC=1 -D ARCH_PMC=UMC \
  -o corefreq-ui.o
cc  -Wall -Wfatal-errors -c corefreq-cli-rsc.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=6575000000 -D UBENCH=0 -D DELAY_TSC=1 -D ARCH_PMC=UMC  \
  -o corefreq-cli-rsc.o
cc  -Wall -Wfatal-errors -c corefreq-cli-json.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=6575000000 -D UBENCH=0 -D DELAY_TSC=1 -D ARCH_PMC=UMC \
  -o corefreq-cli-json.o
cc  -Wall -Wfatal-errors -c corefreq-cli-extra.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=6575000000 -D UBENCH=0 -D DELAY_TSC=1 -D ARCH_PMC=UMC \
  -o corefreq-cli-extra.o
cc  -Wall -Wfatal-errors \
  corefreq-cli.c corefreq-ui.c corefreq-cli-rsc.c \
  corefreq-cli-json.c corefreq-cli-extra.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=6575000000 -D UBENCH=0 -D DELAY_TSC=1 -D ARCH_PMC=UMC  \
  -o corefreq-cli -lm -lrt
make -j1 -C /lib/modules/5.15.0-56-generic/build M=/home/keith/Downloads/CoreFreq-develop modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-56-generic'
  CC [M]  /home/keith/Downloads/CoreFreq-develop/corefreqk.o
  MODPOST /home/keith/Downloads/CoreFreq-develop/Module.symvers
  CC [M]  /home/keith/Downloads/CoreFreq-develop/corefreqk.mod.o
  LD [M]  /home/keith/Downloads/CoreFreq-develop/corefreqk.ko
  BTF [M] /home/keith/Downloads/CoreFreq-develop/corefreqk.ko
Skipping BTF generation for /home/keith/Downloads/CoreFreq-develop/corefreqk.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-56-generic'

KeithMyers, 2022-12-20:
OK, I built the dev package as user first.  Got the same errors so then tried as root to see if it made any difference.

No distro provided corefreq package provided or ever installed so no API mismatch.

I've built custom modules before and they normally build clean except for the ```skipping BTF``` warning.

I've been following the readme so using insmod for the module insertion.

Still not seeing my memory controller with the ```make ARCH_PMC=UMC clean all``` version of the build.

Thanks for the help on the menu commands.  I accidentally brushed the F keys on the keyboard and voila a menu popped up. So good to go now in navigating the cli interface.

Jon0, 2022-12-20:
Heres latest `develop` branch, BIOS set to 6000M/T
```

Processor                                  [AMD Ryzen 9 7950X 16-Core Processor]
|- Architecture                                                   [Zen4/Raphael]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0a601203]
|- Signature                                                           [  AF_61]
|- Stepping                                                            [      2]
|- Online CPU                                                          [ 32/ 32]
|- Base Clock                                                          [100.000]
|- Frequency            (MHz)                      Ratio                        
                 Min   3000.00                    <  30 >                       
                 Max   4500.00                    <  45 >                       
|- Factory                                                             [100.000]
                       4500                       [  45 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   4500.00                    <  45 >                       
   |- CPPC                                                                      
                 Min   3700.00                    <  37 >                       
                 Max   1900.00                    <  19 >                       
                 TGT   1900.00                    <  19 >                       
|- Turbo Boost                                                         [ UNLOCK]
                 XFR   5800.00                    [  58 ]                       
                 CPB   5700.00                    [  57 ]                       
                  1C   3000.00                    <  30 >                       
|- Uncore                                                              [   LOCK]
                 CLK   1000.00                    [  10 ]                       
                 MEM   3000.00                    [  30 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [Y]    AVX512-DQ [Y]  AVX512-IFMA [Y]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [Y]    AVX512-BW [Y]   AVX512-VL [Y] 
|- AVX512-VBMI  [Y] AVX512-VBMI2 [Y]  AVX512-VNNI [Y]  AVX512-ALG [Y] 
|- AVX512-VPOP  [Y] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [Y] AVX-VNNI-VEX [N]    AVX-FP128 [N]   AVX-FP256 [Y] 
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [Y]        UMIP [Y] 
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Advanced Virtual Interrupt Controller                        AVIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- Clear Zero Instruction                                     CLZERO   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Capable]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast Short REP MOVSB                                         FSRM   [Capable]
|- Fast-String Operation                                        ERMS   [Capable]
|- Fused Multiply Add                                     FMA | FMA4   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Hardware P-state control                                      HwP   [Capable]
|- Instruction Based Sampling                                    IBS   [Capable]
[... 229 lines cut ...]
Controller #0                                                Dual Channel  
[... cut ...]
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    1     32768      1024           4096  CMK32GX5M2B6000C40
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    16    1     32768      1024           4096  CMK32GX5M2B6000C40
```

KeithMyers, 2022-12-20:
Just in case it wasn't running the dev version.  Here is the output of version 1.94.0 again 

keith@Pipsqueek:~/Downloads/CoreFreq-develop$ ./corefreq-cli -s -n -C 1 -n -M
Processor                                  [AMD Ryzen 9 7950X 16-Core Processor]
|- Architecture                                                   [Zen4/Raphael]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0a601203]
|- Signature                                                           [  AF_61]
|- Stepping                                                            [      2]
|- Online CPU                                                          [ 32/ 32]
|- Base Clock                                                          [ 99.949]
|- Frequency            (MHz)                      Ratio                        
                 Min   2998.46                    <  30 >                       
                 Max   4497.69                    <  45 >                       
|- Factory                                                             [100.000]
                       4500                       [  45 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   4497.69                    <  45 >                       
   |- CPPC                                                                      
                 Min   3698.10                    <  37 >                       
                 Max   1899.02                    <  19 >                       
                 TGT   1899.02                    <  19 >                       
|- Turbo Boost                                                         [ UNLOCK]
                 XFR   5797.02                    [  58 ]                       
                 CPB   5697.07                    [  57 ]                       
                  1C   2998.46                    <  30 >                       
|- Uncore                                                              [   LOCK]
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [Y]    AVX512-DQ [Y]  AVX512-IFMA [Y]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [Y]    AVX512-BW [Y]   AVX512-VL [Y] 
|- AVX512-VBMI  [Y] AVX512-VBMI2 [Y]  AVX512-VNNI [Y]  AVX512-ALG [Y] 
|- AVX512-VPOP  [Y] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [Y] AVX-VNNI-VEX [N]    AVX-FP128 [N]   AVX-FP256 [Y] 
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [Y]        UMIP [Y] 
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Advanced Virtual Interrupt Controller                        AVIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- Clear Zero Instruction                                     CLZERO   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Capable]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast Short REP MOVSB                                         FSRM   [Capable]
|- Fast-String Operation                                        ERMS   [Capable]
|- Fused Multiply Add                                     FMA | FMA4   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Hardware P-state control                                      HwP   [Capable]
|- Instruction Based Sampling                                    IBS   [Capable]
|- Instruction INVLPGB                                       INVLPGB   [Missing]
|- Instruction INVPCID                                       INVPCID   [Capable]
[... 201 lines cut ...]
[... cut ...]
031 4158.51   207  1.2938   75  000000000000000000    0.000000000   0.000000000

             Package[0]    Cores         Uncore        Memory        Platform
Energy(J): 203.519165039 186.657379150   0.000000000   0.000000000   0.000000000
Power(W) : 203.519165039 186.657379150   0.000000000   0.000000000   0.000000000

                              Zen UMC  [14E0]                              
Controller #0                                                    Disabled 

Still no sign of an enable memory controller.

KeithMyers, 2022-12-20:
Yes, I have IOMMU disabled in the BIOS.  Is that affecting the issue?

$ ./corefreq-cli -B
[ 0] American Megatrends Inc.                                                   
[ 1] 0805                                                                       
[ 2] 11/04/2022                                                                 
[ 3] ASUS                                                                       
[ 4] System Product Name                                                        
[ 5] System Version                                                             
[ 6] S---e---e---l---m---                                                       
[ 7] SKU                                                                        
[ 8] To be filled by O.E.M.                                                     
[ 9] ASUSTeK COMPUTER INC.                                                      
[10] ROG CROSSHAIR X670E HERO                                                   
[11] Rev 1.xx                                                                   
[12] 2---0---7---4--                                                            
[13] Number Of Devices:4\Maximum Capacity:134217728 bytes                       
[14]                                                                            
[15] DIMM 1\P0 CHANNEL A                                                        
[16]                                                                            
[17] DIMM 1\P0 CHANNEL B                                                        
[18]                                                                            
[19] Unknown                                                                    
[20]                                                                            
[21] Unknown                                                                    
[22]                                                                            
[23] F5-6000J3636F16G                                                           
[24]                                                                            
[25] F5-6000J3636F16G                

$ lspci -nn
00:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14d8]
00:01.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14da]
00:01.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14db]
00:01.2 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14db]
00:01.3 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14db]
00:02.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14da]
00:02.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14db]
00:03.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14da]
00:04.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14da]
00:08.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14da]
00:08.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14dd]
00:08.3 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:14dd]
00:14.0 SMBus [0c05]: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller [1022:790b] (rev 71)
00:14.3 ISA bridge [0601]: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge [1022:790e] (rev 51)
00:18.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e0]
00:18.1 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e1]
00:18.2 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e2]
00:18.3 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e3]
00:18.4 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e4]
00:18.5 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e5]
00:18.6 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e6]
00:18.7 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e7]
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GA102 [GeForce RTX 3080 Lite Hash Rate] [10de:2216] (rev a1)
01:00.1 Audio device [0403]: NVIDIA Corporation GA102 High Definition Audio Controller [10de:1aef] (rev a1)
02:00.0 Non-Volatile memory controller [0108]: Shenzhen Longsys Electronics Co., Ltd. Device [1d97:5236] (rev 01)
03:00.0 VGA compatible controller [0300]: NVIDIA Corporation GA102 [GeForce RTX 3080 12GB] [10de:220a] (rev a1)
03:00.1 Audio device [0403]: NVIDIA Corporation GA102 High Definition Audio Controller [10de:1aef] (rev a1)
04:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f4] (rev 01)
05:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
05:08.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
05:0c.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
05:0d.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
07:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f4] (rev 01)
08:01.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
08:02.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
08:03.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
08:04.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43f5] (rev 01)
[... cut ...]
	Part Number: F5-6000J3636F16G
	Rank: 1
	Configured Memory Speed: 6000 MT/s
	Minimum Voltage: 1.1 V
	Maximum Voltage: 1.1 V
	Configured Voltage: 1.1 V
	Memory Technology: DRAM
	Memory Operating Mode Capability: Volatile memory
	Firmware Version:    
	Module Manufacturer ID: Bank 5, Hex 0xCD
	Module Product ID: Unknown
	Memory Subsystem Controller Manufacturer ID: Unknown
	Memory Subsystem Controller Product ID: Unknown
	Non-Volatile Size: None
	Volatile Size: 16 GB
	Cache Size: None
	Logical Size: None

KeithMyers, 2022-12-21:
OK, modified the header file and rebuilt.  Still no memory controller. Enabled the IOMMU in the BIOS.

$ ./corefreq-cli -s -n -C 1 -n -M
Processor                                  [AMD Ryzen 9 7950X 16-Core Processor]
|- Architecture                                                   [Zen4/Raphael]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0a601203]
|- Signature                                                           [  AF_61]
|- Stepping                                                            [      2]
|- Online CPU                                                          [ 32/ 32]
|- Base Clock                                                          [100.037]
|- Frequency            (MHz)                      Ratio                        
                 Min   3001.12                    <  30 >                       
                 Max   4501.68                    <  45 >                       
|- Factory                                                             [100.000]
                       4500                       [  45 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   4501.68                    <  45 >                       
   |- CPPC                                                                      
                 Min   3701.38                    <  37 >                       
                 Max   1900.71                    <  19 >                       
                 TGT   1900.71                    <  19 >                       
|- Turbo Boost                                                         [ UNLOCK]
                 XFR   5802.16                    [  58 ]                       
                 CPB   5702.13                    [  57 ]                       
                  1C   3001.12                    <  30 >                       
|- Uncore                                                              [   LOCK]
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [Y]    AVX512-DQ [Y]  AVX512-IFMA [Y]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [Y]    AVX512-BW [Y]   AVX512-VL [Y] 
|- AVX512-VBMI  [Y] AVX512-VBMI2 [Y]  AVX512-VNNI [Y]  AVX512-ALG [Y] 
|- AVX512-VPOP  [Y] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [Y] AVX-VNNI-VEX [N]    AVX-FP128 [N]   AVX-FP256 [Y] 
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [Y]        UMIP [Y] 
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Advanced Virtual Interrupt Controller                        AVIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- Clear Zero Instruction                                     CLZERO   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Capable]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast Short REP MOVSB                                         FSRM   [Capable]
|- Fast-String Operation                                        ERMS   [Capable]
|- Fused Multiply Add                                     FMA | FMA4   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Hardware P-state control                                      HwP   [Capable]
|- Instruction Based Sampling                                    IBS   [Capable]
|- Instruction INVLPGB                                       INVLPGB   [Missing]
|- Instruction INVPCID                                       INVPCID   [Capable]
[... 442 lines cut ...]
	Locator: DIMM 1
	Bank Locator: P0 CHANNEL B
	Type: DDR5
[... cut ...]
	Part Number: F5-6000J3636F16G
	Rank: 1
	Configured Memory Speed: 6000 MT/s
	Minimum Voltage: 1.1 V
	Maximum Voltage: 1.1 V
	Configured Voltage: 1.1 V
	Memory Technology: DRAM
	Memory Operating Mode Capability: Volatile memory
	Firmware Version:    
	Module Manufacturer ID: Bank 5, Hex 0xCD
	Module Product ID: Unknown
	Memory Subsystem Controller Manufacturer ID: Unknown
	Memory Subsystem Controller Product ID: Unknown
	Non-Volatile Size: None
	Volatile Size: 16 GB
	Cache Size: None
	Logical Size: None

KeithMyers, 2022-12-21:
I won't make any changes until you verify your last post.

I believe you made a mistake and are referencing both the wrong file and  the wrong line number.

You say to make changes in ```corefreqk.c``` at line number ```6558``` and make changes to
```static PCI_CALLBACK AMD_DataFabric_Rembrandt(struct pci_dev *pdev)```

at line number 6558 in this file is just ```{```

I believe you want me to make changes to ```corefreqk.h```  at line number ```2056``` which does have the 
```static PCI_CALLBACK AMD_DataFabric_Rembrandt(struct pci_dev *pdev)``` code

KeithMyers, 2022-12-21:
> @KeithMyers
>
> I think more controllers need to be probed.
> I have encountered this issue with Intel Tiger Lake where the controller is the second
>
> Can you restore original code of develop branch, and replace the below function
> https://github.com/cyring/CoreFreq/blob/e41ec49b5f3ff9085f5c4396ce236182fc8f2711/corefreqk.c#L6558
>
> ... with this code:

Best thing to do is pull the new dev branch code release commit e41ec49
The filename, line number and code matches your post then.

	return AMD_17h_DataFabric(	pdev,
					(const unsigned int[2][2]) {
						{ 0x0, 0x20},
						{0x10, 0x30}
					},
					0x40, 0x90,
					MC_MAX_CTRL, MC_MAX_CHA,
					(const unsigned short[]) {0x18} );
}
(Rembrandt, like Raphael are both for DDR5. That's why this common function)

Rebuild and run CoreFreq, I just need the output of the UMC

Thank you

KeithMyers, 2022-12-21:
> @KeithMyers
> Hello,
>
> I just want to try AMD_17h_DataFabric with a max umc of 8 (aka MC_MAX_CTRL)
>
> Sorry if my instructions were messed up by all the previous changes. You have to pull the develop branch again and go to AMD_DataFabric_Rembrandt function to make the change.

Going to bed.  I'll tackle the new build tomorrow morning.

KeithMyers, 2022-12-21:
No luck still.

$ corefreq-cli -M
                              Zen UMC  [14E0]                              
Controller #0                                                    Disabled  
                                                                           
Controller #1                                                    Disabled  
                                                                           
Controller #2                                                    Disabled  
                                                                           
Controller #3                                                    Disabled  
                                                                           
Controller #4                                                    Disabled

KeithMyers, 2022-12-21:
Yes, I have something similar

Dec 21 09:35:55 Pipsqueek kernel: [ 1169.389122] CoreFreq: AMD_SMN_Read(0, b50d6c) TryLock
Dec 21 09:35:55 Pipsqueek kernel: [ 1169.389127] CoreFreq: AMD_17h_DataFabric() Break UMC(5) probing @ PCI(0x0:0x0:0x80)

Jon0, 2022-12-21:
Looks like something is at 0x502c8:

```
zencli smu 0x502c0
[0x000502c0] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000

zencli smu 0x502c4
[0x000502c4] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000

zencli smu 0x502c8
[0x000502c8] READ(smu) = 0x005a0186 (5898630)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0101 1010 0000 0001 1000 0110

zencli smu 0x502cc
[0x000502cc] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
```

KeithMyers, 2022-12-21:
> > 00:18.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e0]
> > 00:18.1 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e1]
> > 00:18.2 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e2]
> > 00:18.3 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e3]
> > 00:18.4 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e4]
> > 00:18.5 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e5]
> > 00:18.6 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e6]
> > 00:18.7 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:14e7]
> 
> @KeithMyers Can you try the attached archive ? The on-going development is to scan the eight possible controllers.
> 
> [CoreFreq_develop.tar.gz](https://github.com/cyring/CoreFreq/files/10281423/CoreFreq_develop.tar.gz)

Got a weird error for just doing make.   My time is correct.

$ make
cc  -Wall -Wfatal-errors -pthread -c corefreqd.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=6575000000 -D UBENCH=0 -D DELAY_TSC=1 \
  -o corefreqd.o
cc  -Wall -Wfatal-errors -c corefreqm.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=6575000000 -D UBENCH=0 -D DELAY_TSC=1 \
  -o corefreqm.o
cc  -Wall -Wfatal-errors corefreqd.c corefreqm.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=6575000000 -D UBENCH=0 -D DELAY_TSC=1 \
  -o corefreqd -lpthread -lm -lrt
cc  -Wall -Wfatal-errors -c corefreq-cli.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=6575000000 -D UBENCH=0 -D DELAY_TSC=1  \
  -o corefreq-cli.o
cc  -Wall -Wfatal-errors -c corefreq-ui.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=6575000000 -D UBENCH=0 -D DELAY_TSC=1 \
  -o corefreq-ui.o
cc  -Wall -Wfatal-errors -c corefreq-cli-rsc.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=6575000000 -D UBENCH=0 -D DELAY_TSC=1  \
  -o corefreq-cli-rsc.o
cc  -Wall -Wfatal-errors -c corefreq-cli-json.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=6575000000 -D UBENCH=0 -D DELAY_TSC=1 \
  -o corefreq-cli-json.o
cc  -Wall -Wfatal-errors -c corefreq-cli-extra.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=6575000000 -D UBENCH=0 -D DELAY_TSC=1 \
  -o corefreq-cli-extra.o
cc  -Wall -Wfatal-errors \
  corefreq-cli.c corefreq-ui.c corefreq-cli-rsc.c \
  corefreq-cli-json.c corefreq-cli-extra.c \
  -D CORE_COUNT=256 -D TASK_ORDER=5 -D MAX_FREQ_HZ=6575000000 -D UBENCH=0 -D DELAY_TSC=1  \
  -o corefreq-cli -lm -lrt
make -j1 -C /lib/modules/5.15.0-56-generic/build M=/home/keith/Downloads/CoreFreq modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-56-generic'
```make[2]: Warning: File '/home/keith/Downloads/CoreFreq/corefreqk.c' has modification time 1101 s in the future
  CC [M]  /home/keith/Downloads/CoreFreq/corefreqk.o
make[2]: warning:  Clock skew detected.  Your build may be incomplete.```
  MODPOST /home/keith/Downloads/CoreFreq/Module.symvers
  CC [M]  /home/keith/Downloads/CoreFreq/corefreqk.mod.o
  LD [M]  /home/keith/Downloads/CoreFreq/corefreqk.ko
  BTF [M] /home/keith/Downloads/CoreFreq/corefreqk.ko
Skipping BTF generation for /home/keith/Downloads/CoreFreq/corefreqk.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-56-generic'

KeithMyers, 2022-12-21:
Still no luck.

q$ ./corefreq-cli -M
                              Zen UMC  [14E0]                              
Controller #0                                                    Disabled  
                                                                           
Controller #1                                                    Disabled  
                                                                           
Controller #2                                                    Disabled  
                                                                           
Controller #3                                                    Disabled  
                                                                           
Controller #4                                                    Disabled  
                                                                           
Controller #5                                                    Disabled  
                                                                           
Controller #6                                                    Disabled  
                                                                           
Controller #7                                                    Disabled

KeithMyers, 2022-12-22:
./zencli smu 0x050104
 
[0x00050104] READ(smu) = 0xc0800083 (3229614211)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1100 0000 1000 0000 0000 0000 1000 0011

./zencli smu 0x150104
[0x00150104] READ(smu) = 0xc0800083 (3229614211)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1100 0000 1000 0000 0000 0000 1000 0011

./zencli smu 0x250104
[0x00250104] READ(smu) = 0xffffffff (4294967295)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1111 1111 1111 1111 1111 1111 1111 1111

./zencli smu 0x350104
[0x00350104] READ(smu) = 0xffffffff (4294967295)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1111 1111 1111 1111 1111 1111 1111 1111

./zencli smu 0x450104
[0x00450104] READ(smu) = 0xffffffff (4294967295)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1111 1111 1111 1111 1111 1111 1111 1111

./zencli smu 0x550104
[0x00550104] READ(smu) = 0xffffffff (4294967295)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1111 1111 1111 1111 1111 1111 1111 1111

/zencli smu 0x650104
[0x00650104] READ(smu) = 0xffffffff (4294967295)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1111 1111 1111 1111 1111 1111 1111 1111

./zencli smu 0x750104
[0x00750104] READ(smu) = 0xffffffff (4294967295)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1111 1111 1111 1111 1111 1111 1111 1111

KeithMyers, 2022-12-22:
OK, here you go with the json output.
[corefreq.json.gz](https://github.com/cyring/CoreFreq/files/10290326/corefreq.json.gz)

KeithMyers, 2022-12-23:
> @KeithMyers
> 
> Hi, Do you still have CoreFreq errors in the kernel log ? especially `CoreFreq: AMD_SMN_Read(0, b50d6c) TryLock`
Yes, I do.
Dec 22 12:53:47 Pipsqueek kernel: [  294.427042] CoreFreq: AMD_SMN_Read(0, b50d6c) TryLock

KeithMyers, 2022-12-23:
> corefreq-cli -M -n -s

 ./corefreq-cli -M -n -s
                              Zen UMC  [14E0]                              
Controller #0                                                    Disabled  

Processor                                  [AMD Ryzen 9 7950X 16-Core Processor]
|- Architecture                                                   [Zen4/Raphael]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0a601203]
|- Signature                                                           [  AF_61]
|- Stepping                                                            [      2]
|- Online CPU                                                          [ 32/ 32]
|- Base Clock                                                          [100.001]
|- Frequency            (MHz)                      Ratio                        
                 Min   3000.06                    <  30 >                       
                 Max   4500.09                    <  45 >                       
|- Factory                                                             [100.000]
                       4500                       [  45 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   4500.09                    <  45 >                       
   |- CPPC                                                                      
                 Min   3700.07                    <  37 >                       
                 Max   1900.04                    <  19 >                       
                 TGT   3700.07                    <  37 >                       
|- Turbo Boost                                                         [ UNLOCK]
                 XFR   5800.12                    [  58 ]                       
                 CPB   5700.11                    [  57 ]                       
                  1C   3000.06                    <  30 >                       
|- Uncore                                                              [   LOCK]
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [Y]    AVX512-DQ [Y]  AVX512-IFMA [Y]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [Y]    AVX512-BW [Y]   AVX512-VL [Y] 
|- AVX512-VBMI  [Y] AVX512-VBMI2 [Y]  AVX512-VNNI [Y]  AVX512-ALG [Y] 
|- AVX512-VPOP  [Y] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [Y] AVX-VNNI-VEX [N]    AVX-FP128 [N]   AVX-FP256 [Y] 
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [Y]        UMIP [Y] 
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Advanced Virtual Interrupt Controller                        AVIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- Clear Zero Instruction                                     CLZERO   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Capable]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast Short REP MOVSB                                         FSRM   [Capable]
|- Fast-String Operation                                        ERMS   [Capable]
|- Fused Multiply Add                                     FMA | FMA4   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Hardware P-state control                                      HwP   [Capable]
[... 165 lines cut ...]
   |- Power Limit                                                PL1   [    0 W]
   |- Time Window                                                TW1   [   0 ns]
[... cut ...]
   |- Thermal Monitor Trip                                     Limit   [  115 C]
   |- HTC Temperature Limit                                    Limit   [  127 C]
   |- HTC Temperature Hysteresis                           Threshold   [    2 C]
|- Units                                                                        
   |- Power                                               watt   [      Missing]
   |- Energy                                             joule   [  0.000015259]
   |- Window                                            second   [  0.000976562]

KeithMyers, 2022-12-23:
Doesn't seem to have fixed it.

Jon0, 2022-12-23:
Hi, i'll be back home to test anything in about a week

KeithMyers, 2022-12-23:
> > Doesn't seem to have fixed it.
> > `Dec 22 20:00:57 Pipsqueek kernel: [25924.481858] CoreFreq: AMD_SMN_Read(0, b50104) TryLock`
> 
> Within JSON, I noticed `amd-pstate` which is also doing SMN calls. Surely a SMU conflict between drivers. Can you unload, blacklist it, also `k10temp`; and post UMC and log errors?
> 
> EDIT: if it makes things easier, I can provide an ISO image with prerequisites to run CoreFreq ?

No sure what you mean by ISO image.  I have tried every way I could find of disabling amd-pstate but am unable to return to acpi_freq scaling governor. It seems to be locked on this mobo.  There is no CPPC enable/disable in this BIOS like there is on my C7H mobos.

KeithMyers, 2022-12-23:
OK, I get it.  A complete Arch boot image. I am downloading on my slow connection now.
Will give that a try to help you troubleshoot my hardware.

KeithMyers, 2022-12-23:
I tried the initcall blacklist=amd_pstate_init in the grub command line like I found online.  Did not remove amd_pstate scaling governor.

I did not have the amd_pstate.shared_mem=0 statement though.

KeithMyers, 2022-12-23:
Ubuntu 22.04.1 LTS is the distro.

I've never attempted to build a kernel before.  Would not know how unless I found some guide somewhere of exactly what is needed.

KeithMyers, 2022-12-24:
Yes, your Arch ISO image shows the memory controller.  I couldn't figure out a way to run corefreq-cli standalone to get a copy of the memory controller output.

It boots directly into corefreq-cli.  Once I exited the program all I could do was look at the bin dir which had the scripts in it.  I couldn't find the actual corefreq-cli program to launch it.  I got no desktop only command terminal.
I saw no Wiki > LiveCD

KeithMyers, 2022-12-24:
I took a picture with my phone of the memory controller.
[image]

KeithMyers, 2022-12-24:
No I don't have that message in the kernel log.

I do always get these messages.  I don't know their importance.

Dec 23 18:45:49 Pipsqueek kernel: [    0.291364] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.GPP7.UP00.DP40.UP00.DP68], AE_NOT_FOUND (20210730/dswload2-162)
Dec 23 18:45:49 Pipsqueek kernel: [    0.291368] ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20210730/psobject-220)

I don't know what you mean by Experimental mode?

KeithMyers, 2022-12-24:
I don't know how I got the amd_pstate.

I was always running acpi_freq before on the 5950X and C7H board.

I assume I picked up when I changed cpus and mobos.

Unless it got picked up by my one-time experiment with the 6.09 kernel through the mainline ppa app.

Which I reverted when I saw it made no difference in sensors appearing.

KeithMyers, 2022-12-24:
OK after dinner.

KeithMyers, 2022-12-24:
./zencli smu 0x50030
[0x00050030] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000

./zencli smu 0x50034
[0x00050034] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000

./zencli smu 0x50038
[0x00050038] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000

./zencli smu 0x5003c
[0x0005003c] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000

./zencli smu 0x50080
[0x00050080] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000

./zencli smu 0x50084
[0x00050084] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000

./zencli smu 0x50040
[0x00050040] READ(smu) = 0x00150508 (1377544)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 0101 0000 0101 0000 1000

./zencli smu 0x50044
[0x00050044] READ(smu) = 0x00150508 (1377544)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 0101 0000 0101 0000 1000

./zencli smu 0x50048
[0x00050048] READ(smu) = 0x0025060c (2426380)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0010 0101 0000 0110 0000 1100

./zencli smu 0x5004c
[0x0005004c] READ(smu) = 0x00150508 (1377544)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 0101 0000 0101 0000 1000

./zencli smu 0x50090
[0x00050090] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000

./zencli smu 0x50094
[0x00050094] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000

KeithMyers, 2022-12-24:
So how can I test this?  The memory controller is disabled in my distro.
Have you built another Arch distro for me to test so I can boot that?

KeithMyers, 2022-12-24:
> > So how can I test this? The memory controller is disabled in my distro. Have you built another Arch distro for me to test so I can boot that?
> 
> Yes I have refreshed the development ISO for your testings.
> 
> However does a simple driver removing is working on your setup ?
> 
> ```
> modprobe -r amd_pstate
> ```

No because amd_pstate is built into the kernel.  It is not a module

sudo modprobe -r amd_pstate
[sudo] password for keith: 
modprobe: FATAL: Module amd_pstate is builtin.

So is the previous link to the Arch ISO image the same for getting the updated memory controller code?

KeithMyers, 2022-12-25:
Your latest ISO image code changes still isn't showing the correct memory DIMM size.
[image]

KeithMyers, 2022-12-25:
I've been curious about the green bolded values in Settings.  It looks like they may be a toggle of some sort but I had no luck in effecting any change in any of the values.  So I must not know the secret handshake or something.

KeithMyers, 2022-12-25:
I don't know how to get both Settings and Kernel data in the same terminal.  So here is two screenshots instead.

[image]

[image]

Still no memory controller working.

KeithMyers, 2022-12-25:
Where do I get the development version of the ISO?
I realized too late that the link I used last time was the stock ISO image as it didn't have the development naming scheme.

KeithMyers, 2022-12-25:
Well I finally figured out how to disable amd_pstate from the kernel commandline.  I am now running acpi_cpufreq scaling governor.  But it still does not enable the memory controller using your latest  dev commit using the ```0x40``` memory address register.

[image]

KeithMyers, 2022-12-25:
Congratz on figuring out the correct memory address for showing the proper DIMM size.  Here is a screenshot from the dev ISO.
[image]

KeithMyers, 2022-12-26:
You just have to download the Ubuntu Live USB image and burn a stick.  It won't have the same kernel I am currently trying though.  It will have the stock 5.10 kernel. Or it might not , I don't know if they have updated the image to include the 5.15 HWE kernel.  But they usually do that about 6 months after the initial release and the short-lived 22.10 release is already out.  That would have the 5.19 kernel in it.  So should duplicate exactly what  does not show the memory controller.

[https://releases.ubuntu.com/22.04.1/ubuntu-22.04.1-desktop-amd64.iso](url)

or the actual Ubuntu 22.10 ISO dekstop image.
[https://releases.ubuntu.com/kinetic/ubuntu-22.10-desktop-amd64.iso](url)

KeithMyers, 2022-12-26:
> > Well I finally figured out how to disable amd_pstate from the kernel commandline. I am now running acpi_cpufreq scaling governor. But it still does not enable the memory controller using your latest dev commit using the `0x40` memory address register.
> > [image]
> 
> Hard to diagnose this way. Do you know if I can download and boot from USB the same distribution, same kernel, including patches than yours ?
> 
> Meanwhile did you have a chance to try UMC using latest development ISO ?

Yes, I did the dev ISO boot in my previous post.

KeithMyers, 2022-12-26:
I haven't had k10temp loaded since I built the system since it is useless in providing any temps or sensors.

The edac_mce_amd is loaded by default and I can try unloading it.

Apparently I was still getting the trylock error when I was on the acpu_cpufreq scaling governor.

KeithMyers, 2022-12-26:
Still doesn't show the UMC with the ```edac_mce_amd``` module unloaded and removed.

KeithMyers, 2022-12-26:
I can try that tomorrow.  I also downloaded the Ubuntu 22.10 ISO image to try.  I haven't burned anything yet.
But that is for tomorrow.  Time for bed now.

KeithMyers, 2022-12-27:
I gave the Ubuntu 22.10 Live USB image a try.  Still installs the amd_pstate scaling driver by default.  So still no UMC.
You showed you had success with your screenshot, so my issue must be BIOS based? Or cpu based? 
We figured out if I blacklisted the amd_pstate scaling driver it would show the UMC.  But no easy, default configuration shows the UMC with a 7950X installed.  

[image]

KeithMyers, 2022-12-27:
That's the first thing I looked for in the BIOS.  But there is no CPPC toggle in the BIOS for this board.

If CPPC isn't enabled, then the amd_pstate driver can't be loaded as far as the docs tell me.

There is a CPPC toggle for my C7H boards but searching the BIOS with F9 and looking through every menu of the my X670E Hero board does not find any mention of CPPC

KeithMyers, 2022-12-27:
OK, I'll give that a try.  Strange on my C7H boards I have always left the PSS support enabled at its default.  I still am provided with a separate distinct CPPC enabled and preferred cores menu option with the C7H.

None of that is present in the X670E Hero board.  PSS is enabled by default on that board also.

I never upgraded to the X570 platform so am unfamiliar with its BIOS layout.

[58 more replies]

closed 2024-03-29 by cyring