cyring/CoreFreq #412: [AMD] Raphael HSMP and SVI3 addresses

mann1x, 2023-02-07:
Hi @cyring, any luck in accessing the HSMP mailbox and find the address for the SVI3 to read the VDDG voltages?

mann1x, 2023-02-07:
Thanks!

I don't have one as well but yes scanning was attempted with Ryzen SDT on Windows, no luck.
Seems some commands get a rejected cmd error and it's really weird... cause the TestMessage doesn't work.

Probably easier to win lotto than getting docs from AMD... 
I didn't realize at the time what they meant about "Open Firmware".
Seems we and AMD are from different universes :)

irusanov, 2023-02-07:
@cyring I've tried that long time ago, the 0x60000 - 0xfffff range is mostly 0xffffff
There are couple of small segments that have different values though, maybe worth checking - one of them starts at 0x00069000.
I didn't find the needed voltages on the usual 0x50000 - 0x5FFFFF range.

As for HSMP, it should have been the same as on Zen3, but a newer version with some additional commands.
AMD has provided some stripped-down PPR and the addresses are exactly the same, however the mailbox doesn't respond.
https://www.amd.com/en/support/tech-docs/processor-programming-reference-ppr-for-amd-family-19h-model-11h-revision-b1 (volume 3)

mann1x, 2023-03-03:
HSMP seems to be disabled sadly.
Maybe they will enable it in a future AGESA release but most likely is not going to happen.