cyring/CoreFreq #417: DKMS build failure on linux-cachyos-bore 6.2.0-3

gel-crabs, 2023-02-20:
Hey, getting a build failure with the new 6.2 release; the DKMS make log is attached below.

[make.log](https://github.com/cyring/CoreFreq/files/10787385/make.log)

gel-crabs, 2023-02-20:
No, it's actually amd_pstate:
https://github.com/CachyOS/kernel-patches/blob/master/6.2/0008-Implement-amd-pstate-epp-and-amd-pstate-guided-drive.patch

gel-crabs, 2023-02-20:
That fix works!

ptr1337, 2023-02-21:
@cyring 

I think you will have the problem anyways with 6.3 then. We just pulled the recent patches for the amd-pstate from upstream.
amd-pstate-guided will probably get merged with 6.4.

Actually these changes, which you wrote above will land into the 6.3 Kernel.
See here:
https://github.com/CachyOS/linux/commit/418cd789efe547057f7df184bcd641c0f3ae28a7
https://lore.kernel.org/lkml/CAJZ5v0iCmhEQSVZ8nKPZz63uQ+svRst_3N-WoxwOa617H4h9=w@mail.gmail.com/

So, further it need to be fixed anyways, as soon 6.3 rc1 will drop in. Thats a upstream change (soon)