cyring/CoreFreq #418: Loading corefreqk module utterly breaks my system (Kernel block layer?)

WildPenquin, 2023-02-22:
For some reason, loading the corefreqk utterly breaks my system after running with the module for a few hours. The system is stable (with uptimes up to several days) when not loading corefreqk module. I'm running on arch linux, on the -zen kernel branch, and I've installed corefreq with the help of AUR packages. I'm running on the B550 chipset (MSI Tomahawk) and 5950X

The symptoms include things which could all be explained by the block layer of the Kernel breaking down such that the Kernel can not anymore read or write on any block devices, including the root filesystem. Logs can not and will not be written to, journalctl can not be run to check the currently running logs. Running simple commands such as `cat`(from /bin/cat) will result in I/O errors or segmentation faults.

Yesterday, I loaded corefreqk at 17:16:23 and the system broke down at 23:15:00, as at that time the journal ceased to be written to. The system was in an almost unusable state as I've described in the previous paragraph.

I'm a bit reluctant in reproducing the bug since it will take 30 to 300 minutes to trigger, and as it involves the block layer, I fear it might result in data loss (though, so far, I haven't had any data loss because of this bug). But it is indeed the corefreqk module which makes the system unstable, and I can reproduce the bug (eventually, just not at will).

Any ideas as to how to monitor or get useful debug logs before reproducing (if needed)? Would ssh:ing to the computer *before* triggering and running `journalctl -f` (or similar) and saving the output on the other computer be a good idea?

WildPenquin, 2023-02-23:
Hi cyring, thanks for your reply!

One thing before I proceed: browsing previous bug reports I noticed there is the section [Software incompatibilities and workarounds](https://github.com/cyring/CoreFreq/wiki/Software-incompatibilities-and-workarounds). I had k10temp loaded at the same time. Do you think this could be the cause? I am/was using the sensors constantly, as I need that to adjust my cooling system.

Bu I also found an old bug report where it was claimed k10temp can now be used alongside corefreqk, but it's not in the wiki. Which is the current state of things?

cyring, 2024-04-01:
Hi,
Are you still facing a crash using latest `master` branch ?
Can you show screenshots if ok ?

WildPenquin, 2024-04-01:
Hi,

Given all the hassle I've given up on trying to use corefreqk. The last time I tried, yes, I did get the issue on all Arch Kernel flavors I tried; as I'm a bit time-constrained, I don't have the time to debug this in the near future.