cyring/CoreFreq #458: thin lto brakes corefreq

LethalManBoob, 2023-09-12:
cachyos-lto kernel causes this to happen someone reported they were unable to boot.
[image]

cyring, 2023-09-12:
Indeed I can see bad connection issues with the backend:
* SysGate is Off
* No Vcore, neither temperature reported
* Caches and probably others are not displayed

CachyOS was a best effort to support but this kernel mod is too patched for a long term programming environment. 
Thus your guys have to troubleshoot more than that  and will kind to provide the kernel log to investigate CachyOS Patches which may conflict with the Linux mainstream kernel.

LethalManBoob, 2023-09-12:
Turns out its both the RT patches and the LTO THIN brakes dkms
if that helps

LethalManBoob, 2023-09-12:
> @LethalManBoob you might have a better support if you were using any of the Arch Linux kernels from the repos. Can you try with the default Arch Linux real-time kernel to check if this is irelevant of CachyOS patches?
> 
> Otherwise you probably need to ask in CachyOS support or something.

It is capabale on the base aur linux-rt but not cachyos-rt

cyring, 2023-09-12:
@LethalManBoob 
Have a look to these previous solved issues:
* [Failure to build dkms modules](https://github.com/cyring/CoreFreq/issues/435#issuecomment-1536942645)
* [DKMS compilations fails on 6.3](https://github.com/cyring/CoreFreq/issues/422#issuecomment-1472724519)

csutcliff, 2023-09-13:
My experience is that it works fine with normal kernels (including linux-cachyos) and builds fine via DKMS for the LTO variants but as soon as I boot an LTO kernel with the corefreq module enabled I cannot even log in. SDDM launches but after entering my password I get about 20 seconds of black screen before being dumped back at SDDM. If I switch to another TTY and try to login to the console it hangs after I enter username and password, never proceeds to a shell prompt. As I can login via gui or console I don't know how to get logs.

If I disable the corefreq modules the LTO kernels otherwise run fine.

Has anyone got any ideas how to debug further?

gel-crabs, 2023-09-13:
I've been using thin LTO with CoreFreq for a long time with no issues, I'm currently on cachyos 6.6-rc1 built with Clang and thin LTO. I'm almost certain this is a bug which was reported previously. It is fixed by rebuilding your kernel with CONFIG_FORTIFY_SOURCE disabled.

ptr1337, 2023-09-13:
> I've been using thin LTO with CoreFreq for a long time with no issues, I'm currently on cachyos 6.6-rc1 built with Clang and thin LTO. I'm almost certain this is a bug which was reported previously. It is fixed by rebuilding your kernel with CONFIG_FORTIFY_SOURCE disabled.
> 
> See [#422 (comment)](https://github.com/cyring/CoreFreq/issues/422#issuecomment-1483790909)

This still seems not for me for a solution, to disable FORTIFY_SOURCE globally for our config, just because corefreq has issues with.
There should be further debugged, why FORTIFY_SOURCE in combination with clang is breaking it or FORTIFY_SOURCE should be completely disabled in the DKMS module, if clang is used.

Maybe reporting a bug to llvm/clang would be also worth to get some further information.
https://github.com/clangbuiltlinux/linux --> there should be quite easy to get some professional help for this issue, but not sure if they support out of tree modules.