cyring/CoreFreq #461: Alpine Linux

cyring, 2023-09-23:
> @cyring I see you already installed `gmake` but you also need to call it as such since `bmake` is the default `make` on chimera (which only imlements the POSIX make spec afaik)
> 
> Btw testing on https://alpinelinux.org/ may also help as it's using musl too

I believe I have tried all binaries coming with these 2 packages but none is working as expected from a  "standard" GNU `make`
I mean any conditional instruction within my `Makefile` like `ifeq` is not recognized!

_Originally posted by @cyring in https://github.com/cyring/CoreFreq/issues/460#issuecomment-1728166067_

cyring, 2023-09-23:
@JamiKettunen

Based on Virtual x86_64 [alpine-virt-3.18.3-x86_64.iso](https://alpinelinux.org/downloads/)

Host [Arch Linux `6.5.4`]
```
qemu-system-x86_64 -enable-kvm -cpu max,hv-runtime -m 2G -smp 4 -cdrom alpine-virt-3.18.3-x86_64.iso
```

Once VM booted, the following packages have been installed
```
apk add nano git make gcc musl-dev linux-headers
```

CoreFreq `Makefile` is also fixed for symbolic links
```
git clone -b develop https:\\github.com\cyring\CoreFreq.git
cd CoreFreq
nano Makefile
```
```
SYMLINK ?= ln -s
```

Now try to build.
You may have to force clean
```
rm -fr build
make
```
[image]

Missing kernel module: to be continued

JamiKettunen, 2023-09-23:
On top of using `ln -s` instead of making relative symlinks won't you have to fix the instances where `$(SYMLINK)` was used like from:
```
$(SYMLINK) Makefile $(BUILD)/Makefile
```
to:
```
$(SYMLINK) $(PWD)/Makefile $(BUILD)/Makefile
```
in https://github.com/cyring/CoreFreq/blob/develop/Makefile#L109, so that `$(BUILD)/Makefile` won't be a broken symlink?

cyring, 2023-09-23:
* Alpine is now installed to disk to ease Linux Kernel Headers
```sh
apk update
apk upgrade
apk add linux-virt-dev
reboot
```
* CoreFreq is now build and run from a standard user
```sh
cd ~/src/CoreFreq
nano Makefile
```
```
SYMLINK ?= ln -s

prepare:
        @if [ ! -e $(BUILD)/Makefile ]; then \
                $(SYMLINK) $(PWD)/Makefile $(BUILD)/Makefile; \
        fi
```
```
rm -fr build
make
```
[image]

cyring, 2023-09-23:
# [Custom Kernel](https://wiki.alpinelinux.org/wiki/Custom_Kernel#Installing)
```
apk add alpine-sdk
```
* Clone the [aports tree](https://wiki.alpinelinux.org/wiki/Aports_tree)
* Fix write mode of `/var/cache/distfiles/`
* Create User keygen
* Add User to group `abuild`
* Create a branch for custom kernel
```
abuild-keygen -a -i
cd aports
git checkout -b corefreq-kernel
cd main/linux-lts
cp -vi virt.x86_64.config .config
abuild -rk
```

cyring, 2023-09-24:
@JamiKettunen 
With your `Makefile` changes, can you now build and run _CoreFreq_ ?
Can you show the UI with its kernel window and the view in Tasks mode ?

cyring, 2023-11-15:
No more work

closed 2023-11-15 by cyring

cyring, 2023-12-17:
* New parallel build

[image]