cyring/CoreFreq #463: How can i save a configuration to a file and load it?

gundami, 2023-11-13:
I'm using corefreq to overclock my cpu, but i can't find any configuaration file. It's too troublesome to reconfigure every time I start up.
ps: corefreq lacks the ability to adjust the upper current limit, i'm using throttled to implement this function。

cyring, 2023-11-13:
> I'm using corefreq to overclock my cpu, but i can't find any configuaration file. It's too troublesome to reconfigure every time I start up. 

_CoreFreq_ is using kernel module parameters.
Those are usually stored in the common Linux path of `/etc/modprobe.conf`
You can also set parameters in the kernel boot command line.

All possible parameters are managed using standard commands. 
To list them:
```
modinfo /yourpathto/corefreqk.ko
```

> ps: corefreq lacks the ability to adjust the upper current limit, i'm using throttled to implement this function。

Do you know the Register for this **current** limit ?
At best, I'm providing the PL, Power Limiter.
You didn't show me your processor using `corefreq-cli -s`
Does it have PL ?

gundami, 2023-11-14:
> You can also set parameters in the kernel boot command line.

Thanks, i get it.

> Do you know the Register for this current limit ?
I'm  not sure, i find some code in the Throttled repo:
`def writemsr(msr, val):
    msr_list = ['/dev/cpu/{:d}/msr'.format(x) for x in range(cpu_count())]
    if not os.path.exists(msr_list[0]):
        try:
            subprocess.check_call(('modprobe', 'msr'))
        except subprocess.CalledProcessError:
            fatal('Unable to load the msr module.')
    try:
        for addr in msr_list:
            f = os.open(addr, os.O_WRONLY)
            os.lseek(f, MSR_DICT[msr], os.SEEK_SET)
            os.write(f, struct.pack('Q', val))
            os.close(f)
    except (IOError, OSError) as e:
        if TESTMSR:
            raise e
        if e.errno == EPERM or e.errno == EACCES:
            fatal(
                'Unable to write to MSR {} ({:x}). Try to disable Secure Boot '
                'and check if your kernel does not restrict access to MSR.'.format(msr, MSR_DICT[msr])
            )
        elif e.errno == EIO:
            fatal('Unable to write to MSR {} ({:x}). Unknown error.'.format(msr, MSR_DICT[msr]))
        else:
            raise e
`
On my cpu, if i don't increase the current limit, the cpu power will be limited at 100w, though i set PL to 180w.
And this is my processor info, my cpu is Q1HY, the engineering example of i9 13900H.
In my bios i set PL to 200w. The PL below is 500w, set by throttled, just in case.
`Processor                                                [Genuine Intel(R) 0000]
|- Architecture                                                  [Raptor Lake/P]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [0x0000410e]
|- Signature                                                           [  06_BA]
|- Stepping                                                            [      2]
|- Online CPU                                                          [ 20/ 20]
|- Base Clock                                                          [100.426]
|- Frequency            (MHz)                      Ratio                        
                 Min    401.70                    <   4 >                       
                 Max   2611.07                    <  26 >                       
|- Factory                                                             [100.000]
                       2600                       [  26 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   6929.39                    <  69 >                       
   |- HWP                                                                       
                 Min   6929.39                    <  69 >                       
                 Max   6929.39                    <  69 >                       
                 TGT      AUTO                    <   0 >                       
|- Turbo Boost                                                         [ UNLOCK]
                  1C   5423.00                    <  54 >                       
                  2C   5423.00                    <  54 >                       
                  3C   5423.00                    <  54 >                       
                  4C   5423.00                    <  54 >                       
                  5C   5222.15                    <  52 >                       
                  6C   5222.15                    <  52 >                       
                  7C   5222.15                    <  52 >                       
                  8C   5222.15                    <  52 >                       
|- Hybrid                                                              [ UNLOCK]
                  1C   4017.10                    <  40 >                       
                  2C   4017.10                    <  40 >                       
                  3C   3816.24                    <  38 >                       
                  4C   3816.24                    <  38 >                       
                  5C   3816.24                    <  38 >                       
[... 228 lines cut ...]
   |- Time Window                                                TW1   < 976 us>
|- Thermal Design Power                                       Uncore   <Disable>
   |- Power Limit                                                PL1   <    0 W>
   |- Time Window                                                TW1   < 976 us>
|- Thermal Design Power                                         DRAM   [Disable]
   |- Power Limit                                                PL1   [    0 W]
   |- Time Window                                                TW1   [ 976 us]
|- Thermal Design Power                                     Platform   < Enable>
   |- Power Limit                                                PL1   <  200 W>
   |- Time Window                                                TW1   <    1 s>
   |- Power Limit                                                PL2   <  200 W>
   |- Time Window                                                TW2   < 976 us>
|- Electrical Design Current                                     EDC   [Missing]
|- Thermal Design Current                                        TDC   [Missing]
[... cut ...]
|- Package Thermal Point                                                        
   |- DTS Threshold #1                                     Threshold   [Missing]
   |- DTS Threshold #2                                     Threshold   [Missing]
|- Units                                                                        
   |- Power                                               watt   [  0.125000000]
   |- Energy                                             joule   [  0.000061035]
   |- Window                                            second   [  0.000976562]
`

cyring, 2023-11-14:
> I'm not sure, i find some code in the Throttled repo:

Do you have any link to the source code of this repo ?

> `writemsr('MSR_OC_MAILBOX', write_value)`

This MSR Register seems interesting: we need its index value and its protocol specification.

cyring, 2023-11-14:
```
Technologies                                                                    
|- Data Cache Unit                                                              
   |- L1 Prefetcher                                                L1 HW   <OFF>
```

Did you disable this prefetcher on purpose or it's a miss reading of _CoreFreq_ ?

gundami, 2023-11-14:
> > I'm not sure, i find some code in the Throttled repo:
> 
> Do you have any link to the source code of this repo ?
> 
> > `writemsr('MSR_OC_MAILBOX', write_value)`
> 
> This MSR Register seems interesting: we need its index value and its protocol specification.

Here is the link: https://github.com/erpalma/throttled/blob/master/throttled.py

> ```
> Technologies
> |- Data Cache Unit
>    |- L1 [...]

I haven't change it, it seems been set off by default?

cyring, 2023-11-14:
> Here is the link: https://github.com/erpalma/throttled/blob/master/throttled.py
> 

I have a compatible `Nehalem` Xeon W3690 but unfortunately reading Mailbox MSR is not supported.
```
# rdmsr -a 0x150
rdmsr: CPU 0 cannot read MSR 0x00000150
```

cyring, 2023-11-14:
* If supported, you can also try TDC, parameter [`Custom_TDC_Offset`](https://github.com/cyring/CoreFreq/blob/150a2191f37c07f2d0e8024090a16efa72d9e5f5/corefreqk.c#L184)

[image]
[image]

gundami, 2023-11-15:
> * If supported, you can also try TDC, parameter [`Custom_TDC_Offset`](https://github.com/cyring/CoreFreq/blob/150a2191f37c07f2d0e8024090a16efa72d9e5f5/corefreqk.c#L184)
> 
> [image]
> 
> Here after Enabling the feature, I'm adding `3 amps`
> 
> [image]
> 
> You can check support of the [MSR_TURBO_POWER_CURRENT_LIMIT](https://github.com/cyring/CoreFreq/blob/150a2191f37c07f2d0e8024090a16efa72d9e5f5/intel_reg.h#L94) to alter amps. Try the following MSR read like my W3690 :
> 
> ```shell
> rdmsr -a 0x000001ac
> ```
> 
> ```
> 83700410
> 83700410
> 83700410
> 83700410
> 83700410
> 83700410
> 83700410
> 83700410
> 83700410
> 83700410
> 83700410
> 83700410
> ```
> 
> * You may also have a chance with `Thermal Offset` which is added or subtracted to the `TjMax`
> 
> [image]
> 
> It's an architectural register, try to test directly MSR_IA32_TEMPERATURE_TARGET:
> 
> ```
> rdmsr -a 0x000001a2
> ```
> 
> ```
> 651400
> 651400
> 651400
> 651400
> 651400
> 651400
> 651400
> 651400
> 651400
> 651400
> 651400
> 651400
> ```

Sadly, the TDC option on my cpu is missing and i also cannot read MSR_TURBO_POWER_CURRENT_LIMIT. Anyway, thanks for your help. If you would like to test MSR_OC_MAILBOX function, i'm glad to provide a ssh link to my platform for testing.

cyring, 2023-11-15:
https://github.com/cyring/CoreFreq/discussions/333#discussioncomment-2603364

This where we tried to program it

cyring, 2023-11-15:
13th Generation datasheet vol.2 specifies a bit more about BIOS MAILBOX MMIO Registers, although I have not found the possible commands

https://www.intel.com/content/www/us/en/content-details/743846/13th-generation-intel-core-processors-datasheet-volume-2-of-2.html