cyring/CoreFreq #467: AMD Hardware Prefetchers [Zen3 Zen4 EPYC TR]

cyring, 2023-12-01:
In this EPYC 9004 [Tuning Guide](https://www.amd.com/content/dam/amd/en/documents/epyc-technical-docs/tuning-guides/58006-epyc-9004-tg-kafka.pdf) are listed the following Prefetchers

[image]

Those are not programmed in _CoreFreq_ yet and I would like to add them in the Technologies window of the UI

----

The closest definition I've found is the MSR register `0xc0000108` within PPR datasheets:
* `#56713` Family 19h Model 61h 
* `#55898` Family 19h Model 01h

[image]

We are also reading this MSR prerequisite among the CPUID leaf `0x80000021`

[image]

_I have tried to read register from my Ryzen 9 3950X (Family 17h) and it is unsupported as expected._
```
rdmsr -ax 0xc0000108
rdmsr: CPU 31 cannot read MSR 0xc0000108
```

----

I'm calling owners of Zen3, Zen4, EPYC, Threadripper, APU processors to test this MSR and post the results as follow.

* Prerequisite
```sh
corefreq-cli -u | grep 80000021
```
```
|- 80000021:00000000    00000000     00000000     00000000     00000000         
...
|- 80000021:00000000    00000000     00000000     00000000     00000000
```

* Reading test
_please try reading MSR even if bit 13 of above CPUID is disable_
```sh
rdmsr -ax 0xc0000108
## if 0x0 returned, Prefetchers are enabled
```

* Writing test
```sh
wrmsr -a 0xc0000108 0x2f
```

* Verify test
```sh
rdmsr -ax 0xc0000108
## if 0x2f returned, Prefetchers are now disabled
```

Thanks for your help

KeithMyers, 2023-12-02:
Ryzen 5950X Ubuntu 22.04.3 LTS

$ ./corefreq-cli -u | grep 80000021
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000         
|- 80000021:00000000    0000004d     00000000     00000000     00000000 

$ sudo rdmsr -ax 0xc0000108
[sudo] password for keith: 
rdmsr: CPU 0 cannot read MSR 0xc0000108

$ sudo wrmsr -a 0xc0000108 0x2f
wrmsr: CPU 0 cannot set MSR 0xc0000108 to 0x000000000000002f

$ sudo rdmsr -ax 0xc0000108
rdmsr: CPU 0 cannot read MSR 0xc0000108

KeithMyers, 2023-12-02:
Ryzen 7950X Ubuntu 23.10

$ sudo ./corefreq-cli -u | grep 80000021
 
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         
|- 80000021:00000000    00062fcf     0000015c     00000000     00000000         

$ sudo rdmsr -ax 0xc0000108
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0

$ sudo wrmsr -a 0xc0000108 0x2f

$ sudo rdmsr -ax 0xc0000108
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f
2f

KeithMyers, 2023-12-02:
Epyc 7713 Ubuntu 22.04.3 LTS

$ ./corefreq-cli -u | grep 80000021
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
|- 80000021:00000000    0000204d     00000000     00000000     00000000         
[... cut ...]
[1993924.658227] R10: 00007ffee7fbcfac R11: b6859696ff574277 R12: 0000000000000000
[1993924.658228] R13: 00007ffee7fb6ec8 R14: 0000000000000003 R15: 00007ffee7fb6f88
[1993924.658231]  </TASK>
[1993924.658232] ================================================================================
[1994134.090761] msr: Write to unrecognized MSR 0xc0000108 by wrmsr (pid: 1664683).
[1994134.090768] msr: See https://git.kernel.org/pub/scm/linux/kernel/git/tip/tip.git/about for details.
[1994235.632844] CoreFreq: Unload

cyring, 2023-12-02:
So far what it is confirmed by the various architectures is that MSR Prefetch Control only exists if bit 13 of RAX register is set.

Hence CPUID, when last extended index is equal or greater than leaf `0xc0000108`, will be executed prior any MSR `0xc0000108` access.

cyring, 2023-12-02:
@KeithMyers 
Did the `Call Trace` happen while accessing MSR `0xc0000108` ?

KeithMyers, 2023-12-02:
> @KeithMyers Did the `Call Trace` happen while accessing MSR `0xc0000108` ?

I assume so, but I'd have to redo the test with both dmesg terminal and the test terminal open at the same time to verify.

The UBSAN stuff is not unusual for that host on that kernel and BIOS.

Have been told it is mainly benign and fixed in newer kernels.  That host is on the last 6.5-rc7 kernel from the mainline PPA and I can't move to newer kernels until I update to Ubuntu 23.10 because of insufficient glibc level.

KeithMyers, 2023-12-02:
So to the main question I have for this new feature . . .  of what benefit is achieved by toggling on the prefetchers?

KeithMyers, 2023-12-02:
updated to a new BIOS and looked to find all those cache prefetcher settings are available to control.  Default is AUTO which must mean Auto=OFF.

So can set them to ENABLE their also.  But being able to toggle them off via corefreq is useful because you can do it while the host is running your various test applications which you would run to see what if any benefit there is to enabling them.

KeithMyers, 2023-12-02:
Is 7zip benchmark your preferred benchmark? I normally use Geekbench to quickly evaluate changes to hardware and settings because of its varied test loads.

Then of course my final benchmark is my normal BOINC loads to see if the output from my projects goes up, down or no change over daily output.

cyring, 2023-12-02:
> I don't know if you have seen this document, but I quickly scanned through it for the reasons that enabling the prefetchers can be beneficial for some workloads. [https://jbk5155.github.io/publications/MICRO_2020.pdf[](url)](https://jbk5155.github.io/publications/MICRO_2020.pdf)
> 
> The conclusion was this:
> 
> "We experimented with a wide variety of suites covering Cloud, Server and SPEC CPU R 2017 suites."
> 
> "These optimizations combined improve performance by 5.3%, uop cache fetch ratio by 28.8% and dispatch bandwidth by 6.28%, while, reducing the decoder power consumption by 19.4% and branch misprediction latency by 5.23% in our workloads."

Nice reading.
I wish it would specify the low level things: Registers and so on -;)

----

Btw, Geekbench is fine. It just takes longer than 7zip to observe overall performance changes.

----

I have also read a X3D bit in the Milan EPYC documentation. I don't find more explanation but I presume it impacts that cache level enablement.

----

Could you please show me UI screenshots of the Technologies window facing the MSR prefecters ?
I would like to check if bits decoding is correct.

KeithMyers, 2023-12-02:
I had never run the 7zip benchmark before.  It is fast, much faster than Geekbench.  One of the tests in Geekbench is compression/decompression.

Sadly none of my BOINC apps do much decompressing and such.  Mostly FFT's, array sorts, integer type stuff.  The GB suite has more of that type of work in it.

I'll get to the testing later with some screenshots for you.

KeithMyers, 2023-12-02:
Ryzen 7950X Ubuntu 23.10 kernel 6.6.3

Surprised to find that default BIOS has both L1 and L2 Data prefetchers disabled.  Must mean Auto=disabled on the 7950X
The default on the Epyc 7713 is Auto=Enabled
Only the L1 IP prefetcher is enabled by default BIOS settings.

[image]

This is the 7zip benchmark at those default prefetcher settings.
AMD Ryzen 9 7950X 16-Core Processor             (A60F12)
CPU Freq: 32000000 64000000 64000000 64000000 128000000 128000000 256000000 - 2048000000

RAM size:   31796 MB,  # CPU hardware threads:  32
RAM usage:   7060 MB,  # Benchmark threads:     32

                       Compressing  |                  Decompressing
Dict     Speed Usage    R/U Rating  |      Speed Usage    R/U Rating
         KiB/s     %   MIPS   MIPS  |      KiB/s     %   MIPS   MIPS

22:     163611  3077   5172 159161  |    1781134  3026   5020 151893
23:     147821  2963   5082 150612  |    1751867  3071   4937 151591
24:     143255  2953   5216 154028  |    1703316  3040   4918 149505
25:     137982  2948   5344 157543  |    1687789  3070   4893 150202
----------------------------------  | ------------------------------
Avr:            2986   5203 155336  |             3052   4942 150798
Tot:            3019   5073 153067

This is the 7zip benchmark best run with a single prefetcher enabled.  The L1 Stride prefetcher was best.

AMD Ryzen 9 7950X 16-Core Processor             (A60F12)
CPU Freq: 64000000 64000000 64000000 64000000 64000000 - 512000000 1024000000 2048000000

RAM size:   31796 MB,  # CPU hardware threads:  32
RAM usage:   7060 MB,  # Benchmark threads:     32

                       Compressing  |                  Decompressing
Dict     Speed Usage    R/U Rating  |      Speed Usage    R/U Rating
         KiB/s     %   MIPS   MIPS  |      KiB/s     %   MIPS   MIPS

22:     179900  3047   5744 175007  |    2018729  3060   5626 172155
23:     166743  3007   5650 169891  |    1999016  3078   5621 172977
24:     159447  3018   5681 171438  |    1962560  3077   5598 172260
25:     152835  3008   5801 174502  |    1936729  3107   5548 172356
----------------------------------  | ------------------------------
Avr:            3020   5719 172710  |             3081   5598 172437
Tot:            3050   5658 172573

KeithMyers, 2023-12-02:
> > I don't know if you have seen this document, but I quickly scanned through it for the reasons that enabling the prefetchers can be beneficial for some workloads. [https://jbk5155.github.io/publications/MICRO_2020.pdf](https://jbk5155.github.io/publications/MICRO_2020.pdf)
> > The conclusion was this:
> > "We experimented with a wide variety of suites covering Cloud, Server and SPEC CPU R 2017 suites."
> > "These optimizations combined improve performance by 5.3%, uop cache fetch ratio by 28.8% and dispatch bandwidth by 6.28%, while, reducing the decoder power consumption by 19.4% and branch misprediction latency by 5.23% in our workloads."
> 
> Nice reading. I wish it would specify the low level things: Registers and so on -;)
> 
> Btw, Geekbench is fine. It just takes longer than 7zip to observe overall performance changes.
> 
> I have also read a X3D bit in the Milan EPYC documentation. I don't find more explanation but I presume it impacts that cache level enablement.
> 
> Using latest commit in branch `develop`, could you please show me UI screenshots of the Technologies window facing the MSR Prefetchers `0xc0000108` ? I would like to check if bits decoding is correct.

I'm not sure what you mean about a UI screenshot of Technologies window **_facing_** the MSR Prefetchers `0xc0000108` ??

I did run the `rdmsr -ax 0xc0000108` command and the reply was the `2f` that I had before.
Is that what you meant . .  that the registers were being set correctly?

KeithMyers, 2023-12-02:
Enabling the L1 Stride prefetcher definitely hurt both single core and multi-core GB6 scores.

The grey bars are the default baseline from my normal current settings.  The blue bars are from enabling the L1 Stride Prefetcher.  Only see a large improvement in the multi-core Asset Compression test.  Hurt every other test though a lot.

[image]

cyring, 2023-12-02:
> I did run the `rdmsr -ax 0xc0000108` command and the reply was the `2f` that I had before.
> Is that what you meant . . that the registers were being set correctly?

Yes exactly.
Because I don't have your processors I can't verify my programming of the features.

So this a great news: `0x2f`, in binary `101111`, means each bit set to one is a disabled prefetcher **as shown** in the UI

Still in the UI, can you toggle prefetchers individually as will ?
In general, do you confirm my code is working ?

KeithMyers, 2023-12-02:
Yes I toggled every one of the prefetchers individually and ran the 7zip benchmark to get readings.  Then I combined several of them in various combinations together and tested.  None of the combinations was better than a single alone.

And in general the default BIOS Auto setting produces the best GB benchmark.  Only in the L1 Stride p7zip benchmark alone did it really swamp out every other single or combo benchmark.

I can see that setting a prefetcher for a specific workload is possibly preferable.  But who only runs a single type of work that only has that one data type operation?

For general workloads, looks like the BIOS manufacturers set them up for general workloads.  At least for consumer Zen 4 Ryzen.

But on the Asrock Rack Epyc motherboard BIOS, you actually have specific workload tuning settings pre-configured from a picklist.  Very useful and time saving for commercial workloads of a single type.

cyring, 2023-12-02:
> Enabling the L1 Stride prefetcher definitely hurt both single core and multi-core GB6 scores.
>
> The grey bars are the default baseline from my normal current settings. The blue bars are from enabling [...]

Awesome tests.
Let's hope some Users don't have Prefetchers wrongly defined by BIOS!

cyring, 2023-12-02:
> Yes I toggled every one of the prefetchers individually and ran the 7zip benchmark to get readings. Then I combined several of them in various combinations together and tested. None of the [...]

True.
However no general use case according to AMD.
[Chapter 5.1](https://www.amd.com/content/dam/amd/en/documents/epyc-technical-docs/tuning-guides/58006-epyc-9004-tg-kafka.pdf)
[image]

I had also [experimented](https://gist.github.com/cyring/be73a8afee6983bf39fb96c287f646d7) XMRig  benchmark where performance appeared better with `L2 Prefetcher` disabled.
```
Technologies                                                                    
|- Instruction Cache Unit                                                       
   |- L1 IP Prefetcher                                          L1 HW IP   < ON>
|- Data Cache Unit                                                              
   |- L1 Prefetcher                                                L1 HW   < ON>
   |- L2 Prefetcher                                                L2 HW   <OFF>
```

KeithMyers, 2023-12-02:
Revisited the Data Cache L1/L2 prefetchers setting Enabled in the BIOS.  Now running my BOINC workload and will see if it helps, hurts or does nothing for my daily host production.  Will run it for 24 hours and evaluate.

cyring, 2023-12-02:
> Revisited the Data Cache L1/L2 prefetchers setting Enabled in the BIOS. Now running my BOINC workload and will see if it helps, hurts or does nothing for my daily host production. Will run it for 24 hours and evaluate.

Let's hope it will help

Also if not done yet, try to improve BOING with [`1GB Pages`](https://wiki.debian.org/Hugepages#x86_64) set for your kernel

cyring, 2023-12-02:
> I assume so, but I'd have to redo the test with both dmesg terminal and the test terminal open at the same time to verify.
>
> The UBSAN stuff is not unusual for that host on that kernel and BIOS. [...]

Oh thanks for this.
I have never use [UBSAN](https://docs.kernel.org/dev-tools/ubsan.html)

cyring, 2023-12-03:
> This was observed in dmesg.
> 
> ```
> ================================================================================
> [1993924.658030] UBSAN: array-index-out-of-bounds in /home/keith/Downloads/CoreFreq-master/corefreqk.c:13392:3
> [1993924.658033] index 2335 is out of range for type 'TASK_MCB [2335]'
> ...
> [1993924.658232] ================================================================================
> ```

Thanks!
A fix is committed in `develop` branch @ 0829f74f6e0e74113abe25e7e31d8cae0165231c

cyring, 2024-01-18:
Pre-release available for features validation 
https://github.com/cyring/CoreFreq/discussions/472

closed 2024-01-18 by cyring