cyring/CoreFreq #469: Cannot compile on Almalinux 9.3

fishbonesnl, 2023-12-05:
Hi,

I am getting a compilation error on Almalinux 9.3 with the latest version of Corefreq.

[user@machine CoreFreq] make
make -j1 -C /lib/modules/5.14.0-362.8.1.el9_3.x86_64/build M=/home/user/CoreFreq modules
make[1]: Entering directory '/usr/src/kernels/5.14.0-362.8.1.el9_3.x86_64'
  CC [M]  /home/user/CoreFreq/corefreqk.o
/home/user/CoreFreq/corefreqk.c: In function ‘CoreFreqK_FreqDriver_UnInit’:
/home/user/CoreFreq/corefreqk.c:20729:12: error: void value not ignored as it ought to be
20729 |         rc =
      |            ^
compilation terminated due to -Wfatal-errors.
make[2]: *** [scripts/Makefile.build:299: /home/user/CoreFreq/corefreqk.o] Error 1
make[1]: *** [Makefile:1928: /home/user/CoreFreq] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.14.0-362.8.1.el9_3.x86_64'
make: *** [Makefile:86: all] Error 2

All the relevant kernel headers are installed. 

[user@machine CoreFreq] gcc --version
gcc (GCC) 11.4.1 20230605 (Red Hat 11.4.1-2)

What do I do?

cyring, 2023-12-05:
To make AlmaLinux compatible with _CoreFreq_, change its source code with your kernel version at this line:

https://github.com/cyring/CoreFreq/blob/150a2191f37c07f2d0e8024090a16efa72d9e5f5/corefreqk.c#L20728

Like:

```
KERNEL_VERSION(5, 14, 0)
```

Or more appropriated to RHEL, something like:

```
 #if ((LINUX_VERSION_CODE < KERNEL_VERSION(6, 3, 0)) && (!defined(CONFIG_CACHY))) || (RHEL_MAJOR == 11)
```

cyring, 2023-12-05:
You wiil also change version at line 22696:

[image]

cyring, 2023-12-06:
Fix is available on development branch

```
git clone -b develop https://github.com/CyrIng/CoreFreq.git
make clean all
```

It has been compiled against:
* AlmaLinux 9
* AlmaLinux 8
* As regression tests: CentOS, Fedora, Debian, OpenSUSE, Arch, ClearLinux

cyring, 2023-12-13:
Hello,

Are we closing this issue ?

fishbonesnl, 2023-12-13:
Yes, sorry for the late response. Please close it. I got it to work.

closed 2023-12-13 by cyring