cyring/CoreFreq #470: Hardware Prefetch for Atom E-Cores

cyring, 2023-12-13:
In this Intel whitepaper [`357930-001US`](https://t.co/udfFob05be) we are reading new MSR dedicated to Atom _sub-architecture_ within hybrid processors.

Feature|Register
-------|--------
`LLC Streamer`|`0x1320`
`L2 NLP`|`0x1321`
`SELECTION`|`0x1323`

Whereas the known MSR `0x1A4` is said to be available for both P-Cores and E-Cores

----

I'll appreciate if one can `rdmsr` the new MSR on E-Core CPU numbers.

Next, still on E-Cores, toggle the **Enable** bit using a `wrmsr ; rdmsr` modification sequence.

It would be also interesting to check the specs:

* try a read access of those Atom MSR on P-Cores
* verify the scope of writing: MSR should be **module** scope: flipping a MSR bit on an unique E-Core should be reflected on the others. But what is the module cluster formed with ? 

----

Thanks for helping.

cyring, 2023-12-13:
<details>
<summary>contributors</summary>
@gundami @Technologicat @justanerd @BugReporterZ @svmlegacy @kocoman1 @jowa2021 @Betaminos @huajian628 @vitaly-zdanevich @rushvora
</details> 

Can your guys please help to investigate those registers on Hybrid architecture ?

BugReporterZ, 2023-12-13:
I tried this.

```
#rdmsr -a 0x1321
CPU 19: 250122000001
CPU 18: 250122000001
CPU 17: 250122000001
CPU 16: 250122000001
rdmsr: CPU 15 cannot read MSR 0x00001321

#rdmsr -a 0x1323
CPU 19: 1f9cc00000000
CPU 18: 1f9cc00000000
CPU 17: 1f9cc00000000
CPU 16: 1f9cc00000000
rdmsr: CPU 15 cannot read MSR 0x00001323

===============================================

#rdmsr -a 0x1320
CPU 19: 100007e041000004
CPU 18: 100007e041000004
CPU 17: 100007e041000004
CPU 16: 100007e041000004
rdmsr: CPU 15 cannot read MSR 0x00001320

#wrmsr -p 19 0x1320 4
#rdmsr -a -x -0 0x1320
CPU 19: 0000000000000004
CPU 18: 0000000000000004
CPU 17: 0000000000000004
CPU 16: 0000000000000004
rdmsr: CPU 15 cannot read MSR 0x00001320

#wrmsr -p 19 0x1320 1152930164351434756
#rdmsr -a -x -0 0x1320
CPU 19: 100007e041000004
CPU 18: 100007e041000004
CPU 17: 100007e041000004
CPU 16: 100007e041000004
rdmsr: CPU 15 cannot read MSR 0x00001320
```

cyring, 2023-12-13:
> I tried this.

Excellent.
I can code now.
Thank you

cyring, 2023-12-15:
* ROG STRIX Z790 Series BIOS Manual

```console
Hardware Prefetcher
Allows you to enable or disable the MLC streamer prefetcher.
Configuration options: [Disabled] [Enabled]

Adjacent Cache Line Prefetch
Allows you to prefetch adjacent cache lines, reducing the DRAM loading time and improving 
the system performance.
Configuration options: [Disabled] [Enabled]
```

* SUPERMICRO C9Z790 USER MANUAL
```console
Hardware Prefetcher
If set to Enabled, the hardware prefetcher will prefetch streams of data and instructions 
from the main memory to the L2 cache to improve CPU performance. The options are 
Disabled and Enabled.

Adjacent Cache Line Prefetch 
Select Enabled for the CPU to prefetch both cache lines for 128 bytes as comprised. Select 
Disabled for the CPU to prefetch both cache lines for 64 bytes. The options are Disabled 
and Enabled.
```

* ASRock Intel Z790 Motherboard
`<same as above>`

cyring, 2023-12-16:
* MS03-6L0
```console
Enable LP [Global]
Enables Logical processor (Software Method to Enable/Disable Logical 
Processor threads).
Options available: ALL LPs, Single LP. Default setting is ALL LPs.
Hardware Prefetcher Select whether to enable the speculative prefetch unit of the processor.
Options available: Enable, Disable. Default setting is Enable.
L2 RF0 Prefetch Disable Options available: Enable, Disable. Default setting is Disable.
Adjacent Cache Prefetch
When enabled, cache lines are fetched in pairs. When disabled, only 
the required cache line is fetched. 
Options available: Enable, Disable. Default setting is Enable.
DCU Streamer Prefetcher Enable/Disable DCU streamer prefetcher.
Options available: Enable, Disable. Default setting is Enable.
DCU IP Prefetcher
Enable/Disable DCU IP Prefetcher.
Options available: Enable, Disable. Default setting is Enable.
```

cyring, 2023-12-16:
> The Intel Atom cores are placed in a group of four per module with private L1 caches for each core.

cyring, 2023-12-16:
`+` @BugReporterZ : Hello,

[image]

Above is the last commit 3ed6baecd767b6d3dec2159d812f2070e88e789c which adds DCU L1 NLP, a bit of MSR `0x1A4` unveiled in the Whitepaper.

cyring, 2023-12-18:
In ["Hardware LLC prefetch feature 
on 4th Gen Intel® Xeon®
Scalable Processor (Codename 
Sapphire Rapids)"](https://www.google.com/url?sa=t&source=web&rct=j&opi=89978449&url=https://cdrdv2-public.intel.com/780991/780991-Hardware_LLC_prefetch__4th_Gen_Intel_Xeon_Scalable_Processor-R1_0.pdf&ved=2ahUKEwjBya6ayJmDAxWXcKQEHRgJBNMQFnoECCIQAQ&usg=AOvVaw0n-U2vauisbiiRIH7ht0ri) we are reading `L3 prefetch disable` at bit `42` of MSR `0x6D`

Does it work on Desktop or Mobile processors ?

cyring, 2024-01-18:
Pre-release in progress
https://github.com/cyring/CoreFreq/discussions/472
If things are missing, please let me know.

closed 2024-01-18 by cyring