cyring/CoreFreq #473: Clarification on Intel Prerequisites: What's Being Built by `make`?

matte21, 2024-01-23:
Hello,
sorry if it's an obvious question, I'm inexperienced with the Linux kernel.

In the README, the prerequisites section "a - intel only" states:

> a- Intel only: For a better accuracy, disable the Kernel NMI Watchdog
> 
> Add the below [parameter](https://github.com/torvalds/linux/blob/master/Documentation/admin-guide/kernel-parameters.txt) in the kernel boot loader { [Grub](https://www.gnu.org/software/grub/manual/grub/grub.html#GNU_002fLinux), [SysLinux](https://wiki.syslinux.org/wiki/index.php?title=Config#APPEND) } ...
> 
> ```
> nmi_watchdog=0
> ```
> 
> ... and build with the fixed performance counters
> 
> ```
> make MSR_CORE_PERF_UC=MSR_CORE_PERF_FIXED_CTR1 MSR_CORE_PERF_URC=MSR_CORE_PERF_FIXED_CTR2
> ```

Regarding the last sentence/command (i.e. `make`), what is being built? The Linux kernel or this project?

cyring, 2024-01-23:
> The Linux kernel or this project?

Only the _CoreFreq_ project is built

Because kernel is consuming one of the 3 PMU fixed counters within Intel hardware, `nmi_watchdog=0` prevents it to do so, and let the counter free to the usage of other agents.

However, since my initial R&D I have switched to the two recommended APERF/MPERF counters for measurement purposes.
Thus it should be no more required to disable nmi_watchdog.
You simply build with `make` without any other arguments.

cyring, 2024-01-25:
Is the answer ok for you ?

matte21, 2024-01-25:
Yes, thank you so much for the reply.

closed 2024-01-25 by matte21