cyring/CoreFreq #477: Unraid crash on installation - Intel(R) Celeron(R) CPU J3455

ich777, 2024-02-20:
I've link two posts from the Unraid forums [here](https://forums.unraid.net/topic/109314-plugin-corefreq/?do=findComment&comment=1373317) and [here](https://forums.unraid.net/topic/109314-plugin-corefreq/?do=findComment&comment=1373325) (same user).

cyring, 2024-02-20:
[image]

As specified by the [ARK](https://ark.intel.com/content/www/us/en/ark/products/95594/intel-celeron-processor-j3455-2m-cache-up-to-2-3-ghz.html) for Celeron J3455, if processor is not capable of Intel Turbo Boost Technology, there is **no reason** to trigger MSR Turbo Activation Ratio

ich777, 2024-02-20:
This is `lscpu` from the `Diagnostics.zip`: [lscpu.txt](https://github.com/cyring/CoreFreq/files/14351028/lscpu.txt)

cyring, 2024-02-20:
> This is `lscpu` from the `Diagnostics.zip`: [lscpu.txt](https://github.com/cyring/CoreFreq/files/14351028/lscpu.txt)

Thank you.

Fix commit 91ed011ef1f99b416b57735ddcf8fc66a35aea7f is ready in branch `develop`.

Can you please provide a testing version to User ?

ich777, 2024-02-21:
I've compiled the package and let the user know [here](https://forums.unraid.net/topic/109314-plugin-corefreq/?do=findComment&comment=1374508).

ich777, 2024-02-21:
See his comment [here](https://forums.unraid.net/topic/109314-plugin-corefreq/?do=findComment&comment=1374676).

BTW most Unraid users don‘t have a GitHub account.

cyring, 2024-02-21:
> See his comment [here](https://forums.unraid.net/topic/109314-plugin-corefreq/?do=findComment&comment=1374676).
> 
> BTW most Unraid users don‘t have a GitHub account.

For another test, commenting the crashing instructions in this archive
[CoreFreq_Goldmont.tar.gz](https://github.com/cyring/CoreFreq/files/14361584/CoreFreq_Goldmont.tar.gz)

ich777, 2024-02-21:
I'll build him this version and post it on the forums.

cyring, 2024-02-21:
Programming notes:
1. Turbo Boost Technology
* Checking CPUID leaf 0x6 for IDA is not enough to guess Turbo capability
  That bit can be enabled because of LFM support
* If LFM then we should read MSR IA32_MISC_ENABLE (`0x1A0`) to check the status of bit `38`
[image]
[image]
* If bit value is one then don't call MSR_TURBO_ACTIVATION_RATIO

2. DMI string
* If testing turbo is not enough to avoid a crash; my assumption is that MSR_TURBO_ACTIVATION_RATIO is triggering a firmware implementation which is not encoded or activated into QNAP.
* I will then fetch for the Manufacturer string and blacklist MSR_TURBO_ACTIVATION_RATIO conditionally

3. Remaining question
* Does a Goldmont exist with MSR_TURBO_ACTIVATION_RATIO support ?
* I don't have received a _CoreFreq_ execution report of another Goldmont yet ; of architecture `06_5c` ; running on desktop motherboard. Anyone can provide this ?

ich777, 2024-02-21:
Package built and upload to the [forums](https://forums.unraid.net/topic/109314-plugin-corefreq/?do=findComment&comment=1374870).

cyring, 2024-02-21:
> Package built and upload to the [forums](https://forums.unraid.net/topic/109314-plugin-corefreq/?do=findComment&comment=1374870).

Thank you very much

cyring, 2024-02-21:
```console
Processor                              [Intel(R) Celeron(R) CPU J3455 @ 1.50GHz]
|- Architecture                                                  [Atom/Goldmont]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [0x00000048]
|- Signature                                                           [  06_5C]
|- Stepping                                                            [      9]
|- Online CPU                                                          [  4/  4]
|- Base Clock                                                          [ 99.842]
|- Frequency            (MHz)                      Ratio                        
                 Min    798.73                    <   8 >                       
                 Max   1497.62                    <  15 >                       
|- Factory                                                             [100.000]
                       1500                       [  15 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   2296.20                    <  23 >                       
|- Turbo Boost                                                         [   LOCK]
                  1C   2296.36                    <  23 >                       
                  2C   2196.51                    <  22 >                       
                  3C   2196.51                    <  22 >                       
                  4C   2196.51                    <  22 >                       
|- Uncore                                                              [   LOCK]
|- TDP                                                           Level [  0:0  ]
   |- Programmable                                                     [   LOCK]
   |- Configuration                                                    [   LOCK]
   |- Turbo Activation                                                 [   LOCK]
               Turbo      AUTO                    [   0 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [N]          AES [Y]  AVX/AVX2 [N/N] 
|- AMX-BF16     [N]     AMX-TILE [N]     AMX-INT8 [N]    AMX-FP16 [N] 
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N] 
|- AVX512-VBMI  [N] AVX512-VBMI2 [N]  AVX512-VNNI [N]  AVX512-ALG [N] 
|- AVX512-VPOP  [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [N] AVX-VNNI-VEX [N] AVX-VNN-INT8 [N] AVX-NE-CONV [N] 
|- AVX-IFMA     [N]    CMPccXADD [N]      MOVDIRI [N]   MOVDIR64B [N] 
|- BMI1/BMI2  [N/N]         CLWB [N]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [N]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- ENQCMD       [N]         GFNI [N]        OSPKE [N]     WAITPKG [N] 
|- MMX/Ext    [Y/N] MON/MWAITX [N/N]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/N]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [N]         SGX [N] 
|- VAES         [N]   VPCLMULQDQ [N]   PREFETCH/W [Y]       LZCNT [N] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Missing]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Capable]
|- CPL Qualified Debug Store                                  DS-CPL   [Capable]
|- 64-Bit Debug Store                                         DTES64   [Capable]
|- Fast Short REP CMPSB                                         FSRC   [Missing]
|- Fast Short REP MOVSB                                         FSRM   [Missing]
|- Fast Short REP STOSB                                         FSRS   [Missing]
|- Fast Zero-length REP MOVSB                                   FZRM   [Missing]
|- Fast-String Operation                                        ERMS   [Capable]
|- Fused Multiply Add                                            FMA   [Missing]
|- Flexible Return and Event Delivery                           FRED   [Missing]
|- Hardware Feedback Interface                                   HFI   [Missing]
|- Hardware Lock Elision                                         HLE   [Missing]
[... 611 lines cut ...]
[... cut ...]
|- 40000002:00000000    00000003     000000ea     0124f800     00000000         
|- 40000003:00000000    00000003     000000ea     0124f800     00000000         
|- 40000004:00000000    00000003     000000ea     0124f800     00000000         
|- 40000005:00000000    00000003     000000ea     0124f800     00000000         
|- 40000006:00000000    00000003     000000ea     0124f800     00000000
```
* Good news, it works without accessing MSR TURBO_ACTIVATION_RATIO
* Bad news, platform has a full enabled Turbo Boost
* B plan might be the DMI string condition

cyring, 2024-02-21:
```console
Processor                              [Intel(R) Celeron(R) CPU J3455 @ 1.50GHz]
|- Architecture                                                  [Atom/Goldmont]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [0x00000048]
|- Signature                                                           [  06_5C]
|- Stepping                                                            [      9]
|- Online CPU                                                          [  4/  4]
|- Base Clock                                                          [ 99.842]
|- Frequency            (MHz)                      Ratio                        
                 Min    798.73                    <   8 >                       
                 Max   1497.62                    <  15 >                       
|- Factory                                                             [100.000]
                       1500                       [  15 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   2296.20                    <  23 >                       
|- Turbo Boost                                                         [   LOCK]
                  1C   2296.36                    <  23 >                       
                  2C   2196.51                    <  22 >                       
                  3C   2196.51                    <  22 >                       
                  4C   2196.51                    <  22 >                       
|- Uncore                                                              [   LOCK]
|- TDP                                                           Level [  0:0  ]
   |- Programmable                                                     [   LOCK]
   |- Configuration                                                    [   LOCK]
   |- Turbo Activation                                                 [   LOCK]
               Turbo      AUTO                    [   0 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [N]          AES [Y]  AVX/AVX2 [N/N] 
|- AMX-BF16     [N]     AMX-TILE [N]     AMX-INT8 [N]    AMX-FP16 [N] 
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N] 
|- AVX512-VBMI  [N] AVX512-VBMI2 [N]  AVX512-VNNI [N]  AVX512-ALG [N] 
|- AVX512-VPOP  [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [N] AVX-VNNI-VEX [N] AVX-VNN-INT8 [N] AVX-NE-CONV [N] 
|- AVX-IFMA     [N]    CMPccXADD [N]      MOVDIRI [N]   MOVDIR64B [N] 
|- BMI1/BMI2  [N/N]         CLWB [N]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [N]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- ENQCMD       [N]         GFNI [N]        OSPKE [N]     WAITPKG [N] 
|- MMX/Ext    [Y/N] MON/MWAITX [N/N]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/N]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [N]         SGX [N] 
|- VAES         [N]   VPCLMULQDQ [N]   PREFETCH/W [Y]       LZCNT [N] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Missing]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Capable]
|- CPL Qualified Debug Store                                  DS-CPL   [Capable]
|- 64-Bit Debug Store                                         DTES64   [Capable]
|- Fast Short REP CMPSB                                         FSRC   [Missing]
|- Fast Short REP MOVSB                                         FSRM   [Missing]
|- Fast Short REP STOSB                                         FSRS   [Missing]
|- Fast Zero-length REP MOVSB                                   FZRM   [Missing]
|- Fast-String Operation                                        ERMS   [Capable]
|- Fused Multiply Add                                            FMA   [Missing]
|- Flexible Return and Event Delivery                           FRED   [Missing]
|- Hardware Feedback Interface                                   HFI   [Missing]
|- Hardware Lock Elision                                         HLE   [Missing]
[... 199 lines cut ...]
[... cut ...]
CPU Pkg  Apic  Core/Thread  Caches      (w)rite-Back (i)nclusive              
 #   ID   ID    ID     ID  L1-Inst Way  L1-Data Way      L2  Way      L3  Way 
000:BSP    0     0      0    32768  8w    24576  6w  1048576 16w        0  0  
001:  0    2     1      0    32768  8w    24576  6w  1048576 16w        0  0  
002:  0    4     2      0    32768  8w    24576  6w  1048576 16w        0  0  
003:  0    6     3      0    32768  8w    24576  6w  1048576 16w        0  0  

                            GenuineIntel  [   0]                           
```

cyring, 2024-02-21:
```console
[ 0] American Megatrends Inc.                                                   
[ 1] QY47AR54                                                                   
[ 2] 09/21/2017                                                                 
[ 3] Default string                                                             
[ 4] Default string                                                             
[ 5] Default string                                                             
[ 6] D---u---s---n-                                                             
[ 7] Default string                                                             
[ 8] Default string                                                             
[ 9] AMI Corporation                                                            
[10] Aptio CRB                                                                  
[11] Default string                                                             
[12] D---u---s---n-                                                             
[13] Number Of Devices:2\Maximum Capacity:8388608 kilobytes                     
[14] A1_DIMM0\A1_BANK0                                                          
[15] A1_DIMM1\A1_BANK1                                                          
[16]                                                                            
[17]                                                                            
[18] Kingston                                                                   
[19] Kingston                                                                   
[20]                                                                            
[21]                                                                            
[22] A1_AssetTagNum0                                                            
[23] A1_AssetTagNum1                                                            
[24]                                                                            
[25]                                                                            
```

cyring, 2024-02-23:
@ich777 Hello,

Can you please make a plugin package for User based on the following archive:

[CoreFreq_Goldmont_IMC.tar.gz](https://github.com/cyring/CoreFreq/files/14385261/CoreFreq_Goldmont_IMC.tar.gz)

For a future decoder, this will attempt to map the memory controller and dump a range of registers.

```console
--- DEVICE  (DUMP) ---
...
--- MCHBAR (START) ---
...
--- MCHBAR  (STOP) ---
```

Thank you

ich777, 2024-02-23:
Sure, it will take me a few hours.

closed 2024-02-28 by cyring

ich777, 2024-02-28:
@cyring can you give me a little heads up when you release the next version from CoreFreq so that I can compile a new package for Unraid based on the new release please...?

cyring, 2024-02-28:
> @cyring can you give me a little heads up when you release the next version from CoreFreq so that I can compile a new package for Unraid based on the new release please...?

Sure. I will ask User if latest code is stable ?

cyring, 2024-02-28:
@ich777 According to User, J3455 is now working fine. So we can proceed with a plugin creation.
Regards

ich777, 2024-02-28:
@cyring do you still push Releases here to GitHub or at least Tags since my build toolchain is based on Tags. All packages for Unraid are based on CoreFreq [1.96.5](https://github.com/cyring/CoreFreq/releases/tag/1.96.5) currently.
I can of course change that to always use the `master` branch.