cyring/CoreFreq #478: 1C not changing. Xeon E5 1650 v3

UleyeyeUwma, 2024-02-21:
TGT is working fine, I am able to change the values there with no issues, however I am unable to change any of the Turbo Boost C values.

I followed some issues here and the answer seemed to be to raise 1C first, but even if it gives me the option to do so and it says Turbo Boost [UNLOCK] the value never changes, so the tb frequency isn't applied.

I enabled experimental (I don't think it's needed or even does anything in my case) and messed with pretty much every value under the processor window, and it seems to do nothing.

UleyeyeUwma, 2024-02-21:
```
Processor                            [Intel(R) Xeon(R) CPU E5-1650 v3 @ 3.50GHz]
|- PPIN#                                                      [82ef7e64ef306c8d]
|- Architecture                                                     [Haswell/EP]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [0x00000000]
|- Signature                                                           [  06_3F]
|- Stepping                                                            [      2]
|- Online CPU                                                          [ 12/ 12]
|- Base Clock                                                          [ 99.776]
|- Frequency            (MHz)                      Ratio                        
                 Min   1197.30                    <  12 >                       
                 Max   3492.11                    <  35 >                       
|- Factory                                                             [100.000]
                       3500                       [  35 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   4090.76                    <  41 >                       
|- Turbo Boost                                                         [ UNLOCK]
                  1C   3791.44                    <  38 >                       
                  2C   3791.44                    <  38 >                       
                  3C   3591.89                    <  36 >                       
                  4C   3591.89                    <  36 >                       
                  5C   3591.89                    <  36 >                       
                  6C   3591.89                    <  36 >                       
                  7C   3591.89                    <  36 >                       
                  8C   3591.89                    <  36 >                       
                  9C   3591.89                    <  36 >                       
                 10C   3591.89                    <  36 >                       
                 11C   3591.89                    <  36 >                       
                 12C   3591.89                    <  36 >                       
                 13C   3591.89                    <  36 >                       
                 14C   3591.89                    <  36 >                       
                 15C   3591.89                    <  36 >                       
                 16C   3591.89                    <  36 >                       
|- Uncore                                                              [ UNLOCK]
                 Min   1197.30                    <  12 >                       
                 Max   1596.39                    <  16 >                       
|- TDP                                                           Level <  0:1  >
   |- Programmable                                                     [ UNLOCK]
   |- Configuration                                                    [ UNLOCK]
   |- Turbo Activation                                                 [ UNLOCK]
             Nominal   3492.11                    [  35 ]                       
              Level1   3192.79                    [  32 ]                       
               Turbo   4090.76                    <  41 >                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [N]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AMX-BF16     [N]     AMX-TILE [N]     AMX-INT8 [N]    AMX-FP16 [N] 
|- AVX512-F     [N]    AVX512-DQ [N]  AVX512-IFMA [N]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [N]    AVX512-BW [N]   AVX512-VL [N] 
|- AVX512-VBMI  [N] AVX512-VBMI2 [N]  AVX512-VNNI [N]  AVX512-ALG [N] 
|- AVX512-VPOP  [N] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [N] AVX-VNNI-VEX [N] AVX-VNN-INT8 [N] AVX-NE-CONV [N] 
|- AVX-IFMA     [N]    CMPccXADD [N]      MOVDIRI [N]   MOVDIR64B [N] 
|- BMI1/BMI2  [Y/Y]         CLWB [N]      CLFLUSH [Y] CLFLUSH-OPT [N] 
|- CLAC-STAC    [N]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- ENQCMD       [N]         GFNI [N]        OSPKE [N]     WAITPKG [N] 
|- MMX/Ext    [Y/N] MON/MWAITX [Y/N]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [N]      RDTSCP [Y] 
|- SEP          [Y]          SHA [N]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/N]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [N]         SGX [N] 
|- VAES         [N]   VPCLMULQDQ [N]   PREFETCH/W [N]       LZCNT [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
[... 254 lines cut ...]

[... cut ...]
|- Idle Limit                                                 <              C6>
   |- State        POLL      C1     C1E      C3      C6      C7      C8      C9 
   |-           CPUIDLE  HSW-C1 HSW-C1E  HSW-C3  HSW-C6  HSW-C7  HSW-C8  HSW-C9 
   |- Power          -1       0       0       0       0       0       0       0 
   |- Latency         0       2      10      33     133     166     300     600 
   |- Residency       0       2      20     100     400     500     900    1800

```
Here is the output of `corefreq-cli -s -n -m -n -B -n -k`

cyring, 2024-02-21:
1. Do you confirm you have change ratio for all Cores ?
I mean did you make sure to select line `1C`

[image]

----

2. Are the MSR really unlocked ?

Let's start with the MSR_TURBO_RATIO_LIMIT (`0x1ad`)
You can alter register for one frequency ratio using the MSR-Tools
```
rdmsr -ax 0x1ad
```
```
2424242424242626
...
2424242424242626
```
Now add `+1` to `26` and write the register for all Cores
```
wrmsr -a 0x1ad 0x2424242424242627
```
Read again the register to check if value is written. If not MSR is **locked**.
```
rdmsr -ax 0x1ad
```
```
2424242424242627
...
2424242424242627
```
If value has been successfully altered then you should see a TB ratio of `39` in _CoreFreq_. (press star key `*` to force refresh the UI)

At this point let me know how does it work ?

cyring, 2024-02-21:
Is the processor really unlocked ?

Is there any BIOS option which prevent to alter MSR ?
That option could be name semaphore.

cyring, 2024-02-21:
In this [Reddit](https://www.reddit.com/r/overclocking/comments/unidia/intelr_xeonr_cpu_e51650_v3_overclocking_advice/?rdt=58279) it shows a BIOS screenshot where CPU all cores can be changed.
[image]
Can you change ratio in BIOS and see if CoreFreq reads the same afterwards

cyring, 2024-02-21:
> Set individually every Core Ratio Limit to max value, example 38

Source: [level1techs.com](https://winraid.level1techs.com/t/guide-haswell-e-ep-xeon-cpuid-306f2-turbo-unlock/32696)

Ratios OC seems to be a Xeon bug. 
My understanding is that when set to your max ratio in BIOS rather than AUTO then you can increase value from kernel, either _CoreFreq_ either MSR Tools

cyring, 2024-02-24:
I had implemented some Memory Controller [decoders](https://github.com/cyring/CoreFreq/blob/6001d1dfb56d735b0b9c54d5966c0405edf4db99/x86_64/corefreqk.h#L2399) for Haswell; can you take a screenshot or post `corefreq-cli -M` ?
Thanks

UleyeyeUwma, 2024-02-28:
> Is the processor really unlocked ?
> 
> Is there any BIOS option which prevent to alter MSR ? That option could be named semaphore.

Yes the processor is unlocked, however as you probably already figured out, it might've been the bios locking it, I read that message and experimented with different motherboards that I had around (all chinese x99, specifically huananzhi 8m and kllisre v205) and they both seem to be locking it.

> Can you change ratio in BIOS and see if CoreFreq reads the same afterwards

There is no such option in the default bios for these motherboards, there are modified bios that implement it but on the B85 Chipset BIOS OC is not allowed and I am unsure what chipset the other motherboard has.

As such I will close this issue as I suspect the issue is due to the motherboard chipset limitations and not related to this program at all.

closed 2024-02-28 by UleyeyeUwma