cyring/CoreFreq #479: No temp readings on Epyc 9274F

garceri, 2024-03-05:
I can't get any temperature readings under Ubuntu 22.04 running kernel 6.5
[image]

cyring, 2024-03-06:
Hello,

1. Can you post the text output of the CLI command:
```
corefreq-cli -s -n -m -n -B -n -M -n -k -n -C 1
```
2. Do you confirm using latest commits of the `master` branch ?
3. Do you confirm no other agent is monitoring temperature in parallel of _CoreFreq_ ? Is the kernel module `k10temp` running ? (show `lsmod`)
4. The PCI known by _CoreFreq_ are well identified ? Please provide `lspci -nn`

For all these outputs, please employ Markdown formatting.

cyring, 2024-03-16:
I think Genoa and Zen4/Hawk Point (#84)  are missing the temperature for the same reason: no thermal register known.

cyring, 2024-03-16:
Meanwhile I can fix the voltage Vcore if you provide me the CLI output requested above.

cyring, 2024-03-17:
Can you please pull [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch and test temperature ?

garceri, 2024-03-17:
Server is a HPE Proliant DL945 Gen11
k10temp was loaded, unloaded it did not make any difference:

Here is the output for the develop branch: as you can see temps seems to be stuck at 99
[image]

Here is the corefreq-cli output that was also requested (from the develop branh as well)
`# corefreq-cli -s -n -m -n -B -n -M -n -k -n -C 1 -n -i 1 -n -V 1 -n -W 1
Processor                                     [AMD EPYC 9274F 24-Core Processor]
|- PPIN#                                                      [ 2b6a56a45d98031]
|- Architecture                                                     [EPYC/Genoa]
|- Vendor ID                                                      [AuthenticAMD]
|- Firmware                                                        [ 71.121.0-5]
|- Microcode                                                        [0x0a101144]
|- Signature                                                           [  AF_11]
|- Stepping                                                            [      1]
|- Online CPU                                                          [ 48/ 48]
|- Base Clock                                                          [101.099]
|- Frequency            (MHz)                      Ratio
                 Min   1516.48                    <  15 >
                 Max   4043.95                    <  40 >
|- Factory                                                             [100.000]
                       4000                       [  40 ]
|- Performance
   |- P-State
                 TGT   4043.95                    <  40 >
   |- CPPC
                 Min    808.79                    <   8 >
                 Max    101.10                    <   1 >
                 TGT    808.79                    <   8 >
|- Turbo Boost                                                         [   LOCK]
                 XFR    808.79                    [   8 ]
                 CPB    808.79                    [   8 ]
                  1C   2729.67                    <  27 >
                  2C   1516.48                    <  15 >
|- Uncore                                                              [   LOCK]
                 CLK   2426.37                    [  24 ]
                 MEM   2426.37                    [  24 ]

Instruction Set Extensions
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y]
|- AVX512-F     [Y]    AVX512-DQ [Y]  AVX512-IFMA [Y]   AVX512-PF [N]
|- AVX512-ER    [N]    AVX512-CD [Y]    AVX512-BW [Y]   AVX512-VL [Y]
|- AVX512-VBMI  [Y] AVX512-VBMI2 [Y]  AVX512-VNNI [Y]  AVX512-ALG [Y]
|- AVX512-VPOP  [Y] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N]
|- AVX512-BF16  [Y] AVX-VNNI-VEX [N]    AVX-FP128 [N]   AVX-FP256 [Y]
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y]
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y]
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y]
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y]
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y]
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y]
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y]
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [Y]        UMIP [Y]
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y]

Features
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Advanced Virtual Interrupt Controller                        AVIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- LOCK prefix to read CR8                                    AltMov   [Capable]
|- Clear Zero Instruction                                     CLZERO   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Capable]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
[... 683 lines cut ...]
c0:01.2 "0604" "1022" "14ab" -r01 "" ""
c0:01.3 "0604" "1022" "14ab" -r01 "" ""
c0:01.4 "0604" "1022" "14ab" -r01 "" ""
c0:02.0 "0600" "1022" "149f" -r01 "" ""
c0:03.0 "0600" "1022" "149f" -r01 "" ""
c0:03.1 "0604" "1022" "14a5" -r01 "" ""
c0:03.2 "0604" "1022" "14a5" -r01 "" ""
c0:03.3 "0604" "1022" "14a5" -r01 "" ""
c0:04.0 "0600" "1022" "149f" -r01 "" ""
c0:05.0 "0600" "1022" "149f" -r01 "" ""
c0:05.2 "0604" "1022" "14aa" -r01 "" ""
c0:07.0 "0600" "1022" "149f" -r01 "" ""
c0:07.1 "0604" "1022" "14a7" -r01 "" ""
c0:07.2 "0604" "1022" "14a7" -r01 "" ""
c1:00.0 "0880" "103c" "3306" -r08 "1590" "00e4"
c1:00.1 "0300" "102b" "0538" -r03 "1590" "00e4"
c1:00.2 "0880" "103c" "3307" -r08 "1590" "00e4"
c1:00.4 "0c03" "103c" "22f6" -r01 -p20 "1590" "00e4"
c2:00.0 "1300" "1022" "14ac" -r01 "1022" "14ac"
c2:00.1 "0880" "1022" "14dc" "1022" "14dc"
c2:00.4 "0c03" "1022" "14c9" -rda -p30 "1590" "0278"
c3:00.0 "0106" "1022" "7901" -r91 -p01 "1590" "02f0"
c3:00.1 "0106" "1022" "7901" -r91 -p01 "1590" "02f0"
c8:00.0 "0104" "1000" "10e2" "1590" "03c8"
`

cyring, 2024-03-17:
Which temperature are we reading when System is idling ?

```
corefreq-cli -C 1 -n -c 1
```

garceri, 2024-03-17:
Let the server idle for ten minutes, Temp readings seems to be off:
And, are the Mhz numbers correct ? ie can the cpu cores run at 2.11 Mhz) ?

This is what lm-sensors report
```# sensors
power_meter-acpi-0
Adapter: ACPI interface
power1:      531.00 W  (interval = 300.00 s)

bnxt_en-pci-4600
Adapter: PCI adapter
temp1:        +51.0°C

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +43.8°C
Tccd1:        +32.2°C
Tccd2:        +32.0°C
Tccd3:        +33.6°C
Tccd4:        +33.1°C
Tccd5:        +31.8°C
Tccd6:        +30.8°C
Tccd7:        +32.2°C
Tccd8:        +31.1°C

bnxt_en-pci-4601
Adapter: PCI adapter
temp1:        +51.0°C

```
this is what the ILO reports:
[image]

and finally this is what corefreq reports:
```# corefreq-cli -C 1 -n -c 1
CPU Freq(MHz) VID  Vcore  TMP(C)    Accumulator       Energy(J)     Power(W)
000    4.06   193  1.2063   92  000000000000002464    0.037597656   0.037597656
001    1.05   193  1.2063   92  000000000000002008    0.030639648   0.030639648
002    0.90   193  1.2063   92  000000000000002035    0.031051636   0.031051636
003    1.17   193  1.2063   92  000000000000002027    0.030929565   0.030929565
004    6.60   193  1.2063   92  000000000000002387    0.036422729   0.036422729
005    1.52   193  1.2063   92  000000000000002029    0.030960083   0.030960083
006    1.15   193  1.2063   92  000000000000002084    0.031799316   0.031799316
007    1.20   193  1.2063   92  000000000000002044    0.031188965   0.031188965
008    0.92   193  1.2063   92  000000000000007890    0.120391846   0.120391846
009    0.91   193  1.2063   92  000000000000002076    0.031677246   0.031677246
010    1.05   193  1.2063   92  000000000000006834    0.104278564   0.104278564
011    0.69   193  1.2063   92  000000000000002018    0.030792236   0.030792236
012    7.45   193  1.2063   92  000000000000005280    0.080566406   0.080566406
013   22.15   193  1.2063   92  000000000000010238    0.156219482   0.156219482
014   18.18   193  1.2063   92  000000000000037448    0.571411133   0.571411133
015    7.70   193  1.2063   92  000000000000002270    0.034637451   0.034637451
016    1.52   193  1.2063   92  000000000000002144    0.032714844   0.032714844
017    1.75   193  1.2063   92  000000000000003654    0.055755615   0.055755615
018    3.59   193  1.2063   92  000000000000002121    0.032363892   0.032363892
019    1.04   193  1.2063   92  000000000000002005    0.030593872   0.030593872
020    1.05   193  1.2063   92  000000000000002036    0.031066895   0.031066895
021    1.17   193  1.2063   92  000000000000002007    0.030624390   0.030624390
022    1.10   193  1.2063   92  000000000000002004    0.030578613   0.030578613
023    0.63   193  1.2063   92  000000000000002012    0.030700684   0.030700684
024    8.84   193  1.2063   92  000000000000000000    0.000000000   0.000000000
025    0.84   193  1.2063   92  000000000000000000    0.000000000   0.000000000
026    0.96   193  1.2063   92  000000000000000000    0.000000000   0.000000000
027    1.01   193  1.2063   92  000000000000000000    0.000000000   0.000000000
028    0.89   193  1.2063   92  000000000000000000    0.000000000   0.000000000
029    0.92   193  1.2063   92  000000000000000000    0.000000000   0.000000000
030    0.89   193  1.2063   92  000000000000000000    0.000000000   0.000000000
031    2.82   193  1.2063   92  000000000000000000    0.000000000   0.000000000
032   66.44   193  1.2063   92  000000000000000000    0.000000000   0.000000000
033    1.51   193  1.2063   92  000000000000000000    0.000000000   0.000000000
[... 47 lines cut ...]
027    0.74 ( 0.01)   0.02   0.02  99.98   0.00   0.00   0.00  91 / 93:751/104
028    0.55 ( 0.01)   0.01   0.01  99.99   0.00   0.00   0.00  91 / 93:751/104
029    0.65 ( 0.01)   0.02   0.02  99.98   0.00   0.00   0.00  91 / 93:751/104
030    2.98 ( 0.03)   0.07   0.07  99.93   0.00   0.00   0.00  91 / 93:751/104
031    3.06 ( 0.03)   0.08   0.07  99.93   0.00   0.00   0.00  91 / 93:751/104
032   83.08 ( 0.82)   2.05   1.94  98.06   0.00   0.00   0.00  91 / 93:751/104
033    0.97 ( 0.01)   0.02   0.02  99.98   0.00   0.00   0.00  91 / 93:751/104
034    1.16 ( 0.01)   0.03   0.03  99.97   0.00   0.00   0.00  91 / 93:751/104
035    0.87 ( 0.01)   0.02   0.02  99.98   0.00   0.00   0.00  91 / 93:751/104
036    3.56 ( 0.04)   0.09   0.08  99.92   0.00   0.00   0.00  91 / 93:751/104
037   23.61 ( 0.23)   0.58   0.55  99.45   0.00   0.00   0.00  91 / 93:751/104
038 1488.55 (14.72)  36.81  34.67  65.33   0.00   0.00   0.00  91 / 93:751/104
039    3.34 ( 0.03)   0.08   0.08  99.92   0.00   0.00   0.00  91 / 93:751/104
040    4.55 ( 0.05)   0.11   0.11  99.89   0.00   0.00   0.00  91 / 93:751/104
041   42.59 ( 0.42)   1.05   0.99  99.01   0.00   0.00   0.00  91 / 93:751/104
042    0.88 ( 0.01)   0.02   0.02  99.98   0.00   0.00   0.00  91 / 93:751/104
043   34.77 ( 0.34)   0.86   0.81  99.19   0.00   0.00   0.00  91 / 93:751/104
044   22.38 ( 0.22)   0.55   0.52  99.48   0.00   0.00   0.00  91 / 93:751/104
045    1.11 ( 0.01)   0.03   0.03  99.97   0.00   0.00   0.00  91 / 93:751/104
046    1.19 ( 0.01)   0.03   0.03  99.97   0.00   0.00   0.00  91 / 93:751/104
047    1.03 ( 0.01)   0.03   0.02  99.98   0.00   0.00   0.00  91 / 93:751/104

    Averages:        Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)    TjMax:    Pkg:
                      1.04   0.98  99.02   0.00   0.00   0.00       0 C    92 C
```

cyring, 2024-03-18:
First Gen needed an temperature offset. Perhaps same with Genoa.

Can you compile my SMU tool [zencli](https://github.com/cyring/Tips/blob/master/C/zencli.c) ?

```sh
cc zencli.c -o zencli
```

As root, you will peek the thermal  [registers](https://github.com/cyring/CoreFreq/blob/33fa721a97faaf542e0820f8f061a159a9e9cc3a/x86_64/amd_reg.h#L203) I just know:

```sh
## since Zen gen1
zencli smu 0x59800
## per CCD
zencli smu 0x59954
## Family 19h APU
zencli smu 0x59B08
```

----

> And, are the Mhz numbers correct ? ie can the cpu cores run at 2.11 Mhz) ?

As a défaut, _CoreFreq_ is showing the relative frequency.
Press shortkey `!` to  toggle view to absolute frequency.
There is also a CLI argument to start in absolute frequency. See `corefreq-cli -h`

garceri, 2024-03-18:
Here you go:
Here are the zencli outputs you requested:

```
# ./zencli smu 0x59800
[0x00059800] READ(smu) = 0x61e30000 (1642266624)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0110 0001 1110 0011 0000 0000 0000 0000
# ./zencli smu 0x59954
[0x00059954] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
# ./zencli smu 0x59B08
[0x00059b08] READ(smu) = 0x00000ac4 (2756)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 1010 1100 0100

cyring, 2024-03-18:
> The Mhz reported, are relative to what ?

Those are cycles counted by hardware 
When no activity they fall down to no cycle

[image]

Relative comes from the Intel terminlogy about HARDWARE COORDINATION
[image]

Somehow AMD provides the same counters.

cyring, 2024-03-18:
Can you also dump the CCD range ?

```sh
## per CCD
zencli smu 0x59954
zencli smu 0x59958
zencli smu 0x5995C
zencli smu 0x59960
zencli smu 0x59964
zencli smu 0x59968
zencli smu 0x5996C
```

cyring, 2024-03-18:
I see, `k10temp` found them at offset 0x300 of 0x59800

https://elixir.bootlin.com/linux/latest/source/drivers/hwmon/k10temp.c#L475

So SMU address is 0x59B00 + (CCD number * 4)

cyring, 2024-03-18:
Fortunately thermal function was already made for Raphael.
Genoa is now pointing to it.
Please pull and test latest commit 7c926afda6ac5c890373dccebc08d774fc136dfa from `develop`

cyring, 2024-03-18:
It may be a wrong Register address
Still based on `develop` branch, can you edit the header file at that line:
https://github.com/cyring/CoreFreq/blob/87344fae6395c7d3d5dd6b7b0876a6ee3aae413d/x86_64/amd_reg.h#L208

and replace code with this one:

```c

#define SMU_AMD_THM_TCTL_CCD_REGISTER_F19H_61H				\
	(SMU_AMD_THM_TCTL_REGISTER_F17H + 0x300)
```

Next please rebuild, unload, reload and test for temperature

garceri, 2024-03-18:
I pulled and recompiled the commit you mentioned and seems to have improved a bit, still there are some cores whose's temperatures are somehow reported as 0 C..

```# ./corefreq-cli -C 1 -n -c 1
CPU Freq(MHz) VID  Vcore  TMP(C)    Accumulator       Energy(J)     Power(W)
000 4158.57   184  1.1500   38  000000000000254752    3.887207031   3.887207031
001 1155.50   184  1.1500   38  000000000000164422    2.508880615   2.508880615
002  975.78   184  1.1500   38  000000000000153330    2.339630127   2.339630127
003 1101.39   184  1.1500   38  000000000000156543    2.388656616   2.388656616
004 1040.26   184  1.1500   38  000000000000151524    2.312072754   2.312072754
005  976.49   184  1.1500   38  000000000000156197    2.383377075   2.383377075
006 1194.84   184  1.1500   39  000000000000175520    2.678222656   2.678222656
007 1253.40   184  1.1500   39  000000000000171429    2.615798950   2.615798950
008 1078.62   184  1.1500   39  000000000000168110    2.565155029   2.565155029
009 1065.19   184  1.1500    0  000000000000159550    2.434539795   2.434539795
010 1317.53   184  1.1500    0  000000000000173951    2.654281616   2.654281616
011 1144.12   184  1.1500    0  000000000000175765    2.681961060   2.681961060
012 1405.92   184  1.1500   39  000000000000194216    2.963500977   2.963500977
013 1571.54   184  1.1500   39  000000000000195119    2.977279663   2.977279663
014 1792.37   184  1.1500   39  000000000000200345    3.057022095   3.057022095
015 1372.33   184  1.1500    0  000000000000192656    2.939697266   2.939697266
016 1565.62   184  1.1500    0  000000000000198408    3.027465820   3.027465820
017 1351.56   184  1.1500    0  000000000000192128    2.931640625   2.931640625
018 3506.43   184  1.1500   38  000000000000231040    3.525390625   3.525390625
019 3699.47   184  1.1500   38  000000000000333141    5.083328247   5.083328247
020 1384.05   184  1.1500   38  000000000000190392    2.905151367   2.905151367
021 1114.08   184  1.1500   38  000000000000157730    2.406768799   2.406768799
022 1002.75   184  1.1500   38  000000000000276418    4.217803955   4.217803955
023 4158.57   184  1.1500   38  000000000000224629    3.427566528   3.427566528
024 1006.49   184  1.1500   38  000000000000000000    0.000000000   0.000000000
025 1044.69   184  1.1500   38  000000000000000000    0.000000000   0.000000000
026 1007.88   184  1.1500   38  000000000000000000    0.000000000   0.000000000
027 1020.24   184  1.1500   38  000000000000000000    0.000000000   0.000000000
028  979.97   184  1.1500   38  000000000000000000    0.000000000   0.000000000
029 1113.24   184  1.1500   38  000000000000000000    0.000000000   0.000000000
030 1284.31   184  1.1500   39  000000000000000000    0.000000000   0.000000000
031 1109.47   184  1.1500   39  000000000000000000    0.000000000   0.000000000
032 1164.28   184  1.1500   39  000000000000000000    0.000000000   0.000000000
033 1111.59   184  1.1500    0  000000000000000000    0.000000000   0.000000000
034 1189.57   184  1.1500    0  000000000000000000    0.000000000   0.000000000
035 1287.66   184  1.1500    0  000000000000000000    0.000000000   0.000000000
036 1582.32   184  1.1500   39  000000000000000000    0.000000000   0.000000000
037 1416.99   184  1.1500   39  000000000000000000    0.000000000   0.000000000
038 1338.64   184  1.1500   39  000000000000000000    0.000000000   0.000000000
039 1502.98   184  1.1500    0  000000000000000000    0.000000000   0.000000000
040 1464.83   184  1.1500    0  000000000000000000    0.000000000   0.000000000
041 1598.45   184  1.1500    0  000000000000000000    0.000000000   0.000000000
042 1769.44   184  1.1500   38  000000000000000000    0.000000000   0.000000000
043 1120.63   184  1.1500   38  000000000000000000    0.000000000   0.000000000
044 1427.70   184  1.1500   38  000000000000000000    0.000000000   0.000000000
045  972.31   184  1.1500   38  000000000000000000    0.000000000   0.000000000
046 4158.56   184  1.1500   38  000000000000000000    0.000000000   0.000000000
047 1147.79   184  1.1500   38  000000000000000000    0.000000000   0.000000000

             Package[0]    Cores         Uncore        Memory        Platform
Energy(J): 166.636291504  70.912399292  29.650711060   0.000000000   0.000000000
Power(W) : 166.636291504  70.912399292  29.650711060   0.000000000   0.000000000

CPU Freq(MHz) Ratio  Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)  Min TMP:TS  Max
000 4159.27 (41.14) 102.85 100.00   0.00   0.00   0.00   0.00  37 / 44:748/ 50
001  993.01 ( 9.82)  24.56  23.87  76.13   0.00   0.00   0.00  37 / 44:748/ 51
002 1067.17 (10.56)  26.39  25.65  74.35   0.00   0.00   0.00  37 / 44:748/ 51
003 1005.40 ( 9.94)  24.86  24.16  75.84   0.00   0.00   0.00  36 / 38:702/ 45
004 1088.16 (10.76)  26.91  26.15  73.85   0.00   0.00   0.00  36 / 38:698/ 45
005  990.94 ( 9.80)  24.50  23.81  76.19   0.00   0.00   0.00  36 / 38:698/ 45
006  963.20 ( 9.53)  23.82  23.15  76.85   0.00   0.00   0.00  35 / 41:722/ 46
007 1113.60 (11.01)  27.54  26.76  73.24   0.00   0.00   0.00  35 / 41:722/ 46
008 1021.29 (10.10)  25.25  24.55  75.45   0.00   0.00   0.00  35 / 40:714/ 46
009 1265.57 (12.52)  31.30  30.42  69.58   0.00   0.00   0.00  0  /  0:0  /  0
010 1203.92 (11.91)  29.77  28.94  71.06   0.00   0.00   0.00  0  /  0:0  /  0
011 1396.79 (13.82)  34.54  33.58  66.42   0.00   0.00   0.00  0  /  0:0  /  0
[... 49 lines cut ...]
# ./zencli smu 0x50038
[0x00050038] READ(smu) = 0x00000000 (0)
[... cut ...]
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0010 0110 0000 0111 0000 1100
# ./zencli smu 0x50048
[0x00050048] READ(smu) = 0x00150508 (1377544)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 0101 0000 0101 0000 1000
# ./zencli smu 0x5004A
[0x0005004a] READ(smu) = 0x00150508 (1377544)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 0101 0000 0101 0000 1000
```

cyring, 2024-03-18:
> This seems to be the real deal, i modified the file as suggested, with (SMU_AMD_THM_TCTL_REGISTER_F17H + 0x300) and the temperatures are looking much better now

Thank you very much for all your tests.
I'm now waiting for other Zen4 and EPYC Genoa confirmation of non regression before committing this address change.

----

New DIMMs of 24 or 48 GB are not so easy to debug. I have no bit value specified and known to decode with.
That's why they have been respectively decoded as 32 or 64 GB

cyring, 2024-03-18:
The temperature fix is made available in `develop` branch.
Can you pull latest [commits](https://github.com/cyring/CoreFreq/commits/develop/) and verify again ?

garceri, 2024-03-18:
Looks good after recompiling the latest `develop` changes.

cyring, 2024-03-18:
> Looks good after recompiling the latest `develop` changes.

Thank you for your confirmation.

----

In addition to the DIMM Geometry, it also remains misc issues

* Wrong frequency values retrieved by HSMP protocol:

```
|- Turbo Boost                                                         [   LOCK]
                 XFR    808.79                    [   8 ]
                 CPB    808.79                    [   8 ]
```

* CPPC looks odd
```
   |- CPPC
                 Max    101.10                    <   1 >
```

* Need to add the PCI id of IOMMU
```
   |- I/O MMU                                                      AMD-V   [OFF]
   |- Version                                                     [         N/A]
```

* All these ACPI objects were not queried
```
|- Processor Performance Control                                _PCT   [Missing]
|- Performance Supported States                                 _PSS   [Missing]
|- Performance Present Capabilities                             _PPC   [Missing]
|- Continuous Performance Control                               _CPC   [Missing]
```

* Not only yours, Zen4 CPUs are missing the `PU` unit
```
|- Units
   |- Power                                               watt   [      Missing]
```

* And many other `Missing` values in section `Power, Current & Thermal`

garceri, 2024-03-18:
Anything I can do, or information I can provide just let me know.!

cyring, 2024-03-20:
> Anything I can do, or information I can provide just let me know.!

Yes, please, I have various requests in next comments

cyring, 2024-03-20:
* Thermal Trip or PROC_HOT
```
./zencli smu 0x59808
```

Value above zero may happened when System is under high load

garceri, 2024-03-20:
Here you go:

```# ./zencli smu 0x59808
[0x00059808] READ(smu) = 0x00012921 (76065)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 0010 1001 0010 0001
```

cyring, 2024-03-20:
* Boosted coefficient of frequency

Either:
```
./zencli smu 0x5d2c4
```
Or:
```
./zencli smu 0x5d324
```

Fyi, to compute the boosted ratio, use the hexa returned above, instead of `0x12345678` in this formula:

```sh
echo $(( ((0x12345678 >> 17) & 0xff) >> 2 ))
```

garceri, 2024-03-20:
```
# ./zencli smu 0x5d324
[0x0005d324] READ(smu) = 0x06403900 (104872192)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0110 0100 0000 0011 1001 0000 0000
# ./zencli smu 0x5d2c4
[0x0005d2c4] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
# echo $(( ((0x06403900 >> 17) & 0xff) >> 2 ))
8
```

cyring, 2024-03-20:
> ```
> # ./zencli smu 0x5d324
> [0x0005d324] READ(smu) = 0x06403900 (104872192)
>    60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
>  0000 0000 0000 0000 0000 0000 0000 0000 [...]

Register is specified here:
https://github.com/cyring/CoreFreq/blob/49e4e2bc812f0a10467a00a6f9fb2c06e860a760/x86_64/amd_reg.h#L2205
But it was known as is for Zen2

Either address and/or bit fields for Zen4/EPYC have changed

According to [9274F](https://www.amd.com/en/products/cpu/amd-epyc-9274f), I'm looking for  a ratio of `43`

cyring, 2024-03-20:
Commit b465f42d8d5fd42978572be2ad4f70c286cdb5a8 in branch `develop` you will a fix to not decode the boosted ratio from SMU but from table I'm providing.
Can you please give a test and check results in Processor window at `XFR` and `CPB` ?

garceri, 2024-03-21:
Compiled that commit:
Is this what you are looking for ?
[image]

cyring, 2024-03-21:
> Is this what you are looking for ?

Exactly what I was expecting.

Thank you.

cyring, 2024-03-21:
In your PCI list I was expecting the IOMMU
For example, mine is the identifier `1022:1481`:
```sh
lspci -nn|grep -i iommu
```
```console
00:00.2 IOMMU [0806]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse IOMMU [1022:1481]
```

If enabled in BIOS, yours should be [`1022:149e`](https://github.com/cyring/CoreFreq/blob/b465f42d8d5fd42978572be2ad4f70c286cdb5a8/x86_64/corefreq-api.h#L1508) but not found in your list neither as a bus:device.function `00:00.2`

As the closest references, I found this [Dell](https://linux-hardware.org/?id=pci:1022-149e-1028-0af9) and [table](https://github.com/linuxhw/LsPCI/blob/master/Server/README.md), could HP have a different mapping ?

garceri, 2024-03-21:
I'd have to check if IOMMU is enabled in BIOS, I can't reboot the server right now, I'll check the BIOS setting as soon as I can find a time window to reboot it.

cyring, 2024-03-21:
Do you have other software, even for made for Windows, which can read the geometry of your 48 GB DIMM ?
I would like to know what I have to obtain in Banks, Ranks, Columns and Rows

garceri, 2024-03-21:
Does `dmidecode` output helps:
```
Memory Device
        Array Handle: 0x0016
        Error Information Handle: Not Provided
        Total Width: 80 bits
        Data Width: 64 bits
        Size: 96 GB
        Form Factor: DIMM
        Set: 5
        Locator: PROC 1 DIMM 6
        Bank Locator: Not Specified
        Type: DDR5
        Type Detail: Synchronous Registered (Buffered)
        Speed: 4800 MT/s
        Manufacturer: Samsung
        Serial Number: 04CED650
        Asset Tag: Not Specified
        Part Number: M321RYGA0BB0-CQKZJ
        Rank: 2
        Configured Memory Speed: 4800 MT/s
        Minimum Voltage: 1.1 V
        Maximum Voltage: 1.1 V
        Configured Voltage: 1.1 V
        Memory Technology: DRAM
        Memory Operating Mode Capability: Volatile memory
        Firmware Version: Not Specified
        Module Manufacturer ID: Bank 1, Hex 0xCE
        Module Product ID: Unknown
        Memory Subsystem Controller Manufacturer ID: Unknown
        Memory Subsystem Controller Product ID: Unknown
        Non-Volatile Size: None
        Volatile Size: 96 GB
        Cache Size: None
        Logical Size: None
```

cyring, 2024-03-21:
> Here is the manufacturer [spec sheet](https://semiconductor.samsung.com/emea/dram/module/rdimm/m321ryga0bb0-cqk/):
> 
> Does `dmidecode` output help?:

Thanks.

They give some hints
* Rank x Organization: 2R x 4
* ? Component Composition (4G x 4) x 40 

I'm searching for a deeper datasheet like JEDEC specs

cyring, 2024-03-22:
# P-States
You can reprogram P-States from UI but also using the driver parameters.

For example, your EPYC default P-States were discovered as bellow.
```
|- Turbo Boost                                                         [   LOCK]
                 XFR   4347.25                    [  43 ]
                 CPB   4347.25                    [  43 ]
                  1C   2729.67                    <  27 >
                  2C   1516.48                    <  15 >
```

Now suppose you want to alter these two **enabled** P-States to respectively `12` and `30` frequency ratios:
```ini
insmod build/corefreqk.ko Register_ClockSource=1 \
Register_Governor=1 Register_CPU_Idle=1 Register_CPU_Freq=1 \
Turbo_Ratio_Unlock=1 TurboBoost_Enable="0,1" \
Ratio_Boost="-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,12,30"
```

## Remarks
* `Ratio_Boost` is the P-States array from `18C` down to `1C`.
  `-1` means " don't touch value "
  This EPYC has only `1C` and `2C` enabled by hardware
* `Turbo_Ratio_Unlock` allows reprogrammation of P-States
* `TurboBoost_Enable="0,1"` is first disabled, hence value `"0"`, next P-States are reprogrammed, finally `"1"` means enable Turbo Boost

----

I hope you won't find this method too complicated.
Please try it out of Production and post the results.

garceri, 2024-03-26:
It will be some time until I can test it, the server is running production workloads and I'll need to take it offline to test this.
BTW, I could not find the IOMMU option on BIOS.

cyring, 2024-03-26:
> It will be some time until I can test it, the server is running production workloads and I'll need to take it offline to test this. BTW, I could not find the IOMMU option on BIOS.

Actions are transferred to the [todo list](https://github.com/cyring/CoreFreq/discussions/424#discussioncomment-8918475)

To refresh the [EPYC 9274F Wiki](https://gist.github.com/cyring/427d90c133afd88bfc8bd69a05fc045d), can you however pull the latest from `master` branch and post the output of:
```sh
corefreq-cli -s -n -m -n -z -n -B -n -M -n -k -n -C 1 -n -i 1 -n -V 1 -n -W 1 -n -R
```

garceri, 2024-03-27:
Here ya go:
```
# ./corefreq-cli -s -n -m -n -z -n -B -n -M -n -k -n -C 1 -n -i 1 -n -V 1 -n -W 1 -n -R
Processor                                     [AMD EPYC 9274F 24-Core Processor]
|- PPIN#                                                      [ 2b6a56a45d98031]
|- Architecture                                                     [EPYC/Genoa]
|- Vendor ID                                                      [AuthenticAMD]
|- Firmware                                                        [ 71.121.0-5]
|- Microcode                                                        [0x0a101144]
|- Signature                                                           [  AF_11]
|- Stepping                                                            [      1]
|- Online CPU                                                          [ 48/ 48]
|- Base Clock                                                          [101.099]
|- Frequency            (MHz)                      Ratio
                 Min   1516.49                    <  15 >
                 Max   4043.97                    <  40 >
|- Factory                                                             [100.000]
                       4000                       [  40 ]
|- Performance
   |- P-State
                 TGT   4043.97                    <  40 >
   |- CPPC
                 Min   4347.27                    <  43 >
                 Max    404.40                    <   4 >
                 TGT   4347.27                    <  43 >
|- Turbo Boost                                                         [   LOCK]
                 XFR   4347.27                    [  43 ]
                 CPB   4347.27                    [  43 ]
                  1C   2729.68                    <  27 >
                  2C   1516.49                    <  15 >
|- Uncore                                                              [   LOCK]
                 CLK   2426.38                    [  24 ]
                 MEM   2426.38                    [  24 ]

Instruction Set Extensions
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y]
|- AVX512-F     [Y]    AVX512-DQ [Y]  AVX512-IFMA [Y]   AVX512-PF [N]
|- AVX512-ER    [N]    AVX512-CD [Y]    AVX512-BW [Y]   AVX512-VL [Y]
|- AVX512-VBMI  [Y] AVX512-VBMI2 [Y]  AVX512-VNNI [Y]  AVX512-ALG [Y]
|- AVX512-VPOP  [Y] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N]
|- AVX512-BF16  [Y] AVX-VNNI-VEX [N]    AVX-FP128 [N]   AVX-FP256 [Y]
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y]
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y]
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y]
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y]
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y]
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y]
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y]
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [Y]        UMIP [Y]
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y]

Features
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Advanced Virtual Interrupt Controller                        AVIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- LOCK prefix to read CR8                                    AltMov   [Capable]
|- Clear Zero Instruction                                     CLZERO   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Capable]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast Short REP MOVSB                                         FSRM   [Capable]
|- Fast-String Operation                                        ERMS   [Capable]
[... 1112 lines cut ...]
#24   0   0   1   0   0   1   0   0   1   1   1   0   0
#25   0   0   1   0   0   1   0   0   1   1   1   0   0
#26   0   0   1   0   0   1   0   0   1   1   1   0   0
#27   0   0   1   0   0   1   0   0   1   1   1   0   0
#28   0   0   1   0   0   1   0   0   1   1   1   0   0
#29   0   0   1   0   0   1   0   0   1   1   1   0   0
#30   0   0   1   0   0   1   0   0   1   1   1   0   0
#31   0   0   1   0   0   1   0   0   1   1   1   0   0
#32   0   0   1   0   0   1   0   0   1   1   1   0   0
#33   0   0   1   0   0   1   0   0   1   1   1   0   0
#34   0   0   1   0   0   1   0   0   1   1   1   0   0
#35   0   0   1   0   0   1   0   0   1   1   1   0   0
#36   0   0   1   0   0   1   0   0   1   1   1   0   0
#37   0   0   1   0   0   1   0   0   1   1   1   0   0
#38   0   0   1   0   0   1   0   0   1   1   1   0   0
#39   0   0   1   0   0   1   0   0   1   1   1   0   0
#40   0   0   1   0   0   1   0   0   1   1   1   0   0
#41   0   0   1   0   0   1   0   0   1   1   1   0   0
#42   0   0   1   0   0   1   0   0   1   1   1   0   0
#43   0   0   1   0   0   1   0   0   1   1   1   0   0
#44   0   0   1   0   0   1   0   0   1   1   1   0   0
#45   0   0   1   0   0   1   0   0   1   1   1   0   0
#46   0   0   1   0   0   1   0   0   1   1   1   0   0
#47   0   0   1   0   0   1   0   0   1   1   1   0   0
```

cyring, 2024-03-30:
Commit dc6280f215c92f288f161db144897d4580d8c3ac fixes the spacing in the Power Monitoring bottom area, especially when a 3 digits power is measured.

Format:
```
corefreq-cli -W
```
```console
CPU Freq(MHz)    Accumulator      Min  Energy(J) Max    Min  Power(W)  Max
000 3798.08  000000000000411306    0.00   6.28   8.87    0.00   6.28   8.87
001 3682.77  000000000000496376    0.00   7.57   8.20    0.00   7.57   8.20
002 3850.84  000000000000417714    0.00   6.37   9.53    0.00   6.37   9.53
003 3830.85  000000000000496311    0.00   7.57   8.59    0.00   7.57   8.59
004 3860.38  000000000000437088    0.00   6.67   8.64    0.00   6.67   8.64
005 3826.94  000000000000448859    0.00   6.85   8.19    0.00   6.85   8.19
006 3951.14  000000000000407283    0.00   6.21   8.10    0.00   6.21   8.10
007 3820.92  000000000000512015    0.00   7.81   8.20    0.00   7.81   8.20
008 3838.89  000000000000453518    0.00   6.92   7.74    0.00   6.92   7.74
009 3807.62  000000000000450171    0.00   6.87   8.38    0.00   6.87   8.38
010 3969.38  000000000000451074    0.00   6.88   7.74    0.00   6.88   7.74
011 4025.41  000000000000339855    0.00   5.19   7.88    0.00   5.19   7.88
012 3892.19  000000000000417007    0.00   6.36   7.74    0.00   6.36   7.74
013 3905.71  000000000000386342    0.00   5.90   7.68    0.00   5.90   7.68
014 3726.84  000000000000527824    0.00   8.05   9.04    0.00   8.05   9.04
015 3908.34  000000000000494423    0.00   7.54   8.21    0.00   7.54   8.21
016 3984.11  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
017 3749.12  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
018 3783.17  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
019 3755.73  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
020 3774.03  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
021 3810.57  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
022 4013.79  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
023 3953.22  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
024 3711.33  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
025 4015.93  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
026 3878.68  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
027 3880.01  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
028 3820.70  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
029 3956.70  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
030 3721.36  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
031 3987.81  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00

Energy(J)  Package[0]         Cores               Uncore              Memory
 17.68 145.5 145.50   0.02 109.1 126.85  10.50  17.1  18.67   0.00   0.0   0.00
Power(W)
 17.68 145.5 145.50   0.02 109.1 126.85  10.50  17.1  18.67   0.00   0.0   0.00
```

Can you please show me yours ?

garceri, 2024-04-05:
Compiled with that commit:

```
CPU Freq(MHz)    Accumulator      Min  Energy(J) Max    Min  Power(W)  Max
000 4158.46  000000000000287548    3.96   4.39   4.58    3.96   4.39   4.58
001  815.78  000000000000140811    0.99   2.15   3.20    0.99   2.15   3.20
002 1242.39  000000000000168693    0.99   2.57   3.03    0.99   2.57   3.03
003  822.40  000000000000134472    1.45   2.05   3.48    1.45   2.05   3.48
004  867.54  000000000000128800    1.06   1.97   3.56    1.06   1.97   3.56
005 4158.45  000000000000213139    1.61   3.25   3.49    1.61   3.25   3.49
006 1218.80  000000000000175362    1.39   2.68   3.88    1.39   2.68   3.88
007 1174.40  000000000000183114    1.41   2.79   3.61    1.41   2.79   3.61
008 1059.32  000000000000163275    1.34   2.49   3.43    1.34   2.49   3.43
009 1020.70  000000000000156604    1.77   2.39   4.36    1.77   2.39   4.36
010 1076.38  000000000000163866    1.43   2.50   3.77    1.43   2.50   3.77
011 1161.18  000000000000166150    1.35   2.54   3.44    1.35   2.54   3.44
012 1554.54  000000000000208391    1.59   3.18   3.88    1.59   3.18   3.88
013 1557.12  000000000000193716    1.50   2.96   3.97    1.50   2.96   3.97
014 1593.36  000000000000219877    1.92   3.36   4.09    1.92   3.36   4.09
015 2129.36  000000000000213482    2.09   3.26   3.95    2.09   3.26   3.95
016 1363.04  000000000000194069    2.33   2.96   3.79    2.33   2.96   3.79
017 2811.99  000000000000261894    2.65   4.00   4.36    2.65   4.00   4.36
018 1393.42  000000000000195983    1.56   2.99   3.38    1.56   2.99   3.38
019 1330.79  000000000000188547    2.06   2.88   3.46    2.06   2.88   3.46
020 1442.29  000000000000199629    1.42   3.05   3.31    1.42   3.05   3.31
021 1127.02  000000000000171821    1.02   2.62   4.11    1.02   2.62   4.11
022 1405.24  000000000000200195    0.55   3.05   3.93    0.55   3.05   3.93
023 2114.84  000000000000269604    2.28   4.11   5.29    2.28   4.11   5.29
024  987.82  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
025  881.87  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
026  998.60  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
027  923.60  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
028  898.90  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
029 1102.39  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
030 1150.28  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
031 1320.70  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
032 1094.68  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
033 1050.02  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
034 1147.96  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
035 1094.74  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
036 1929.81  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
037 1689.75  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
038 1725.64  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
039 1429.20  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
040 1668.44  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
041 2993.88  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
042 1368.23  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
043 1432.18  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
044 1646.60  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
045 1360.20  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
046 1654.89  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
047 3433.35  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00

Energy(J)  Package[0]         Cores               Uncore              Memory
137.49 166.2 178.57  12.81  70.2  83.26   0.46  29.7  88.04   0.00   0.0   0.00
Power(W)
137.49 166.2 178.57  12.81  70.2  83.26   0.46  29.7  88.04   0.00   0.0   0.00
```

cyring, 2024-04-05:
> Compiled with that commit:

Thank you.
It looks as expected.

Version [`1.97.1`](https://github.com/cyring/CoreFreq/releases/tag/1.97.1) is now released.

cyring, 2024-04-07:
It looks like some bits have architecturally been fixed since kernel `6.8.4`

I'm now reading the `PSE` of `CR4` register set to `1` on all Cores rather than the only Bootstrap Core

[image]

Is it still case on [Genoa](https://gist.github.com/cyring/427d90c133afd88bfc8bd69a05fc045d) ?

cyring, 2024-04-07:
Hello,

In this AMD HSMP source code, I'm reading an [address exception](https://github.com/amd/amd_hsmp/blob/f34b98fb70b67a7908f32d4d6da5a2ee89857b08/amd_hsmp.c#L47) for Zen family 1A

Can you edit and replace value of `SMU_HSMP_CMD`

https://github.com/cyring/CoreFreq/blob/3a25d369777c2924b578721bec5e3994dc95c412/x86_64/amd_reg.h#L222

with this value:

```c
#define SMU_HSMP_CMD		0x3b10934
```

I would also like to put the `case AMD_Zen4_Genoa`

https://github.com/cyring/CoreFreq/blob/3a25d369777c2924b578721bec5e3994dc95c412/x86_64/corefreqk.c#L7892

as below:

```c

	switch (PUBLIC(RO(Proc))->ArchID) {
	case AMD_Zen4_PHX2:
	case AMD_Zen4_HWK:
	case AMD_Zen4_PHX:
	case AMD_Zen4_RPL:
	case AMD_Zen3Plus_RMB:
	case AMD_Zen3_VMR:
	case AMD_Zen2_MTS:
		Core_AMD_SMN_Read(XtraCOF,
				SMU_AMD_F17H_MATISSE_COF,
				PRIVATE(OF(Zen)).Device.DF);
		break;
	case AMD_Zen4_Bergamo:
	case AMD_EPYC_Rome_CPK:
	case AMD_Zen4_Genoa:
		Core_AMD_SMN_Read(XtraCOF,
				SMU_AMD_F17H_ZEN2_MCM_COF,
				PRIVATE(OF(Zen)).Device.DF);
		break;
	}
```

Rebuild and reload _CoreFreq_

Finally post the output of `corefreq-cli -s`

Thank you for helping.

garceri, 2024-04-07:
Will do as  soon as I can, pleased to help!

garceri, 2024-04-18:
Here you go, sorry for the delay:

`
Processor                                     [AMD EPYC 9274F 24-Core Processor]
|- PPIN#                                                      [ 2b6a56a45d98031]
|- Architecture                                                     [EPYC/Genoa]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0a101144]
|- Signature                                                           [  AF_11]
|- Stepping                                                            [      1]
|- Online CPU                                                          [ 48/ 48]
|- Base Clock                                                          [101.099]
|- Frequency            (MHz)                      Ratio
                 Min   1516.48                    <  15 >
                 Max   4043.95                    <  40 >
|- Factory                                                             [100.000]
                       4000                       [  40 ]
|- Performance
   |- P-State
                 TGT   4043.95                    <  40 >
   |- CPPC
                 Min    808.79                    <   8 >
                 Max    101.10                    <   1 >
                 TGT    808.79                    <   8 >
|- Turbo Boost                                                         [   LOCK]
                 XFR    808.79                    [   8 ]
                 CPB    808.79                    [   8 ]
                  1C   2729.67                    <  27 >
                  2C   1516.48                    <  15 >
|- Uncore                                                              [   LOCK]
                 CLK   2426.37                    [  24 ]
                 MEM   2426.37                    [  24 ]

Instruction Set Extensions
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y]
|- AVX512-F     [Y]    AVX512-DQ [Y]  AVX512-IFMA [Y]   AVX512-PF [N]
|- AVX512-ER    [N]    AVX512-CD [Y]    AVX512-BW [Y]   AVX512-VL [Y]
|- AVX512-VBMI  [Y] AVX512-VBMI2 [Y]  AVX512-VNNI [Y]  AVX512-ALG [Y]
|- AVX512-VPOP  [Y] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N]
|- AVX512-BF16  [Y] AVX-VNNI-VEX [N]    AVX-FP128 [N]   AVX-FP256 [Y]
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y]
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y]
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y]
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y]
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y]
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y]
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y]
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [Y]        UMIP [Y]
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y]

Features
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Advanced Virtual Interrupt Controller                        AVIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- LOCK prefix to read CR8                                    AltMov   [Capable]
|- Clear Zero Instruction                                     CLZERO   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Capable]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast Short REP MOVSB                                         FSRM   [Capable]
|- Fast-String Operation                                        ERMS   [Capable]
|- Fused Multiply Add                                           FMA4   [Missing]
[... 154 lines cut ...]
   |- Time Window                                                TW1   [   0 ns]
|- Thermal Design Power                                       Uncore   [Disable]
   |- Power Limit                                                PL1   [    0 W]
   |- Time Window                                                TW1   [   0 ns]
|- Thermal Design Power                                         DRAM   [Disable]
   |- Power Limit                                                PL1   [    0 W]
   |- Time Window                                                TW1   [   0 ns]
|- Thermal Design Power                                     Platform   [Disable]
   |- Power Limit                                                PL1   [    0 W]
   |- Time Window                                                TW1   [   0 ns]
   |- Power Limit                                                PL2   [    0 W]
   |- Time Window                                                TW2   [   0 ns]
[... cut ...]
|- Package Thermal Point
   |- Thermal Monitor Trip                                     Limit   [  115 C]
   |- HTC Temperature Limit                                    Limit   [    0 C]
   |- HTC Temperature Hysteresis                           Threshold   [    0 C]
|- Units
   |- Power                                               watt   [      Missing]
   |- Energy                                             joule   [  0.000015259]
   |- Window                                            second   [  0.000976562]
`

cyring, 2024-04-19:
Thanks.
Looking at CPB ratio and PL1, PL2, it appears that changing the address as below is a regression on HSMP
```
#define SMU_HSMP_CMD		0x3b10934
```

cyring, 2024-04-19:
* Reading once again the [commit](https://github.com/amd/amd_hsmp/commit/49534a1f73bcf1d986ce8b0e5a3670abb50d09b9), it concerns the future Zen5/EPYC 

* Redhat [boot issue](https://access.redhat.com/solutions/7064896) on Genoa.

* Nothing else found